fwupd/fwupd #8504: Option to entirely skip bootloader management for UEFI capsule updates

nicholasbishop, 2025-02-25:
**Is your feature request related to a problem? Please describe.**

On ChromeOS Flex, we'd like our UEFI bootloader to manage installing firmware updates rather than chaining to [fwupd-efi].

This is already implemented in our bootloader; it can read/write the same UEFI vars and file paths as fwupd-efi, and calls [`UpdateCapsule`] to initiate the actual update. Among other things, doing it in our bootloader simplifies our signing process and reduces the amount of C code in our boot stack. (I can go into more detail on our boot stack and some of the constraints thereof, if it would be helpful.)

On the fwupd side, there are a few actions that interfere with this approach:
1. [`fu_uefi_capsule_device_check_asset`] will fail if secure boot is enabled and a signed `fwupd-efi` is not present.
2. [`fu_uefi_nvram_device_write_firmware`] calls [`fu_uefi_bootmgr_bootnext`], which (depending on secure boot status and various configs) may try to copy bootloader files, and will add a "Linux Firmware Updater" `BootNNNN` entry, and modify `BootNext`.
3. [`fu_uefi_nvram_device_get_results`] calls [`fu_uefi_bootmgr_verify_fwupd`], which will fail if the expected boot entry doesn't exist.

**Describe the solution you'd like**

A potential solution would be to add a new `[uefi_capsule]` config called `ManageBootloaders`. This would default to true (i.e. keep existing behavior unchanged). If false, the unwanted checks and actions described above would be skipped.

**Describe alternatives you've considered**

A variant of the above idea, instead of adding conditionals to `fu-uefi-nvram-device.c` we could add a new `FuUefiCapsuleDevice` backend and conditionally use that, depending on the config. I'm not sure if there are any advantages to adding a new backend though.

**Additional context**

I've implemented the proposed change in this [WIP branch] and verified it works for my use case.

However, I'm not wedded to this particular proposal, happy to hear any feedback you might have whether positive or negative, and happy to provide more details if it would be helpful!

[ChromeOS Flex]: https://chromeos.google/products/chromeos-flex/
[fwupd-efi]: https://github.com/fwupd/fwupd-efi
[`UpdateCapsule`]: https://uefi.org/specs/UEFI/2.11/08_Services_Runtime_Services.html#updatecapsule
[`fu_uefi_nvram_device_write_firmware`]: https://github.com/fwupd/fwupd/blob/949f57f32f18120febb241e2f2ba8d50f08a9539/plugins/uefi-capsule/fu-uefi-nvram-device.c#L115-L126
[`fu_uefi_bootmgr_bootnext`]: https://github.com/fwupd/fwupd/blob/949f57f32f18120febb241e2f2ba8d50f08a9539/plugins/uefi-capsule/fu-uefi-bootmgr.c#L340
[`fu_uefi_nvram_device_get_results`]: https://github.com/fwupd/fwupd/blob/949f57f32f18120febb241e2f2ba8d50f08a9539/plugins/uefi-capsule/fu-uefi-nvram-device.c#L21C1-L21C33
[`fu_uefi_bootmgr_verify_fwupd`]: https://github.com/fwupd/fwupd/blob/949f57f32f18120febb241e2f2ba8d50f08a9539/plugins/uefi-capsule/fu-uefi-bootmgr.c#L58
[`fu_uefi_capsule_device_check_asset`]: https://github.com/fwupd/fwupd/blob/949f57f32f18120febb241e2f2ba8d50f08a9539/plugins/uefi-capsule/fu-uefi-capsule-device.c#L471
[WIP branch]: https://github.com/nicholasbishop/fwupd/commits/bishop-manage-bootloaders-flag/

hughsie, 2025-02-25:
> and calls [UpdateCapsule](https://uefi.org/specs/UEFI/2.11/08_Services_Runtime_Services.html#updatecapsule) to initiate the actual update

Can you describe the desired flow? I'm having a hard time working out how this works without UEFI variables. It feels a lot like it's a different fu-uefi-xxxx-device.c object.

hughsie, 2025-02-25:
It's more the BootNext and BootOrder bits I'm confused about.

nicholasbishop, 2025-02-25:
Ah gotcha. The BootNext and BootOrder vars don't need to be modified because our regular bootloader can check on each boot if there are any firmware updates to install (by checking for `fwupd-*` variables in the `ATTEMPT_UPDATE` state). If firmware updates are ready to be attempted the bootloader does that, otherwise it boots the OS.

superm1, 2025-02-25:
I'm that sense it's more like how windows does it.

hughsie, 2025-02-26:
@superm1 what do you think about the config name; that jarred slightly. I wondered if rather than turning off searching for fwupdx64.efi we should just check for the *right* thing -- e.g. chromeosfw64.efi or whatever the binary is called.

hughsie, 2025-02-26:
I don't think we involve nmbl for `fwupdx64.efi` -- i.e. we'd still be chainloaded from `shimx64.efi`.

superm1, 2025-02-26:
> I don't think we involve nmbl for `fwupdx64.efi` -- i.e. we'd still be chainloaded from `shimx64.efi`.

Currently; but is that the long term plan? It's a new bootloader. I do generally like this idea, and especially that it means not having to write Boot#### variables to work.

nicholasbishop, 2025-03-06:
> I wondered if rather than turning off searching for fwupdx64.efi we should just check for the _right_ thing -- e.g. chromeosfw64.efi or whatever the binary is called.

I'll try to implement that. I think that does make sense, and would presumably be easy to extend to nbml or other bootloaders in the future; just add additional names to a list of bootloaders-that-do-capsule-updates.

superm1, 2025-03-06:
> I'll try to implement that in a new branch. I think that does make sense, and would presumably be easy to extend to nbml or other bootloaders in the future; just add additional names to a list of bootloaders-that-do-capsule-updates.

The one challenge I see is how do you detect you have the right _version_ of a bootloader for such support?  Can you have a pkgconfig file (for a build time decision) or touch an nvram variable or something else?

> Currently; but is that the long term plan? It's a new bootloader. I do generally like this idea, and especially that it means not having to write Boot#### variables to work.

As I like this idea; I've added some RFE bugs for nmbl and systemd-boot to consider this approach too and referenced this one.

nicholasbishop, 2025-03-06:
> The one challenge I see is how do you detect you have the right version of a bootloader for such support? Can you have a pkgconfig file (for a build time decision) or touch an nvram variable or something else?

A UEFI var is a good idea. It wouldn't even need to be in nvram right? Have the bootloader create a variable on each boot like `BootloaderSupportsFwupd` (with GUID `FU_EFIVARS_GUID_FWUPDATE`?), and attrs `EFI_VARIABLE_RUNTIME_ACCESS`. Then we avoid unnecessarily taking any nvram space.

superm1, 2025-03-06:
Yeah that sounds workable to me!

closed 2025-03-11 by hughsie

nicholasbishop, 2025-03-11:
Thank you both for your help in finding the right shape for this feature :)