fwupd/fwupd #8510: Regression: ESRT min version doesn't use format

JohnAZoidberg, 2025-02-27:
**fwupd version information**
Built from 0b7fb83ba640bdfd5667aa87dd2f38448baeb836, also occurs on 2.0.6 from dnf in fedora rawhide.
---

CSME version is correctly rendered but for the system firmware only the current, but not the minimum version is rendered correctly.
```
fwupdtool get-devices --plugins uefi-capsule
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Framework Laptop 12 (13th Gen Intel Core)
│                                      
├─Intel Management Engine:     
│     Device ID:          6b850c6cee19a6261e19846c7e71c0da64b053cd
│     Summary:            UEFI System Resource Table device (Updated via capsule-on-disk)
│     Current version:    0.0.0.2473                                          
│     Minimum Version:    0.0.0.1000                                          
│     Vendor:             Framework (DMI:INSYDE Corp.)      
│     Update State:       Success                                                                                                                            
│     GUID:               0f74c56d-d5ba-4942-96fa-d37560f40554
│     Device Flags:       • Internal device
│                         • Updatable                                                                                                                        
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│                                      
└─System Firmware:                     
      Device ID:          384231dd4d582f9575f1c0dde71f2ec239a869ff
      Summary:            UEFI System Resource Table device (Updated via capsule-on-disk)
      Current version:    0.0.2.1 
      Minimum Version:    512  
      Vendor:             Framework (DMI:INSYDE Corp.)
      Update State:       Success                                                                                                                                  GUID:               6bc0986c-d281-5ba3-965c-2f8d13e1eee8
      Device Flags:       • Internal device                                                                                                                                            • Updatable                                         
                          • System requires external power source
```

This prevents me from updating my firmware:

```
Selected device: System Firmware
10:16:31.645 FuEngine             6bc0986c-d281-5ba3-965c-2f8d13e1eee8 matched 3 components
10:16:31.645 FuRelease            remote framework-embargo has kind=download and so marking as trusted
10:16:31.645 FuEngine             failed to set release for component: Specified firmware is older than the minimum required version '0.0.2.0 < 512'
10:16:31.645 FuRelease            remote framework-embargo has kind=download and so marking as trusted
10:16:31.645 FuEngine             failed to set release for component: Specified firmware is older than the minimum required version '0.0.2.1 < 512'
10:16:31.645 FuRelease            remote framework-embargo has kind=download and so marking as trusted
10:16:31.645 FuEngine             failed to set release for component: Specified firmware is older than the minimum required version '0.0.2.2 < 512'
10:16:31.645 FuEngine             6bc0986c-d281-5ba3-965c-2f8d13e1eee8 matched 0 releases
10:16:31.645 FuMain               No releases found
```

JohnAZoidberg, 2025-02-27:
I can workaround this with creating a quirk:

```
[6bc0986c-d281-5ba3-965c-2f8d13e1eee8]
VersionFormat=quad
```

but usually the format is just fetched from the updates on LVFS, so no quirk is needed.

closed 2025-02-27 by hughsie

JohnAZoidberg, 2025-03-13:
Could we release a new version soon @hughsie this bug prevents us from updating our firmware and I'm sure it affects users as well.

hughsie, 2025-03-13:
@superm1 is on holiday this week, so realistically it'll be next week at soonest.

JohnAZoidberg, 2025-04-01:
Thanks. For future reference, this is fixed in 2.07 and 1.9.29