fwupd/fwupd #8512: Capsule header fix up is missing in Capsule-on-disk method

jebasimmanuvel07, 2025-02-27:
**Describe the bug**
In the Capsule-on-memory method, there is a capsule header fix up when the capsule header is missing or invalid.

Ref: https://github.com/fwupd/fwupd/blob/main/plugins/uefi-capsule/fu-uefi-nvram-device.c#L92

But in other hand, for Capsule-on-disk method, there is no capsule header fix up.  

Ref:  https://github.com/fwupd/fwupd/blob/main/plugins/uefi-capsule/fu-uefi-cod-device.c#L181

**Steps to Reproduce**
Tried to pass capsule without EFI_CAPSULE_HEADER in capsule-on-disk method. The tool didn't fix up capsule header.

**Expected behavior**
Fwupd have to fixed up capsule header in capsule on disk also.

hughsie, 2025-02-27:
I was hoping that for devices new enough to support CoD we'd also get a valid capsule header -- the fwupd heuristic is just that -- some guesses that might be right. Is there any reason you need the fixup?

jebasimmanuvel07, 2025-02-27:
@hughsie   

In Windows, OS loader will add a capsule header by information taken from the ESRT table.  [https://learn.microsoft.com/en-us/windows-hardware/drivers/bringup/processing-updates](https://learn.microsoft.com/en-us/windows-hardware/drivers/bringup/processing-updates)

The Fwupd tool also added a capsule header in the capsule-on-memory method if the header is not there, but it does not add it in the capsule-on-disk method. It causes inconsistency.

superm1, 2025-02-27:
I didn't realize Windows will add the header too.  If that's the case; compat makes sense to me.

closed 2025-02-27 by hughsie