fwupd/fwupd #8522: Crash after some "too many open files" error

hexchain, 2025-03-02:
**Describe the bug**
Since Feb 15th, fwupd has started to emit a lot of "Too many open files" messages:

```
Feb 21 10:15:30 host fwupd[2246]: 02:15:30.666 FuBackend            failed to create device: failed to probe: failed to open /sys/devices/pci0000:00/0000:00:08.1/uevent: Too many open files
Feb 21 10:15:30 host fwupd[2246]: 02:15:30.666 FuBackend            failed to create device: failed to probe: failed to open /sys/devices/pci0000:00/0000:00:08.1/0000:63:00.0/uevent: Too many open files
Feb 21 10:15:30 host fwupd[2246]: 02:15:30.666 FuBackend            failed to create device: failed to probe: failed to open /sys/devices/pci0000:00/0000:00:08.1/uevent: Too many open files
Feb 21 10:15:30 host fwupd[2246]: 02:15:30.668 FuBackend            failed to create device: failed to probe: failed to open /sys/devices/pci0000:00/0000:00:08.1/uevent: Too many open files
Feb 21 10:15:30 host fwupd[2246]: 02:15:30.668 FuBackend            failed to create device: failed to probe: failed to open /sys/devices/pci0000:00/0000:00:08.1/0000:63:00.0/uevent: Too many open files
```

And eventually it crashed after a few days:
```
Feb 21 10:54:09 host fwupd[2246]: 02:54:09.466 GLib                 Creating pipes for GWakeup: Too many open files
Feb 21 10:54:10 host systemd-coredump[323800]: [🡕] Process 2246 (fwupd) of user 0 dumped core.
```

Here is the backtrace of the last crash:
<details>

```
(gdb) thread apply all bt

Thread 4 (Thread 0x7a5682ffe6c0 (LWP 235106)):
#0  __syscall_cancel_arch () at ../sysdeps/unix/sysv/linux/x86_64/syscall_cancel.S:56
#1  0x00007a56891f1e33 in __internal_syscall_cancel (a1=<optimized out>, a2=<optimized out>, a3=<optimized out>, a4=<optimized out>, a5=a5@entry=8, a6=a6@entry=0, nr=271) atcancellation.c:49
#2  0x00007a56891f1e74 in __syscall_cancel (a1=<optimized out>, a2=<optimized out>, a3=<optimized out>, a4=<optimized out>, a5=a5@entry=8, a6=a6@entry=0, nr=271) at cancellation.c:75
#3  0x00007a568926ca06 in __GI_ppoll (fds=fds@entry=0x7a5678000b90, nfds=nfds@entry=2, timeout=<optimized out>, timeout@entry=0x0, sigmask=sigmask@entry=0x0) at ../sysdeps/unix/sysv/linux/ppoll.c:42
#4  0x00007a56894406a7 in ppoll (__fds=0x7a5678000b90, __nfds=2, __timeout=0x0, __ss=0x0) at /usr/include/bits/poll2.h:101
#5  g_main_context_poll_unlocked (priority=<optimized out>, context=0x601b1acb40b0, timeout_usec=<optimized out>, fds=0x7a5678000b90, n_fds=2) at ../glib/glib/gmain.c:4591
#6  g_main_context_iterate_unlocked.isra.0 (context=0x601b1acb40b0, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4268
#7  0x00007a56893de647 in g_main_loop_run (loop=0x601b1ad97ea0) at ../glib/glib/gmain.c:4475
#8  0x00007a5689643fc4 in gdbus_shared_thread_func (user_data=0x601b1ad99760) at ../glib/gio/gdbusprivate.c:284
#9  0x00007a5689412f66 in g_thread_proxy (data=0x601b1ad97ec0) at ../glib/glib/gthread.c:892
#10 0x00007a56891f570a in start_thread (arg=<optimized out>) at pthread_create.c:448
#11 0x00007a56892798b4 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:100

Thread 3 (Thread 0x7a56845ff6c0 (LWP 235104)):
#0  __syscall_cancel_arch () at ../sysdeps/unix/sysv/linux/x86_64/syscall_cancel.S:56
#1  0x00007a56891f1e33 in __internal_syscall_cancel (a1=<optimized out>, a2=<optimized out>, a3=<optimized out>, a4=<optimized out>, a5=a5@entry=8, a6=a6@entry=0, nr=271) atcancellation.c:49
#2  0x00007a56891f1e74 in __syscall_cancel (a1=<optimized out>, a2=<optimized out>, a3=<optimized out>, a4=<optimized out>, a5=a5@entry=8, a6=a6@entry=0, nr=271) at cancellation.c:75
#3  0x00007a568926ca06 in __GI_ppoll (fds=fds@entry=0x601b1acb17e0, nfds=nfds@entry=2, timeout=<optimized out>, timeout@entry=0x7a56845fead0, sigmask=sigmask@entry=0x0) at ../sysdeps/unix/sysv/linux/ppoll.c:42
#4  0x00007a56894406a7 in ppoll (__fds=0x601b1acb17e0, __nfds=2, __timeout=0x7a56845fead0, __ss=0x0) at /usr/include/bits/poll2.h:101
#5  g_main_context_poll_unlocked (priority=<optimized out>, context=0x601b1acb1580, timeout_usec=<optimized out>, fds=0x601b1acb17e0, n_fds=2) at ../glib/glib/gmain.c:4591
#6  g_main_context_iterate_unlocked.isra.0 (context=context@entry=0x601b1acb1580, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4268
#7  0x00007a56893dce15 in g_main_context_iteration (context=0x601b1acb1580, may_block=may_block@entry=1) at ../glib/glib/gmain.c:4338
#8  0x00007a56893dce72 in glib_worker_main (data=<optimized out>) at ../glib/glib/gmain.c:6537
#9  0x00007a5689412f66 in g_thread_proxy (data=0x601b1acad780) at ../glib/glib/gthread.c:892
#10 0x00007a56891f570a in start_thread (arg=<optimized out>) at pthread_create.c:448
#11 0x00007a56892798b4 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:100

Thread 2 (Thread 0x7a56837ff6c0 (LWP 235105)):
#0  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
#1  0x00007a568940ef90 in g_cond_wait_impl (cond=0x601b1ad69ae8, mutex=0x601b1ad69ae0) at ../glib/glib/gthread-posix.c:1007
#2  g_cond_wait (cond=0x601b1ad69ae8, mutex=0x601b1ad69ae0) at ../glib/glib/gthread.c:1665
#3  0x00007a56893a5d0c in g_async_queue_pop_intern_unlocked (queue=0x601b1ad69ae0, wait=1, end_time=-1) at ../glib/glib/gasyncqueue.c:375
#4  0x00007a56894155b7 in g_thread_pool_spawn_thread (data=<optimized out>) at ../glib/glib/gthreadpool.c:297
#5  0x00007a5689412f66 in g_thread_proxy (data=0x601b1ad8e7d0) at ../glib/glib/gthread.c:892
#6  0x00007a56891f570a in start_thread (arg=<optimized out>) at pthread_create.c:448
#7  0x00007a56892798b4 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:100

Thread 1 (Thread 0x7a56875b0a80 (LWP 235098)):
#0  _g_log_abort (breakpoint=<optimized out>) at ../glib/glib/gmessages.c:426
#1  g_logv (log_domain=0x7a56894a23cf "GLib", log_level=G_LOG_LEVEL_ERROR, format=<optimized out>, args=args@entry=0x7ffe0e3de000) at ../glib/glib/gmessages.c:1273
#2  0x00007a56893e3354 in g_log (log_domain=<optimized out>, log_level=<optimized out>, format=<optimized out>) at ../glib/glib/gmessages.c:1315
#3  0x00007a568943a1f3 in g_wakeup_new () at ../glib/glib/gwakeup.c:162
#4  0x00007a56893daf2c in g_main_context_new_with_flags (flags=flags@entry=G_MAIN_CONTEXT_FLAGS_NONE) at ../glib/glib/gmain.c:663
#5  0x00007a56893dafcc in g_main_context_new () at ../glib/glib/gmain.c:608
#6  0x00007a5689631a7c in g_dbus_connection_send_message_with_reply_sync (connection=0x7a566c000ba0, message=0x7a567800ad70, flags=G_DBUS_SEND_MESSAGE_FLAGS_NONE, timeout_msec=2000, out_serial=0x0, cancellable=0x0, error=0x7ffe0e3de1f0) at ../glib/gio/gdbusconnection.c:2290
#7  0x00007a568963e891 in g_dbus_connection_call_sync_internal (connection=0x7a566c000ba0, bus_name=0x601b2111b680 "org.freedesktop.DBus", object_path=0x601b1ae09e20 "/org/freedesktop/DBus", interface_name=0x601b1ae0d990 "org.freedesktop.DBus", method_name=0x601af4aa84f6 "GetConnectionUnixUser", parameters=0x601b21a49ee0, reply_type=0x7a568969d441, flags=G_DBUS_CALL_FLAGS_NONE, timeout_msec=2000, fd_list=0x0, out_fd_list=0x0, cancellable=0x0, error=0x7ffe0e3de360) at ../glib/gio/gdbusconnection.c:6582
[... 343 lines cut ...]
│   
└─Unifying Receiver:
      Device ID:          97cdbea0c50eb8670b1e55df727adb04bc294bd9
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR24.11_B0036
      Bootloader Version: BOT03.03_B0009
      Vendor:             Logitech, Inc. (USB:0x046D, HIDRAW:0x046D)
      Install Duration:   18 seconds
      GUIDs:              aa995882-da42-574a-8338-c8dfdca447e3 ← UFY\VID_046D&PID_C52B
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
                          cc4cbfa9-bf9d-540b-b92b-172ce31013c1 ← USB\VID_046D&PID_AAAC
      Device Flags:       • Updatable
                          • Supported on remote server
                          • Signed Payload
                          • Can tag for emulation
    
```

</details>

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? No

hughsie, 2025-03-02:
When fwupd is running, what does `sudo ls /proc/`pidof fwupd`/fd/ | wc -l` say?

hexchain, 2025-03-04:
```
# ls /proc/`pidof fwupd`/fd/ | wc -l
112
```

It appears that a set of particular files are opened multiple times:
<details>

```
/proc/298802/fd # ll
total 0
lr-x------ 1 root root 64 Mar  2 21:28 0 -> /dev/null
lrwx------ 1 root root 64 Mar  2 21:28 1 -> 'socket:[1712381]'
lrwx------ 1 root root 64 Mar  2 21:28 2 -> 'socket:[1712381]'
lrwx------ 1 root root 64 Mar  2 21:28 3 -> 'anon_inode:[eventfd]'
lr-x------ 1 root root 64 Mar  2 21:28 4 -> anon_inode:inotify
lrwx------ 1 root root 64 Mar  4 14:18 5 -> 'anon_inode:[eventfd]'
lrwx------ 1 root root 64 Mar  4 14:18 6 -> /var/lib/fwupd/pending.db
lrwx------ 1 root root 64 Mar  4 14:18 7 -> /var/cache/fwupd/quirks.db
lrwx------ 1 root root 64 Mar  4 14:18 8 -> 'anon_inode:[eventfd]'
lrwx------ 1 root root 64 Mar  4 14:18 9 -> 'anon_inode:[eventfd]'
lrwx------ 1 root root 64 Mar  4 14:18 10 -> 'anon_inode:[timerfd]'
lrwx------ 1 root root 64 Mar  4 14:18 11 -> 'socket:[1712399]'
lrwx------ 1 root root 64 Mar  4 14:18 12 -> 'socket:[1715557]'
lr-x------ 1 root root 64 Mar  4 14:18 13 -> /sys/firmware/acpi/tables/UEFI
lr-x------ 1 root root 64 Mar  4 14:18 14 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 15 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/descriptors
lr-x------ 1 root root 64 Mar  2 21:28 16 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/9-1.1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 17 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 18 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 19 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/9-1.1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 20 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 21 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 22 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/9-1.1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 23 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 24 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 25 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/9-1.2/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 26 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 27 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 28 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/9-1.2/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 29 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 30 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 31 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/9-1.2/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 32 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 33 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 34 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/9-1.1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 35 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 36 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 37 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/9-1.1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 38 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 39 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 40 -> /sys/devices/pci0000:00/0000:00:08.1/0000:63:00.4/usb3/3-3/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 41 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/9-1.1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 42 -> /sys/devices/pci0000:00/0000:00:08.1/0000:63:00.3/usb1/1-4/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 43 -> /sys/devices/pci0000:00/0000:00:08.1/0000:63:00.3/usb1/1-4/bos_descriptors
lr-x------ 1 root root 64 Mar  4 14:18 44 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 45 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 46 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/9-1.1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 47 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 48 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 49 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/9-1.2/descriptors
lr-x------ 1 root root 64 Mar  4 14:18 50 -> /sys/devices/pci0000:00/0000:00:08.3/0000:64:00.4/usb9/9-1/descriptors
[... cut ...]
  9-1               1a40:0101 09 1IF  [USB 2.00,   480 Mbps, 100mA] (Terminus Technology Inc. Hub) hub
    9-1.1             046d:c52b 00 3IFs [USB 2.00,    12 Mbps,  98mA] (Logitech USB Receiver)
    9-1.2             05ac:0220 ef 3IFs [USB 2.00,    12 Mbps, 100mA] (Milsky LL Dongle 66CF5E573AD0)
```
This is the built-in USB hub on my external monitor, connected to the computer via a USB cable. Each time the monitor wakes up from DPMS, the USB hub gets disconnected -> connected -> disconnected -> connected.

hughsie, 2025-03-04:
I'm looking into this now.

closed 2025-03-04 by hughsie