fwupd/fwupd #8523: Problem of executed fwupdmgr to update FW error

Chris679745, 2025-03-02:
**Describe the bug**
Updating BIOS with Fwupdmgr then get error message of "No supported devices found".

**Steps to Reproduce**
test step as follows:
1. Filled the ESRT GUID and built bios.
2. Create *.metainfo.xml to meet ESRT GUID
3. Generate Cabinet file 
4. Offline update binary image on SUT (boot in Ubuntu 24.04.)
5. Type the command "fwupdmgr install xxx.cab"
- This time will see error message of "No supported devices found" result.
[root@localhost Documents]# fwupdmgr install  AAA_BIOS_xxxxxx.cab
Decompressing…           [   -                                   ]
No supported devices found

6. Type command "fwupdmgr get-devices --show-all-devices"
root@aaa-SERVER:/home/aaa/Desktop# fwupdmgr get-devices --show-all-devices
AAA AAA SERVER
│
├─Unknown Device:
│     Device ID:          b5881e53bf55f4e454a88a354e1f9432e5780820
│     Serial Number:      1458
│     GUID:               2615c3d5-abac-5f05-b178-fb917ed9d272 ← DRM\VEN_BNQ&DEV_7692
│   
├─AMD EPYC 9475F 48-Core Processor:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0b00211e
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              a9020c87-24da-572e-a209-fb025ea50d2a ← CPUID\PRO_0&FAM_1A&MOD_02
│ │                       588d70fb-e905-5e24-a67c-e19cfe4536a7 ← CPUID\PRO_0&FAM_1A&MOD_02&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ └─Secure Processor:
│       Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│       Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│       GUIDs:            e86aaf84-671f-5186-a7d6-11960d1ebaec ← PCI\VEN_1022&DEV_156E
│                         f637121e-514f-59a5-ac60-ef73434f1180 ← PCI\VEN_1022&DEV_156E&SUBSYS_102214CA
│       Device Flags:     • Internal device
│     
├─ASPEED Graphics Family:
│     Device ID:          806f785700b71a9934455bb120fef51bd2586cca
│     Current version:    52
│     Vendor:             ASPEED Technology, Inc. (PCI:0x1A03)
│     GUIDs:              237d7407-66ed-5408-852b-2d5d2647b131 ← PCI\VEN_1A03&DEV_2000
│                         b4fcd116-9b58-5bc1-b7aa-fe5510ef4b47 ← PCI\VEN_1A03&DEV_2000&SUBSYS_1A032000
│                         2174e084-c2d9-5055-a67d-d528634abfc3 ← PCI\VEN_1A03&DEV_1150
│                         bd916a57-0a2c-599b-a41f-2d5b226b96a7 ← PCI\VEN_1A03&DEV_1150&SUBSYS_1A031150
│     Device Flags:       • Internal device
│                         • Supported on remote server
│                         • Cryptographic hash verification is available
│   
├─GPIO controller:
│     Device ID:          f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│   
├─System Firmware:
│ │   Device ID:          18ffc26172df91d9c8f84b215029f6f340147a94
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    9018
│ │   Minimum Version:    9018
│ │   Vendor:             AAA 
│ │   Update State:       Success
│ │   GUID:               6095ac68-d66c-4b2d-9428-7a8a1bebe7b5
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  371
│       Minimum Version:  371
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            d07ff664-b0e1-5f4e-a723-d7fbcbfcb94f ← UEFI\CRT_3CD3F0309EDAE228767A976DD40D9F4AFFC4FBD5218F2E8CC3C9DD97E8AC6F9D&ARCH_X64
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Custom PK
      GUID:               9b259c30-edf1-5153-8227-c90924d39bbf ← UEFI\CRT_11AC6974A07B23373D9B6B8B7829CA3F3458EAA1
      
root@aaa-SERVER:/home/aaa/Desktop# 

7. "metainfo.xml file" is as below.
[image]

hughsie, 2025-03-02:
Hmm, nothing obvious jumps out at me. Can you attach the output of `sudo fwuptool install AAA_BIOS_xxxxxx.cab -vv` please

Chris679745, 2025-03-03:
attach log.

[fwupdmgr_install_cab_dump.txt](https://github.com/user-attachments/files/19046887/fwupdmgr_install_cab_dump.txt)

closed 2025-03-04 by Chris679745