fwupd/fwupd #8529: Can't install DBX update, Error opening file /boot/efi/EFI/ubuntu/shimx64.efi: Permission denied

poshul, 2025-03-03:
**Describe the bug**
When I try to install the latest DBX update on my X1C12 thinkpad I get the following: 
`failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /boot/efi/EFI/ubuntu/shimx64.efi: Error opening file /boot/efi/EFI/ubuntu/shimx64.efi: Permission denied
`. This looks like fwupdmgr is helpfully trying to check to make sure that it doesn't brick my machine on the DBX update (https://github.com/fwupd/fwupd/wiki/Blocked-executable-in-the-ESP,-ensure-grub-and-shim-are-up-to-date) but it can't load the shimx64.efi file to check whether the signature is still good post-DBX-update.

**Steps to Reproduce**
On an X1 carbon 12th gen run `fwupdmgr update`, with DBX 20230501 and an update to DBX 20241101 pending, select Y at perform operation.

**Expected behavior**
Update installs cleanly

**fwupd version information**
Please provide the version of the daemon and client.
`CompileVersion(io.snapcraft.fwupd): 7013
CompileVersion(org.freedesktop.fwupd): 2.0.5
`
`
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**System UEFI configuration**
Please provide the output of the following commands:

```shell
efibootmgr -v
```

```shell
efivar -l | grep fw
```

```shell
tree /boot
```

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?
- Are you using an NVMe disk?
- Is secure boot enabled?
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?

valentindavid, 2025-03-03:
What version of snapd is running? Could it be https://github.com/canonical/snapd/pull/14987 ?

poshul, 2025-03-03:
It was running snapd 2.67. Updating to the beta went smoothly and fixed the issue.

closed 2025-03-03 by poshul