fwupd/fwupd #8536: Update failed "no valid release URIs" with fwupd 1.7.9

russellxiao, 2025-03-04:
**Describe the bug**
I am verifying the firmware update from our embargo remote, I used fwupdmgr update to install the update, but after click y to perform an update, it has a error "no valid release URIs".
[image]

BTW,  I manually downloaded the .cab file from embargo remote and installed on the target successfully.

**Steps to Reproduce**

Copy nvidia-embargo.conf to /etc/fwupd/remotes.d/.
Run fwupdmgr refresh --force.
Run fwupdmgr get-updates, checked the available updates are there.
Run fwupdmgr update
Click y to perform update.

**Expected behavior**
I expect the firmware installed on target.

**fwupd version information**
Please provide the version of the daemon and client.
# fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.168-tegra
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10

```shell
fwupdtool get-report-metadata
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

apt
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

# fwupdmgr get-devices --show-all-devices
�│
�├�─G1M15M:
�│     Device ID:          a3ef198bd9356a5c55e7784412afec09581bce68
�│     Current version:    0x3634515430304741
�│     Vendor:             Micron (EMMC:0x000013)
�│     Serial Number:      0x1ebca2a5
�│     GUIDs:              4069750d-b244-5498-b4f1-47b182d59c6b �← EMMC\NAME_G1M15M
�│                         61de7eba-6573-5196-aef5-467ecab3b9fb �← EMMC\0019&0334
�│                         d21b74dc-09a8-56b6-9395-f3d5779e3b4f �← EMMC\MAN_0013&OEM_014E&NAME_G1M15M
�│     Device Flags:       �• Internal device
�│                         �• Updatable
�│   
�└�─System Firmware:
      Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
      Summary:            UEFI ESRT device
      Current version:    0.36.4.0
      Minimum Version:    0.36.4.0
      Vendor:             NVIDIA (DMI:EDK II)
      Update State:       Success
      GUIDs:              bf0d4599-20d4-414e-b2c5-3595b1cda402
                          230c8b18-8d9b-53ec-838b-6cfc0383493a �← main-system-firmware
      Device Flags:       �• Internal device
                          �• Updatable
                          �• System requires external power source
                          �• Supported on remote server
                          �• Needs a reboot after installation
                          �• Device is usable for the duration of the update
                          �• Signed Payload
    

**Additional questions**

I have modified the out-embargo.conf with my mail address and generated User token.
Also, I check the output of fwupdmgr get-releases --json, the output as following:

# fwupdmgr get-releases --json
{
  "Releases" : [
    {
      "AppstreamId" : "com.nvidia.test.lvfs.firmware",
      "ReleaseId" : "108533",
      "RemoteId" : "nvidia-embargo",
      "Name" : "LVFS Test",
      "Summary" : "Firmware update for the System Firmware",
      "Description" : "<p>New feature update, improve the performance and stability of the device.</p>",
      "Version" : "0.36.4.2",
      "Filename" : "8eaa0b77de8832b61801ab68dbe007d641fa1230",
      "Protocol" : "org.uefi.capsule",
      "Categories" : [
        "X-Device"
      ],
      "Checksum" : [
        "66ca1807b3804fde9ccd0e0ebe008a5d16787477",
        "7bb8afafba501cb2636c6a3156d8431ca9bf953c9ab481c4086216a6404bb0cc"
      ],
      "License" : "LicenseRef-proprietary",
      "Size" : 44122731,
      "Created" : 1740528000,
      "Locations" : [
        "https://my_mail_address:my_user_token@fwupd.org/downloads/7bb8afafba501cb2636c6a3156d8431ca9bf953c9ab481c4086216a6404bb0cc-lvfs_test_r36.4.2_v1.cab"
      ],
      "Uri" : "https://my_mail_address:my_user_token@fwupd.org/downloads/7bb8afafba501cb2636c6a3156d8431ca9bf953c9ab481c4086216a6404bb0cc-lvfs_test_r36.4.2_v1.cab",
      "Homepage" : "https://developer.nvidia.com",
      "Vendor" : "NVIDIA",
      "Flags" : [
        "is-upgrade"
      ]
    }
}

I can download the .cab file with the Uri in Chrome, why does fwupdmgr still output "no valid release URIs"? Thanks.

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

hughsie, 2025-03-04:
Hey @russellxiao -- 1.7.x [is now EOL](https://github.com/fwupd/fwupd/blob/main/SECURITY.md) -- would it be possible for you to update fwupd to 1.8.x, 1.9.x or 2.0.x? If not, then we can continue to debug but I think updating would be the easiest thing.

If I had to guess, I would guess that `my_mail_address` needs to be escaped, e.g. `russellxiao%40nvidia.com` -- if you get the idea.

russellxiao, 2025-03-04:
Hi @hughsie Thanks for quick reply. I checked the URI, it looks like correct:
"Uri" : "https://russellx@nvidia.com:my_user_token@fwupd.org/downloads/7bb8afafba501cb2636c6a3156d8431ca9bf953c9ab481c4086216a6404bb0cc-lvfs_test_r36.4.2_v1.cab",

would it be possible for you to update fwupd to 1.8.x, 1.9.x or 2.0.x?
=> I think I can try the 1.9.24 with Ubuntu 24.04. But I have an issue about esp configuration issue with this new version. Not sure this is a good place to ask this question.
The capsule offline test error is "/boot/efi does not have sufficient space, required 109.4▒ MB, got 66.0▒ MB". I checked the source code of fwupd 1.9.24, when the RequireESPFreeSpace is not configured, fwupd calculates the size <sz_reqd = fu_firmware_get_size(firmware) * 2 + (20u * 1024 * 1024);>. I'd like to configure the RequireESPFreeSpace, should I create a file such as uefi_capsule.conf to the /etc/fwupd/? Or do you have any suggestion? A new config file is better for me.

hughsie, 2025-03-04:
@russellxiao can you try changing it to `https://russellx%40nvidia.com:my_user_token@fwupd.org`....

> I'd like to configure the RequireESPFreeSpace

`man fwupd.conf` should help -- for 1.9.x you just need to edit `/etc/fwupd/fwupd.conf` --  you might also be interested in https://github.com/fwupd/fwupd/pull/8463 although that would just be for 2.0.7 and newer.

russellxiao, 2025-03-04:
Thanks, @hughsie change it to %40 works for fwupd 1.7.9.
I'll also try the newer fwupd.
Close the issue.

closed 2025-03-04 by russellxiao