fwupd/fwupd #8541: fwupd 1.9.13 could not upgrade LFVS packages on RHEL9.4

kean0048, 2025-03-05:
**Describe the bug**
After downloading the LVFS packages the 1.9.13 version, the system rebooted, and could not finish the update, and prompt like " WARNING: No updates to process, exiting in 10 seconds." However, there are updated files for the ESP partition.

**Steps to Reproduce**
1) install RHEL 9.4
2) fwupdmgr get-updates and fwupdmgr update, and input Y
3) reboot and Press F12 and select 'Linux-Firmware-Updater'
4) reproduced as 'WARNING: No updates to process, exiting in 10 seconds.'

**Expected behavior**
upgrade sucess.

**fwupd version information**
fwupd-efi version 1.9.13

```shell
fwupdtool get-report-metadata
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
dnf

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version: RHEL9.4
- Have you tried rebooting? YES
- Is this a regression? YES

kean0048, 2025-03-05:
Hi,

Please help to check and is there any way to show more debug info during this process?

Thanks.

hughsie, 2025-03-05:
hey @kean0048 -- the `No updates to process` is shown when the fwupdx64.efi binary can't find any updates to process. The list comes from the fwupd-prefixed EFI key with the GUID of 0abba7dc-e516-4167-bbf5-4d9d1c739416 -- e.g. `/sys/firmware/efi/efivars/fwupd-*-*-0abba7dc-e516-4167-bbf5-4d9d1c739416` should exist when you get the "An update requries a reboot" prompt. Can you confirm that EFI key exists when you've scheduled a capsule update with `fwupdmgr update`? And if the fwupdx64.efi binary refuses to run, can you see if the variable exists in the EFI shell?

kean0048, 2025-03-07:
Hi Richard,

Thanks for your response.
Please help to check the test results I have attached here.
It looks like not correct, I will check the directory '/sys/firmware/efi/efivars/' again.

[ECR647151.zip](https://github.com/user-attachments/files/19117853/ECR647151.zip)

kean0048, 2025-03-07:
Upload the /sys/firmware/efi/efivars/ output.
There is no item which we are testing.

[list.txt](https://github.com/user-attachments/files/19122837/list.txt)

Thanks.