fwupd/fwupd #8557: `PciPspNoHsi`, `PciPspWithHsi`, and `VeryBasicTest` tests fail

Makonede, 2025-03-07:
**Describe the bug**
Installation fails with `PciPspNoHsi.test_pci_psp_device`, `PciPspNoHsi.test_pci_psp_hsi`, `PciPspWithHsi.test_pci_psp_device`, `PciPspWithHsi.test_pci_psp_hsi`, `VeryBasicTest.test_get_devices`, and `VeryBasicTest.test_properties` tests failing with exit status `1`.

**Steps to Reproduce**
1. Attempt to install the package

**Expected behavior**
The package installs.

**fwupd version information**
`fwupd-git-1.9.10.r1196.gb8b45b190-1` from the AUR

<details>

<summary>**fwupd device information**</summary>

```
Framework Laptop 16 (AMD Ryzen 7040 Series)
│
├─Unknown Device:
│     Device ID:          e9fde5043538ed8b8974e7726598b453c6768b90
│     GUID:               bf413d0a-f2ac-5dfa-83b1-01846d036cf0 ← GPIO\ID_cros-ec-gpio.3.auto
│     Device Flags:       • Can tag for emulation
│   
├─AMD Ryzen 9 7940HS w/ Radeon 780M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a704107
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
│ │                       5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─AMD Radeon 780M:
│ │ │   Device ID:        83eba1fa14bdefee1f6b4bcf06069b298e1c338f
│ │ │   Summary:          AMD AMD_PHOENIX_GENERIC
│ │ │   Current version:  1
│ │ │   Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │ │   GUID:             4a1501b7-b500-5255-9d9c-41d652a4d5bc ← AMD\113-PHXGEN
│ │ │   Device Flags:     • Internal device
│ │ │                     • Can tag for emulation
│ │ │ 
│ │ └─NE160QDM-NZ6:
│ │       Device ID:      aec1a869eb0df71b7cea6b3ac71d39b830faf164
│ │       Vendor:         PNP:BOE
│ │       GUID:           62710139-0372-57e5-a59e-f83260745941 ← DRM\VEN_BOE&DEV_0BC9
│ │       Device Flags:   • Internal device
│ │                       • Can tag for emulation
│ │     
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Current version:  00.2d.00.7d
│ │     Bootloader Version: 00.2d.00.7d
│ │     Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUID:             9eb6a793-7f97-5fb5-b49c-139d2ce3ee46 ← PCI\VEN_1022&DEV_15C7
│ │     Device Flags:     • Internal device
│ │                       • Can tag for emulation
│ │   
│ ├─System Management Unit (SMU):
│ │     Device ID:        db0330716216c629bb2c07256e5d018f499eb6ce
│ │     Summary:          Microcontroller used within CPU/APU program 0
│ │     Current version:  76.87.0
│ │     Vendor:           Advanced Micro Devices, Inc.
│ │     GUID:             3ac3159e-6eef-5f6b-bc8d-67686b238747 ← /sys/devices/platform/AMDI0009:00
│ │     Device Flags:     • Internal device
│ │                       • Can tag for emulation
│ │   
│ └─TPM:
│       Device ID:        1d8d50a4dbc65618f5c399c2ae827b632b3ccc11
│       Current version:  6.3.1.615
│       Vendor:           Microsoft (TPM:MSFT)
[... 101 lines cut ...]
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│                         • Can tag for emulation
│     
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            frame.work-LaptopAMDPK
      GUID:               c701846d-5461-5c19-a7ae-2d89b2f44c14 ← UEFI\CRT_D610D8205624E644D9435383826B2DBF7298859C
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
                          • Can tag for emulation
```

</details>

**Additional questions**

- Operating system and version: Arch Linux `x86_64`, Zen Kernel `6.13.5.zen1-1`
- Have you tried rebooting? Yes
- Is this a regression? Yes

superm1, 2025-03-07:
Isn't this a problem with the AUR package? Not an upstream problem?

We run those tests just fine in CI.

Makonede, 2025-03-07:
Why would the tests in the package be any different than the ones upstream?

superm1, 2025-03-07:
I would say environment, or missing dependencies (like gobject introspection).  Here is a recent test run we did on arch.  You can look at the logs and see it working.

https://github.com/fwupd/fwupd/actions/runs/13726080761/job/38392824427

closed 2025-04-03 by superm1