fwupd/fwupd #8563: Wavlink WL-ST348 (JMicron JMS586U) enclosure drives not recognised

aioue, 2025-03-09:
**Describe the bug**
Drives inside the Wavlink WL-ST348 (JMicron JMS586U) are not recognised. FuEngine error.

Dual NVMe -> USB-C enclosure

**Steps to Reproduce**
```
flatpak run org.freedesktop.fwupd get-devices
```

**Expected behavior**
NVMe drives are show as present

**fwupd version information**

```
#  flatpak run org.freedesktop.fwupd --version
Loading…                 [************************************** ]
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
WARNING: This package has not been validated, it may not work properly.
compile   info.libusb                   1.0.23
compile   org.freedesktop.fwupd         2.0.6
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.5
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.8.12-8-pve
runtime   org.freedesktop.fwupd         2.0.6
```

```shell
# fwupdtool get-report-metadata
Loading…                 [*****                                  ]17:55:01.924 FuPluginUefiCapsule  cannot find default ESP: No ESP or BDP found
Loading…                 [*******************                    ]17:55:02.135 FuUdevDevice         reading subsystem_device for 0xffff was invalid: value 65535 was above maximum 65534
Loading…                 [*******************                    ]17:55:02.195 FuUdevDevice         reading subsystem_device for 0xffff was invalid: value 65535 was above maximum 65534
Loading…                 [********************                   ]17:55:02.219 FuUdevDevice         reading subsystem_device for 0xffff was invalid: value 65535 was above maximum 65534
Loading…                 [*********************                  ]17:55:02.225 FuUdevDevice         reading subsystem_device for 0xffff was invalid: value 65535 was above maximum 65534
Loading…                 [*************************************  ]
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
WARNING: Will measure elements of system integrity around an update
See https://github.com/fwupd/fwupd/wiki/PluginFlag:measure-system-integrity for more information.
Waiting…                 [*************************************  ]17:55:02.315 FuVolume             fu_volume_get_mount_point: assertion 'FU_IS_VOLUME(self)' failed

BootTime:               1741523634
CompileVersion(com.hughsie.libjcat):0.1.9
CompileVersion(com.hughsie.libxmlb):0.3.10
CompileVersion(org.freedesktop.fwupd):1.8.12
CompileVersion(org.freedesktop.gusb):0.3.10
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 5 PRO 5650G with Radeon Graphics
DistroId:               debian
DistroName:             Debian GNU/Linux
DistroPrettyName:       Debian GNU/Linux 12 (bookworm)
DistroVersion:          12
FwupdSupported:         True
HostBaseboardManufacturer:ASRock
HostBaseboardProduct:   X300M-STX
HostBiosMajorRelease:   05
HostBiosMinorRelease:   11
HostBiosVendor:         American Megatrends International, LLC.
HostBiosVersion:        P1.90C
HostEnclosureKind:      3
HostFamily:             To Be Filled By O.E.M.
HostFirmwareMajorRelease:ff
HostFirmwareMinorRelease:ff
HostProduct:            To Be Filled By O.E.M.
HostSku:                To Be Filled By O.E.M.
[... 133 lines cut ...]
└─WDC PC SN530 256G:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    HPS2
      Vendor:             PCI:0x15B7
      Serial Number:      foo
      GUIDs:              dd75b173-c635-57e1-9569-c9da6dca84bb ← NVME\VEN_15B7&DEV_5009
                          83579426-bdbb-5944-9af0-393070f66a99 ← NVME\VEN_15B7&DEV_5009&SUBSYS_15B75009
                          4b4d8fa2-61cb-572d-9bf2-a092c8e84c2f ← WDC PC SN530 SDBPNPZ-256G-1006
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Can tag for emulation
```

</details>

**Additional questions**

- Operating system and version:  Linux proxmox 6.8.12-8-pve #1 SMP PREEMPT_DYNAMIC PMX 6.8.12-8 (2025-01-24T12:32Z) x86_64 GNU/Linux

- Have you tried rebooting?
- Is this a regression?

hughsie, 2025-03-09:
> RuntimeVersion(org.freedesktop.fwupd):1.8.12

Why and how are you running such an old fwupd version using flatpak?

aioue, 2025-03-10:
> > RuntimeVersion(org.freedesktop.fwupd):1.8.12
> 
> Why and how are you running such an old fwupd version using flatpak?

Isn't 2.0.6 the latest release?

`runtime   org.freedesktop.fwupd         2.0.6`

The one bundled with my system's package manager is old.

hughsie, 2025-03-10:
> fwupdtool get-report-metadata

Ohh if you're using flatpak that needs to be `flatpak run org.freedesktop.fwupd get-report-metadata` instead I guess. I'd also need `flatpak run org.freedesktop.fwupd get-devices -vv` in that case please.

aioue, 2025-03-30:
Returned this item due to lack of TRIM support. I am no longer able to debug and issue can be closed if you wish.

closed 2025-03-30 by hughsie