fwupd/fwupd #8569: fwupd does not play nice with GNOME 48 (girepository 3.0)

DimStar77, 2025-03-11:
While trying to build fwupd 2.0.6 against GNOME 48 (pygobject 3.52), Have encountered these errors in the build log:

```
[  165s] Traceback (most recent call last):
[  165s]   File "/home/abuild/rpmbuild/BUILD/fwupd-2.0.6-build/fwupd-2.0.6/data/tests/fwupd_test.py", line 63, in setUpClass
[  165s]     override_gi_search_path()
[  165s]     ~~~~~~~~~~~~~~~~~~~~~~~^^
[  165s]   File "/home/abuild/rpmbuild/BUILD/fwupd-2.0.6-build/fwupd-2.0.6/data/tests/fwupd_test.py", line 27, in override_gi_search_path
[  165s]     gi.require_version("GIRepository", "2.0")
[  165s]     ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
[  165s]   File "/usr/lib64/python3.13/site-packages/gi/__init__.py", line 132, in require_version
[  165s]     raise ValueError('Namespace %s not available for version %s' %
[  165s]                      (namespace, version))
[  165s] ValueError: Namespace GIRepository not available for version 2.0
[  165s] 
[  165s] ----------------------------------------------------------------------
```

With Pygobject 3.52, we need GIRepository = 3.0, which has some minor API changes

There is a migration guide available at https://docs.gtk.org/girepository/migrating-gi.html

hughsie, 2025-03-11:
Isn't GIRepository now provided by Glib? At least on my system I have parallel installed GIRepository versions, e.g.

	$ ls /usr/lib64/girepository-1.0/GIRepository-?.0.typelib -l
	-rw-r--r--. 1 root root 29424 Sep 16 01:00 /usr/lib64/girepository-1.0/GIRepository-2.0.typelib
	-rw-r--r--. 1 root root 24708 Jan 22 00:00 /usr/lib64/girepository-1.0/GIRepository-3.0.typelib

	$ rpm -qf /usr/lib64/girepository-1.0/GIRepository-?.0.typelib
	gobject-introspection-1.82.0-1.fc41.x86_64
	glib2-2.83.2-6.fc42.x86_64

DimStar77, 2025-03-11:
Correct - GIRepository now comes from glib

DimStar77, 2025-03-11:
some WIP patch (but does not yet work):

```
diff --git a/data/tests/fwupd_test.py b/data/tests/fwupd_test.py
index 4840fc52..e517c732 100755
--- a/data/tests/fwupd_test.py
+++ b/data/tests/fwupd_test.py
@@ -24,13 +24,21 @@ from gi.repository import UMockdev
 
 def override_gi_search_path():
     if "LIBFWUPD_BUILD_DIR" in os.environ:
-        gi.require_version("GIRepository", "2.0")
-        from gi.repository import GIRepository
+        try:
+            gi.require_version("GIRepository", "3.0")
+            from gi.repository import GIRepository
+            repo = GIRepository.Repository()
+        except ValueError:
+            gi.require_version("GIRepository", "2.0")
+            from gi.repository import GIRepository
 
-        GIRepository.Repository.prepend_search_path(
+            repo = GIRepository.Repository
+
+        repo.prepend_search_path(
             os.path.join(os.environ["LIBFWUPD_BUILD_DIR"])
         )
-        GIRepository.Repository.prepend_library_path(
+
+        repo.prepend_library_path(
             os.path.join(os.environ["LIBFWUPD_BUILD_DIR"])
         )
 ```

Still works on GIR 2.0 case - still fails in GIR 3.0 case :(

PR #8583 (2025-03-13): Fix build against pygobject 3.52
With the introduction of pygobject 3.52, accessing GIRepository 2.0 is
no longer permitted.

Instead of complicating the test code with conditionalizing between
GIRepository 2.0 and GIRepository 3.0 in override_gi_search_path,
we can drop override_gi_search_path entirely and rely on GIRepository's
own envvar GI_TYPELIB_PATH to help it find our freshly built typelib

Fixes #8569

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [ ] Feature
- [ ] Documentatio [... cut]
commit 73f46ad6 (2025-03-13): Fix build against pygobject 3.52

With the introduction of pygobject 3.52, accessing GIRepository 2.0 is
no longer permitted.

Instead of complicating the test code with conditionalizing between
GIRepository 2.0 and GIRepository 3.0 in override_gi_search_path,
we can drop override_gi_search_path entirely and rely on GIRepository's
own envvar GI_TYPELIB_PATH to help it find our freshly built typelib

Fixes #8569
closed 2025-03-13 by hughsie