fwupd/fwupd #8577: fwupd not running on reboot (another shim problem?)

tmo1, 2025-03-12:
When trying to apply "ZBook Fury G8 Mobile Workstation PC T95 System Update" (LVFS: 0x01120000 -> 0x01140000), fwupd reports that a reboot is required, but it doesn't run on reboot as long as Secure Boot is enabled on the machine. Disabling Secure Boot allows it to run.

```
~# fwupdmgr --version 
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.6
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.kernel                    6.12.17-amd64
runtime   org.freedesktop.fwupd         2.0.6

~# fwupdtool get-report-metadata
Loading…                 [********************                   ]14:14:22.369 FuPluginUefiCapsule  failed to get screen size: failed to setup: failed to get DRM resources
Waiting…                 [*************************************  ]14:14:22.604 FuDeviceLocker       failed to close device: cannot close device, refcount already zero

BatteryLevel:           99
BatteryThreshold:       50
BootTime:               1741786657
CompileVersion(com.hughsie.libjcat): 0.2.0
CompileVersion(com.hughsie.libxmlb): 0.3.21
CompileVersion(info.libusb): 1.0.27
CompileVersion(org.freedesktop.fwupd): 2.0.6
CpuArchitecture:        x86_64
CpuModel:               Intel 11th Gen Intel Core™ i7-11850H @ 2.50GHz
DisplayState:           connected
DistroId:               debian
DistroName:             Debian GNU/Linux
DistroPrettyName:       Debian GNU/Linux trixie/sid
FwupdSupported:         True
HostBaseboardManufacturer: HP
HostBaseboardProduct:   8870
HostBiosMajorRelease:   14
HostBiosMinorRelease:   00
HostBiosVendor:         HP
HostBiosVersion:        T95 Ver. 01.20.00
HostEnclosureKind:      a
HostFamily:             103C_5336AN HP ZBook
HostFirmwareMajorRelease: 52
HostFirmwareMinorRelease: 41
HostProduct:            HP ZBook Fury 15.6 inch G8 Mobile Workstation PC
HostSku:                5D482UC#ABA
HostVendor:             HP
KernelName:             Linux
KernelVersion:          6.12.17-amd64
LidState:               open
PlatformArchitecture:   x86_64
PowerState:             ac-fully-charged
RuntimeVersion(com.hughsie.libjcat): 0.2.0
RuntimeVersion(com.hughsie.libxmlb): 0.3.21
RuntimeVersion(org.freedesktop.fwupd): 2.0.6
RuntimeVersion(org.freedesktop.fwupd-efi): 1.7
RuntimeVersion(org.kernel): 6.12.17-amd64
DeviceId:               18012ed77a0e72e3f0df6904b7d6085d890f162f
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x1140000
DeviceId:               b8e2f4eb5953baa0de8e6075d2ecefb92cf23616
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
[... 198 lines cut ...]
│     Update State:       Success
│     GUID:               ca1146a5-1403-4385-9661-619f4e9eefbb
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            HP Inc. PK 2016 CA
      Vendor:             HP Inc.
      GUID:               77c6cc94-a10e-5208-916d-2a805c661547 ← UEFI\CRT_9FD5D4B39F4374A3C6CAA212D4007AB905C58591
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
                          • Can tag for emulation
    
────────────────────────────────────────────────
Devices that have been updated successfully:
 • UEFI dbx (20230501 → 20241101)
 • System Firmware (0x01120000 → 0x01140000)

```
fwupd installed via apt/aptitude. I got the idea to disable Secure Boot from [the wiki](https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot), but the issues / PRs linked there are years old and seemingly resolved. Is there a current problem with shim / fwupd and Secure Boot?

hughsie, 2025-03-12:
> fwupd installed via apt/aptitude

On what distro?

tmo1, 2025-03-12:
Debian Sid - sorry.

hughsie, 2025-03-12:
We don't do much testing on Debian at all I'm afraid. Would it be possible to test the firmware update again with Ubuntu, Fedora or Arch to rule out a Debian-specific problem?

hughsie, 2025-03-12:
You can reinstall the current version quite safely; the same mechanism runs as an update. Unfortunately it does need to be bare metal, and if it helps I have multiple NVMe drives around with various Linux distros and Windows versions :/

tmo1, 2025-03-18:
> Unfortunately it does need to be bare metal,

Will fwupd run correctly from something like a live instance of Ubuntu, or does it need a writable filesystem?

xuanruiqi, 2025-03-23:
I am experiencing the same issue on Arch.

```
xuanrui@xuanrui:~ % fwupdmgr --version           
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.6
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.9
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.0
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.Passim        0.1.9
runtime   org.kernel                    6.13.7-1-cachyos
runtime   org.freedesktop.fwupd         2.0.6
xuanrui@xuanrui:~ % sudo fwupdtool get-report-metadata
Waiting…                 [*************************************  ]
BatteryLevel:           100
BatteryThreshold:       10
BootTime:               1742734062
CompileVersion(com.hughsie.libjcat): 0.2.2
CompileVersion(com.hughsie.libxmlb): 0.3.21
CompileVersion(info.libusb): 1.0.27
CompileVersion(org.freedesktop.Passim): 0.1.9
CompileVersion(org.freedesktop.fwupd): 2.0.6
CpuArchitecture:        x86_64
CpuModel:               Intel 11th Gen Intel Core™ i7-11800H @ 2.30GHz
DisplayState:           connected
DistroId:               arch
DistroName:             Arch Linux
DistroPrettyName:       Arch Linux
FwupdSupported:         True
HostBaseboardManufacturer: Dell Inc.
HostBaseboardProduct:   1V4T3
HostBiosMajorRelease:   01
HostBiosMinorRelease:   23
HostBiosVendor:         Dell Inc.
HostBiosVersion:        1.35.0
HostEnclosureKind:      a
HostFamily:             XPS
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            XPS 15 9510
HostSku:                A61
HostVendor:             Dell Inc.
KernelCmdline:          lsm=lockdown,landlock,yama,safesetid,apparmor,bpf,smack nowatchdog nvidia-drm.modeset=1 rd.udev.log_level=3 psi=1 vt.global_cursor_default=0 nvidia-drm.fbdev=1
KernelName:             Linux
KernelVersion:          6.13.7-1-cachyos
LidState:               open
PassimDownloadSaving:   9638109
PlatformArchitecture:   x86_64
PowerState:             ac-fully-charged
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.6
RuntimeVersion(org.freedesktop.fwupd-efi): 1.0
RuntimeVersion(org.kernel): 6.13.7-1-cachyos
DeviceId:               5d7db2206ec1bc7aed22f9ecc219438e13b944cb
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x0
DeviceId:               ef9c68af09e25eb2f3db75d9da05ef78beb1409d
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x0
DeviceId:               b0094b2e160d33b6abf1a83c357e06dcdcd3ba0a
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x0
DeviceId:               362301da643102b9f38477387e2193e57abaa590
  pre:
      EfivarsNvramUsed: 107645
      LinuxLockdown:    none
      Pcr0_SHA256:      0ed9abc1667cc75aede04e1faf30a7ac247d0c6b07f78b8957e4985a9fe48602
      TpmEventLog:      0x00000003 0000000000000000000000000000000000000000000000000000000000000000 [U3RhcnR1cExvY2FsaXR5AAM=]
      0x00000007 22e2fa6a9ffc81497801319df6b98e2dfd8ba8e18e790bb73f128bb3ee28e22d [Qm9vdCBHdWFyZCBNZWFzdXJlZCBTLUNSVE0A]
      0x00000008 96a296d224f285c67bee93c30f8a309157f0daa35dc5b87e410b78630a09cfc7 [AAA=]
      0x80000008 39d2b2762c7fc6783b48abe552b5ed5c3cf69e2935dce185b3025332ee0ffe1f [AADS/wAAAAAAAAgAAAAAAA==]
      0x80000008 eaa7349b969385028ef75e611a80ed5bfaec8c621bbde375c662afe2211682e6 [AAAc/wAAAAAAAC4AAAAAAA==]
      0x80000008 431be0a033de6155298bf7280554e16df6adf3b83c69cf00e20a7fa63cd825cf [AABe/wAAAAAAAA4AAAAAAA==]
      0x80000008 b426cd6ef279cdf0b1587e9849a855162dda7a96f00394077a06c28d46d725df [AABX/wAAAAAAAAcAAAAAAA==]
      0x80000008 c7f8c3a88b27eab7d3b1c767ab21f3a47aceaf5518114037f97ecc3c6a8cac30 [AABK/wAAAAAAAA0AAAAAAA==]
      0x80000008 374e08d4a28ba9bd8122a2dbb24bc6d64d1592af3d27f1490f7ce4eac15a0298 [AADL/wAAAAAAAAcAAAAAAA==]
      0x80000008 ab131abcf90735bfe570a74fcde198363c69b9eccce591c037e949dca63cf602 [AAB8/wAAAAAAAAQAAAAAAA==]
      0x80000008 5e74510edbaaba80d1483e05ec8205446e18bf7ee8609b6e1cba26bc3ee7570a [AAAU/wAAAAAAAAIAAAAAAA==]
      0x80000008 70b551b6f9f4655c200ca3947c53e92dda40e16fd56c027e2493076a8d52261c [AABs/wAAAAAAABAAAAAAAA==]
      0x80000008 af65ea05adcc4a3ddb6d1a620b5daa8bba04b9c7c4f02776fb6bf0dd7965144c [AADj/wAAAAAAAAYAAAAAAA==]
      0x00000001 6dae428960b60a24cf6af581f8ec9b2d047d0aa546c87ea2ef85ab429b539c63 [AABdWwAAAAAAAEMAAAAAAA==]
      0x00000011 b66aba2749be1c31114c33c6cdc363a70f15ba23837acc6f325d8b5f15aedda4 [SW50ZWxDU3hFSW5mb0V2ZW50AAABAAAAhoDgQwEAAAA=]
      0x00000004 df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119 [AAAAAA==]
      PCR0: 0ed9abc1667cc75aede04e1faf30a7ac247d0c6b07f78b8957e4985a9fe48602
      TpmFamily:        2.0
      EfivarsNvramUsed: 107645
      SecureBoot:       Enabled
      UEFIUXCapsule:    Enabled
```

danielcb, 2025-04-02:
Same issue on fedora rawhide. The boot entry created by fwupdmgr just contains shimx64.efi. Shouldn't there be a reference to fwupdx64.efi somewhere in there?

`Boot0001* Linux Firmware Updater        HD(1,GPT,141658f1-f646-4364-98a9-9b841272434f,0x800,0x12c000)/\EFI\fedora\shimx64.efi `

yann-soubeyrand, 2025-04-04:
@danielcb where do you see the boot entry? On my system, the `grub.cfg` file is static and the boot entries are supplied via BLS (files in `/boot/loader/entries`), but I do not see a boot entry for for `fwupdmgr` there.

danielcb, 2025-04-04:
I'm currently using fedora with secure boot enabled. I list them via `efibootmgr`. The entry is created by `fwupdmgr upgrade` (in case there actually is an upgrade).

yann-soubeyrand, 2025-04-04:
Ha yes, thanks, I see the same:
```
Boot0001* Linux-Firmware-Updater	HD(1,GPT,6a16fcc8-e75b-473a-b748-06aefa06d816,0x800,0x64000)/\EFI\fedora\shimx64.efi
```

Taipan67, 2025-04-04:
Same issue on openSUSE Tumbleweed, and fwupd just updated to 2.0.7 a couple of hours ago. It's downloading the update-files it needs to EFI\opensuse\fw\ but then rebooting straight back into Linux via Grub when I confirm I want to complete the update after said reboot.

I too noticed the EFI boot entry for firmware updates pointing to shim.efi the same as the Linux entry - is fwupdmgr supposed to use efibootmgr to send a 'bootnext' instruction along with some additional arguments to shim so SecureBoot is honoured? Is it worth deleting that menu entry to force the creation of a new one? I don't really want to have to disable SecureBoot unless I have no option.

On my Dell Vostro 3520, it looks like I've previously done 2 updates successfully in October and December of last year, it's just this latest one from February that's not worked yet...

hughsie, 2025-04-04:
I think `sudo fwupdtool efiboot-info` might be more useful -- but there's certainly something weird going on if the fwupdx64.efi path isn't being appended. Test PR incoming!

hughsie, 2025-04-04:
Can anyone affected please test https://github.com/fwupd/fwupd/pull/8651 and see if that fixes the issue? If so, I'll tag a new release next week.

yann-soubeyrand, 2025-04-07:
> Can anyone affected please test [#8651](https://github.com/fwupd/fwupd/pull/8651) and see if that fixes the issue? If so, I'll tag a new release next week.

Hello @hughsie, I tried to install fwupd, fwupd-plugin-uefi-capsule-data, fwupd-plugin-flashrom and fwupd-plugin-modem-manager on my Fedora Silverblue 42, but I get the following errors in the fwupd.service logs after reboot:
```
avril 07 09:45:23 wrk117 systemd[1]: Starting fwupd.service - Firmware update daemon...
avril 07 09:45:23 wrk117 fwupd[5739]: Failed to load daemon: syscall filtering is configured but not working
avril 07 09:45:23 wrk117 fwupd[5739]: profiling:/root/rpmbuild:Cannot create directory
avril 07 09:45:23 wrk117 fwupd[5739]: profiling:/root/rpmbuild/BUILD/fwupd-2.0.8-build/fwupd-2.0.8/redhat-linux-build/src/fwupd.p/meson-generated_.._fu-engine-struct.c.gcda:Skip
avril 07 09:45:23 wrk117 fwupd[5739]: profiling:/root/rpmbuild:Cannot create directory
avril 07 09:45:23 wrk117 fwupd[5739]: profiling:/root/rpmbuild/BUILD/fwupd-2.0.8-build/fwupd-2.0.8/redhat-linux-build/src/fwupd.p/meson-generated_.._fu-daemon-struct.c.gcda:Skip
avril 07 09:45:23 wrk117 fwupd[5739]: profiling:/root/rpmbuild:Cannot create directory
…
```

closed 2025-04-07 by hughsie

hughsie, 2025-04-07:
@yann-soubeyrand I've built a package in https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/ -- can you try that?

danielcb, 2025-04-07:
Just tested it. Before:
`Boot0001* Linux Firmware Updater        HD(1,GPT,141658f1-f646-4364-98a9-9b841272434f,0x800,0x12c000)/\EFI\fedora\shimx64.efi`

After:
`Boot0001* Linux Firmware Updater        HD(1,GPT,141658f1-f646-4364-98a9-9b841272434f,0x800,0x12c000)/\EFI\fedora\shimx64.efi5c00660077007500700064007800360034002e006500660069000000`

Looks like it's working again. Thanks! :)

yann-soubeyrand, 2025-04-07:
> [@yann-soubeyrand](https://github.com/yann-soubeyrand) I've built a package in https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/ -- can you try that?

@hughsie I confirm that I was able to install my two pending firmware updates using your package. Thanks!

tmo1, 2025-04-07:
@hughsie Thank you! I'll be happy to test as soon as I can get my hands on a deb package.

superm1, 2025-04-07:
> @hughsie Thank you! I'll be happy to test as soon as I can get my hands on a deb package.

There are artifacts on GitHub actions for the deb build you might be able to try. They're built against Debian unstable.

Taipan67, 2025-04-20:
The update to 2.0.8 has resolved the issue I posted about in openSUSE Tumbleweed - many thanks.

dionorgua, 2025-04-22:
I can confirm that it works now on Debian Unstable. Just got new BIOS update from Dell and it was applied as needed