fwupd/fwupd #8578: 2.0.6 - Both USB-C ports stuck in TBT mode, no device attached. No charging, no data

rishabh-chromium, 2025-03-12:
**Describe the bug**
```
screebo-rev4 ~ # ectool usbpdmuxinfo
Port 0: USB=0 DP=0 POLARITY=NORMAL HPD_IRQ=0 HPD_LVL=0 SAFE=0 TBT=1 USB4=0 
Port 1: USB=0 DP=0 POLARITY=NORMAL HPD_IRQ=0 HPD_LVL=0 SAFE=0 TBT=1 USB4=0 
```
Seems to be in "No device attached" mode

NDA (no device attach) configuration shows up in /sys/bus/thunderbolt/devices.

```
screebo-rev4 /sys/bus/thunderbolt/devices # ls
0-0  0-0:3.1  1-0  1-0:3.1  domain0  domain1
Device nodes 0-0:3.1 and 1-0:3.1 represent the two Hayden Bridge retimers on the USB-C ports.
```

They don't usually show up when there's nothing attached to the ports. The fact that they do right now when nothing is plugged in points to fwupd potentially trying to probe for the retimers and getting them stuck in this mode.

I watched the EC console while this was happening by the way. This is it disabling C1 while enabling retimer power.

```
25-03-12 02:47:24.219 [27.962200 C1: Disabled]
25-03-12 02:47:24.222 [28.016200 C1: Retimer I2C read err=1]
25-03-12 02:47:24.276 [28.059700 C1: retimer power ena![28.062600 event clear 0x0000000008000000]
25-03-12 02:47:24.324 ![28.065900 ACPI query = 28]
25-03-12 02:47:24.325 ble success]
25-03-12 02:47:24.326 [28.061100 event set 0x0000000008000000]
25-03-12 02:47:24.329 [28.134800 event set 0x000000000![28.137200 event clear 0x0000000008000000]
25-03-12 02:47:24.395 ![28.140900 ACPI query = 28]
25-03-12 02:47:24.398 8000000]
25-03-12 02:47:24.398 [28.190400 event![28.191800 event clear 0x0000000008000000]
25-03-12 02:47:24.451 ![28.195200 ACPI query = 28]
25-03-12 02:47:24.454  set 0x0000000008000000]
25-03-12 02:47:24.454 [31.998000 Battery 77% (Display 77.0 %) / 2h:19 to empty]
```

**Steps to Reproduce**
Flash and initialize fwupd 2.0.6

**Expected behavior**
Ports work

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdtool get-report-metadata
```
fwupd-2.0.6

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version: chromeos
- Have you tried rebooting? yes
- Is this a regression? yes

rishabh-chromium, 2025-03-12:
Only affecting boards with Intel Hayden Bridge USB4 retimer.
I am crawling through commits to find the culprit

hughsie, 2025-03-12:
I'd bet 2dc288e4ca81b0032461a886bcee929f5b1a3758 personally. I asked Intel for some Screebo/Rex hardware months ago but so far nothing.

bleungatchromium, 2025-03-12:
Thanks Richard for taking a look at this. I found the issue yesterday, and good that we caught it pretty early after our uprev here.

rishabh-chromium, 2025-03-12:
Yes, I verified that commit is responsible. I have reverted that commit on the ChromeOS branch. I will wait for you land a fix whenever you have some time. I will also try to find out if we can get a Screebo device to you

hughsie, 2025-03-12:
I'm confused why it's enabled, it's allowlisted on the Screebo/Rex guids. Did intel reuse the guid on a different model?

bleungatchromium, 2025-03-12:
Hi @hughsie,

The GUIDs you've got in the thunderbolt.quirks file, where did they come from, and is there a quick way for me to check that i'm getting a match on a Screebo system I have here?

hughsie, 2025-03-13:
> where did they come from

It's a HWID -- so if you do `sudo fwupdtool hwids` then one of those must be matching.

bleungatchromium, 2025-08-29:
Hi @hughsie : This issue wasn't tagged in the PR https://github.com/fwupd/fwupd/pull/9171, but your change fixed this for me on ChromeOS when I pulled it.

closed 2025-08-29 by hughsie

bleungatchromium, 2025-10-22:
I found that this wasn't fixed, and dug into it. I figured out what's going on. Short summary,  `fu_thunderbolt_controller_setup_usb4` sets a 5 second timeout to defer the onlining step, but the whole fu-thunderbolt-controller finalizes almost immediately, and we remove the timer and never let it fire.

bleungatchromium, 2025-10-23:
Fixed with #9438. Thanks all.