fwupd/fwupd #8598: Updating EM120R GL fails "no firehose port provided for filename"

neilzone, 2025-03-19:
**Describe the bug**
Updating the firmware on a EM120R GL from EM120RGLAPR02A07M4G_20.020.20.020 to EM120RGLAPR02A07M4G_25.025.25.025 fails with

```
failed to write-firmware: no firehose port provided for filename
```

**Steps to Reproduce**
I ran `fwupdmgr update` on my ThinkPad X1 Yoga Gen 6 running Debian testing.

It showed me a firmware update:

```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade EM120R GL from EM120RGLAPR02A07M4G_20.020.20.020
 to                ║
║ EM120RGLAPR02A07M4G_25.025.25.025?                                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Firmware release EM120RGLAPR02A07M4G_25.025.25.025 packaged for LVFS.        ║
║                                                                              ║
║ • Performance improvements                                                   ║
║ • Stability improvements                                                     ║
║                                                                              ║
║ EM120R GL and all connected devices may not be usable while updating.        ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y

```

It download and (seemingly) wrote, but failed:

```
Successfully installed firmware
Successfully uploaded report
failed to write-firmware: no firehose port provided for filename

```

**Expected behavior**
Firmware to update.

<details>
**fwupd version information**

```
fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.6
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.kernel                    6.12.17-amd64
runtime   org.freedesktop.fwupd         2.0.6
```

fwupdtool get-report-metadata: https://pad.neilzone.co.uk/p/qCGQLEBuFp1PiM1NZ-ma

fwupdmgr get-devices --show-all-devices: https://pad.neilzone.co.uk/p/qCGQLEBuFp1PiM1NZ-ma

</details>

neilzone, 2025-03-19:
> I'm actually in the process of ripping apart and glueing back together the modem support -- see [#8551](https://github.com/fwupd/fwupd/pull/8551) for progress so far.

I'm so sorry - I had missed that you were already working on this. Thanks for all you do with fwupdmgr and lvfs.

jan-sarao, 2025-06-18:
I get the same for on a Thinkpad X1 Carbon Gen 12

```
$sudo fwupdmgr update
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade EM160R GL from EM160RGLAPR02A10M4G_01.206.01.206 to                  ║
║ EM160RGLAPR02A10M4G_01.212.01.212?                                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Firmware release EM160RGLAPR02A10M4G_01.212.01.212 packaged for LVFS.        ║
║                                                                              ║
║ • Performance improvements                                                   ║
║ • Stability improvements                                                     ║
║                                                                              ║
║ EM160R GL and all connected devices may not be usable while updating.        ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Writing…                 [****                                   ] Less than one minute remaining…
failed to write-firmware: no firehose port provided for filename
```

neilzone, 2025-06-18:
> I get the same for on a Thinkpad X1 Carbon Gen 12

My understanding - could be wrong - is that this might be fixed in fwupd 2.0.9.

jan-sarao, 2025-06-18:
> > I get the same for on a Thinkpad X1 Carbon Gen 12
> 
> My understanding - could be wrong - is that this might be fixed in fwupd 2.0.9.

Thank you. I am running Debian 13 with fwupd 2.0.8 but got fwupd (and libfwupd) 2.0.10 from Debian's Experimental repository. The error changed to:
```
$sudo fwupdmgr update                                                                                         
╔══════════════════════════════════════════════════════════════════════════════╗                                                                   
║ Upgrade EM160R GL from EM160RGLAPR02A10M4G_01.206.01.206 to                  ║                                                                   
║ EM160RGLAPR02A10M4G_01.212.01.212?                                           ║                                                                   
╠══════════════════════════════════════════════════════════════════════════════╣                                                                   
║ Firmware release EM160RGLAPR02A10M4G_01.212.01.212 packaged for LVFS.        ║                                                                   
║                                                                              ║                                                                   
║ • Performance improvements                                                   ║                                                                   
║ • Stability improvements                                                     ║                                                                   
║                                                                              ║                                                                   
║ EM160R GL and all connected devices may not be usable while updating.        ║                                                                   
╚══════════════════════════════════════════════════════════════════════════════╝                                                                   
Perform operation? [Y|n]: y                                                                                                                        
Restarting device…       [*************************************  ] Less than one minute remaining…                                                 
                                                                                                                                                   
                                                                                                                                                   
failed to attach: failed to wait for attach replug: device 8e92eda1ab953dd9f27d11cd320c36e5d1127030 did not come back
```

hughsie, 2025-11-12:
@neilzone do you have more luck with a recent 2.0.x version? Thanks.

neilzone, 2025-11-12:
I have tried to use the flatpak version, to give me 2.0.17, but, while I can install the flaktpak, I can't work out how to use it, I'm afraid!