fwupd/fwupd #8605: `ERROR: Unknown options: "launchd, plugin_powerd"`

Makonede, 2025-03-21:
**Describe the bug**
Installation fails with this message:

```
fwupd-git/meson.build:1:0: ERROR: Unknown options: "launchd, plugin_powerd"

A full log can be found at /home/mako/.cache/paru/clone/fwupd-git/src/build/meson-logs/meson-log.txt
==> ERROR: A failure occurred in prepare().
    Aborting...
error: failed to build 'fwupd-git-1.9.10.r1196.gb8b45b190-1':
error: packages failed to build: fwupd-git-1.9.10.r1196.gb8b45b190-1
```

**Steps to Reproduce**
1. Attempt to install the package

**Expected behavior**
The package installs.

**fwupd version information**
`fwupd-git-1.9.10.r1196.gb8b45b190-1` from the AUR

```
BatteryLevel:           100
BatteryThreshold:       10
BootTime:               1742573518
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.21
CompileVersion(info.libusb): 1.0.27
CompileVersion(org.freedesktop.fwupd): 2.0.7
CompileVersion(org.freedesktop.fwupd.source): 2.0.6-58-g2edc0a3e2
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 9 7940HS w/ Radeon 780M Graphics
DisplayState:           connected
DistroId:               arch
DistroName:             Arch Linux
DistroPrettyName:       Arch Linux
FwupdSupported:         True
HostBaseboardManufacturer: Framework
HostBaseboardProduct:   FRANMZCP09
HostBiosMajorRelease:   03
HostBiosMinorRelease:   05
HostBiosVendor:         INSYDE Corp.
HostBiosVersion:        3.05
HostEnclosureKind:      a
HostFamily:             16in Laptop
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            Laptop 16 (AMD Ryzen 7040 Series)
HostSku:                FRAGAACP0J
HostVendor:             Framework
KernelName:             Linux
KernelVersion:          6.13.7-zen1-1-zen
LidState:               open
PlatformArchitecture:   x86_64
PowerState:             ac-fully-charged
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.7
RuntimeVersion(org.freedesktop.fwupd-efi): 1.7
RuntimeVersion(org.kernel): 6.13.7-zen1-1-zen
DeviceId:               7010981714ee5c9258de42216cae163ef272c7ab
  pre:
      CapsuleApplyMethod: cod
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /efi
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x305
[... 210 lines cut ...]
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│                         • Can tag for emulation
│     
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            frame.work-LaptopAMDPK
      GUID:               c701846d-5461-5c19-a7ae-2d89b2f44c14 ← UEFI\CRT_D610D8205624E644D9435383826B2DBF7298859C
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
                          • Can tag for emulation
```

</details>

**Additional questions**

- Operating system and version: Arch Linux `x86_64`, Zen Kernel `6.13.7.zen1-1`
- Have you tried rebooting? Yes
- Is this a regression? Yes

superm1, 2025-03-21:
Intended behavior.  Those were just dropped recently.

https://github.com/fwupd/fwupd/commit/a5d548931f75c359b3df52221f825c8c36853d22
https://github.com/fwupd/fwupd/commit/1f7032d712b27d91983fb216b9fd64a767636bd3

Please update your AUR repo.

superm1, 2025-03-21:
and context: https://github.com/fwupd/fwupd/pull/8591

closed 2025-03-21 by superm1