fwupd/fwupd #8608: Getting "failed to open device for prepare: failed to subclass open: no proxy", then fwupd gets stuck in "Firmware Update in Progress" state

eliasrudberg, 2025-03-23:
Getting this problem on a XPS 13 9340. Running Debian trixie, with fwupd installed using apt.

```
# fwupdmgr update
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade WD19S from 01.01.00.07 to 01.01.00.10?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ -Fixed the issue where the system automatically shuts down when you plug or  ║
║ unplug the dock to the system.                                               ║
║                                                                              ║
║ -Fixed the issue where the USB device and Preboot Execution Environment      ║
║ (PXE) boot option are not displayed. This issue occurs when you restart or   ║
║ power on the system and press the F12 key to enter the BIOS Setup or Boot    ║
║ Menu.                                                                        ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Waiting…                 [***************************************] Less than one minute remaining…
Successfully installed firmware
failed to open device for prepare: failed to subclass open: no proxy
#
```

Several problems:

First the issue that the error message is about, "failed to open device for prepare: failed to subclass open: no proxy". How to solve that, or how to troubleshoot further?

Then I think it's also a problem that it said "Successfully installed firmware" when in fact it seems the firmware was not installed, because if I reboot and run the same "fwupdmgr update" command again, it still asks to install the same update. So it seems like the "Successfully installed firmware" message is printed too early, it is printed before the installation is complete.

Then when trying to reboot afterwards, that is inhibited with the message "Firmware Update in Progress" which seems misleading because nothing is actually in progress, it does not matter how long I wait:
```
# /sbin/reboot
Operation inhibited by "fwupd" (PID 3556 "fwupd", user root), reason is "Firmware Update in Progress".
User elias is logged in on tty2.
Please retry operation after closing inhibitors and logging out other users.
'systemd-inhibit' can be used to list active inhibitors.
Alternatively, ignore inhibitors and users with 'systemctl reboot -i'.
#
```

The reboot command only works after I kill the fwupd process.

It seems like when the "failed to open device" error happens, fwupd gets stuck in some bad state where it never clears the "Firmware Update in Progress" information.

This is all reproducible for me, everything happens the same way if I reboot and try again. So if anyone has ideas on how to get more relevant troubleshooting information, please let me know, I could do more tests.

**fwupd version information**

```
# apt list fwupd
fwupd/testing,now 2.0.6-7 amd64 [installed,automatic]

# fwupdmgr --version
compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.6
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libjcat           0.2.3
runtime   com.hughsie.libxmlb           0.3.22
runtime   org.kernel                    6.12.19-amd64
runtime   org.freedesktop.fwupd         2.0.6

# fwupdtool get-report-metadata
Loading…                 [********************                   ]09:28:01.278 FuBackend            failed to create device from /sys/devices/platform/intel_vsc/mei/mei1: failed to probe: no parent with subsystem pci
Waiting…                 [*************************************  ]
failed to subclass open: no proxy
```

superm1, 2025-03-25:
@CragW can you please look?

CragW, 2025-03-25:
This issue is similar to #8368 that will impact WD19/WD22 docks across fwupd 2.0.4, 2.0.5 and 2.0.6.   The next stable release 2.0.7 should have #8481 included to address this bug.

klemmster, 2025-03-27:
Just writing to confirm that `2.0.7` indeed fixed the bug for me.

parkerlreed, 2025-04-25:
Anyone had luck on 2.0.8?

```
Perform operation? [Y|n]: 
Erasing…                 [                                       ]
WD19S is pending activation; use fwupdmgr activate to complete the update.
Successfully uploaded report
failed to write-firmware: failed to erase bank: failed after 5 retries: failed to SetReport: USB error: Operation timed out [-7]
[parker@parker-framework ~]$ sudo fwupdmgr activate
No firmware to activate
[parker@parker-framework ~]$ pacman -Q fwupd
fwupd 2.0.8-1
```

alumni, 2025-07-06:
> This issue is similar to [#8368](https://github.com/fwupd/fwupd/issues/8368) that will impact WD19/WD22 docks across fwupd 2.0.4, 2.0.5 and 2.0.6. The next stable release 2.0.7 should have [#8481](https://github.com/fwupd/fwupd/pull/8481) included to address this bug.

I stared encountering this bug in 2.0.12:
```
$ sudo fwupdmgr update 4c495025a00c4d621540d9494bbc78853570a1d0
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade WD19TB from 01.01.00.10 to 01.01.00.14?                              ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This package contains the firmware update for Dell Dock WD19, WD22, HD22,    ║
║ WD25, and SD25 series. The dock firmware update package contains utility     ║
║ and firmware for updating the dock components.                               ║
║                                                                              ║
║ • Improved the stability of the system.                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall RTS5413 in Dell dock to 01.23?                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This package contains the firmware update for Dell Dock WD19, WD22, HD22,    ║
║ WD25, and SD25 series. The dock firmware update package contains utility     ║
║ and firmware for updating the dock components.                               ║
║                                                                              ║
║ • Improved the stability of the system.                                      ║
║                                                                              ║
║ RTS5413 in Dell dock must remain connected for the duration of the update    ║
║ to avoid damage.                                                             ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Waiting…                 [***************************************]
failed to open device for prepare: failed to subclass open: no proxy
```

CragW, 2025-07-07:
An improvement 28d686212 was added a few days ago that addresses some conditions like this.  Can you try a build with the mentioned fix?

CoryMnvl, 2025-07-26:
I have a Dell WD19TB and just tried updating its firmware with fwupd 2.0.13 and am still receiving the `failed to open device for prepare: failed to subclass open: no proxy` message.
If I'm reading the commit correctly it should be included in version 2.0.13.

CoryMnvl, 2025-07-27:
I'm fairly certain my dock is faulty (I will be returning it in the next few days), but there still is a discrepancy between what I was able to flash between Windows 10 and Linux on the same PC. 
- Under Windows I was unable to flash the Thunderbolt (WD19 TBT Dock NVM) and MST firmware, everything else I was able to downgrade and upgrade as I pleased. 
- Under Linux I kept getting the `failed to open device for prepare: failed to subclass open: no proxy` error. Both when the dock was connected to the thunderbolt 4 capable ports and the USB 3.2 Gen 1 only port. 

I confirmed that Thunderbolt was enabled and as unrestricted as possible in the BIOS, but the device never appeared in boltctl, nor did the Thunderbolt Control Center in Windows list the device. This is one of the problems that lead me to believe it is faulty.