fwupd/fwupd #8609: Search multiple devices for capsule updates

Geblaat, 2025-03-23:
**Is your feature request related to a problem? Please describe.**
1. Use fwupd from a Fedora live ISO on a USB thumb drive. (In this case fwupd 1.9.29).
2. Apply updates
3. Reboot
4. Following message shows up
```
fwupd-efi version 1.6
WARNING: Could not apply capsule update: Not found
WARNING: Could not apply capsules: Not found
Reset System
```
The process assumes the capsules are on the thumb drive and if it does not find them it aborts.
User needs to reboot into the USB thumbdrive, redo the updates, reboot and remove the USB thumb drive on time.

**Describe the solution you'd like**
If capsules are not found on the thumb, check the next device.

**Describe alternatives you've considered**
Not applicable

**Additional context**
The default behviour is unclear, leading users to file bug reports:
https://github.com/fwupd/fwupd/issues/989#issuecomment-2567009988
https://github.com/fwupd/firmware-lenovo/issues/257#issuecomment-2258857591
(There may be more, found those 2 just after a very quick search.

superm1, 2025-03-23:
Can you please explain what exactly happens today in your situation better? 

FYI - This is my expectation for what happens.

1. You boot a live key
2. You run an update that stages a capsule
3. The capsule is copied to an (existing) ESP on the disk.
4. The bootloader is copied to an (existing) ESP on the disk.
5. A boot entry is added to point to that newly copied bootloader.

Is that not what happens?  If so; please show us `efibootmgr -v` output after it's staged but before you've rebooted as well as the `tree` output from all potential ESPs on the system.

Geblaat, 2025-03-30:
@superm1 
Yes, it seems to me that is what happens. When I reboot it starts fwupd-efi v1.6, so everything seems succesful. However, it only works if I remove the live key during reboot. If I don't I get the error message at .4 in the OP. 
So it seems to be that the bootloader boots succesfully from the internal disk, but then tries to find the actual firmware update files on the live key instead of the internal disk. And because they're not on the live key, it aborts with an error instead of searching the internal disk as well.

Commands after staging update:
```
efibootmgr -v
BootNext: 0006
BootCurrent: 001D
Timeout: 0 seconds
BootOrder: 0005,0004,0003,001D,0023,0001,0000,0010,0011,0012,0013,0014,0015,0016,0017,0002,001A,001B,001E,001F,0020,0021,001C,0006
Boot0000  ubuntu	HD(1,GPT,614719d9-3cfe-48d4-97e6-2adf02e28f1b,0x800,0x100000)/\EFI\ubuntu\shimx64.efi
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 00 10 00 00 00 00 00 d9 19 47 61 fe 3c d4 48 97 e6 2a df 02 e2 8f 1b 02 02 / 04 04 34 00 5c 00 45 00 46 00 49 00 5c 00 75 00 62 00 75 00 6e 00 74 00 75 00 5c 00 73 00 68 00 69 00 6d 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0001  Windows Boot Manager	HD(1,GPT,1e5e69a0-6c48-427c-abc3-ed353f2d7e7c,0x800,0x32000)/\EFI\Microsoft\Boot\bootmgfw.efi57494e444f5753000100000088000000780000004200430044004f0042004a004500430054003d007b00390064006500610038003600320063002d0035006300640064002d0034006500370030002d0061006300630031002d006600330032006200330034003400640034003700390035007d00000068000100000010000000040000007fff0400
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 20 03 00 00 00 00 00 a0 69 5e 1e 48 6c 7c 42 ab c3 ed 35 3f 2d 7e 7c 02 02 / 04 04 46 00 5c 00 45 00 46 00 49 00 5c 00 4d 00 69 00 63 00 72 00 6f 00 73 00 6f 00 66 00 74 00 5c 00 42 00 6f 00 6f 00 74 00 5c 00 62 00 6f 00 6f 00 74 00 6d 00 67 00 66 00 77 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
    data: 57 49 4e 44 4f 57 53 00 01 00 00 00 88 00 00 00 78 00 00 00 42 00 43 00 44 00 4f 00 42 00 4a 00 45 00 43 00 54 00 3d 00 7b 00 39 00 64 00 65 00 61 00 38 00 36 00 32 00 63 00 2d 00 35 00 63 00 64 00 64 00 2d 00 34 00 65 00 37 00 30 00 2d 00 61 00 63 00 63 00 31 00 2d 00 66 00 33 00 32 00 62 00 33 00 34 00 34 00 64 00 34 00 37 00 39 00 35 00 7d 00 00 00 68 00 01 00 00 00 10 00 00 00 04 00 00 00 7f ff 04 00
Boot0002* Qubes OS	HD(1,GPT,dc60a392-4c45-444b-a143-5eaae18c2df1,0x800,0x12c000)/\EFI\qubes\grubx64.efi
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 c0 12 00 00 00 00 00 92 a3 60 dc 45 4c 4b 44 a1 43 5e aa e1 8c 2d f1 02 02 / 04 04 32 00 5c 00 45 00 46 00 49 00 5c 00 71 00 75 00 62 00 65 00 73 00 5c 00 67 00 72 00 75 00 62 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0003  QUBES4.1	HD(1,GPT,86a75fba-5344-4b24-b84d-1bac40e8fa49,0x800,0x12c000)/\EFI\qubes\grubx64.efi
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 c0 12 00 00 00 00 00 ba 5f a7 86 44 53 24 4b b8 4d 1b ac 40 e8 fa 49 02 02 / 04 04 32 00 5c 00 45 00 46 00 49 00 5c 00 71 00 75 00 62 00 65 00 73 00 5c 00 67 00 72 00 75 00 62 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0004  Fedora	HD(2,GPT,c16424e0-ff65-4961-a244-5afa171d24f6,0x463120,0xf000)/\EFI\fedora\shimx64.efi
      dp: 04 01 2a 00 02 00 00 00 20 31 46 00 00 00 00 00 00 f0 00 00 00 00 00 00 e0 24 64 c1 65 ff 61 49 a2 44 5a fa 17 1d 24 f6 02 02 / 04 04 34 00 5c 00 45 00 46 00 49 00 5c 00 66 00 65 00 64 00 6f 00 72 00 61 00 5c 00 73 00 68 00 69 00 6d 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0005* Fedora	HD(2,GPT,c16424e0-ff65-4961-a244-5afa171d24f6,0x463120,0xf000)/\EFI\fedora\shimx64.efi
      dp: 04 01 2a 00 02 00 00 00 20 31 46 00 00 00 00 00 00 f0 00 00 00 00 00 00 e0 24 64 c1 65 ff 61 49 a2 44 5a fa 17 1d 24 f6 02 02 / 04 04 34 00 5c 00 45 00 46 00 49 00 5c 00 66 00 65 00 64 00 6f 00 72 00 61 00 5c 00 73 00 68 00 69 00 6d 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0006* Linux-Firmware-Updater	HD(1,GPT,dc60a392-4c45-444b-a143-5eaae18c2df1,0x800,0x12c000)/\EFI\fedora\fwupdx64.efi
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 c0 12 00 00 00 00 00 92 a3 60 dc 45 4c 4b 44 a1 43 5e aa e1 8c 2d f1 02 02 / 04 04 36 00 5c 00 45 00 46 00 49 00 5c 00 66 00 65 00 64 00 6f 00 72 00 61 00 5c 00 66 00 77 00 75 00 70 00 64 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
      dp: 04 06 14 00 66 8b 1c 72 6c 42 86 4e 8e 99 34 57 c4 6a b0 b9 / 7f ff 04 00
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
      dp: 04 06 14 00 2d 76 6a 12 58 57 ca 4f 85 31 20 1a 7f 57 f8 50 / 7f ff 04 00
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
      dp: 04 06 14 00 a6 d9 d8 a7 b0 6a eb 4a ad 9d 16 3e 59 a7 a3 80 / 7f ff 04 00
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
      dp: 04 06 14 00 5b 61 7e 3f 45 0d 80 4f 88 dc 26 b2 34 95 85 60 / 7f ff 04 00
Boot0014  Asset Information	FvFile(da465b87-a26f-4c12-b78a-0361428fa026)
      dp: 04 06 14 00 87 5b 46 da 6f a2 12 4c b7 8a 03 61 42 8f a0 26 / 7f ff 04 00
Boot0015  Regulatory Information	FvFile(e4a83242-deee-f12e-15ff-0102036cc3ce)
      dp: 04 06 14 00 42 32 a8 e4 ee de 2e f1 15 ff 01 02 03 6c c3 ce / 7f ff 04 00
Boot0016  ThinkShield secure wipe	FvFile(3593a0d5-bd52-43a0-808e-cbff5ece2477)
[... cut ...]
│       ├── f348dfd1c88e4a589d3547dabc40f64b-0-rescue.conf
│       └── f348dfd1c88e4a589d3547dabc40f64b-6.13.8-200.fc41.x86_64.conf
├── symvers-6.11.4-301.fc41.x86_64.xz
├── symvers-6.13.8-200.fc41.x86_64.xz
├── System.map-6.11.4-301.fc41.x86_64
├── System.map-6.13.8-200.fc41.x86_64
├── vmlinuz-0-rescue-f348dfd1c88e4a589d3547dabc40f64b
├── vmlinuz-6.11.4-301.fc41.x86_64
└── vmlinuz-6.13.8-200.fc41.x86_64

12 directories, 35 files

```
(Not sure how to do `tree` on the internal disk, I'm not that familiar with command line.)

superm1, 2025-03-31:
I'm actually not sure if there is anything we can do here.  Let me explain; the error code you saw came from the UEFI `UpdateCapsule` runtime service.  Here is where that comes from:

https://github.com/fwupd/fwupd-efi/blob/e2ced295211991500ef6f86347def94cfc9fd3ff/efi/fwupdate.c#L480

That is the _firmware_ is what failed and returned this error.

closed 2025-04-03 by superm1