fwupd/fwupd #8612: PopOS `/recovery` partition is flagged as ESP

superm1, 2025-03-24:
> Could the amended/extended ignore list get considered/implemented please? Both new and existing installs of PopOS are triggering this message, as they create a "Recovery" partition which is *not* an actual ESP, but it's being IDed as such by fwupdmgr and triggering the complaint. I do not want the ESP flag set on that partition, or it treated as one, as it doesn't have a complete and usable ESP layout.
> 
> 16:16:31.237 FuVolume             mounting /org/freedesktop/UDisks2/block_devices/nvme2n1p1
> 16:16:31.343 FuPluginUefiCapsule  not a Linux ESP: did not find grubx64.efi|shimx64.efi|systemd-bootx64.efi in /boot/efi
> 16:16:31.343 FuVolume             unmounting /org/freedesktop/UDisks2/block_devices/nvme2n1p1
> 16:16:31.378 FuPluginUefiCapsule  not a Linux ESP: did not find grubx64.efi|shimx64.efi|systemd-bootx64.efi in /recovery
> 16:16:31.378 FuPluginUefiCapsule  more than one ESP possible:
>  - 0xfff:	/org/freedesktop/UDisks2/block_devices/nvme2n1p4
>  - 0x800:	/org/freedesktop/UDisks2/block_devices/nvme2n1p1
> 16:16:31.389 FuDevice             using a45df35ac0e948ee180fe216a5f703f32dda163f for /sys/firmware/efi/esrt/entries/entry0
> 16:16:31.389 FuPlugin             emit added from uefi_capsule: a45df35ac0e948ee180fe216a5f703f32dda163f 

 _Originally posted by @Forge64 in [#6035](https://github.com/fwupd/fwupd/issues/6035#issuecomment-2714966297)_

superm1, 2025-03-24:
What exactly is on that partition?  I'm wondering how we can know to ignore it.

```
/org/freedesktop/UDisks2/block_devices/nvme0n1p2:
  org.freedesktop.UDisks2.Block:
    Configuration:              [('fstab', {'fsname': <b'PARTUUID=dea95449-e032-4984-ba4a-3c64ae01bcd5'>, 'dir': <b'/recovery'>, 'type': <b'vfat'>, 'opts': <b'umask=0077'>, 'freq': <0>, 'passno': <0>})]
    CryptoBackingDevice:        '/'
    Device:                     /dev/nvme0n1p2
    DeviceNumber:               66306
    Drive:                      '/org/freedesktop/UDisks2/drives/APPLE_SSD_AP0256J_C028306049KHJ9V1D'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 true
    Id:                         by-id-nvme-APPLE_SSD_AP0256J_C028306049KHJ9V1D-part2
    IdLabel:                    
    IdType:                     vfat
    IdUUID:                     02C7-B62F
    IdUsage:                    filesystem
    IdVersion:                  FAT32
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/nvme0n1p2
    ReadOnly:                   false
    Size:                       34359734272
    Symlinks:                   /dev/disk/by-diskseq/9-part2
                                /dev/disk/by-id/nvme-APPLE_SSD_AP0256J_C028306049KHJ9V1D-part2
                                /dev/disk/by-id/nvme-APPLE_SSD_AP0256J_C028306049KHJ9V1D_1-part2
                                /dev/disk/by-id/nvme-nvme.106b-433032383330363034394b484a39563144-4150504c4520535344204150303235364a-00000001-part2
                                /dev/disk/by-partlabel/recovery
                                /dev/disk/by-partuuid/dea95449-e032-4984-ba4a-3c64ae01bcd5
                                /dev/disk/by-path/pci-0000:01:00.0-nvme-1-part2
                                /dev/disk/by-uuid/02C7-B62F
    UserspaceMountOptions:      
  org.freedesktop.UDisks2.Filesystem:
    MountPoints:        /recovery
    Size:               0
  org.freedesktop.UDisks2.Partition:
    Flags:              0
    IsContained:        false
    IsContainer:        false
    Name:               recovery
    Number:             2
    Offset:             8589934592
    Size:               34359734272
    Table:              '/org/freedesktop/UDisks2/block_devices/nvme0n1'
    Type:               ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
    UUID:               dea95449-e032-4984-ba4a-3c64ae01bcd5
```

Can you get us `tree` output from a mounted partition so we can see if there are some hints we can use to know to ignore it?

Forge64, 2025-03-24:
Sure thing. I'm attaching tree listings of the recovery partition from both a 24.04 PopOS (glenn) system, as well as a legacy 22.04 install (ipop). Both trigger the warning in fwupd, and contents are largely the same, so any ID that is positive on both should positive on future PopOS recovery partitions as well.

Thanks!

[recovery.glenn.tree.txt](https://github.com/user-attachments/files/19437952/recovery.glenn.tree.txt)
[recovery.ipop.tree.txt](https://github.com/user-attachments/files/19437951/recovery.ipop.tree.txt)

Edit: Holy crap Github is dumb sometimes. Added .txt suffix to get my attachments allowed.

superm1, 2025-03-24:
@Forge64 any files that start with `.` which tree is excluding? 

@hughsie well it looks like these are basically dumps of a USB live media key.  Resemble a Ubuntu-ish layout.
Maybe we should make sure that the ESP has a `EFI/` directory to consider it an ESP?

Forge64, 2025-03-24:
@superm1 There is a directory on the top level, .disk, which contains a single file, "info", which contains:

Pop_OS 24.04 "Noble Numbat" - Release amd64 (20240924)

Pop_OS 22.04 "Jammy Jellyfish" - Release amd64 (20240325)

superm1, 2025-03-24:
OK; that really is like a live media then.  So we have two directions to go IMO.
1. We can exclude things that look like live media  (look for `.disk/info`) file.
2. We can only include things that have a top level directory of `/EFI`.

I've done a PR above that follows second direction.  Can you please test it?

If @hughsie prefers the first direction I'll swap it over.

Forge64, 2025-03-24:
I personally prefer the second, as it catches more future corner cases, but both seem reasonable. I'll pull the PR, build, and test.

commit 657ba2a9 (2025-03-25): trivial: Ensure that a volume contains /EFI to be considered an ESP

Closes: #8612
PR #8613 (2025-03-25): trivial: Ensure that a volume contains /EFI to be considered an ESP
Closes: #8612

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [ ] Feature
- [ ] Documentation
commit 7ea468af (2025-03-25): trivial: Ensure that a volume contains /EFI to be considered an ESP

Closes: #8612
closed 2025-03-25 by superm1