fwupd/fwupd #8629: fwupdmgr refresh failed

tipuraneo, 2025-03-28:
**Describe the bug**
Download failed because of checksum verification of downloaded file failed.

**Steps to Reproduce**
1. ```fwupdmgr refresh -vv --force```

**Expected behavior**
checksum verification is correct

**fwupd version information**
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-134-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10

```shell
# fwupdmgr refresh -vv --force
(fwupdmgr:3325657): GLib-DEBUG: 08:58:46.789: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:3325657): GLib-GIO-DEBUG: 08:58:46.796: _g_io_module_get_default: Found default implementation dummy (GDummyProxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.798: Emitting ::status-changed() [idle]
WARNING: UEFI firmware can not be updated in legacy BIOS mode
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information.
Updating lvfs
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.801: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.801: Emitting ::status-changed() [downloading]
Downloading…             [    \                                  ](fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.933: download progress: 100%
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.933: download progress: 100%
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.933: download progress: 100%
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.933: download progress: 100%
Downloading…             [***************************************]
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.933: Emitting ::status-changed() [idle]
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.933: status-code was 200
Downloading…             [***************************************]
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.933: changing metadata URI from https://cdn.fwupd.org/downloads/firmware.xml.gz to https://cdn.fwupd.org/downloads/firmware-07857-stable.xml.gz
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.933: downloading https://cdn.fwupd.org/downloads/firmware-07857-stable.xml.gz
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.933: Emitting ::status-changed() [downloading]
Downloading…             [***************************************]
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.996: download progress: 0%
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.996: download progress: 0%
Downloading…             [ -                                     ](fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.997: download progress: 0%
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.997: download progress: 0%
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.997: download progress: 0%
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.998: download progress: 0%
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.998: download progress: 1%
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.998: download progress: 1%
Downloading…             [                                       ](fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.998: download progress: 1%
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:46.998: download progress: 1%
[..]
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:47.042: download progress: 100%
Downloading…             [***************************************]
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:47.042: Emitting ::status-changed() [idle]
(fwupdmgr:3325657): Fwupd-DEBUG: 08:58:47.042: status-code was 200
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 7e81a28bfd4d8c1ccfa9c5e54e6f8e3ca7f1126f
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
# fwupdmgr get-devices --show-all-devices
[... 141 lines cut ...]
# these are only required when the SMBIOS or Device Tree data is invalid or missing
#Manufacturer=
#ProductName=
#ProductSku=
#Family=
#EnclosureKind=
#BaseboardProduct=
#BaseboardManufacturer=
```

</details>

**Additional questions**

- Operating system and version:
```shell
# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy
``` 
- Have you tried rebooting? no
- Is this a regression? no

fmonts, 2025-03-28:
I confirm I have the same issues on 10+ servers

hughsie, 2025-03-28:
Will investigate now; thanks.

vladget, 2025-03-28:
Same issue on a lot of servers:
```

# sudo fwupdmgr refresh
WARNING: UEFI firmware can not be updated in legacy BIOS mode
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information.
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected e2ab73978426e577c02ff02aba20530af88f9bcc
```

hughsie, 2025-03-28:
I've manually refreshed the metadata and purged the CDN; can you give it a go now please? I'll investigate later what happened.

fmonts, 2025-03-28:
`service fwupd-refresh start` still fails, the expected checksum is changed for me from `e2ab73978426e577c02ff02aba20530af88f9bcc` to `87178af7a1ff1fa4189615ebd735bb4ef9d3fd1e`

hughsie, 2025-03-28:
> https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat

What fwupd version is that?

vladget, 2025-03-28:
Not sure I understand your question, but anyway:
```
# zcat firmware.xml.gz.jcat 
{"JcatVersionMajor": 0, "JcatVersionMinor": 1, "Items": [{"Id": "firmware-07859-stable.xml.gz", "AliasIds": ["firmware.xml.gz"], "Blobs": [{"Kind": 4, "Flags": 1, "Timestamp": 1743153009, "Data": "0ef05feba95af0af422403d032b1bd95b681b8dc"}, {"Kind": 1, "Flags": 1, "Timestamp": 1743153009, "Data": "48ce0e85636c20d5607dd45ab730b8c3ea8c6aaf12c05f8cc97ce3c8aaaf972c"}, {"Kind": 2, "Flags": 1, "Timestamp": 1743153009, "Data": "-----BEGIN PGP SIGNATURE-----\n\niQEzBAABCAAdFiEEP8a4BEEO0IQNjy+XSKbYDkU4usIFAmfmZ3EACgkQSKbYDkU4\nusJxiQgAmtjteGunjTNLcle/d1bb8ein2ZfyWOlRgN7ZypHq5/ExlgEnz3S62d3j\nDT4APqfIq6gU1ZvMo25yVpzyA+UnSUwC5UAb+ELdJXTjDG5jAarEZEdcm6sPS9uJ\n+KI8r/b1njDru1v5o4D3eXit2kcYtEOslQkEc+41dN4xfGf3skapX1a/Sbu4uLZr\nvyfdUbiM1nphoNP5yBeLLGoP54pZk6MvjyJSkG5l+kCaWOvzZawXIsuvnKjeJ42E\nR0XxTK8ScDqZw4Dde+QCmi7xvkX9uL+hfoqSkP7UQQnvj1N6GIFy7tgDMQ1PCYh9\nVhkXoQx0MW3L3c6UW6WdeV6thNtkPQ==\n=cRbu\n-----END PGP SIGNATURE-----\n"}, {"Kind": 3, "Flags": 1, "Timestamp": 1743153009, "Data": "-----BEGIN PKCS7-----\nMIIGUgYJKoZIhvcNAQcCoIIGQzCCBj8CAQExDTALBglghkgBZQMEAgEwCwYJKoZI\nhvcNAQcBoIIEOjCCBDYwggKeoAMCAQICDFprhisibP88kP07YjANBgkqhkiG9w0B\nAQsFADA6MRAwDgYDVQQDEwdMVkZTIENBMSYwJAYDVQQKEx1MaW51eCBWZW5kb3Ig\nRmlybXdhcmUgUHJvamVjdDAeFw0xODAxMTYwMDAwMDBaFw0yODAxMTYwMDAwMDBa\nMBkxFzAVBgNVBAMTDlJpY2hhcmQgSHVnaGVzMIIBIjANBgkqhkiG9w0BAQEFAAOC\nAQ8AMIIBCgKCAQEArR5nseG3+Zs+o41P9LTspiVSGVcp6ifNHhbKKxBvZZXsy0gX\n+P/VRlsHLiKQrFulQ8GbPODytFv0o+y/0MkiJxv/fY3yEZ2bwNpsSeXFQSGHI6yz\njaVNeNCu8lnnDtD7kiC8UNUNHcnA4+2h/Yv4k+KPqYF+Qb6nWAEIID1ObMnjeJUb\ndbPPvy12aasV3gZcZ+goYNFc0ua6OU/CNEuzAVVCTAJ/EpCdGpll1+6BGU5ImIUG\nTlMTWq2xmCfCPugakHrmWA66yHWwE6LmC/U7qQDWFemsSNnmzyBB9HPkqsW1DjHr\n+ZmNUPj3+q2UGnNwP/Cne462XbsZB569w7pnzQIDAQABo4HcMIHZMAwGA1UdEwEB\n/wQCMAAwNwYDVR0RBDAwLoYXaHR0cHM6Ly9md3VwZC5vcmcvbHZmcy+BE3JpY2hh\ncmRAaHVnaHNpZS5jb20wEwYDVR0lBAwwCgYIKwYBBQUHAwMwDwYDVR0PAQH/BAUD\nAweAADAdBgNVHQ4EFgQUMcrnDWk3DM1GJCqJK7EbAycgB+UwHwYDVR0jBBgwFoAU\nsY3q5COnfgmOte4x4GrdnjQ3ZawwKgYDVR0fBCMwITAfoB2gG4YZaHR0cDovL3d3\ndy5md3VwZC5vcmcvcGtpLzANBgkqhkiG9w0BAQsFAAOCAYEAjoXBOtVb1qPLuE7I\nShdSkk66JMNmzZnODbun3BaViUF3PPhuRiJ2y7Bu7loCnxHwzKpq5Hn4Untg1WhQ\nSsPEv6H2oc2E4HU1Gds2HEE1UL0VzulPEwaOFX0OZ37LmJ2VFvfgwfNmjAWZ+ngN\nsjuff3PWeHNsOwbS3To0CBlHqz9LlcA7Gxpxbz4DbUEvgTTKZFbasgUS6/QmbpPy\n880ThgD93yCg6Q002AooPMw8zW6gGUP7/2D4aKBjgM/IJDrJEicTdKDFtUacsN+d\nCfIEcn6F1rzPgwo0B/yLYVMpbL8WzxSS7cX5dBlkreH6Nf7g+A2NTryU/6FxoFxo\nnWJmLssNQCgx42Ywc8CgJRt1yWXxoZXOPap8zPvFAAmrfUlqfZlr2Wzcf2nxpSZJ\noQRzGDXDRiYNt28xjX408cJ+Wy+0oWwgnXCkfzMDnJQB8MX6+5Ah4v4mS/XjmOIi\nQ+hKCT99Yf0Eq3TWzFhyYducqLU904FbrrUSoi07DfxXa/8RMYIB3jCCAdoCAQEw\nSjA6MRAwDgYDVQQDEwdMVkZTIENBMSYwJAYDVQQKEx1MaW51eCBWZW5kb3IgRmly\nbXdhcmUgUHJvamVjdAIMWmuGKyJs/zyQ/TtiMAsGCWCGSAFlAwQCAaBpMBgGCSqG\nSIb3DQEJAzELBgkqhkiG9w0BBwEwHAYJKoZIhvcNAQkFMQ8XDTI1MDMyODA5MTAw\nOVowLwYJKoZIhvcNAQkEMSIEIEjODoVjbCDVYH3UWrcwuMPqjGqvEsBfjMl848iq\nr5csMA0GCSqGSIb3DQEBAQUABIIBAKts6etiNqHn1i8egbqvrI2d8fCcWvJVwO0S\n68r+h4RshkCAShkmLkB969PxY1FpRCAoEsP5xe86IFhX7VhAylytp9ve1DR1E4lJ\n193Ib6zy7xNoSIQLpVm/Un/MbITnWldbbHf7UNDSpzK7UP8FZxE6P/0BayrjzkY1\nwQba+/aMO0HP+VHYu5lN0NzJU+u/60im4ojZYONozxVywD6Q7UqtGLFEfbxIct8Z\nUY2aFbu0fVRxR5nCYd/g+Z8AcaZsepPfQTgrYobGRq1t5RbYiTTprsIyS+A7hEOk\n5G7l+Yz/ggtWkXp2LmPnK7gjZ89VX3qMVBkoU52r2z/2IsjbH78=\n-----END PKCS7-----\n"}]}]}
```

and

```
# fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.15.0-92-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

zxqro1, 2025-03-28:
Same problem here, on multiple hosts (all VMs)

```
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 87178af7a1ff1fa4189615ebd735bb4ef9d3fd1e
```

```
wupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-133-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

YashikKhunt, 2025-03-28:
I have same problem received this 

[image]

Simon-HS, 2025-03-28:
I had the same issue. I fixed the problem with `fwupdtool refresh`. After this `fwupdmgr refresh --force` was succesfully. The service `fwupd-refresh` was also fixed after restart.

jacq34326562, 2025-03-28:
> I had the same issue. I fixed the problem with `fwupdtool refresh`. After this `fwupdmgr refresh --force` was succesfully. The service `fwupd-refresh` was also fixed after restart. It was on Ubuntu 22.04

This worked for me too for all devices ty

j1mc, 2025-03-28:
Here is some relevant output, if it is helpful. This is on Ubuntu 22.04 with the stock fwupd package at version 1.7.9-1~22.04.3.

```
root@laptop:~# fwupdmgr get-updates
Firmware metadata has not been updated for 30 days and may not be up to date.

Update now? (Requires internet connection) [y|N]: y
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Downloading…             [***************************************]
Failed to update metadata for lvfs: checksum failure: failed to verify data, expected ed3275d5427a3b1845c72b60ad2c6019294a8043

root@laptop:~# fwupdtool refresh
Loading…                 [***************************************]
18:16:01:0937 FuEngine             failed to get releases for UEFI dbx: No releases found: Not compatible with org.freedesktop.fwupd version 1.7.9, requires >= 1.9.1, Not compatible with org.freedesktop.fwupd version 1.7.9, requires >= 1.9.28

root@laptop:~# fwupdmgr get-updates
Devices with no available firmware updates: 
 • Integrated Camera
<snip>
 • UEFI Device Firmware
<snip>
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Intel Management Engine
 • System Firmware
 • UEFI Device Firmware
No updates available
```

fmonts, 2025-03-29:
Just to know, any chance this will fix itself or with an apt upgrade, or do we have to enter every server and run the command?

hughsie, 2025-03-29:
It should fix itself. Please also note that 1.7.x versions are EOL now, and not supported by upstream: https://github.com/fwupd/fwupd/blob/main/SECURITY.md

an0key, 2025-03-29:
It would seem that the version via Ubuntu LTS is indeed EOL. Raised a bug there. https://bugs.launchpad.net/ubuntu/+source/fwupdate/+bug/2105405

superm1, 2025-03-29:
Anyone coming here to report this on Ubuntu 22.04 with fwupd 1.7.9, this is EOL.  Ubuntu's SRU team needs to review https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/2056548 to upgrade to a current version.

closed 2025-03-29 by superm1

hughsie, 2025-03-29:
I wonder if the gz metadata is too large. We had that 1MB issue a looong time ago and I wonder if we're tripping into that issue again.

superm1, 2025-03-29:
> I wonder if the gz metadata is too large. We had that 1MB issue a looong time ago and I wonder if we're tripping into that issue again.

Even if that's the case, this will never get fixed in Ubuntu if we keep doing bandages.  I sent up the package for an SRU over a year ago and no one did anything with it.

hughsie, 2025-03-30:
The workaround for Ubuntu is to cherry pick 0729ab5cb013dd90d72f8010a7d826a38d2ea474 -- but the real fix is to backport a new version as 1.7.x isn't even getting security updates now.

hughsie, 2025-03-30:
Of course, what I could do is just generate "empty" metadata for 1.7.x -- which means the problem won't even happen again, but it does mean anyone on an EOL version will get zero firmware updates.

superm1, 2025-03-31:
> The workaround for Ubuntu is to cherry pick https://github.com/fwupd/fwupd/commit/0729ab5cb013dd90d72f8010a7d826a38d2ea474 -- but the real fix is to backport a new version as 1.7.x isn't even getting security updates now.

Yeah that requires that they do _something_. 

> Of course, what I could do is just generate "empty" metadata for 1.7.x -- which means the problem won't even happen again, but it does mean anyone on an EOL version will get zero firmware updates.

I have mixed feelings about this.  Doing "nothing" on LVFS means it's more likely something gets done in Ubuntu 22.04 as more people complain.  Doing that on LVFS means it's a very clear message that we don't support EOL versions of fwupd.

If doing that I would go as far as to say ANY EOL version should get zero updates, not just 1.7.x.