fwupd/fwupd #8634: fwupdtool refresh -> no libcurl support

bolemo, 2025-03-30:
**Describe the bug**
When running `fwupdtool refresh` it returns `no libcurl support`

**Steps to Reproduce**
```shell
# fwupdtool refresh
Loading…                 [************************************** ]
no libcurl support
```

**Expected behavior**
It should work.
Moreover, libcurl is installed:
```shell
# opkg status libcurl4
Package: libcurl4
Version: 8.10.1-r1
Depends: libc, libmbedtls21, libnghttp2-14, ca-bundle
Provides: libcurl
Status: install ok installed
Architecture: x86_64
Installed-Time: 1738624177
```

**fwupd version information**
```shell
# fwupdtool get-report-metadata
Waiting…                 [*************************************  ]
BatteryLevel:           101
BatteryThreshold:       10
BootTime:               1743304424
CompileVersion(com.hughsie.libjcat): 0.2.1
CompileVersion(com.hughsie.libxmlb): 0.3.19
CompileVersion(info.libusb): 1.0.27
CompileVersion(org.freedesktop.fwupd): 2.0.1
CpuArchitecture:        x86_64
DisplayState:           unknown
DistroId:               openwrt
DistroName:             OpenWrt
DistroPrettyName:       OpenWrt 24.10.0
DistroVersion:          24.10.0
FwupdSupported:         True
HostBaseboardManufacturer: Techvision
HostBaseboardProduct:   TVI7309X
HostBiosMajorRelease:   05
HostBiosMinorRelease:   13
HostBiosVendor:         Techvision, LLC.
HostBiosVersion:        5.19
HostEnclosureKind:      3
HostFamily:             Default string
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            TVI7309X
HostSku:                Default string
HostVendor:             Techvision
KernelCmdline:          rootwait noinitrd
KernelName:             Linux
KernelVersion:          6.6.73
LidState:               unknown
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat): 0.2.1
RuntimeVersion(com.hughsie.libxmlb): 0.3.19
RuntimeVersion(org.freedesktop.fwupd): 2.0.1
RuntimeVersion(org.kernel): 6.6.73
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

```shell
# opkg install fwupd-tools
Installing fwupd-tools (2.0.1-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/fwupd-tools_2.0.1-r1_x86_64.ipk
Installing libdbus (1.14.10-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/libdbus_1.14.10-r1_x86_64.ipk
Installing dbus (1.14.10-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/dbus_1.14.10-r1_x86_64.ipk
Installing libxmlb (0.3.19-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/libxmlb_0.3.19-r1_x86_64.ipk
Installing json-glib0 (1.8.0-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/json-glib0_1.8.0-r1_x86_64.ipk
Installing libgpg-error (1.49-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/libgpg-error_1.49-r1_x86_64.ipk
Installing libassuan (2.5.6-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/libassuan_2.5.6-r1_x86_64.ipk
Installing libgpgme (1.21.0-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/libgpgme_1.21.0-r1_x86_64.ipk
Installing libjcat (0.2.1-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/libjcat_0.2.1-r1_x86_64.ipk
Installing libprotobuf-c (1.4.1-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/libprotobuf-c_1.4.1-r1_x86_64.ipk
Installing libarchive (3.7.4-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/libarchive_3.7.4-r1_x86_64.ipk
Installing libmbim (1.30.0-r2) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/libmbim_1.30.0-r2_x86_64.ipk
Installing libqrtr-glib (1.2.2-r3) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/libqrtr-glib_1.2.2-r3_x86_64.ipk
Installing libqmi (1.34.0-r2) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/libqmi_1.34.0-r2_x86_64.ipk
Installing modemmanager (1.22.0-r20) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/modemmanager_1.22.0-r20_x86_64.ipk
Installing fwupd-libs (2.0.1-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.0/packages/x86_64/packages/fwupd-libs_2.0.1-r1_x86_64.ipk
Configuring json-glib0.
Configuring libgpg-error.
Configuring libassuan.
Configuring libgpgme.
Configuring libjcat.
Configuring libdbus.
Configuring libprotobuf-c.
Configuring dbus.
Configuring libxmlb.
Configuring libarchive.
Configuring libmbim.
Configuring libqrtr-glib.
Configuring libqmi.
Configuring modemmanager.
Configuring fwupd-libs.
Configuring fwupd-tools.
```

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
# fwupdtool get-devices --show-all-devices
Loading…                 [************************************** ]
Techvision TVI7309X
│
└─Unknown Device:
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Current version:    01
      Vendor:             PCI:0x8086
      GUID:               06006e7d-13d0-5716-90a3-a12c4c18d860 ← PCI\VEN_8086&DEV_4E61
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
                          • Can tag for emulation
```

</details>

**Additional questions**

- Operating system and version:
```shell
# uname -a
Linux Zeus 6.6.73 #0 SMP PREEMPT_DYNAMIC Mon Feb  3 23:09:37 2025 x86_64 GNU/Linux
# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='24.10.0'
DISTRIB_REVISION='r28427-6df0e3d02a'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 24.10.0 r28427-6df0e3d02a'
DISTRIB_TAINTS='no-all busybox'
```

superm1, 2025-03-30:
It sounds like it has been _compiled_ with libcurl support..
You can confirm by looking at the build log.

bolemo, 2025-03-30:
> It sounds like it hasn't been _compiled_ with libcurl support.. You can confirm by looking at the build log.

Possible, then I would have to contact the fwupd package maintainer for openwrt.

I prefer to use the package provided version, but I also tried to compile directly, but I got another problem as openwrt is not a common platform, and as far as I know, is not directly based on any of the proposed choices:
```shell
# git clone https://github.com/fwupd/fwupd.git
Cloning into 'fwupd'...
remote: Enumerating objects: 118197, done.
remote: Counting objects: 100% (761/761), done.
remote: Compressing objects: 100% (426/426), done.
remote: Total 118197 (delta 411), reused 335 (delta 335), pack-reused 117436 (from 4)
Receiving objects: 100% (118197/118197), 58.10 MiB | 27.58 MiB/s, done.
Resolving deltas: 100% (93388/93388), done.
# cd fwupd/
# ./contrib/setup
Using OS profile openwrt to setup
Setting up virtualenv
/usr/bin/python3: No module named pip
usage: fwupd_setup_helpers.py [-h] [-o {fedora,centos,debian,ubuntu,arch,darwin}] [-v VARIANT] [-y] [-d]
                              {get-dependencies,test-markdown,test-jinja2,test-meson,detect-profile,install-dependencies,install-pip}
fwupd_setup_helpers.py: error: argument -o/--os: invalid choice: 'openwrt' (choose from 'fedora', 'centos', 'debian', 'ubuntu', 'arch', 'darwin')
```

hughsie, 2025-03-30:
> invalid choice: 'openwrt' (choose from 'fedora', 'centos', 'debian', 'ubuntu', 'arch', 'darwin')

I'm not against adding openwrt, but you'd need to add all the opgk packages to https://github.com/fwupd/fwupd/blob/main/contrib/ci/dependencies.xml and also add the opgk bits to https://github.com/fwupd/fwupd/blob/main/contrib/ci/fwupd_setup_helpers.py#L195

superm1, 2025-03-30:
> > It sounds like it hasn't been _compiled_ with libcurl support.. You can confirm by looking at the build log.
> 
> Possible, then I would have to contact the fwupd package maintainer for openwrt.

Not just possible; it's the root cause. Every other distro compiles with curl support.

> I prefer to use the package provided version, but I also tried to compile directly, but I got another problem as openwrt is not a common platform, and as far as I know, is not directly based on any [...]

Yeah this isn't something we know all the deps needed to build on this distro.

You can skip the dependencies resolution step of the setup script if you know you installed everything you need.