fwupd/fwupd #8639: fwupd executes twice during DFU upgrade

AVer-Jierger, 2025-04-02:
Using fwupd for Device Updates with DFU: Why Does It Execute Twice Quickly in ChromeOS?

What mechanisms or possibilities could cause this behavior?

[image]

hughsie, 2025-04-02:
Sounds weird. Can you attach the `fwupdtool install -vv ...`  (with two verbose `v`'s) output to this issue please.

rcyeh, 2025-04-03:
[install238on206.log](https://github.com/user-attachments/files/19579618/install238on206.log)

This log was collected on a Chromebox running R132-16093.68.12 and fwupd 2.0.6 with patches. It seems to end with `failed write-firmware 'failed to write chunk 47103: cannot download data: USB error: Input/Output Error [-1]': FuDfuDevice:`

hughsie, 2025-04-03:
@rcyeh it's not writing twice; it writes all the chunks from 0000 to b7fe successfully, and then fails with:

    20:29:26.859 FuEngine             failed write-firmware 'failed to write chunk 47103: cannot download data: USB error: Input/Output Error [-1]': FuDfuDevice:

Is this expected behaviour? It looks like the device self-reset before the update was complete.

rcyeh, 2025-04-03:
We're able to use fwupd 1.9.21 to do the update (possibly with the 2x behavior from the original report; which might be due to Chromebox-for-Meetings udev rules triggering on the camera's reappearance).

I suspect that there may be a problem across DFU-based updates on fwupd 2.0.6 and 2.0.7.

hughsie, 2025-04-03:
> which might be due to Chromebox-for-Meetings udev rules triggering

Is the udev rule triggering a fwupdtool command? If so, I've told them that's a bad idea as it's not going to work very well with DFU. I'm guessing using fwupdmgr works correctly?

rcyeh, 2025-04-30:
Sorry, do you mind if I repurpose this issue completely? Retitling from "fwupd executes twice during DFU upgrade" to "avocor-avm-6570-firmware with dfuclient fails to install on fwupd 2".

* They all install successfully on fwupd 1.9.21, but not fwupd 2.0.6, both on ChromeOS.
* It does look like the camera just disappears. Have attached the latest log with `fwupdtool`, with the end pasted at the bottom of this message. [atlas_camera_240_on_206.log](https://github.com/user-attachments/files/19984023/atlas_camera_240_on_206.log)
* This fails even when we have disabled/deleted the udev rule.

```
19:56:58.278 FuPluginDfu          no dnload-timeout, using default of 5ms
19:56:58.278 FuPluginDfu          refreshed status=ok and state=dfu-dnload-idle (dnload=5)
19:56:58.279 FuPluginDfu          writing #b984 chunk of size 0x40
19:56:58.279 FuPluginDfu          Message:
ce 33 fa be 7f d1 88 6b 36 5b 1f f5 59 93 ca 29 26 ac d5 75 5b 15 51 24 f2 d0 15 22 f7 79 6f 80 
66 28 c8 8a 72 90 42 61 a7 c8 60 39 e8 24 f7 71 0c ab bf 92 a2 69 ba 12 44 18 7b bd 53 ce ca 37 
19:56:58.279 FuPluginDfu          sleeping for 5ms...
19:56:58.285 FuPluginDfu          no dnload-timeout, using default of 5ms
19:56:58.285 FuPluginDfu          refreshed status=ok and state=dfu-dnload-idle (dnload=5)
19:56:58.285 FuPluginDfu          writing #b985 chunk of size 0x40
19:56:58.285 FuPluginDfu          Message:
f4 63 15 eb 98 0f 22 d9 15 17 e5 a4 1f 2b 03 3b 32 ff 76 74 3d 3d a1 43 db 05 e3 64 25 94 af ab 
64 48 12 e8 7f 0c ae 51 cc ad c8 90 cd 63 88 12 2c ed cf 03 83 c1 9a 77 76 5e ed 42 7e 63 9a 51 
19:56:58.285 FuPluginDfu          sleeping for 5ms...
19:56:58.291 FuPluginDfu          no dnload-timeout, using default of 5ms
19:56:58.291 FuPluginDfu          refreshed status=ok and state=dfu-dnload-idle (dnload=5)
19:56:58.291 FuPluginDfu          writing #b986 chunk of size 0x40
19:56:58.291 FuPluginDfu          Message:
3e c8 45 32 17 e1 8e 45 3c ee 17 02 ee e9 3f c0 fd 5a c0 3d 2c e0 7e f5 07 b8 5b 00 d7 be f7 5f 
f7 ae a8 0f 65 e7 81 3d 9e 6b 5f 1d 88 e7 40 7f 48 d3 a8 9a 93 f7 9d 6e 45 6b e7 87 7b 3f 00 7e 
19:56:58.291 FuPluginDfu          sleeping for 5ms...
19:56:58.323 FuHistory            modifying device Series One Board 65 Camera DFU Mode [d787669ee4a103fe0b361fe31c10ea037c72f27c]
19:56:58.324 FuEngine             ignoring: no device d787669ee4a103fe0b361fe31c10ea037c72f27c
19:56:58.324 FuMain               ignoring Series One Board 65 Camera DFU Mode with status unknown
19:56:58.324 FuEngine             failed write-firmware 'failed to write chunk 47494: cannot wait for busy: cannot get device state: USB error: Input/Output Error [-1]': FuDfuDevice:
  DeviceId:             d787669ee4a103fe0b361fe31c10ea037c72f27c
  Name:                 Series One Board 65 Camera DFU Mode
  Percentage:           5
  Guid:                 337e9034-5da4-5a43-8072-23f0ce1a00af <- USB\VID_34AD&PID_0006
  Serial:                Mode
  Plugin:               dfu
  Protocol:             org.usb.dfu
  Flags:                updatable|is-bootloader|can-emulation-tag
  Vendor:               Avocor Technologies USA, Inc.
  VendorId:             USB:0x34AD
  Version:              2.34
  VersionFormat:        bcd
  VersionRaw:           0x00000234
  Created:              2025-04-30
  UpdateState:          failed
  UpdateError:          Device has been removed, An update is in progress
  InstanceId[qu,ge]:    8cc540e8-f987-51ec-9688-ee09012ca887 <- USB\VID_34AD
  InstanceId[vi,qu,ge]: 337e9034-5da4-5a43-8072-23f0ce1a00af <- USB\VID_34AD&PID_0006
  InstanceId[qu,ge]:    c418fefd-0db6-5152-a4da-7a7527183cd1 <- USB\CLASS_FE
  InstanceId[qu,ge]:    ab182806-032b-5b7b-994f-b0713031d2b6 <- USB\CLASS_FE&SUBCLASS_01
  InstanceId[qu,ge]:    a55d6961-d2c3-5659-a26c-528521d001cd <- USB\CLASS_FE&SUBCLASS_01&PROT_02
  PhysicalId:           1-1
  BackendId:            /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1
  Vid:                  0x34ad
  Pid:                  0x6
  RemoveDelay:          60000
  AcquiesceDelay:       2500
  CustomFlags:          detach-for-attach
  Order:                0
  BestDevice:           /dev/urandom
  PossiblePlugin:       dfu
  PrivateFlags:         add-counterpart-guids,replug-match-guid,md-set-signed,md-set-flags,add-instance-id-rev,detach-for-attach,can-upload,can-download,registered
  Events:
  Subsystem:            usb
  Devtype:              usb_device
  BindId:               1-1
  DeviceFile:           /dev/bus/usb/001/004
  OpenFlags:            read,write
[... 141 lines cut ...]
        Order:          0
        BestDevice:     /dev/urandom
        PossiblePlugin: dfu
        PrivateFlags:   add-counterpart-guids,replug-match-guid,md-set-signed,md-set-flags,add-instance-id-rev,detach-for-attach,can-upload,can-download,registered
        Events:
      AltIdx:           0x5
        Idx00:          Zone:0, Sec#:0, Addr:0x00000000, Size:0x0000, Caps:0x3 [readable,writeable]

19:56:58.325 FuDeviceLocker       ignoring close error on aborted open: device 34ad:0006 has not been opened
19:56:58.325 FuEngine             failed to update-cleanup after failed update: failed to open device for cleanup: failed to subclass open: failed to open /dev/bus/usb/001/004: No such file or directory
19:56:58.333 FuPlugin             composite_cleanup(dell_dock)
19:56:58.333 FuPlugin             composite_cleanup(dell_kestrel)
19:56:58.333 FuPlugin             composite_cleanup(logitech_tap)
[... cut ...]
failed to write-firmware: failed to write chunk 47494: cannot wait for busy: cannot get device state: USB error: Input/Output Error [-1]
19:56:58.337 FuProgress           raw timing data was { 0.002, 0.000, 0.001, 0.003, 0.000 } -- steps were set as [ 1 1 3 1 94 ] but should have been [ 30 2 12 55 1 ] at ../fwupd-2.0.6/libfwupdplugin/fu-context.c:1090
19:56:58.342 FuProgress           raw timing data was { 1.873, 1.570, 0.000, 0.000, 0.000 } -- steps were set as [ 0 1 88 1 10 ] but should have been [ 54 46 0 0 0 ] at ../fwupd-2.0.6/plugins/dfu/fu-dfu-device.c:1468
```

hughsie, 2025-05-03:
This is fixed in 656f352fa5853b582ca5156e31311fa562a21a5b I think.

closed 2025-05-03 by hughsie

rcyeh, 2025-05-03:
Thanks! I'll give it a try.