fwupd/fwupd #8641: fwupdtool get-devices - segmentation fault after switching from 1.9.26 to 2.0.6 / 2.0.7

bbo-igel, 2025-04-02:
**Describe the bug**

We are running our own Linux distribution called IGEL OS.
It is mainly a read-only filesystem with an overlay for persistence.

fwupd 1.9.26 and previous versions worked fine for several years.

We tried to switch to 2.06/2.07, but fwupdmgr get-devices only shows minimal devices (like DBX), no more system firmware from the uefi-plugin

The journal shows a segfault.

Further analysis showed that fwupdtool get-devices -v segfaults at FU_IS_VOLUME

```shell
(gdb) bt
#0  0x00007f69f0abe149 in FU_IS_VOLUME (ptr=0x3e435760) at ../libfwupdplugin/fu-volume.h:16
#1  0x00007f69f0ac030f in fu_volume_locker (self=0x3e435760, error=0x7ffcb81f6920) at ../libfwupdplugin/fu-volume.c:899
#2  0x00007f69f0a1bb5f in fu_context_get_default_esp (self=0x3e2d0f90, error=0x7ffcb81f6920) at ../libfwupdplugin/fu-context.c:1932
#3  0x00007f69f0ec6114 in fu_uefi_dbx_plugin_constructed (obj=0x3e35ddd0) at ../plugins/uefi-dbx/fu-uefi-dbx-plugin.c:111
#4  0x00007f69f0a8b5d6 in fu_plugin_runner_init (self=0x3e35ddd0) at ../libfwupdplugin/fu-plugin.c:963
#5  0x00007f69f0ca49f8 in fu_engine_plugins_init (self=0x3e2d51f0, progress=0x3e2e5eb0, error=0x7ffcb81f6d10) at ../src/fu-engine.c:7308
#6  0x00007f69f0ca8ab3 in fu_engine_load (self=0x3e2d51f0, flags=366, progress=0x3e2c8aa0, error=0x7ffcb81f6d10) at ../src/fu-engine.c:8415
#7  0x000000000040e089 in fu_util_start_engine (priv=0x3e29c6d0, flags=366, progress=0x3e2c8aa0, error=0x7ffcb81f6d10) at ../src/fu-tool.c:249
#8  0x000000000041042e in fu_util_get_devices (priv=0x3e29c6d0, values=0x3e2dd0d0, error=0x7ffcb81f6d10) at ../src/fu-tool.c:883
#9  0x00007f69f0bd17f9 in fu_util_cmd_array_run (array=0x3e29c7a0, priv=0x3e29c6d0, command=0x7ffcb81f81ac "get-devices", values=0x7ffcb81f72d8, error=0x7ffcb81f6d10) at ../src/fu-util-common.c:504
#10 0x000000000041d914 in main (argc=2, argv=0x7ffcb81f72c8) at ../src/fu-tool.c:5629
```

I am looking for any idea what might have changed here between 1.9.26 and 2.0.7
This seems to be a side effect of our IGEL OS architecture, since the issue is not happening in stock debian.

Further digging via strace showed that fwupd tries to access/mount our special block device /dev/igf0 , possibly because it is not marked at read-only (?).
When deleting the device, the segfault is gone.

**Steps to Reproduce**
```shell
mount /dev/nvme0n1p2 /boot/efi
fwupdtool get-devices -v
```

**Expected behavior**
No segfault

**fwupd version information**
```shell
compile   info.libusb                   1.0.26
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         2.0.7
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.12.19
runtime   org.freedesktop.fwupd         2.0.7
```

```shell
fwupdtool get-report-metadata  
Also segfaults at FU_IS_VERSION
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices

LENOVO 21H6S05J00
│
├─AMD Ryzen 3 PRO 7330U with Radeon Graphics:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             Advanced Micro Devices, Inc.
│     GUIDs:              79759cdc-94db-5098-be7b-eb02521fbbec ← CPUID\PRO_0&FAM_19&MOD_50
│                         20b595b0-5892-5870-8e4c-688133ad6e34 ← CPUID\PRO_0&FAM_19&MOD_50&STP_0
│     Device Flags:       • Internal device
│
├─EM05-G:
│     Device ID:          3217f195a5f3e13fc15a924ce00b2cbefd3cefb9
│     Summary:            Mobile broadband device
│     Current version:    EM05GFAR07A07M1G_01.002.01.002
│     Vendor:             Quectel (USB:0x2C7C)
│     GUIDs:              0322d845-8497-5cbe-bfd5-2c2a6c9ea129 ← USB\VID_2C7C&PID_030A&REV_0200
│                         df71f261-0f4c-55d5-8c0c-e6d2840feee5 ← USB\VID_2C7C&PID_030A
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Unsigned Payload
│
├─UEFI Platform Key:
│     Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│     Summary:            Lenovo Ltd. PK CA 2012
│     Vendor:             Lenovo Ltd.
│     GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│                         • Can tag for emulation
│
└─UEFI dbx:
      Device ID:          362301da643102b9f38477387e2193e57abaa590
      Summary:            UEFI revocation database
      Current version:    20241101
      Minimum Version:    20241101
      Vendor:             UEFI:Microsoft
      Install Duration:   1 second
      GUIDs:              5971a208-da00-5fce-b5f5-1234342f9cf7 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649&ARCH_X64
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
      Device Flags:       • Internal device
                          • Updatable
                          • Needs a reboot after installation
                          • Cryptographic hash verification is available
                          • Device is usable for the duration of the update
                          • Only version upgrades are allowed
                          • Signed Payload
                          • Can tag for emulation

```

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

bbo-igel, 2025-04-02:
> I am worried that hint is applied to ESP as well. Can you double check?

Yes you are right, ESP is also using this.

hughsie, 2025-04-02:
Yup, my ESP is `HintIgnore:true` - I think that last time I looked at udisks that boolean was more "should we hide this from the UI" rather than "pretend this wasn't here". I'm more worried about the segfault; that seems bad.

hughsie, 2025-04-02:
@bbo-igel your backtrace:

    #2  0x00007f69f0a1bb5f in fu_context_get_default_esp (self=0x3e2d0f90, error=0x7ffcb81f6920) at ../libfwupdplugin/fu-context.c:1932

What fwupd version is that from? That doesn't seem to match up with the git tag for 2.0.7.

superm1, 2025-04-02:
Also; what exactly _is_ this block device?  Is it always at `/dev/igf0`?  Or is there sometimes `/dev/igf1`?  If we need to exclude it I'm not against that (we do some exclusions for RAID already).  But I would like to better understand why the existing code flops over on it.

bbo-igel, 2025-04-02:
> [@bbo-igel](https://github.com/bbo-igel) your backtrace:
> 
> ```
> #2  0x00007f69f0a1bb5f in fu_context_get_default_esp (self=0x3e2d0f90, error=0x7ffcb81f6920) at ../libfwupdplugin/fu-context.c:1932
> ```
> 
> What fwupd version is that from? That doesn't seem to match up with the git tag for 2.0.7.

You are correct, the bt was from a version that did have one of our patches applied. I forgot to roll back to the unmodified version.

My first assumption was that one of our patches was the culprit, that was the first thing I checked. 
I will provide the same bt without any patches and edit this commend.

superm1, 2025-04-02:
> @hughsie I have to apologize, it seems like there was a mistake on my side. It does not crash without our patches, therefore the error must be on our side. I will close the ticket once I have found the root cause.

_Ideally_ we don't want to see distros carrying out of tree patches.  It makes the vendor ecosystem to map out to versions on LVFS.  Vendors can set up their firmware to only be targeted to a version or newer for example.

Can you please nominate your patches for upstream (after fixing the issue 😁)?

bbo-igel, 2025-04-03:
I was not the original author of our patch, further analysis shows that the whole reason for the patch was that the ESP check fails for our ESP due to the "efi" folder on the mount path being lowercase (efi instead of EFI)

I did a PR for that:
https://github.com/fwupd/fwupd/pull/8644