fwupd/fwupd #8647: fwupdmgr get-devices shows way less then fwupdtool get-devices

bbo-igel, 2025-04-03:
**Describe the question**
This is another issue we are facing with in our linux distribution after migration from 1.9.26 to 2.0.7
Any suggestion what to check would be helpful.  👍 

We are facing the following issue:
fwupdmgr get-devices does only shows a handful of devices. Same with Fwupd.Client used in python (GObject introspection) 

fwupdtool get-devices shows all devices.  (!)

```shell
fwupdmgr get-devices -vv
(fwupdmgr:16411): GLib-GIO-DEBUG: 13:51:16.235: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:16411): GVFS-RemoteVolumeMonitor-DEBUG: 13:51:16.235: Error: The connection is closed

(fwupdmgr:16411): GLib-GIO-DEBUG: 13:51:16.235: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.235: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.235: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: unwatch_fast: "/system/proxy/" (active: 1, establishing: 0)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: unwatch_fast: "/system/proxy/http/" (active: 1, establishing: 0)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: unwatch_fast: "/system/proxy/https/" (active: 1, establishing: 0)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: unwatch_fast: "/system/proxy/ftp/" (active: 1, establishing: 0)
(fwupdmgr:16411): dconf-DEBUG: 13:51:16.236: unwatch_fast: "/system/proxy/socks/" (active: 1, establishing: 0)
(fwupdmgr:16411): GLib-GIO-DEBUG: 13:51:16.244: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:16411): GLib-GIO-DEBUG: 13:51:16.245: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:16411): Fwupd-DEBUG: 13:51:16.246: Emitting ::status-changed() [idle]
FuMain-INFO: 13:51:16.249: FwupdDevice:
  DeviceId:             3217f195a5f3e13fc15a924ce00b2cbefd3cefb9
  Name:                 EM05-G
  Guid:                 0322d845-8497-5cbe-bfd5-2c2a6c9ea129 ← USB\VID_2C7C&PID_030A&REV_0200
  Guid:                 df71f261-0f4c-55d5-8c0c-e6d2840feee5 ← USB\VID_2C7C&PID_030A
  Summary:              Mobile broadband device
  Plugin:               modem_manager
  Protocol:             com.qualcomm.firehose
  Flags:                updatable|require-ac|unsigned-payload
  Vendor:               Quectel
  VendorId:             USB:0x2C7C
  Version:              EM05GFAR07A07M1G_01.002.01.002
  VersionFormat:        plain
  Icon:                 modem
  Created:              2025-04-03

FuMain-INFO: 13:51:16.249: FwupdDevice:
  DeviceId:             6469856584e2f5873b2f148302e46c9313c7d054
  Name:                 SBAT
  Guid:                 9947c120-b97b-52d6-9d83-738a1c342e23 ← UEFI\OS_ubuntu&VAR_SbatLevelRT
  Summary:              Generation number based revocation mechanism
  Plugin:               uefi_sbat
  Protocol:             com.uefi.sbat
  Flags:                updatable|needs-reboot|signed-payload
  VendorId:             OS:ubuntu
  Version:              1.7.4
  VersionFormat:        triplet
  Created:              2025-04-03

FuMain-INFO: 13:51:16.249: FwupdDevice:
  DeviceId:             362301da643102b9f38477387e2193e57abaa590
  Name:                 UEFI dbx
  Guid:                 5971a208-da00-5fce-b5f5-1234342f9cf7 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649&ARCH_X64
  Guid:                 f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
  Summary:              UEFI revocation database
  Plugin:               uefi_dbx
  Protocol:             org.uefi.dbx2
[... 440 lines cut ...]
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x0
DeviceId:               362301da643102b9f38477387e2193e57abaa590
  pre:
      EfivarsNvramUsed: 50700
      LinuxLockdown:    integrity
      Pcr0_SHA256:      c4be5ba22b39e0c54a8488598830fb14a5774ec3362e55ffd2d2b19af44605d7
      Pcr0_SHA384:      f2c4eb19c4f71680b3b5530467a38e1959c4c7ab35533ef6895b4d58ee99410f2f69fa666f6d3f3313eca26fc45550b2
      TpmEventLog:      0x00000008 486470411e2c314db8e65f45f0f748ef1f3f27774ff7621d462dc096f4887501 [UgAyADUARQBUADQAMABXACAAAAA=]
      0x80000008 d56e9823dfb0fe21a27de40bd84fa9a4888a4adbd418a55f48c663f0ea3dae79 [AADACQAAAAAAABsAAAAAAA==]
      0x00000001 260e61c6434e72e34823716943afea42c7bc7fae53d70f02f99a6755c713f1cb [QUNQSSBEQVRB]
      0x00000001 608072d8953921f15718897cbb0a47623e0c29fa7286d20beb0d733756acb643 [QUNQSSBEQVRB]
      0x00000004 df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119 [AAAAAA==]
      PCR0: c4be5ba22b39e0c54a8488598830fb14a5774ec3362e55ffd2d2b19af44605d7
      PCR0: f2c4eb19c4f71680b3b5530467a38e1959c4c7ab35533ef6895b4d58ee99410f2f69fa666f6d3f3313eca26fc45550b2
      TpmFamily:        2.0
      BootMgrDesc:      legacy
      EfivarsNvramUsed: 50700
      SecureBoot:       Enabled
      UEFIUXCapsule:    Enabled
```

bbo-igel, 2025-04-03:
Hi @hughsie  We are  using the fwupd .deb package from debian sid.

I booted into a live testing and there the situation was the other way around:
Debian:
fwupdmgr get-devices shows all devices.
fwupdtool get-devices shows only a handful of devices. 

But with debian, I could solve this by running fwupdtool with sudo.

Is there any way to get more verbose debugging from fwupdmgr ?
fwupdtool -vv has tons of information, fwupdmgr -vv only adds very little.

superm1, 2025-04-03:
@bbo-igel please add `-vv` to the `fwupd.service` `ExecStart` and then share the journal output for `fwupd.service`.

bbo-igel, 2025-04-03:
Hi @superm1 Thanks for the advice!

I think we are getting closer:

```shell
[...]
Apr 03 19:24:38 ITCF4A80D794D89 fwupd[12097]: 17:24:38.018 FuEngine             disabling plugin because: failed to coldplug using uefi_capsule: /sys/firmware/efi/efivars is read only and no CoD
```

Seems like uefi_capsule update has been disabled.
We are indeed using DisableCapsuleUpdateOnDisk=true in fwupd.conf, because we had some issues with certain ThinClients that would not update.
But before starting the fwupd.service, we always make sure to mount efivars as rw.

```shell
mount | grep efivars
efivarfs on /sys/firmware/efi/efivars type efivarfs (rw,nosuid,nodev,noexec,relatime)
```

Gotta have to look in the code what is checked to determine that efivars is read only

Complete journal output since restart:
```shell
Apr 03 19:24:13 ITCF4A80D794D89 systemd[1]: Reloading finished in 894 ms.
Apr 03 19:24:35 ITCF4A80D794D89 systemd[1]: Starting modprobe@sd_mod.service - Load Kernel Module sd_mod...
Apr 03 19:24:35 ITCF4A80D794D89 systemd[1]: modprobe@sd_mod.service: Deactivated successfully.
Apr 03 19:24:35 ITCF4A80D794D89 systemd[1]: Finished modprobe@sd_mod.service - Load Kernel Module sd_mod.
Apr 03 19:24:35 ITCF4A80D794D89 systemd[1]: Starting fwupd.service - Firmware update daemon...
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.886 FuDebug              verbose to info (on console 0)
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.887 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.887 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.887 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.888 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.888 FuEngine             client certificate now exists: nothing to do
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.893 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.893 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS Manufacturer=LENOVO
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS EnclosureKind=a
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS Family=ThinkPad L14 Gen 4
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS ProductName=21H6S05J00
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS ProductSku=LENOVO_MT_21H6_BU_Think_FM_ThinkPad L14 Gen 4
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS BiosVendor=LENOVO
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS BiosVersion=R25ET40W (1.21 )
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS BiosMajorRelease=01
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS BiosMinorRelease=15
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS FirmwareMajorRelease=01
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS FirmwareMinorRelease=0a
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS BaseboardManufacturer=LENOVO
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            SMBIOS BaseboardProduct=21H6S05J00
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.894 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb
Apr 03 19:24:35 ITCF4A80D794D89 fwupd[12097]: 17:24:35.939 FuContext            added udev subsystem watch of firmware-attributes
Apr 03 19:24:37 ITCF4A80D794D89 fwupd[12097]: 17:24:37.658 FuBiosSettings       loaded 88 BIOS settings
Apr 03 19:24:37 ITCF4A80D794D89 fwupd[12097]: 17:24:37.658 FuEngine             loading metadata for remote 'vendor-directory'
Apr 03 19:24:37 ITCF4A80D794D89 fwupd[12097]: 17:24:37.659 FuEngine             ignoring: /usr/share/fwupd/remotes.d/vendor/firmware/README.md
Apr 03 19:24:37 ITCF4A80D794D89 fwupd[12097]: 17:24:37.659 FuEngine             ignoring: Error opening directory "/var/lib/fwupd/local.d": No such file or directory
Apr 03 19:24:37 ITCF4A80D794D89 fwupd[12097]: 17:24:37.659 FuEngine             ignoring: Error opening directory "/usr/share/fwupd/local.d": No such file or directory
Apr 03 19:24:37 ITCF4A80D794D89 fwupd[12097]: 17:24:37.666 FuContext            added udev subsystem watch of usb
Apr 03 19:24:37 ITCF4A80D794D89 fwupd[12097]: 17:24:37.666 FuEngine             failed to setup backend udev: failed to set up netlink: bind to udev socket failed: Address already in use
Apr 03 19:24:37 ITCF4A80D794D89 fwupd[12097]: 17:24:37.669 FuContext            added udev subsystem watch of tty
Apr 03 19:24:37 ITCF4A80D794D89 fwupd[12097]: 17:24:37.669 FuContext            added udev subsystem watch of usbmisc
Apr 03 19:24:37 ITCF4A80D794D89 fwupd[12097]: 17:24:37.669 FuContext            added udev subsystem watch of wwan
[... cut ...]
Apr 03 19:24:38 ITCF4A80D794D89 fwupd[12097]: 17:24:38.023 FuDeviceList         ::added 362301da643102b9f38477387e2193e57abaa590 [UEFI dbx]
Apr 03 19:24:38 ITCF4A80D794D89 fwupd[12097]: 17:24:38.025 FuDeviceList         ::added 6924110cde4fa051bfdc600a60620dc7aa9d3c6a [UEFI Platform Key]
Apr 03 19:24:38 ITCF4A80D794D89 fwupd[12097]: 17:24:38.025 FuEngine             resetting update motd timeout
Apr 03 19:24:38 ITCF4A80D794D89 fwupd[12097]: 17:24:38.027 FuMain               Daemon ready for requests (locale en_US.UTF-8)
Apr 03 19:24:38 ITCF4A80D794D89 systemd[1]: Started fwupd.service - Firmware update daemon.
```

superm1, 2025-04-03:
Are you modifying the systemd unit in any other way perhaps?

superm1, 2025-04-03:
>  we had some issues with certain ThinClients that would not update.

Can you please share more about this?  We've got safety measures in fwupd to ensure CoD works properly.  There was a bug on older Insyde based BIOS, but we have a way to detect it's [functional now on Insyde](https://github.com/fwupd/fwupd/blob/d45f027b4d344ea8b9fc5f7cb65787127eb0f4c7/plugins/uefi-capsule/fu-acpi-uefi.c#L69).

bbo-igel, 2025-04-03:
> Can you please share more about this? We've got safety measures in fwupd to ensure CoD works properly. There was a bug on older Insyde based BIOS, but we have a way to detect it's [functional now on [...]

It was an older version of this devices firmware:
https://fwupd.org/lvfs/devices/com.lenovo.M2W.BIOS.firmware

It seems like Lenovo has removed the faulty firmware by now:
https://fwupd.org/downloads/7a40fc316662d933933793f8ddb09b3927f696832f6fffa896ec3d47698d3c41-ThinkCentre-Tiny8-SystemFirmware-M43KT28A.cab

Back then, one of our customers was only able to update by setting disabling CoD.

superm1, 2025-04-03:
> Back then, one of our customers was only able to update by setting disabling CoD.

Do you know what version of fwupd you had at that time?  Our quirk detection is [fwupd 1.8.11 or newer](https://github.com/fwupd/fwupd/commit/2b96b9e93358f361c0003a290e1bc0ea3032631c).

bbo-igel, 2025-04-03:
> > Back then, one of our customers was only able to update by setting disabling CoD.
> 
> Do you know what version of fwupd you had at that time? Our quirk detection is [fwupd 1.8.11 or newer](https://github.com/fwupd/fwupd/commit/2b96b9e93358f361c0003a290e1bc0ea3032631c).

According to my PR from back then, it was 1.9.1
But the BIOS was also not from Insyde but from AMI, therefore I am not sure if the quirk was even applicable.

superm1, 2025-04-03:
>  But the BIOS was also not from Insyde but from AMI, therefore I am not sure if the quirk was even applicable.

Ah then it's an issue we were never made aware of upstream.  In the future if cases like this crop up please bring them upstream, it's best we dig into them while _they're hot_ and get quirks put together if applicable.

bbo-igel, 2025-04-03:
With the old fwupd.service, the problem is gone!

Thanks for all the help @superm1   👍

superm1, 2025-04-03:
> With the old fwupd.service, the problem is gone!

Great.  Please narrow down what line is causing it so we can determine next steps.

bbo-igel, 2025-04-04:
The ProtectSystem=full seems to be the culprit.
https://www.freedesktop.org/software/systemd/man/latest/systemd.exec.html#ProtectSystem=

For our OS, I found two possible workarounds:
1. Set it to strict (which does not make sense, because strict should have even less rw access then full)
2. Add -/sys/firmware/efi/efivars to ReadWritePaths=

I suspect this might be a bug with ProtectSystem=full in the systemd (254.22-1~bpo12+1) debian bookworm-backport package.

closed 2025-04-04 by superm1

superm1, 2025-04-04:
> I suspect this might be a bug with ProtectSystem=full in the systemd (254.22-1~bpo12+1) debian bookworm-backport package.

Yeah it must be if this works in Trixie.