fwupd/fwupd #8654: `fu_console_input_bool` does not terminate on EOF or ^C

zaz, 2025-04-07:
**Describe the bug**
`fu_console_input_bool` does not terminate on EOF or ^C. 

**Steps to Reproduce**

```console
$ fwupdmgr --json get-updates
Firmware metadata has not been updated for 30 days and may not be up to date.
Update now? (Requires internet connection) [y|N]: ^C^Z
```

pressing <kbd>Ctrl</kbd> + <kbd>d</kbd> or  <kbd>Ctrl</kbd> + <kbd>c</kbd> does not terminate the program.

**Expected behavior**
Exit with a special error code. This may be helpful [downstream](https://github.com/QubesOS/qubes-issues/issues/9820#issuecomment-2760925054).

**fwupd version information**
1.8.17

**Additional questions**

- Operating system and version:  Qubes OS 4.2.4
- Have you tried rebooting?  Yes
- Is this a regression?  Unlikely

hughsie, 2025-04-07:
We're doing a pretty dumb `fgets` at the moment -- do you have any references on how to switch to a non-blocking model? Last time I looked at this there were dragons everywhere.

superm1, 2025-04-07:
I see another problem here though - we shouldn't be prompting ANYTHING in `--json` mode.  That's meant for a computer to use, not a user.

zaz, 2025-04-07:
I'm afraid this is really not my specialty. Is there a reason to implement the UI so low-level? If not, maybe you could use readline or, better yet, move as much UI code as possible into the Python program (which would make it easier for downstream developers to interface with your program more smoothly). If it does need to be this low-level, then it's such a common task that I'm sure you could find good code for it somewhere (perhaps readline?). Sorry I couldn't be more help.

zaz, 2025-04-07:
[Related downstream commit](https://github.com/QubesOS/qubes-fwupd/commit/73cf5c881e5de1d034c13a4af0540117f107c052):

> 'fwupdmgr get-updates' needs to think it's running interactively to _list_ all updates, also those requiring user interaction. But at the same time, avoid hanging on a prompt, by connecting stdin to /dev/null.

superm1, 2025-04-07:
> I've opened this to adjust the --json behavior for the prompts you saw.
> 
> [#8656](https://github.com/fwupd/fwupd/pull/8656)

That's merged, so `--json` shouldn't prompt you anymore to do updates.

As for using readline(), can you see if https://github.com/fwupd/fwupd/pull/8659 behaves more like you expect it to?

closed 2025-04-10 by superm1