fwupd/fwupd #8657: boot entry for Linux-Firmware-Updates - Missing file to  does not

bbo-igel, 2025-04-07:
**Describe the question**
Still working on the fwupd migration from 1.9.26 to 2.0.7 on igel os.
With SecureBoot off, everything is working fine.
With SecureBoot enabled, the update is not triggered. The reboot boot into the OS without any error message.

I compared with 1.9.26 (where fw updates with Secure Boots did work fine on igel os)

```shell
1.9.26
efibootmgr (after running fwupdmgr local-install xxx.cab)

BootNext: 0002
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0001,0010,0011,0012,0013,0014,0015,0016,0017,0018,001B,001C,001D,001E,001F,0020,0021,0022,0023,0002
Boot0000* IGEL_EFI_2	HD(3,GPT,93531061-2588-4985-9f44-4ade0a806636,0x69800,0x64000)/File(\efi\boot\bootx64.efi)
Boot0001* IGEL_EFI_1	HD(2,GPT,71654c9d-fa48-47d2-9453-cac46c31efa8,0x5800,0x64000)/File(\efi\boot\bootx64.efi)
Boot0002* Linux-Firmware-Updater	HD(2,GPT,71654c9d-fa48-47d2-9453-cac46c31efa8,0x5800,0x64000)/File(\EFI\boot\shimx64.efi) File(.\fwupdx64.efi)
[...]
```
This is working and seems logical to me, boot0002 is using shimx64.efi to chainload the fwupdx64.efi, which is the signed bootloader for fwupd.

```shell
2.0.7
efibootmgr (after running fwupdmgr local-install xxx.cab)

BootNext: 0002
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0001,0010,0011,0012,0013,0014,0015,0016,0017,0018,001B,001C,001D,001E,001F,0020,0021,0022,0023,0002
Boot0000* IGEL_EFI_2	HD(3,GPT,93531061-2588-4985-9f44-4ade0a806636,0x69800,0x64000)/File(\efi\boot\bootx64.efi)
Boot0001* IGEL_EFI_1	HD(2,GPT,71654c9d-fa48-47d2-9453-cac46c31efa8,0x5800,0x64000)/File(\efi\boot\bootx64.efi)
Boot0002* Linux-Firmware-Updater	HD(2,GPT,71654c9d-fa48-47d2-9453-cac46c31efa8,0x5800,0x64000)/File(\EFI\boot\shimx64.efi)
[...]
```
Is this the expected result for 2.0.7? There is no mention of the fwupdx64.efi

Output of the verbose journal:
```shell
Apr 07 14:40:11 ITCF4A80D794D89 systemd[1]: Starting fwupd.service - Firmware update daemon...
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.911 FuDebug              verbose to info (on console 0)
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.912 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.912 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.912 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.912 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.913 FuEngine             client certificate now exists: nothing to do
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.918 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.918 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS Manufacturer=LENOVO
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS EnclosureKind=a
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS Family=ThinkPad L14 Gen 4
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS ProductName=21H6S05J00
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS ProductSku=LENOVO_MT_21H6_BU_Think_FM_ThinkPad L14 Gen 4
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS BiosVendor=LENOVO
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS BiosVersion=R25ET42W (1.23 )
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS BiosMajorRelease=01
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS BiosMinorRelease=17
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS FirmwareMajorRelease=01
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS FirmwareMinorRelease=0c
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS BaseboardManufacturer=LENOVO
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            SMBIOS BaseboardProduct=21H6S05J00
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.919 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb
Apr 07 14:40:11 ITCF4A80D794D89 fwupd[14787]: 12:40:11.966 FuContext            added udev subsystem watch of firmware-attributes
Apr 07 14:40:13 ITCF4A80D794D89 fwupd[14787]: 12:40:13.686 FuBiosSettings       loaded 88 BIOS settings
Apr 07 14:40:13 ITCF4A80D794D89 fwupd[14787]: 12:40:13.686 FuEngine             loading metadata for remote 'vendor-directory'
Apr 07 14:40:13 ITCF4A80D794D89 fwupd[14787]: 12:40:13.686 FuEngine             ignoring: /usr/share/fwupd/remotes.d/vendor/firmware/README.md
Apr 07 14:40:13 ITCF4A80D794D89 fwupd[14787]: 12:40:13.686 FuEngine             ignoring: Error opening directory "/var/lib/fwupd/local.d": No such file or directory
Apr 07 14:40:13 ITCF4A80D794D89 fwupd[14787]: 12:40:13.686 FuEngine             ignoring: Error opening directory "/usr/share/fwupd/local.d": No such file or directory
Apr 07 14:40:13 ITCF4A80D794D89 fwupd[14787]: 12:40:13.693 FuContext            added udev subsystem watch of usb
Apr 07 14:40:13 ITCF4A80D794D89 fwupd[14787]: 12:40:13.694 FuEngine             failed to setup backend udev: failed to set up netlink: bind to udev socket failed: Address already in use
[... 357 lines cut ...]
Apr 07 14:40:31 ITCF4A80D794D89 fwupd[14787]: 12:40:31.384 FuDeviceList         no replug or re-enumerate required
Apr 07 14:40:31 ITCF4A80D794D89 fwupd[14787]: 12:40:31.384 FuEngine             install phase now setup
Apr 07 14:40:31 ITCF4A80D794D89 fwupd[14787]: 12:40:31.384 FuEngine             resetting update motd timeout
Apr 07 14:40:35 ITCF4A80D794D89 polkitd[6930]: Unregistered Authentication Agent for unix-process:14855:302800 (system bus name :1.127, object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8) (disconnected from bus)

[...]
```

**fwupd version information**
```shell
fwupdmgr --version
compile   info.libusb                   1.0.26
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         2.0.7
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.12.20
runtime   org.freedesktop.fwupd         2.0.7
```

```shell
fwupdtool get-report-metadata
```

superm1, 2025-04-07:
Can you please try to reproduce on main?  Or at least backport this patch https://github.com/cysk003/fwupd/commit/5dc6d189ad3e5f72afb389cd0c42305a7f9a9e53?
We have a regression in 2.0.x that is supposed to be fixed in the upcoming 2.0.8, planning to tag this week.

bbo-igel, 2025-04-07:
Yes I will backport the patch and report the result here.

bbo-igel, 2025-04-08:
Backport solved the issue

closed 2025-04-08 by bbo-igel