fwupd/fwupd #8660: `error: unknown enumerated scalar` when committing main on Fedora 41

Arno500, 2025-04-07:
**Describe the bug**
When committing something, the check hooks tries to run "clang-format", but it fails with:
```
Failed to run ['clang-format-diff', '-i', '-regex', '^.*\\.(c|h|proto)$', '-p1']
/home/arno/custom-code/fwupd/.clang-format:39:11: error: unknown enumerated scalar
Language: 'C'
          ^~~
Error reading /home/arno/custom-code/fwupd/.clang-format: Invalid argument
```

**Steps to Reproduce**
Just try to commit something

Removing the "Language: C" section from the `.clang-format` file unblock the formatting.

**Expected behavior**
Work :P

**Additional questions**

- Operating system and version: Fedora 41, up to date. (clang-format version 19.1.7 (Fedora 19.1.7-3.fc41), which seems the latest in the packages)

hughsie, 2025-04-07:
Urgh, @superm1 should I revert https://github.com/fwupd/fwupd/commit/cc58c36b0f58bf6d9a74ac89734923d4914d12b1 until we work out a solution for both old and new clang's?

superm1, 2025-04-07:
Yes; please do.

closed 2025-04-07 by hughsie