fwupd/fwupd #8661: Redfish apply mode choice

Arno500, 2025-04-07:
**Is your feature request related to a problem? Please describe.**
When installing a Redfish firmware, the hardcoded `@Redfish.OperationApplyTime` is `Immediate` which is fiiiiiine I guess, but if you install a firmware that requires a reboot (like basically most firmware on Dell machines), then the machine shutdown directly by itself at the end of the upload of one firmware (and does not ask you anything, the reboot request is sent by the BMC). Which I suppose may not necessarily be desired. 

**Describe the solution you'd like**
I think a flag, or a config option may be in order to allow the user to chose what to do here. Or move to OnReset and have a specific way to request a reboot to update the firmware (but from what I gather, the reboot obligation is not always correctly specified in the metadata catalogs on my side).

**Describe alternatives you've considered**
None

**Additional context**
https://www.dmtf.org/sites/default/files/standards/documents/DSP2062_1.0.2.pdf page 16

hughsie, 2025-04-07:
The intention is that the update is *applied* immediately, but a reboot immediately isn't quite what we expected.

I don't think we want a config option, this stuff just has to "just work" magically. I think at the very least the metadata needs to be correct.

Arno500, 2025-04-07:
That seems fair. For HPE servers the option (using their proprietary implem) does not have the setting (basically immediately applies, never forces to reboot, but can indicates it needs to afterwards). For Dell, I will test tomorrow how the "OnReset" applies to things like iDRAC that do not require a system reset.
Usually firmware on Dell servers are apparently installed using a UEFI capsule (it seems) that takes the file from somewhere (???) and flashes it directly on the device, thus needing a reboot (and that will reboot afterward). It seems that contrary to HPE, they cannot really (or don't want to) update the devices directly from the BMC using PLDM despite their metadata catalog saying so (and the PLDM files included in the archive).

Arno500, 2025-04-15:
Alright, I tested the `OnReset` and it does what I thought: for the iDRAC it installed it immediately, and for other firmware it did not force a restart. I do not know if this behavior is similar amongst all manufacturer. Maybe I should just "if" that in the code to make it it only applies on Dell servers?

hughsie, 2025-07-03:
@Arno500 we can close this now?

Arno500, 2025-07-03:
Yes, sorry, forgot to clean this one

closed 2025-07-03 by Arno500