fwupd/fwupd #8667: Offline update - Version format from .cab not used for device - number vs triplet

bbo-igel, 2025-04-08:
**Describe the bug**
Offline update does not inherit the version format from the .cab file
Lenovo L14 GEN 4 (AMD)

**Steps to Reproduce**
- Download cab file from fwupd.org
- fwupdtool install MyFirmware.cab

```shell
07:54:33.235 FuDebug              verbose to info (on console 0)
07:54:33.235 FuEngine             starting fwupd 2.0.7...
[GLoading…                 [ -                                     ][GLoading…                 [ -                                     ]07:54:33.238 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
07:54:33.239 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
07:54:33.239 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
07:54:33.239 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
07:54:33.239 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
[GLoading…                 [                                       ][GLoading…                 [                                       ]07:54:33.245 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
07:54:33.245 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
07:54:33.246 FuContext            SMBIOS Manufacturer=LENOVO
07:54:33.246 FuContext            SMBIOS EnclosureKind=a
07:54:33.246 FuContext            SMBIOS Family=ThinkPad L14 Gen 4
07:54:33.246 FuContext            SMBIOS ProductName=21H6S05J00
07:54:33.246 FuContext            SMBIOS ProductSku=LENOVO_MT_21H6_BU_Think_FM_ThinkPad L14 Gen 4
07:54:33.246 FuContext            SMBIOS BiosVendor=LENOVO
07:54:33.246 FuContext            SMBIOS BiosVersion=R25ET42W (1.23 )
07:54:33.246 FuContext            SMBIOS BiosMajorRelease=01
07:54:33.246 FuContext            SMBIOS BiosMinorRelease=17
07:54:33.246 FuContext            SMBIOS FirmwareMajorRelease=01
07:54:33.246 FuContext            SMBIOS FirmwareMinorRelease=0c
07:54:33.246 FuContext            SMBIOS BaseboardManufacturer=LENOVO
07:54:33.246 FuContext            SMBIOS BaseboardProduct=21H6S05J00
07:54:33.246 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb
07:54:33.293 FuContext            added udev subsystem watch of firmware-attributes
07:54:35.014 FuBiosSettings       loaded 88 BIOS settings
[GLoading…                 [*                                      ]07:54:35.014 FuEngine             loading metadata for remote 'vendor-directory'
07:54:35.014 FuEngine             ignoring: /usr/share/fwupd/remotes.d/vendor/firmware/README.md
07:54:35.014 FuEngine             ignoring: Error opening directory "/var/lib/fwupd/local.d": No such file or directory
07:54:35.014 FuEngine             ignoring: Error opening directory "/usr/share/fwupd/local.d": No such file or directory
07:54:35.020 FuContext            added udev subsystem watch of usb
07:54:35.020 FuEngine             failed to setup backend udev: failed to set up netlink: bind to udev socket failed: Address already in use
[GLoading…                 [*                                      ]07:54:35.023 FuContext            added udev subsystem watch of tty
07:54:35.023 FuContext            added udev subsystem watch of usbmisc
07:54:35.023 FuContext            added udev subsystem watch of wwan
07:54:35.023 FuContext            added udev subsystem watch of drm
07:54:35.023 FuContext            added udev subsystem watch of block
07:54:35.023 FuContext            added udev subsystem watch of block:disk for plugin algoltek_usbcr
07:54:35.023 FuContext            added udev subsystem watch of i2c
07:54:35.024 FuContext            added udev subsystem watch of platform
07:54:35.024 FuContext            added udev subsystem watch of pci
07:54:35.024 FuContext            added udev subsystem watch of block:partition for plugin android_boot
07:54:35.024 FuContext            added udev subsystem watch of drm_dp_aux_dev for plugin bnr_dp
07:54:35.024 FuContext            added udev subsystem watch of i2c-dev
07:54:35.024 FuContext            added udev subsystem watch of hidraw
07:54:35.024 FuContext            added udev subsystem watch of gpio for plugin gpio
07:54:35.024 FuContext            added udev subsystem watch of mei
07:54:35.024 FuContext            added udev subsystem watch of iommu for plugin iommu
07:54:35.024 FuContext            added udev subsystem watch of video4linux
07:54:35.024 FuContext            added udev subsystem watch of msr for plugin msr
07:54:35.024 FuContext            added udev subsystem watch of mtd for plugin mtd
07:54:35.024 FuContext            added udev subsystem watch of nvme for plugin nvme
07:54:35.024 FuContext            added udev subsystem watch of serio
07:54:35.025 FuContext            added udev subsystem watch of thunderbolt
07:54:35.025 FuContext            added udev subsystem watch of tpm for plugin tpm
07:54:35.064 FuVolume             Looking for volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
07:54:35.076 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p4, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: 
07:54:35.091 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
07:54:35.098 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: ebd0a0a2-b9e5-4433-87c0-68b6b72699c7, internal: 1, fs: 
07:54:35.136 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
07:54:35.194 FuVolume             Looking for volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
07:54:35.206 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p4, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: 
[... 213 lines cut ...]
      BootMgrDesc:      legacy
      EfivarsNvramUsed: 50700
      SecureBoot:       Enabled
      UEFIUXCapsule:    Enabled
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
debian sid .deb package integrated into igel os
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version: igel os 12.7
- Have you tried rebooting? yes
- Is this a regression? from 1.9.26

hughsie, 2025-04-08:
> fwupdtool install MyFirmware.cab

Why `fwuptool` instead of `fwupdmgr`? Does the latter work?

> verified metadata

So we're reading everything correctly, I think that `fwuptool` is just loading the release without the engine wrapper (that calls fu_device_ensure_from_component and sets the verfmt). PR incoming.

bbo-igel, 2025-04-08:
> > fwupdtool install MyFirmware.cab
> 
> Why `fwuptool` instead of `fwupdmgr`? Does the latter work?

Hi @hughsie Just for convenience, because it is easier to get debug output.
fwupdmgr install xxx.cab shows the same problem.

```shell
journal
[...]
Apr 08 13:39:18 ITCF4A80D794D89 systemd[1]: Started fwupd.service - Firmware update daemon.
Apr 08 13:39:40 ITCF4A80D794D89 polkitd[6125]: Registered Authentication Agent for unix-process:12047:23046 (system bus name :1.101 [/usr/bin/pkttyagent --notify-fd 6 --fallback], object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8)
Apr 08 13:39:40 ITCF4A80D794D89 fwupd[12028]: 11:39:40.804 FuPluginLinuxLockdown failed to ensure attribute fix flags: missing executable grubby in PATH
Apr 08 13:39:40 ITCF4A80D794D89 fwupd[12028]: 11:39:40.804 FuPluginMsr          vulnerable to sinkclose
Apr 08 13:39:40 ITCF4A80D794D89 fwupd[12028]: 11:39:40.819 FuEngine             skipping writing HSI attrs to database as unchanged
Apr 08 13:39:40 ITCF4A80D794D89 fwupd[12028]: 11:39:40.868 FuCabinet            verified metadata firmware.metainfo.xml: 3
Apr 08 13:39:40 ITCF4A80D794D89 fwupd[12028]: 11:39:40.868 FuCabinet            processing file: firmware.metainfo.xml
Apr 08 13:39:40 ITCF4A80D794D89 fwupd[12028]: 11:39:40.869 FuCabinet            processing release: 0.1.12
Apr 08 13:39:40 ITCF4A80D794D89 fwupd[12028]: 11:39:40.886 FuCabinet            verified payload firmware.bin: 3
Apr 08 13:39:40 ITCF4A80D794D89 polkitd[6125]: Unregistered Authentication Agent for unix-process:12047:23046 (system bus name :1.101, object path /org/freedesktop/PolicyKit1/AuthenticationAgent, locale en_US.UTF-8) (disconnected from bus)
```

```shell
command line:
Decompressing...
Specified firmware is older than installed '0.1.12 < 65548'
```

hughsie, 2025-04-08:
> Specified firmware is older than installed '0.1.12 < 65548'

Ohh I see the problem -- I'll update the other PR.

bbo-igel, 2025-04-08:
> Is this CAB from lvfs or directly from the vendor?

It is directly from LVFS.