fwupd/fwupd #8676: fwupdmgr Error opening file /boot/efi/EFI/ubuntu/shimx64.efi: Permission denied

david-robson, 2025-04-08:
**Describe the bug**

Running "sudo fwupdmgr update" gives the output:
``` bash
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 20230501 to 20241101?                                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ An insecure version of Howyar's SysReturn software was added, due to a       ║
║ security vulnerability that allowed an attacker to bypass UEFI Secure Boot.  ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Verifying…               [                                       ]
Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /boot/efi/EFI/ubuntu/shimx64.efi: Error opening file /boot/efi/EFI/ubuntu/shimx64.efi: Permission denied
```

Running "fwupdmgr get-device" gives the output:

``` bash
└─System Firmware:
  │   Device ID:          69c2e00c41ae1d9de59f8189e14a063ce5a5bbe2
  │   Summary:            UEFI System Resource Table device (updated via NVRAM)
  │   Current version:    69632
  │   Minimum Version:    69632
  │   Vendor:             Dell Inc. (DMI:Dell Inc.)
  │   Update State:       Success
  │   GUID:               6f47d9b5-41a8-422b-8dd7-9d779c74ca55
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │   Device Requests:    • Message
  │ 
  └─UEFI dbx:
        Device ID:        362301da643102b9f38477387e2193e57abaa590
        Summary:          UEFI revocation database
        Current version:  20230501
        Minimum Version:  20230501
        Vendor:           UEFI:Microsoft
        Install Duration: 1 second
        Update State:     Failed
        Update Error:     Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /boot/efi/EFI/ubuntu/shimx64.efi: Error opening file /boot/efi/EFI/ubuntu/shimx64.efi: Permission denied
        Last modified:    2025-04-08
        GUIDs:            4a6cd2cb-8741-5257-9d1f-89a275dacca7 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9&ARCH_X64
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
        Device Flags:     • Internal device
                          • Updatable
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Cryptographic hash verification is available
                          • Device is usable for the duration of the update
                          • Only version upgrades are allowed
                          • Signed Payload
                          • Can tag for emulation

────────────────────────────────────────────────
Devices that were not updated correctly:
 • UEFI dbx (20230501 → 20241101)
Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Review and upload report now? (Requires internet connection) [Y|n]:
```
I uploaded a report.

**Steps to Reproduce**

``` bash
[... 882 lines cut ...]
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo

```
- Have you tried rebooting?
Yes
- 
- Is this a regression?
I believe so, the UEFI update has run previously.

- Are you using an NVMe disk?
Yes.
 
- Is secure boot enabled?
No.

- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
No.

This is a dual boot system with Windows 10. It upgraded to Windows 22H2 a few days ago with no issue.

superm1, 2025-04-09:
This is the same as https://github.com/fwupd/fwupd/issues/8529.  You'll need to upgrade to a new enough snapd.

closed 2025-04-09 by superm1

david-robson, 2025-04-09:
Sorry I missed the duplication.

Before update: snap was version **2.67**
Updated snapd using `snap refresh --beta snapd`
After update: snap is version **2.68.4**
Running `sudo fwupdmgr update` without rebooting gives the same error. 
Rebooted.
The update now runs without error.  However, the output of `sudo fwupdmgr get-updates` shows a warning/error:
``` shell
Dell Inc. Inspiron 7786
│
└─UEFI dbx:
  │   Device ID:          362301da643102b9f38477387e2193e57abaa590
  │   Summary:            UEFI revocation database
  │   Current version:    20230501
  │   Minimum Version:    20230501
  │   Vendor:             UEFI:Microsoft
  │   Install Duration:   1 second
  │   Update Error:       Snapd integration for DBX update is not available
  │   GUIDs:              4a6cd2cb-8741-5257-9d1f-89a275dacca7 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9&ARCH_X64
  │                       f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
...
```

Is this an issue?

superm1, 2025-04-09:
> Is this an issue?

That means that the dbx snapd integration isn't present.  Open another ticket and we can pull in those guys who worked on it to confirm details.

superm1, 2025-10-22:
what version of snapd are you on?  This should have been fixed by a new enough snapd.