fwupd/fwupd #8677: No VENDOR for PK on Lenovo Workstation

superm1, 2025-04-09:
On my P620 workstation I'm seeing this.

```
04:39:35.790 FuEngine             failed to add device 8be4df61-93ca-11d2-aa0d-00e098032b8c-PK: failed to probe: no value for VENDOR
```

I double checked with a debugger and I only see

```0x5555557d5810 "CN=Lenovo_Workstation"```

so there really is no `O=` key.

iddpd, 2025-04-11:
I'm seeing the same message, accompanied by "failed to setup" concerning KEK and db, after updating to 2.0.8 on ArchLinux.

```
07:11:27.919 FuEngine             failed to add device 8be4df61-93ca-11d2-aa0d-00e098032b8c-KEK: failed to setup: no value for VENDOR
07:11:27.921 FuEngine             failed to add device 8be4df61-93ca-11d2-aa0d-00e098032b8c-PK: failed to probe: no value for VENDOR
07:11:27.923 FuEngine             failed to add device d719b2cb-3d3a-4596-a3bc-dad00e67656f-db: failed to setup: no value for VENDOR 
```

My system is set up with custom secure boot keys, created and enrolled with the help of **sbctl**.

```
Variable PK, length 1300
PK: List 0, type X509
    Signature 0, size 1272, owner 2d2c8d0b-c85c-4b55-9a63-33e24515f0f1
        Subject:
            C=Platform Key, CN=Platform Key
        Issuer:
            C=Platform Key, CN=Platform Key
Variable KEK, length 2876
KEK: List 0, type X509
    Signature 0, size 1288, owner 2d2c8d0b-c85c-4b55-9a63-33e24515f0f1
        Subject:
            C=Key Exchange Key, CN=Key Exchange Key
        Issuer:
            C=Key Exchange Key, CN=Key Exchange Key
KEK: List 1, type X509
    Signature 0, size 1532, owner 77fa9abd-0359-4d32-bd60-28f4e78f784b
        Subject:
            C=US, ST=Washington, L=Redmond, O=Microsoft Corporation, CN=Microsoft Corporation KEK CA 2011
        Issuer:
            C=US, ST=Washington, L=Redmond, O=Microsoft Corporation, CN=Microsoft Corporation Third Party Marketplace Root
Variable db, length 4444
db: List 0, type X509
    Signature 0, size 1273, owner 2d2c8d0b-c85c-4b55-9a63-33e24515f0f1
        Subject:
            C=Database Key, CN=Database Key
        Issuer:
            C=Database Key, CN=Database Key
db: List 1, type X509
    Signature 0, size 1572, owner 77fa9abd-0359-4d32-bd60-28f4e78f784b
        Subject:
            C=US, ST=Washington, L=Redmond, O=Microsoft Corporation, CN=Microsoft Corporation UEFI CA 2011
        Issuer:
            C=US, ST=Washington, L=Redmond, O=Microsoft Corporation, CN=Microsoft Corporation Third Party Marketplace Root
db: List 2, type X509
    Signature 0, size 1515, owner 77fa9abd-0359-4d32-bd60-28f4e78f784b
        Subject:
            C=US, ST=Washington, L=Redmond, O=Microsoft Corporation, CN=Microsoft Windows Production PCA 2011
        Issuer:
            C=US, ST=Washington, L=Redmond, O=Microsoft Corporation, CN=Microsoft Root Certificate Authority 2010
```

I noticed that in this state `org.fwupd.hsi.Uefi.Pk` is no longer reported: [fwupd-security.json](https://github.com/user-attachments/files/19703173/fwupd-security.json). In turn this seems to cause GNOME Settings seems to be confused about the state of secure boot as well.

[image]

hughsie, 2025-04-11:
If we have no organisation then how do we restrict the vendors that are allowed to propose updates? I know it's not going to work to deploy the Lenovo update on a self-enrolled PK. Maybe we just use CN in this case as a fallback?

closed 2025-04-11 by superm1