fwupd/fwupd #8681: Expected e75bfb656efc107d10c7a9a7e289f3dd7935431c2c6d4f1f9d0709df775d0953 and got 66efc2261f1b9c9167ae998b9d12a23adda95f8ee76e3b790b70c3807fa89d3e

CummingCowGirl, 2025-04-09:
Running Garuda Linux, did garuda-updates, ran topgrade which does this particular update and got

~~~
── 03:14:37 - Firmware upgrades ────────────────────────────────────────────────
Updating lvfs
Downloading…             [************************************** ]
metadata checksum expected e75bfb656efc107d10c7a9a7e289f3dd7935431c2c6d4f1f9d0709df775d0953 and got 66efc2261f1b9c9167ae998b9d12a23adda95f8ee76e3b790b70c3807fa89d3e
Firmware upgrades failed:
0: Command failed: `/usr/bin/fwupdmgr refresh`
1: `/usr/bin/fwupdmgr` failed: exit status: 1
~~~
**fwupd 2.0.7-1**
Please provide the version of the daemon and client.

```~  sudo fwupdtool get-report-metadata                                                                                                                    1 ✘  03:32:38 AM
[sudo] password for cccp:
Loading…                 [****                                   ]10:33:06.970 FuPluginUefiCapsule  skipping device that failed coldplug: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid
Loading…                 [**********************                 ]10:33:08.060 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:14:00.0/usb3/3-1/3-1.1/3-1.1:1.0/host12/target12:0:0/12:0:0:0/block/sda: failed to setup: failed to read version: command fail with status 2, senseKey 0x05, asc 0x20, ascq 0x00
10:33:08.075 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:14:00.0/usb3/3-1/3-1.2/3-1.2:1.0/host13/target13:0:0/13:0:0:0/block/sdb: failed to setup: failed to read version: command fail with status 2, senseKey 0x05, asc 0x20, ascq 0x00
10:33:08.091 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:0c.0/0000:14:00.0/usb3/3-1/3-1.3/3-1.3:1.0/host14/target14:0:0/14:0:0:0/block/sdc: failed to setup: failed to read version: command fail with status 2, senseKey 0x05, asc 0x20, ascq 0x00
Waiting…                 [*************************************  ]
BatteryLevel:           101
BatteryThreshold:       10
BootTime:               1744050466
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.Passim): 0.1.9
CompileVersion(org.freedesktop.fwupd): 2.0.7
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 7 7700X 8-Core Processor
DisplayState:           connected
DistroId:               garuda
DistroName:             Garuda Linux
DistroPrettyName:       Garuda Linux
FwupdSupported:         True
HostBaseboardManufacturer: Gigabyte Technology Co., Ltd.
HostBaseboardProduct:   X670 AORUS ELITE AX
HostBiosMajorRelease:   05
HostBiosMinorRelease:   22
HostBiosVendor:         American Megatrends International, LLC.
HostBiosVersion:        FB1
HostEnclosureKind:      3
HostFamily:             X670 MB
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            X670 AORUS ELITE AX
HostSku:                Default string
HostVendor:             Gigabyte Technology Co., Ltd.
KernelCmdline:          ibt=off
KernelName:             Linux
KernelVersion:          6.13.8-zen1-1-zen
LidState:               unknown
PassimDownloadSaving:   0
PlatformArchitecture:   x86_64
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.7
RuntimeVersion(org.freedesktop.fwupd-efi): 1.7
RuntimeVersion(org.kernel): 6.13.8-zen1-1-zen
LinuxLockdown:    none
Pcr0_SHA256:      a0e3787752f002877ad3a30560135f11b6064ec3e16ffb20d84a3a12060e647a
TpmEventLog:      0x00000008 d681837b3b8f5fd685bfad042ca78e6884e10e1c8f272dbccd2fd10be738a7fa [BSsQp8fZZUGBQCrd6Ur2PA==]
0x8000000a 107795fa88790c2287f632fce2ec776719c9e91de1f650f43c98f702b101b545 [D0ZWX01BSU5fV1JBUFBFUgAAAAEAAAAAAPC6AAAAAAA=]
0x00000004 df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119 [AAAAAA==]
PCR0: a0e3787752f002877ad3a30560135f11b6064ec3e16ffb20d84a3a12060e647a
TpmFamily:        2.0
EfivarsNvramUsed: 38219
SecureBoot:       Disabled
[... 155 lines cut ...]
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│                         • Can tag for emulation
│
└─TPM:
Device ID:          1d8d50a4dbc65618f5c399c2ae827b632b3ccc11
Current version:    6.31.0.6
Vendor:             Advanced Micro Devices, Inc. (TPM:AMD)
GUIDs:              9305de1c-1e12-5665-81c4-37f8e51219b8 ← TPM\VEN_AMD&DEV_0001
78a291ae-b499-5b0f-8f1d-74e1fefd0b1c ← TPM\VEN_AMD&MOD_AMD
65a3fced-b423-563f-8098-bf5c329fc063 ← TPM\VEN_AMD&DEV_0001&VER_2.0
5e704f0d-83cb-5364-8384-f46d725a23b8 ← TPM\VEN_AMD&MOD_AMD&VER_2.0
Device Flags:       • Internal device
• System requires external power source
• Needs a reboot after installation
• Device can recover flash failures
• Full disk encryption secrets may be invalidated when updating
• Signed Payload
• Can tag for emulation
```

Garuda Linux - Plasma desktop 
Rebooting after this report

CummingCowGirl, 2025-04-09:
OK just did my reboot and tried again and same error message.

FeelTheLemon, 2025-04-09:
same problem
fwupd 1.9.29 on Fedora 41

david-robson, 2025-04-09:
Updating to the beta version of snap fixed this issue for me.
Before update: `snap --version` was 2.67
Updated snapd using snap refresh --beta snapd
After update: `snap --version` is 2.68.4
Rebooted.
Now the update runs ok: `sudo fwupdmgr refresh --force && sudo fwupdmgr get-updates -y && sudo fwupdmgr update` 

Maybe the package was updated in between times. Maybe is is the new snap version...

I have created the alias below to update the apt libs, the snaps and the firmwares:
``` shell
alias update='sudo apt update && sudo aptitude full-upgrade && sudo snap refresh && fwupdmgr refresh --force && sudo fwupdmgr get-updates -y && sudo fwupdmgr update'
```

CummingCowGirl, 2025-04-10:
david took the below from you last post and worked like a charm. Ran it then topgrade and no issues.

~~~
fwupdmgr refresh --force && sudo fwupdmgr get-updates -y && sudo fwupdmgr update
~~~

closed 2026-03-02 by hughsie