fwupd/fwupd #8682: fwupd cannot find EFI partition despite being specified in configuration file and in a common location

mid-at-coding, 2025-04-09:
**Describe the bug**
A clear and concise description of what the bug is.
fwupd cannot find EFI partition, despite being in a common location (/boot), mounted, and specified in /etc/fwupd/fwupd.conf. Very sorry if there was some documentation I missed, but I feel like I went through all the steps I could find.

**Steps to Reproduce**
Steps to reproduce the behavior.
Mount boot partition, install grub, boot from EFISTUB instead because it's faster (might be my issue?)

**Expected behavior**
A clear and concise description of what you expected to happen.
fwupd to find my EFI partition or give useful debug information

**fwupd version information**
Please provide the version of the daemon and client.
compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.7
compile   com.hughsie.libxmlb           0.3.22
compile   org.freedesktop.Passim        0.1.9
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.9
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd         2.0.7
runtime   org.kernel                    6.13.8-zen1-1-zen

```shell
fwupdtool get-report-metadata
[sudo] password for emelia:
Loading…                 [****                                   ]13:15:06.255 FuPluginUefiCapsule  cannot find default ESP: No ESP or BDP found
Loading…                 [*********************                  ]13:15:07.021 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:c1:00.0/drm/card1/card1-eDP-1/drm_dp_aux0: failed to setup: failed to read device firmware header: failed to write 7 bytes to 16: Input/output error
Loading…                 [**********************                 ]13:15:07.145 FuPluginDfu          failed to parse interface data: no image idx 33 found in firmware
13:15:07.145 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
Loading…                 [*************************************  ]
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Waiting…                 [*************************************  ]
BatteryLevel:           42
BatteryThreshold:       10
BootTime:               1744203312
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.Passim): 0.1.9
CompileVersion(org.freedesktop.fwupd): 2.0.7
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 5 7640U w/ Radeon 760M Graphics
DisplayState:           connected
DistroId:               arch
DistroName:             Arch Linux
DistroPrettyName:       Arch Linux
FwupdSupported:         True
HostBaseboardManufacturer: Framework
HostBaseboardProduct:   FRANMDCP05
HostBiosMajorRelease:   03
HostBiosMinorRelease:   05
HostBiosVendor:         INSYDE Corp.
HostBiosVersion:        3.05
HostEnclosureKind:      a
HostFamily:             Laptop
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            Laptop 13 (AMD Ryzen 7040Series)
HostSku:                FRANDRCP05
HostVendor:             Framework
KernelName:             Linux
KernelVersion:          6.13.8-zen1-1-zen
LidState:               open
PassimDownloadSaving:   0
PlatformArchitecture:   x86_64
[... 636 lines cut ...]
│       ├── wrmsr.mod
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initramfs-linux-zen-fallback.img
├── initramfs-linux-zen.img
└── vmlinuz-linux-zen

9 directories, 371 files
```

**Additional questions**

- Operating system and version: Arch linux, kernel version 6.13.8 zen
- Have you tried rebooting? Yes
- Is this a regression? No
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

mid-at-coding, 2025-04-09:
udiskctl isn't a command, despite udisks2 being installed, is there some other package I have to install? Why isn't it a dependency?

mid-at-coding, 2025-04-09:
oops nevermind misread -- here is the output

[out.txt](https://github.com/user-attachments/files/19669459/out.txt)

superm1, 2025-04-09:
what does `fwupdtool esp-list -v` show?

mid-at-coding, 2025-04-09:
Of course!
```
15:25:17.293 FuDebug              verbose to info (on console 1)
15:25:17.294 FuEngine             starting fwupd 2.0.7…
Loading…                 [*                                      ]15:25:17.321 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
15:25:17.321 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
Loading…                 [*                                      ]15:25:17.323 FuContext            SMBIOS Manufacturer=Framework
15:25:17.323 FuContext            SMBIOS EnclosureKind=a
15:25:17.323 FuContext            SMBIOS Family=Laptop
15:25:17.323 FuContext            SMBIOS ProductName=Laptop 13 (AMD Ryzen 7040Series)
15:25:17.323 FuContext            SMBIOS ProductSku=FRANDRCP05
15:25:17.323 FuContext            SMBIOS BiosVendor=INSYDE Corp.
15:25:17.323 FuContext            SMBIOS BiosVersion=03.05
15:25:17.323 FuContext            SMBIOS BiosMajorRelease=03
15:25:17.323 FuContext            SMBIOS BiosMinorRelease=05
15:25:17.323 FuContext            SMBIOS FirmwareMajorRelease=ff
15:25:17.323 FuContext            SMBIOS FirmwareMinorRelease=ff
15:25:17.323 FuContext            SMBIOS BaseboardManufacturer=Framework
15:25:17.323 FuContext            SMBIOS BaseboardProduct=FRANMDCP05
15:25:17.323 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb
15:25:17.328 FuContext            added udev subsystem watch of firmware-attributes
Loading…                 [**                                     ]15:25:17.328 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory
15:25:17.328 FuEngine             ignoring: Error opening directory “/usr/share/fwupd/local.d”: No such file or directory
Loading…                 [***                                    ]15:25:17.329 FuContext            added udev subsystem watch of tty
15:25:17.329 FuContext            added udev subsystem watch of usbmisc
15:25:17.329 FuContext            added udev subsystem watch of wwan
15:25:17.329 FuContext            added udev subsystem watch of drm
15:25:17.330 FuContext            added udev subsystem watch of block
15:25:17.330 FuContext            added udev subsystem watch of block:disk for plugin algoltek_usbcr
15:25:17.330 FuContext            added udev subsystem watch of i2c
15:25:17.330 FuContext            added udev subsystem watch of platform
15:25:17.330 FuContext            added udev subsystem watch of pci
15:25:17.330 FuContext            added udev subsystem watch of block:partition for plugin android_boot
15:25:17.331 FuContext            added udev subsystem watch of drm_dp_aux_dev for plugin bnr_dp
15:25:17.331 FuContext            added udev subsystem watch of i2c-dev
15:25:17.331 FuContext            added udev subsystem watch of hidraw
15:25:17.331 FuContext            added udev subsystem watch of gpio for plugin gpio
15:25:17.331 FuContext            added udev subsystem watch of mei
15:25:17.331 FuContext            added udev subsystem watch of iommu for plugin iommu
15:25:17.331 FuContext            added udev subsystem watch of video4linux
15:25:17.331 FuContext            added udev subsystem watch of msr for plugin msr
15:25:17.332 FuContext            added udev subsystem watch of mtd for plugin mtd
15:25:17.332 FuContext            added udev subsystem watch of nvme for plugin nvme
Loading…                 [***                                    ]15:25:17.333 FuContext            added udev subsystem watch of serio
15:25:17.333 FuContext            added udev subsystem watch of thunderbolt
15:25:17.333 FuContext            added udev subsystem watch of tpm for plugin tpm
15:25:17.336 FuVolume             Looking for volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
15:25:17.338 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
15:25:17.339 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: vfat
15:25:17.344 FuVolume             Looking for volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
15:25:17.345 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
15:25:17.348 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: vfat
15:25:17.350 FuPluginUefiDbx      cannot find default ESP: No ESP or BDP found
15:25:17.351 FuEngine             plugins disabled: test, test_ble
Loading…                 [***                                    ]15:25:17.352 FuContext            battery threshold now 10
Loading…                 [************************************** ]15:25:17.352 FuEngine             resetting update motd timeout

WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
15:25:17.356 FuVolume             Looking for volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
15:25:17.357 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
15:25:17.360 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: vfat
15:25:17.365 FuVolume             Looking for volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
15:25:17.366 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
15:25:17.369 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: vfat
No ESP or BDP found
15:25:17.373 FuMain               No ESP or BDP found
```

superm1, 2025-04-09:
Your ESP is marked incorrectly.  It's marked as a Linux partition:
```
    Type:               0fc63daf-8483-4772-8e79-3d69d8477de4
```

This isn't something that we support.  We only support ESP or BDP types.
```
c12a7328-f81f-11d2-ba4b-00a0c93ec93b
```
or
```
ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
```

mid-at-coding, 2025-04-09:
Is there any way to change the marking? the partition is VFAT, and it does work fine

superm1, 2025-04-09:
You can do it with `fdisk` (or similar tools)

closed 2025-04-15 by superm1