fwupd/fwupd #8684: UEFI dbx: Update Error:  Snapd integration for DBX update is not available

david-robson, 2025-04-09:
The output of `sudo fwupdmgr get-devices` gives the error in the title:

``` shell
sudo fwupdmgr get-devices
Dell Inc. Inspiron 7786
│
├─WD19S:
│ │   Device ID:          3d96cd52fdc0a44bc4fcf85c5aeafa22173687cd
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.10
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      F6KZ4Q3/3152467764095082
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │ 
│ ├─Package level of Dell dock:
│ │     Device ID:        d9754b92dc3efa4b2e738901501b7cfc037faaa0
│ │     Summary:          A representation of dock update status
│ │     Current version:  01.00.44.01
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 5 seconds
│ │     GUID:             8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Unsigned Payload
│ │   
│ ├─RTS5413 in Dell dock:
│ │     Device ID:        2664d8d73c7e603fd0bbcbf5b5915e204218546a
│ │     Summary:          USB 3.1 Generation 1 Hub
│ │     Current version:  01.23
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 14 seconds
│ │     GUIDs:            b27d25f1-019d-5718-b41a-02ddaefe5577 ← USB\VID_413C&PID_B06F
│ │                       ac5b774c-b49d-566b-9255-85f0f7f8a4ed ← USB\VID_413C&PID_B06F&hub
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │   
│ ├─RTS5487 in Dell dock:
│ │     Device ID:        06a27d566066ac90defbe5e5fde01145542b4101
│ │     Summary:          USB 3.1 Generation 2 Hub
│ │     Current version:  01.59
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 3 seconds
│ │     GUIDs:            acfcd89b-105d-55b9-b85b-08bf8508f38c ← USB\VID_413C&PID_B06E
│ │                       568ffa1e-a0db-5287-9ea3-872b60f7730b ← USB\VID_413C&PID_B06E&hub
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │   
│ └─VMM5331 in Dell dock:
│       Device ID:        67888d1ef0e0c7359923637e278541b53381ae09
│       Summary:          Multi Stream Transport controller
│       Current version:  05.07.07
│       Vendor:           Dell Inc. (USB:0x413C)
│       Install Duration: 6 minutes
│       GUID:             89fec0b6-6b76-5008-b82c-5e5c6c164007 ← MST-panamera-vmm5331-259
│       Device Flags:     • Updatable
│                         • Supported on remote server
│                         • Device stages updates
│                         • Device is usable for the duration of the update
│                         • Unsigned Payload
│     
├─Display Device:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             PCI:0x10DE
│     GUID:               7991065c-5075-5809-8414-df6c2a42b6a8 ← PCI\VEN_10DE&DEV_1D10
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│                         • Can tag for emulation
│   
├─Samsung SSD 970 PRO 512GB:
│     Device ID:          3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:            NVM Express solid state drive
│     Current version:    1B2QEXP7
│     Vendor:             PCI:0x144D
│     Serial Number:      S463NF0K921227P
│     GUIDs:              47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                         c9d531ea-ee7d-5562-8def-c64d0d144813 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801
│                         d356e2f6-25bb-5121-b968-a05719ccdbfa ← Samsung SSD 970 PRO 512GB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│                         • Can tag for emulation
│   
└─System Firmware:
  │   Device ID:          69c2e00c41ae1d9de59f8189e14a063ce5a5bbe2
  │   Summary:            UEFI System Resource Table device (updated via NVRAM)
  │   Current version:    69632
  │   Minimum Version:    69632
  │   Vendor:             Dell Inc. (DMI:Dell Inc.)
  │   Update State:       Success
  │   GUID:               6f47d9b5-41a8-422b-8dd7-9d779c74ca55
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │   Device Requests:    • Message
  │ 
  └─UEFI dbx:
        Device ID:        362301da643102b9f38477387e2193e57abaa590
        Summary:          UEFI revocation database
        Current version:  20230501
        Minimum Version:  20230501
        Vendor:           UEFI:Microsoft
        Install Duration: 1 second
        Update Error:     Snapd integration for DBX update is not available
        GUIDs:            4a6cd2cb-8741-5257-9d1f-89a275dacca7 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9&ARCH_X64
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
        Device Flags:     • Internal device
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Cryptographic hash verification is available
                          • Device is usable for the duration of the update
                          • Updatable
                          • Only version upgrades are allowed
                          • Signed Payload
                          • Can tag for emulation
      
```

See  #[8676](https://github.com/fwupd/fwupd/issues/8676) for more details.

Ticket opened as per @superm1 comment:
> That means that the dbx snapd integration isn't present.  Open another ticket and we can pull in those guys who worked on it to confirm details. 

 _Originally posted by @superm1 in [#8676](https://github.com/fwupd/fwupd/issues/8676#issuecomment-2791002360)_

Let me know if you need more info and thanks in advance for your help.

superm1, 2025-04-10:
@valentindavid @bboozzoo can you guys comment?

bboozzoo, 2025-04-10:
@david-robson please attach the output of `snap version`, and then run the following command and attach its output:
```
echo '{"action":"efi-secureboot-update-startup"}' | \
        sudo snap debug api -X POST -H 'Content-Type: application/json' /v2/system-secureboot
```

bboozzoo, 2025-04-10:
Thanks, looks like this might be an issue with snapd. I'll investigate.

david-robson, 2025-04-11:
Dziękuję. Thank you. 

I will let you know the result when I receive the new version.

david-robson, 2025-04-14:
Thanks for your help here!

I think I got the update, here are the results...

In short, still the same message "**Update Error:       Snapd integration for DBX update is not available**"

Ran this to get the update: `sudo snap refresh --edge snapd`
The version is:
``` shell
➜ snap --version                                                                  
snap    2.68.4+g260.e7f3e8b
snapd   2.68.4+g260.e7f3e8b
series  16
ubuntu  24.04
kernel  6.5.0-28-generic
```

``` shell
➜ fwupdmgr refresh --force && sudo fwupdmgr get-updates -y && sudo fwupdmgr update
Updating lvfs
Downloading…             [************************               ]
Successfully downloaded new metadata: Updates have been published for 6 of 7 local devices
WD19S is pending activation; use /snap/fwupd/7172/bin/fwupdmgr activate to complete the update.
[sudo] password for david: 
Devices with no available firmware updates: 
 • Samsung SSD 970 PRO 512GB
 • System Firmware
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19S
Dell Inc. Inspiron 7786
│
└─UEFI dbx:
  │   Device ID:          362301da643102b9f38477387e2193e57abaa590
  │   Summary:            UEFI revocation database
  │   Current version:    20230501
  │   Minimum Version:    20230501
  │   Vendor:             UEFI:Microsoft
  │   Install Duration:   1 second
  │   Update Error:       Snapd integration for DBX update is not available
  │   GUIDs:              4a6cd2cb-8741-5257-9d1f-89a275dacca7 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9&ARCH_X64
  │                       f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
  │   Device Flags:       • Internal device
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │                       • Updatable
  │                       • Only version upgrades are allowed
  │                       • Signed Payload
  │                       • Can tag for emulation
  │ 
  └─Secure Boot dbx Configuration Update:
        New version:      20241101
        Remote ID:        lvfs
        Release ID:       105821
        Summary:          UEFI Secure Boot Forbidden Signature Database
        Variant:          x64
        License:          Proprietary
        Size:             15.1 kB
        Created:          2025-01-17
        Urgency:          High
        Vendor:           Linux Foundation
        Duration:         1 second
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:      
        This updates the list of forbidden signatures (the "dbx") to the latest release from Microsoft.
        
        An insecure version of Howyar's SysReturn software was added, due to a security vulnerability that allowed an attacker to bypass UEFI Secure Boot.
        Issues:           529659
                          CVE-2024-7344
        Checksum:         d661d4a0aaca09dfa9e56967ca2467b0575fc07cb704d182fa8c68225452957f
      
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19S
Devices with no available firmware updates: 
 • Samsung SSD 970 PRO 512GB
 • System Firmware

``` 

Having the same error message, I rebooted and tried again. **I got the same message after rebooting - no change.**

...

Concerning the "activation" message: `WD19S is pending activation; use /snap/fwupd/7172/bin/fwupdmgr activate to complete the update.`
- This is a message from a previous update today.
- I did the following:
``` shell
➜ sudo fwupdmgr activate
No firmware to activate

➜ /snap/fwupd/7172/bin/fwupdmgr activate
/snap/fwupd/7172/bin/fwupdmgr: error while loading shared libraries: libfwupd.so.3: cannot open shared object file: No such file or directory

➜ sudo /snap/fwupd/7172/bin/fwupdmgr activate
/snap/fwupd/7172/bin/fwupdmgr: error while loading shared libraries: libfwupd.so.3: cannot open shared object file: No such file or directory
``` 

What about this new message to activate the firmware? The device is a Dell Docking station. The update is activated by forcing a reboot: unplugging it and then plugging it back it works fine. I did this and I think the new firmware was successfully loaded.

bboozzoo, 2025-04-15:
> Having the same error message, I rebooted and tried again. **I got the same message after rebooting - no change.**
> 

Can you clarify what you mean by *the same error*? Can you run the command I provided earlier (`snap debug api..`) and paste the output?

> 
> ➜ sudo fwupdmgr activate
> No firmware to activate
> 
> ➜ /snap/fwupd/7172/bin/fwupdmgr activate
> /snap/fwupd/7172/bin/fwupdmgr: error while loading shared libraries: libfwupd.so.3: cannot open shared object file: No such file or directory
> 
This is not going to work. You're attempting to run fwupd which is built for a specific runtime against the libraries provided by your host.

bboozzoo, 2025-04-16:
Are you sure you're running the right snapd version? Can you run `cat /var/lib/snapd/device/fde/sealed-keys` ?

david-robson, 2025-04-16:
> Are you sure you're running the right snapd version?

The output above was with **beta:**
``` shell
➜ snap --version 
snap    2.68.4
snapd   2.68.4
series  16
ubuntu  24.04
kernel  6.5.0-28-generic
``` 

Moving to **edge:**
``` shell
➜ sudo snap refresh --edge snapd                 
2025-04-16T21:31:25+02:00 INFO Waiting for automatic snapd restart...
2025-04-16T21:31:26+02:00 INFO Waiting for automatic snapd restart...
2025-04-16T21:31:27+02:00 INFO Waiting for automatic snapd restart...
2025-04-16T21:31:28+02:00 INFO Waiting for automatic snapd restart...
snapd (edge) 2.69+g47.de527b2 from Canonical✓ refreshed

➜ snap --version                                                                  
snap    2.69+g47.de527b2
snapd   2.69+g47.de527b2
series  16
ubuntu  24.04
kernel  6.5.0-28-generic
```

... Rebooting ...

Run:
``` shell
fwupdmgr refresh --force && sudo fwupdmgr get-updates -y && sudo fwupdmgr update
``` 
Same message as above: **"Update Error:       Snapd integration for DBX update is not available"**

``` shell
➜ echo '{"action":"efi-secureboot-update-startup"}' | \                           
        sudo snap debug api -X POST -H 'Content-Type: application/json' /v2/system-secureboot
{
  "result": null,
  "status": "OK",
  "status-code": 200,
  "type": "sync"
}

➜ fwupdmgr --version 
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   io.snapcraft.fwupd            7172
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         2.0.7
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.kernel                    6.5.0-28-generic
runtime   org.freedesktop.fwupd         2.0.7
```

> Can you run cat /var/lib/snapd/device/fde/sealed-keys ?

``` shell
➜ sudo cat /var/lib/snapd/device/fde/sealed-keys
cat: /var/lib/snapd/device/fde/sealed-keys: No such file or directory

➜ sudo ls -l /var/lib/snapd/device/ 
total 4
drwxr-xr-x 2 root root 4096 jui 31  2023 private-keys-v1
```

david-robson, 2025-04-16:
The reboot after updating snapd would have caused a restart.

In any case, running `snap restart fwupd` then `fwupdmgr refresh --force && sudo fwupdmgr get-updates -y && sudo fwupdmgr update` gives the same error message: **"Update Error: Snapd integration for DBX update is not available"**.

bboozzoo, 2025-04-16:
I'm puzzled tbh. You're clearly running the latest version. The API does not return any errors, so we can rule that out. 

Can you check if there are any denials in the logs? I'm thinking apparmor mostly. If so attach them.

Next, please attach the output of `snap connections fwupd`.

And lastly, you can try this to enable debug logging in snapd:
```
mkdir -p /etc/systemd/system/snapd.service.d
cat <<'EOF' > /etc/systemd/system/snapd.service.d/debugs.conf
[Service]
Environment=SNAPD_DEBUG=1
EOF
systemctl daemon-reload
systemctl restart snapd
```
and then run:

```
fwupd.fwupdtool --daemon-verbose=FuPluginUefiDbx --daemon-verbose=FuPlugin get-updates
```
you should be able to observe snapd generate some new logs. Please paste that output.

david-robson, 2025-04-16:
For the denials, you're probably on to something:
- I re-ran the update `fwupdmgr refresh --force && sudo fwupdmgr get-updates -y && sudo fwupdmgr update' 
- Looking in dmesg, I skipped the previous runs:

```shell
➜ sudo dmesg | less  
...
[ 9228.187861] audit: type=1326 audit(1744841766.947:308): auid=1026 uid=1026 gid=1026 ses=4 subj=snap.fwupd.fwupdmgr pid=29328 comm="pkttyagent" exe="/snap/fwupd/7172/usr/bin/pkttyagent" sig=0 arch=c000003e syscall=434 compat=0 ip=0x75613b3ac25d code=0x50000
[ 9228.187867] audit: type=1326 audit(1744841766.947:309): auid=1026 uid=1026 gid=1026 ses=4 subj=snap.fwupd.fwupdmgr pid=29328 comm="pkttyagent" exe="/snap/fwupd/7172/usr/bin/pkttyagent" sig=0 arch=c000003e syscall=434 compat=0 ip=0x75613b3ac25d code=0x50000
[ 9228.194000] audit: type=1107 audit(1744841766.955:310): pid=2843 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.PolicyKit1.Authority" member="Changed" name=":1.14" mask="receive" pid=29328 label="snap.fwupd.fwupdmgr" peer_pid=2854 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[ 9228.317474] audit: type=1107 audit(1744841767.079:311): pid=2843 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/NetworkManager" interface="org.freedesktop.DBus.Properties" member="GetAll" mask="send" name=":1.12" pid=29288 label="snap.fwupd.fwupdmgr" peer_pid=2914 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[ 9228.894183] audit: type=1326 audit(1744841767.655:312): auid=1026 uid=0 gid=0 ses=4 subj=snap.fwupd.fwupdmgr pid=29409 comm="pkttyagent" exe="/snap/fwupd/7172/usr/bin/pkttyagent" sig=0 arch=c000003e syscall=434 compat=0 ip=0x721a7534025d code=0x50000
[ 9228.894189] audit: type=1326 audit(1744841767.655:313): auid=1026 uid=0 gid=0 ses=4 subj=snap.fwupd.fwupdmgr pid=29409 comm="pkttyagent" exe="/snap/fwupd/7172/usr/bin/pkttyagent" sig=0 arch=c000003e syscall=434 compat=0 ip=0x721a7534025d code=0x50000
[ 9228.899791] audit: type=1107 audit(1744841767.659:314): pid=2843 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.PolicyKit1.Authority" member="Changed" name=":1.14" mask="receive" pid=29409 label="snap.fwupd.fwupdmgr" peer_pid=2854 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[ 9229.292392] audit: type=1326 audit(1744841768.051:315): auid=1026 uid=0 gid=0 ses=4 subj=snap.fwupd.fwupdmgr pid=29482 comm="pkttyagent" exe="/snap/fwupd/7172/usr/bin/pkttyagent" sig=0 arch=c000003e syscall=434 compat=0 ip=0x7e5f84f6225d code=0x50000
[ 9229.292398] audit: type=1326 audit(1744841768.051:316): auid=1026 uid=0 gid=0 ses=4 subj=snap.fwupd.fwupdmgr pid=29482 comm="pkttyagent" exe="/snap/fwupd/7172/usr/bin/pkttyagent" sig=0 arch=c000003e syscall=434 compat=0 ip=0x7e5f84f6225d code=0x50000
[ 9229.296799] audit: type=1107 audit(1744841768.055:317): pid=2843 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.PolicyKit1.Authority" member="Changed" name=":1.14" mask="receive" pid=29482 label="snap.fwupd.fwupdmgr" peer_pid=2854 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
```

``` shell
➜ snap connections fwupd
Interface         Plug                    Slot               Notes
bluez             fwupd:bluez             :bluez             -
dbus              -                       fwupd:fwupd-dbus   -
fwupd             -                       fwupd:fwupd        -
fwupd             fwupd:fwupdmgr          -                  -
hardware-observe  fwupd:hardware-observe  :hardware-observe  -
home              fwupd:home              :home              -
modem-manager     fwupd:modem-manager     :modem-manager     -
network           fwupd:network           :network           -
opengl            fwupd:opengl            :opengl            -
polkit            fwupd:polkit            :polkit            -
raw-usb           fwupd:raw-usb           :raw-usb           -
shutdown          fwupd:shutdown          :shutdown          -
udisks2           fwupd:udisks2           :udisks2           -
upower-observe    fwupd:upower-observe    :upower-observe    -

➜ sudo su -
root@____:~# mkdir -p /etc/systemd/system/snapd.service.d
cat <<'EOF' > /etc/systemd/system/snapd.service.d/debugs.conf
[Service]
Environment=SNAPD_DEBUG=1
EOF
systemctl daemon-reload
systemctl restart snapd
root@goanna:~# exit
logout

➜ sudo fwupd.fwupdtool --daemon-verbose=FuPluginUefiDbx --daemon-verbose=FuPlugin get-updates
Loading…                 [*                                      ]22:29:17.300 FuPlugin             load(/snap/fwupd/7172/lib/x86_64-linux-gnu/fwupd-2.0.7/libfu_plugin_modem_manager.so)
Loading…                 [***                                    ]22:29:18.262 FuPlugin             constructed(modem_manager)
22:29:18.262 FuPlugin             constructed(uefi_capsule)
22:29:18.264 FuPlugin             constructed(acpi_phat)
[... cut ...]
                          • Is upgrade
        Description:      
        This updates the list of forbidden signatures (the "dbx") to the latest release from Microsoft.
        
        An insecure version of Howyar's SysReturn software was added, due to a security vulnerability that allowed an attacker to bypass UEFI Secure Boot.
        Issues:           529659
                          CVE-2024-7344
        Checksum:         d661d4a0aaca09dfa9e56967ca2467b0575fc07cb704d182fa8c68225452957f
      
```

bboozzoo, 2025-04-17:
Thanks for the logs.

> For the denials, you're probably on to something:
> ...
> ➜ snap connections fwupd
> Interface         Plug                    Slot               Notes
> bluez             fwupd:bluez             :bluez             -
> dbus              -                       fwupd:fwupd-dbus   -
> fwupd             -                       fwupd:fwupd        -
> fwupd             fwupd:fwupdmgr          -                  -
> hardware-observe  fwupd:hardware-observe  :hardware-observe  -
> home              fwupd:home              :home              -
> modem-manager     fwupd:modem-manager     :modem-manager     -
> network           fwupd:network           :network           -
> opengl            fwupd:opengl            :opengl            -
> polkit            fwupd:polkit            :polkit            -
> raw-usb           fwupd:raw-usb           :raw-usb           -
> shutdown          fwupd:shutdown          :shutdown          -
> udisks2           fwupd:udisks2           :udisks2           -
> upower-observe    fwupd:upower-observe    :upower-observe    -

This looks wrong. The `fwupd` interface should be self connected, i.e. the sandbox allows fwupd snap to act both as a fwupd  daemon and the consumer of the daemon's APIs. The connection is made automatically during installation of the fwupd snap. This is what I see in core24 system where no issues are being reported:

```
$ snap connections fwupd
Interface         Plug                    Slot               Notes
bluez             fwupd:bluez             -                  -
dbus              -                       fwupd:fwupd-dbus   -
fwupd             fwupd:fwupdmgr          fwupd:fwupd        -
hardware-observe  fwupd:hardware-observe  :hardware-observe  -
home              fwupd:home              -                  -
modem-manager     fwupd:modem-manager     -                  -
network           fwupd:network           :network           -
opengl            fwupd:opengl            :opengl            -
raw-usb           fwupd:raw-usb           :raw-usb           -
shutdown          fwupd:shutdown          :shutdown          -
udisks2           fwupd:udisks2           udisks2:udisks2    -
upower-observe    fwupd:upower-observe    -                  -
```

First, run `sudo snap connect fwupd:fwupdmgr fwupd:fwupd`, next restart fwupd (`snap restart fwupd`) or run `fwupd.fwupdtool`.

> 
> ➜ sudo fwupd.fwupdtool --daemon-verbose=FuPluginUefiDbx --daemon-verbose=FuPlugin get-updates
> ...
> 22:29:18.268 FuPlugin             constructed(uefi_dbx)
> 22:29:18.270 FuPluginUefiDbx      snapd request failed with status 403, response: {"type":"error","status-code":403,"status":"Forbidden","result":{"message":"access denied","kind":"login-required"}}2
> 22:29:18.270 FuPluginUefiDbx      snapd integration non-functional: failed to notify snapd of startup: snapd request failed with status 403
> ...

yeah, this indicates that the interface is disconnected

david-robson, 2025-04-17:
The error has changed...

``` shell
➜ sudo snap connect fwupd:fwupdmgr fwupd:fwupd

➜ sudo snap restart fwupd                     
Restarted.

➜ snap connections fwupd                                                                     
Interface         Plug                    Slot               Notes
bluez             fwupd:bluez             :bluez             -
dbus              -                       fwupd:fwupd-dbus   -
fwupd             fwupd:fwupdmgr          fwupd:fwupd        manual
hardware-observe  fwupd:hardware-observe  :hardware-observe  -
home              fwupd:home              :home              -
modem-manager     fwupd:modem-manager     :modem-manager     -
network           fwupd:network           :network           -
opengl            fwupd:opengl            :opengl            -
polkit            fwupd:polkit            :polkit            -
raw-usb           fwupd:raw-usb           :raw-usb           -
shutdown          fwupd:shutdown          :shutdown          -
udisks2           fwupd:udisks2           :udisks2           -
upower-observe    fwupd:upower-observe    :upower-observe    -

➜ sudo fwupd.fwupdtool --daemon-verbose=FuPluginUefiDbx --daemon-verbose=FuPlugin get-updates
Loading…                 [*                                      ]19:16:59.845 FuPlugin             load(/snap/fwupd/7172/lib/x86_64-linux-gnu/fwupd-2.0.7/libfu_plugin_modem_manager.so)
Loading…                 [***                                    ]19:17:00.720 FuPlugin             constructed(modem_manager)
19:17:00.720 FuPlugin             constructed(uefi_capsule)
19:17:00.722 FuPlugin             constructed(acpi_phat)
19:17:00.722 FuPlugin             constructed(algoltek_usb)
19:17:00.722 FuPlugin             constructed(algoltek_usbcr)
19:17:00.722 FuPlugin             constructed(amd_kria)
19:17:00.722 FuPlugin             constructed(amd_pmc)
19:17:00.722 FuPlugin             constructed(analogix)
19:17:00.723 FuPlugin             constructed(android_boot)
19:17:00.723 FuPlugin             constructed(asus_hid)
19:17:00.723 FuPlugin             constructed(ata)
19:17:00.723 FuPlugin             constructed(aver_hid)
19:17:00.723 FuPlugin             constructed(bcm57xx)
19:17:00.723 FuPlugin             constructed(bnr_dp)
19:17:00.723 FuPlugin             constructed(ccgx)
19:17:00.723 FuPlugin             constructed(ccgx_dmc)
19:17:00.723 FuPlugin             constructed(cfu)
19:17:00.723 FuPlugin             constructed(ch341a)
19:17:00.723 FuPlugin             constructed(ch347)
19:17:00.723 FuPlugin             constructed(corsair)
19:17:00.723 FuPlugin             constructed(cpu)
19:17:00.723 FuPlugin             constructed(cros_ec)
19:17:00.723 FuPlugin             constructed(dell)
19:17:00.723 FuPlugin             constructed(dell_dock)
19:17:00.723 FuPlugin             constructed(dell_kestrel)
19:17:00.723 FuPlugin             constructed(dfu)
19:17:00.723 FuPlugin             constructed(ebitdo)
19:17:00.723 FuPlugin             constructed(elantp)
19:17:00.723 FuPlugin             constructed(elanfp)
19:17:00.723 FuPlugin             constructed(elan_kbd)
19:17:00.723 FuPlugin             constructed(emmc)
19:17:00.723 FuPlugin             constructed(ep963x)
19:17:00.723 FuPlugin             constructed(fastboot)
19:17:00.723 FuPlugin             constructed(focalfp)
19:17:00.723 FuPlugin             constructed(fpc)
19:17:00.723 FuPlugin             constructed(fresco_pd)
19:17:00.723 FuPlugin             constructed(genesys)
19:17:00.724 FuPlugin             constructed(genesys_gl32xx)
19:17:00.724 FuPlugin             constructed(goodixmoc)
19:17:00.724 FuPlugin             constructed(goodixtp)
19:17:00.724 FuPlugin             constructed(gpio)
19:17:00.724 FuPlugin             constructed(hailuck)
19:17:00.724 FuPlugin             constructed(hpi_cfu)
19:17:00.724 FuPlugin             constructed(huddly_usb)
[... 470 lines cut ...]

Excerpt from dmesg:
``` shell
...
[84931.886207] audit: type=1400 audit(1744917471.177:369): apparmor="DENIED" operation="open" class="file" profile="snap.fwupd.fwupd" name="/media/root/ESP/EFI/Microsoft/Boot/bootmgfw.efi" pid=54788 comm="fwupd" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[85224.801097] audit: type=1326 audit(1744917764.053:370): auid=1026 uid=1026 gid=1026 ses=4 subj=snap.fwupd.fwupdmgr pid=61472 comm="pkttyagent" exe="/snap/fwupd/7172/usr/bin/pkttyagent" sig=0 arch=c000003e syscall=434 compat=0 ip=0x70903a38a25d code=0x50000
[85224.801103] audit: type=1326 audit(1744917764.053:371): auid=1026 uid=1026 gid=1026 ses=4 subj=snap.fwupd.fwupdmgr pid=61472 comm="pkttyagent" exe="/snap/fwupd/7172/usr/bin/pkttyagent" sig=0 arch=c000003e syscall=434 compat=0 ip=0x70903a38a25d code=0x50000
[85224.806688] audit: type=1107 audit(1744917764.057:372): pid=2843 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.PolicyKit1.Authority" member="Changed" name=":1.14" mask="receive" pid=61472 label="snap.fwupd.fwupdmgr" peer_pid=2854 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[85224.956816] audit: type=1107 audit(1744917764.209:373): pid=2843 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/NetworkManager" interface="org.freedesktop.DBus.Pro [... cut ...]
[85422.160510] audit: type=1400 audit(1744917961.406:380): apparmor="DENIED" operation="open" class="file" profile="snap.fwupd.fwupd" name="/media/root/ESP/" pid=63180 comm="fwupd" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[85422.281241] audit: type=1400 audit(1744917961.526:381): apparmor="DENIED" operation="open" class="file" profile="snap.fwupd.fwupd" name="/media/root/ESP/" pid=63180 comm="fwupd" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
```

It looks like 1 step forward and 2 steps backwards.

david-robson, 2025-04-17:
I tried restarting snapd and running the update again and got a different output:
``` shell
➜ sudo systemctl restart snapd

Work/Sorel/sorel-connect took 3.4s …
➜ fwupdmgr refresh --force && sudo fwupdmgr get-updates -y && sudo fwupdmgr update
Updating lvfs
Downloading…             [***********************                ]
Successfully downloaded new metadata: Updates have been published for 6 of 8 local devices
WD19S is pending activation; use /snap/fwupd/7172/bin/fwupdmgr activate to complete the update.
Devices with no available firmware updates: 
 • Samsung SSD 970 PRO 512GB
 • System Firmware
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19S
Dell Inc. Inspiron 7786
│
└─UEFI dbx:
  │   Device ID:          362301da643102b9f38477387e2193e57abaa590
  │   Summary:            UEFI revocation database
  │   Current version:    20230501
  │   Minimum Version:    20230501
  │   Vendor:             UEFI:Microsoft
  │   Install Duration:   1 second
  │   GUIDs:              4a6cd2cb-8741-5257-9d1f-89a275dacca7 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9&ARCH_X64
  │                       f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │                       • Only version upgrades are allowed
  │                       • Signed Payload
  │                       • Can tag for emulation
  │ 
  └─Secure Boot dbx Configuration Update:
        New version:      20241101
        Remote ID:        lvfs
        Release ID:       105821
        Summary:          UEFI Secure Boot Forbidden Signature Database
        Variant:          x64
        License:          Proprietary
        Size:             15.1 kB
        Created:          2025-01-17
        Urgency:          High
        Vendor:           Linux Foundation
        Duration:         1 second
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:      
        This updates the list of forbidden signatures (the "dbx") to the latest release from Microsoft.
        
        An insecure version of Howyar's SysReturn software was added, due to a security vulnerability that allowed an attacker to bypass UEFI Secure Boot.
        Issues:           529659
                          CVE-2024-7344
        Checksum:         d661d4a0aaca09dfa9e56967ca2467b0575fc07cb704d182fa8c68225452957f
      
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 20230501 to 20241101?                                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ An insecure version of Howyar's SysReturn software was added, due to a       ║
║ security vulnerability that allowed an attacker to bypass UEFI Secure Boot.  ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Verifying…               [                                       ]
Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /media/root/ESP/EFI/Microsoft/Boot/bootmgfw.efi: Error opening file /media/root/ESP/EFI/Microsoft/Boot/bootmgfw.efi: Permission denied

``

bboozzoo, 2025-04-18:
> Blocked executable in the ESP, ensure grub and shim are up to date: failed to load /media/root/ESP/EFI/Microsoft/Boot/bootmgfw.efi: Error opening file /media/root/ESP/EFI/Microsoft/Boot/bootmgfw.efi: Permission denied

`/media/root/ESP` is a very unusual place for mounting the ESP. Normally we expect it somewhere under `/boot/efi/`, which is allowed be relevant sandbox profiles. Looking at https://github.com/fwupd/fwupd/issues/8676 I assumed this is a Ubuntu Core system, which sets the mounts up as expected, but your setup looks quite different tbh and is not something we can accommodate in snapd upstream.

See if you can tweak your setup to use /boot/efi, if not, it really doesn't make sense to fight with the sandbox, so you can perhaps remove `fwupd` snap and install it with the permissive devmode confinement, i.e. `snap install fwupd --devmode`.

david-robson, 2025-04-18:
Actually, the ESP is under /boot/efi. 
I don't know why it is looking at /media/root, which exists but is empty.
That's the issue here: where is it getting that crazy path?

Coming back to the "connection" issue above, what was the cause? Even if it was a pre-existing issue, when I installed the newer version, it should have created the connections correctly, right?

bboozzoo, 2025-04-18:
> Actually, the ESP is under /boot/efi. I don't know why it is looking at /media/root, which exists but is empty. That's the issue here: where is it getting that crazy path? I wouldn't know how to tweak it to use another path.

Do you maybe have an external drive with another OS which just happens to be connected and mounted?

> 
> Coming back to the "connection" issue above, what was the cause? Even if it was a pre-existing issue, when I installed the newer version, it should have created the connections correctly, right?

If the interface is disconnected manually, snapd will keep disconnected during an update.

david-robson, 2025-04-18:
> Do you maybe have an external drive with another OS which just happens to be connected and mounted?

No. I have 2 internal drives:
``` shell
➜ lsblk | grep -v loop
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    0 931.5G  0 disk 
├─sda1        8:1    0   500M  0 part 
├─sda2        8:2    0   128M  0 part 
├─sda3        8:3    0   350G  0 part /mnt/win10
├─sda4        8:4    0 564.4G  0 part /mnt/windata
├─sda5        8:5    0   990M  0 part 
├─sda6        8:6    0  14.3G  0 part 
└─sda7        8:7    0   1.2G  0 part 
nvme0n1     259:0    0 476.9G  0 disk 
├─nvme0n1p1 259:1    0   487M  0 part /boot/efi
└─nvme0n1p2 259:2    0 476.5G  0 part /var/snap/firefox/common/host-hunspell
                                      /
```

> If the interface is disconnected manually, snapd will keep disconnected during an update.

Ok. I have no idea what would have disconnected it. 

Thanks for your efforts to try and resolve this.

paddylandau, 2025-04-27:
I have encountered this same problem. My hardware is also Dell, in case that makes a difference.

This was discussed on [Ubuntu Discourse](https://discourse.ubuntu.com/t/how-do-i-run-firmware-updates-with-the-new-snap-store/59895), which includes output from the command, and the error message seen when booting the computer.

The repository version 1.7.9 worked flawlessly, after the snap version 2.0.7 repeatedly failed.

bboozzoo, 2025-04-28:
> I have encountered this same problem. My hardware is also Dell, in case that makes a difference.
>
> This was discussed on [Ubuntu [...]

The issue you had is already fixed in snapd 2.68+ which unfortunately is in the candidate channel, but should be available in stable later this week.

david-robson, 2025-05-05:
Update: 
The issue with fwupdmgr not updating because of the error with the path "/media/root/ESP/EFI/Microsoft/Boot/bootmgfw.efi" is now resolved.

[This post](https://askubuntu.com/questions/1445560/unable-to-update-uefi-dbx) indicates the following workaround, which corrected the issue:
``` shell
sudo fwupdmgr update --force -y
```
**NB: the system IMMEDIATELY rebooted, be sure to save any work in progress BEFORE running this.**

Looking at the issue, it is not clear where this path "/media/root/" is coming from. Maybe the EFI manager can see the EFI files for Windows on drives that are not mounted (see the "lsblk" and "efibootmgr" below). This is beyond my skill & patience and could be something else entirely.

For reference:

``` shell
➜ sudo lsblk  | grep -v loop  
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    0 931.5G  0 disk 
├─sda1        8:1    0   500M  0 part 
├─sda2        8:2    0   128M  0 part 
├─sda3        8:3    0   350G  0 part /mnt/win10
├─sda4        8:4    0 564.4G  0 part /mnt/windata
├─sda5        8:5    0   990M  0 part 
├─sda6        8:6    0  14.3G  0 part 
└─sda7        8:7    0   1.2G  0 part 
nvme0n1     259:0    0 476.9G  0 disk 
├─nvme0n1p1 259:1    0   487M  0 part /boot/efi
└─nvme0n1p2 259:2    0 476.5G  0 part /var/snap/firefox/common/host-hunspell
                                      /

➜ efibootmgr   
BootCurrent: 000B
Timeout: 0 seconds
BootOrder: 0002,0006,0008,000B,0005,0000,0001,0004,0003
Boot0000* Windows Boot Manager  HD(1,GPT,3a0bb2e7-74d2-44be-b8e4-48936a0c9a3e,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)57494e444f5753000100000088000000780000004200430044004f0042004a004500430054003d007b00390064006500610038003600320063002d0035006300640064002d0034006500370030002d0061006300630031002d006600330032006200330034003400640034003700390035007d00000000000100000010000000040000007fff0400
Boot0001* Recovery      PciRoot(0x0)/Pci(0x17,0x0)/Sata(1,32768,0)/HD(2,GPT,2d56c839-3199-30c2-f42b-eda4f93e8e74,0x40800,0x1ef000)/File(\recovery\windowsre\boot.sdi)
Boot0002* Ubuntu        HD(1,GPT,5727a2c8-0ba8-4275-a692-ab7776d9e9d6,0x800,0xf3800)/File(\EFI\ubuntu\shimx64.efi)
Boot0003* Win Recovery  PciRoot(0x0)/Pci(0x17,0x0)/Sata(1,32768,0)/HD(2,GPT,2d56c839-3199-30c2-f42b-eda4f93e8e74,0x40800,0x1ef000)/File(\recovery\windowsre\winre.wim)
Boot0004* Dell SOS      PciRoot(0x0)/Pci(0x17,0x0)/Sata(1024,32768,0)/HD(1,GPT,1b65a0d8-306d-424f-acae-8b4e4237df74,0x800,0x145000)/File(\EFI\dell\SOS\en-GB\bootmgr.efi.mui)
Boot0005* Windows Boot Manager  HD(1,GPT,3a0bb2e7-74d2-44be-b8e4-48936a0c9a3e,0x28,0xf9ffa)/File(\EFI\Microsoft\Boot\bootmgfw.efi)57494e444f5753000100000088000000780000004200430044004f0042004a004500430054003d007b00390064006500610038003600320063002d0035006300640064002d0034006500370030002d0061006300630031002d006600330032006200330034003400640034003700390035007d00000000000100000010000000040000007fff0400
Boot0006* Ubuntu        HD(1,GPT,3a0bb2e7-74d2-44be-b8e4-48936a0c9a3e,0x800,0xf3800)/File(\EFI\ubuntu\shimx64.efi)
Boot0008* rEFInd Boot Manager   HD(1,GPT,3a0bb2e7-74d2-44be-b8e4-48936a0c9a3e,0x800,0xf3800)/File(\EFI\refind\refind_x64.efi)
Boot000B* UEFI: Samsung SSD 970 PRO 512GB, Partition 1  HD(1,GPT,5727a2c8-0ba8-4275-a692-ab7776d9e9d6,0x800,0xf3800)/File(\EFI\Boot\BootX64.efi)0000424f
Boot000D* Onboard NIC   BBS(Network,Onboard NIC,0x0)0000424f
```

superm1, 2025-08-24:
@bboozzoo can you look again considering above?

bboozzoo, 2025-08-27:
I had a chat with folks who may know what is going on and the likely issue is that there are 2 slots providing fwupd interface in the system. One from `snapd` and another from `fwupd` snap. We think the solution is to update the snap declaration, perhaps adding a more specific clause allowing plug connections from fwupd's own snap ID.

@superm1 as a snap publisher, can you open a topic in the snapcraft forum in `store-requests` category, asking for the snap declaration to be updated and link back to this issue? I'll ping the relevant folks to have a look.

bboozzoo, 2025-09-02:
The snap-declaration has been updated. The connection is established automatically and snapd integration is available:
```
root@u1:~# snap install fwupd
Warning: /snap/bin was not found in your $PATH. If you've not restarted your session since you
         installed snapd, try doing that. Please see https://forum.snapcraft.io/t/9469 for more
         details.

fwupd 2.0.11 from Richard Hughes installed
root@u1:~# snap connections
Interface         Plug                    Slot               Notes
bluez             fwupd:bluez             :bluez             -
fwupd             fwupd:fwupdmgr          fwupd:fwupd        -
hardware-observe  fwupd:hardware-observe  :hardware-observe  -
home              fwupd:home              :home              -
modem-manager     fwupd:modem-manager     :modem-manager     -
network           fwupd:network           :network           -
opengl            fwupd:opengl            :opengl            -
polkit            fwupd:polkit            :polkit            -
raw-usb           fwupd:raw-usb           :raw-usb           -
shutdown          fwupd:shutdown          :shutdown          -
udisks2           fwupd:udisks2           :udisks2           -
upower-observe    fwupd:upower-observe    :upower-observe    -
```
and then querying updates:
```
root@u1:~# fwupd.fwupdtool get-updates --daemon-verbose=FuPluginUefiDbx
Loading…                 [***                                    ]09:23:05.086 FuPluginUefiDbx      snapd integration enabled
Loading…                 [*******************                    ]09:23:05.191 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:01.1/0000:02:00.0/virtio6/host0/target0:0:0/0:0:0:1/block/sda: failed to subclass open: failed to open /dev/sda: Operation not permitted
09:23:05.194 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:01.1/0000:02:00.0/virtio6/host0/target0:0:0/0:0:0:1/block/sda: failed to subclass open: failed to open /dev/sda: Operation not permitted
Loading…                 [************************               ]09:23:05.218 FuEngine             failed to add device /sys/devices/virtual/msr/msr0: failed to add device using on msr: failed to subclass open: failed to open /dev/cpu/0/msr: Operation not permitted
Loading…                 [************************************** ]
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Devices with no available firmware updates:
 • KEK CA
 • SBAT
 • UEFI CA
 • UEFI dbx
 • Ubuntu OVMF Secure Boot (PK/KEK key)
 • Windows Production PCA
No updates available for remaining devices
```
and relevant request is logged by snapd:
```
Sep 02 09:23:05 u1 snapd[2086]: daemon.go:246: DEBUG: pid=2386;uid=0;socket=/run/snapd-snap.socket; POST /v2/system-secureboot 136.065µs 200
```
AFAICT everything should be in order now.

ewenmcneill, 2025-09-03:
By way of confirmation, after doing:

```
sudo snap remove fwupd
sudo snap install --channels latest/stable fwupd
```

I can confirm (a) the `fwupd:fwupdmgr          fwupd:fwupd ` connection was automatically present (didn't have to add it manually; no longer flagged as manual), and (b) both `fwupdmgr get-devices` and `fwupdmgr get-updates` ran without the previous "UEFI dbx: Update Error: Snapd integration for DBX update is not available" error.

So looks fixed to me (on Ubuntu 22.04 LTS).

Thanks for figuring out the root cause, and getting someone to fix the snap metadata.

Ewen

PS: I suspect this issue could be closed again, since the underlying issue seems solved this time, not just worked around.

closed 2025-09-03 by superm1