fwupd/fwupd #8691: Building without systemd / consolekit fails

mjk-gh, 2025-04-14:
**Describe the bug**
Building fwupd without systemd / consolekit, as outlined in #1408, does not work.

**Steps to Reproduce**
Follow build instructions in issue #1408, the first one being:
```
$ meson build -Dlibexecdir=/usr/lib/ \
                -Dprefix=/usr/ \
                -Ddocs=false \
                -Dnls=false \
                -Dgtkdoc=false \
                -Dstemmer=true \
                -Dvapi=false \
                -Dconsolekit=false \
                -Dsystemd=false \
                -Dgcab:docs=false \
                -Dgcab:vapi=false \
                -Dgusb:docs=false \
                -Dgusb:vapi=false
The Meson build system
Version: 1.5.1
Source dir: /tmp/fwupd-2.0.8
Build dir: /tmp/fwupd-2.0.8/build
Build type: native build

meson.build:1:0: ERROR: Value "false" (of type "string") for option "None" is not one of the choices. Possible choices are (as string): "enabled", "disabled", "auto".

A full log can be found at /tmp/fwupd-2.0.8/build/meson-logs/meson-log.txt
WARNING: Running the setup command as `meson [options]` instead of `meson setup [options]` is ambiguous and deprecated.
$
```
Logfile: [meson-log.txt](https://github.com/user-attachments/files/19734977/meson-log.txt)

**Expected behavior**
I expect it to build just fine.

**fwupd version information**
2.0.8 as .tar.gz from github.

**Additional questions**

- Operating system and version:
  Devuan Daedalus (systemd-less fork of Debian 12 (bookworm))

superm1, 2025-04-14:
Make sure you double check meson_options.txt.

Iirc we dropped building without CK so you shouldn't have that in your build line.

superm1, 2025-04-14:
Here is the commit I was thinking of.
https://github.com/fwupd/fwupd/commit/4d287fe2088dc3fadb63d3f04bc521473c8c6231

Please re-open if you have a problem after correcting your build config.

closed 2025-04-14 by superm1

mjk-gh, 2025-04-14:
> Make sure you double check meson_options.txt.

If only I knew what to look for ... the string "console" is not in that file. The meson_options.txt is the unaltered version that came with the released 2.0.8.tar.gz.

> Iirc we dropped building without CK so you shouldn't have that in your build line.

Removed "-Dconsolekit=false" from the "meson build" call, to no avail, same error message. I also tried to replace various occurrences of "false" with "disabled", also to no avail.

As the error says:

```
meson.build:1:0: ERROR: Value "false" (of type "string") for option "None" is not one of the choices
```
I have looked for "None" in my build command and for an option "None" in the ominous meson_options.txt, but found nothing.

mjk-gh, 2025-04-14:
> Here is the commit I was thinking of. [4d287fe](https://github.com/fwupd/fwupd/commit/4d287fe2088dc3fadb63d3f04bc521473c8c6231)

Sooo ... are you telling me that the version of fwupd before above commit is the last version that will be able to run with systemd (and its dependencies)? No criticism, just trying to understand a bit (I do not know too much about systemd and nothing about meson).

> Please re-open if you have a problem after correcting your build config.

Might do. :-}

superm1, 2025-04-14:
> Sooo ... are you telling me that the version of fwupd before above commit is the last version that will be able to run with systemd (and its dependencies)? No criticism, just trying to understand a bit (I do not know too much about systemd and nothing about meson).

No I'm not saying that.    You can still build with systemd support or without it.  But consolekit is not a configurable option anymore.  Drop it from your build config.

mjk-gh, 2025-04-14:
> No I'm not saying that. You can still build with systemd support or without it.

Aaah, that's good news, thanks!

> But consolekit is not a configurable option anymore. Drop it from your build config.

Hm, I guess my build config consists of "meson_options.txt" and the shell call to "meson build -D...", right?

My build command line now looks like this:
```$ meson build -Dlibexecdir=/usr/lib/ \
                -Dprefix=/usr/ \
                -Ddocs=disabled \
                -Dnls=disabled \
                -Dgtkdoc=false \
                -Dstemmer=true \
                -Dvapi=false \
                -Dsystemd=false \
                -Dgcab:docs=false \
                -Dgcab:vapi=false \
                -Dgusb:docs=false \
                -Dgusb:vapi=false
```
... i.e. without the previously used "-Dconsolekit=false" option.

And my meson_options.txt is the default from the 2.0.8 release (.tar.gz): [meson_options.txt](https://github.com/user-attachments/files/19739230/meson_options.txt)

Do both these things look right to you? Error message is still the same:

```
The Meson build system
Version: 1.5.1
Source dir: /tmp/fwupd-2.0.8
Build dir: /tmp/fwupd-2.0.8/build
Build type: native build

meson.build:1:0: ERROR: Value "false" (of type "string") for option "None" is not one of the choices. Possible choices are (as string): "enabled", "disabled", "auto".

A full log can be found at /tmp/fwupd-2.0.8/build/meson-logs/meson-log.txt
WARNING: Running the setup command as `meson [options]` instead of `meson setup [options]` is ambiguous and deprecated.

```

mjk-gh, 2025-04-14:
> Systemd is not a Boolean. It's a feature. You should be using "disabled".

Thanks, changed that! Still same(!) error message. I tried to replace "false" with "disabled" for various other meson build options, but either got the same error message or the error message changed to:
```
meson.build:1:0: ERROR: Unknown options: "gtkdoc, nls, stemmer, vapi"
```

How do I infer from the message
```
ERROR: Value "false" (of type "string") for option "None" is not one of the choices. Possible choices are (as string): "enabled", "disabled", "auto".
```
**which** option has to be corrected from true/false to enabled/disabled? Whatever I tentatively try, it always claims it is about option "None" (yeah, right).

superm1, 2025-04-14:
> for various other meson build options, but either got the same error message or the error message changed to:

Drop those options from your build config.

> How do I infer from the message

Cross reference all your options against meson_options.txt.  If they're missing, drop them.  If they're present check their types are right.  Boolean needs true/false.  Feature needs enabled/disabled.

mjk-gh, 2025-04-14:
> > for various other meson build options, but either got the same error message or the error message changed to:
> 
> Drop those options from your build config.

Done, now it starts building, yay! :)

In the process, it failed with downloading cmocka and a patch:
```
flashrom| Downloading cmocka source from https://cmocka.org/files/1.1/cmocka-1.1.5.tar.xz                                                                            
flashrom| <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1006)>                           
flashrom| WARNING: failed to download with error: could not get https://cmocka.org/files/1.1/cmocka-1.1.5.tar.xz is the internet available?. Trying after a delay...                                                                            
```
I hotfixed that by downloading the two requested files and replaced the two URLs in cmocka.wrap with appropriate local `file:///tmp/...` URLs.

After that meson build(?), I executed `ninja -C build`, and after a few cycles of build fails and installing missing packages, I got stuck here:

```
$ ninja -C build
ninja: Entering directory `build'
[16/747] Linking target plugins/flashrom/libfu_plugin_flashrom.so
FAILED: plugins/flashrom/libfu_plugin_flashrom.so
cc  -o plugins/flashrom/libfu_plugin_flashrom.so plugins/flashrom/libfu_plugin_flashrom.so.p/fu-flashrom-device.c.o plugins/flashrom/libfu_plugin_flashrom.so.p/fu-flashrom-plugin.c.o plugins/flashrom/libfu_plugin_flashrom.so.p/fu-flashrom-cmos.c.o -Wl,--as-needed -Wl,--allow-shlib-undefined -shared -fPIC -Wl,-z,relro -Wl,-z,defs -Wl,-z,now -Wl,-z,ibt,-z,shstk '-Wl,-rpath,$ORIGIN/../../libfwupd:$ORIGIN/../../subprojects/libjcat/libjcat:$ORIGIN/../../subprojects/json-glib/json-glib:$ORIGIN/../../libfwupdplugin:$ORIGIN/../../subprojects/libxmlb/src' -Wl,-rpath-link,/tmp/fwupd-2.0.8/build/libfwupd -Wl,-rpath-link,/tmp/fwupd-2.0.8/build/subprojects/libjcat/libjcat -Wl,-rpath-link,/tmp/fwupd-2.0.8/build/subprojects/json-glib/json-glib -Wl,-rpath-link,/tmp/fwupd-2.0.8/build/libfwupdplugin -Wl,-rpath-link,/tmp/fwupd-2.0.8/build/subprojects/libxmlb/src -Wl,--start-group libfwupd/libfwupd.so.3.0.0 libfwupdplugin/libfwupdplugin.so subprojects/json-glib/json-glib/libjson-glib-1.0.so.0.1000.1 subprojects/libxmlb/src/libxmlb.so.2.0.0 /usr/lib/x86_64-linux-gnu/libgio-2.0.so /usr/lib/x86_64-linux-gnu/libgobject-2.0.so /usr/lib/x86_64-linux-gnu/libglib-2.0.so /usr/lib/x86_64-linux-gnu/libgnutls.so -Wl,--export-dynamic /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so -pthread /usr/lib/x86_64-linux-gnu/libusb-1.0.so /usr/lib/x86_64-linux-gnu/libarchive.so /usr/lib/x86_64-linux-gnu/liblzma.so /usr/lib/x86_64-linux-gnu/libzstd.so /usr/lib/x86_64-linux-gnu/libdrm_amdgpu.so -Wl,--end-group
/usr/bin/ld: plugins/flashrom/libfu_plugin_flashrom.so.p/fu-flashrom-device.c.o: in function `fu_flashrom_device_open':
/tmp/fwupd-2.0.8/build/../plugins/flashrom/fu-flashrom-device.c:80: undefined reference to `flashrom_flash_getsize'
[...]
```
I assume I need a newer version of libflashrom, but that shoves me into dependency hell (I'm running Devuan stable, and newer libflashrom versions are in testing or unstable) or into the rabbit hole of infinite fractal dependency compilations. I cannot be scolded for not trying, but I think trying to get fwupd to work is starting to take a toll on my mental health, so I should better stop here.

Thank you very much for trying to make it work, I appreciate it! ❤️ 

> > How do I infer from the message
> 
> Cross reference all your options against meson_options.txt. If they're missing, drop them. If they're present check their types are right. Boolean needs true/false. Feature needs enabled/disabled.
Thanks for the explanation!

FWIW: Although I specified "-Ddocs=disabled", it still complained about missing gtkdocs-scan:

```
subprojects/libxmlb/meson.build:184:15: ERROR: Program 'gtkdoc-scan' not found or not executable
```

superm1, 2025-04-14:
Most distro packagers don't allow subprojects while building.  I think you should avoid that if you can.

> I assume I need a newer version of libflashrom, but that shoves me into dependency hell (I'm running Devuan stable, and newer libflashrom versions are in testing or unstable) or into the rabbit hole [...]

I would just disable flashrom.  It's only used for a small subset of devices and there are some pretty major issues with it.  See https://github.com/fwupd/fwupd/issues/8251 for an example.

> FWIW: Although I specified "-Ddocs=disabled", it still complained about missing gtkdocs-scan:

That's for the subproject `libxmlb`.   It seems to me you don't have libxmlb-dev installed otherwise it wouldn't try to install it.