fwupd/fwupd #8696: UEFI dbx update never actually installs

mauke, 2025-04-15:
**Describe the bug**
My system keeps nagging me that a new UEFI dbx version (20241101) is available. When I try to install it, the system tells me to reboot, but that just ... reboots. And then nags me again that an update is available.

**Steps to Reproduce**
Run `fwupdmgr upgrade`, hit enter when it asks to "Upgrade UEFI dbx from 20230501 to 20241101?" with default Y, then reboot.

**Expected behavior**
I expected the reboot to actually upgrade something. Or at least give me an error message.

**fwupd version information**
Please provide the version of the daemon and client.

```
$ fwupdtool get-report-metadata
BatteryLevel:           94
BatteryThreshold:       25
BootTime:               1744724568
CompileVersion(com.hughsie.libjcat):0.2.0
CompileVersion(com.hughsie.libxmlb):0.3.18
CompileVersion(org.freedesktop.fwupd):1.9.28
CompileVersion(org.freedesktop.gusb):0.4.8
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 5 5600U with Radeon Graphics
DisplayState:           connected
DistroId:               ubuntu
DistroName:             Ubuntu
DistroPrettyName:       Ubuntu 24.04.2 LTS
DistroVersion:          24.04
FwupdSupported:         True
HostBaseboardManufacturer:LENOVO
HostBaseboardProduct:   20X7004KGE
HostBiosMajorRelease:   01
HostBiosMinorRelease:   20
HostBiosVendor:         LENOVO
HostBiosVersion:        R1KET47W (1.32 )
HostEnclosureKind:      a
HostFamily:             ThinkPad L15 Gen 2a
HostFirmwareMajorRelease:01
HostFirmwareMinorRelease:0d
HostProduct:            20X7004KGE
HostSku:                LENOVO_MT_20X7_BU_Think_FM_ThinkPad L15 Gen 2a
HostVendor:             LENOVO
KernelName:             Linux
KernelVersion:          6.11.0-21-generic
LidState:               open
PowerState:             ac-fully-charged
RuntimeVersion(com.hughsie.libjcat):0.2.0
RuntimeVersion(com.hughsie.libxmlb):0.3.x
RuntimeVersion(org.freedesktop.fwupd):1.9.28
RuntimeVersion(org.freedesktop.fwupd-efi):1.4
RuntimeVersion(org.freedesktop.gusb):0.4.8
RuntimeVersion(org.kernel):6.11.0-21-generic
DeviceId:               80e05e5ac9671644c7c66be138467ae315395857
  pre:
      CapsuleApplyMethod:nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader:False
  post:
      LastAttemptStatus:0x0
      LastAttemptVersion:0x0
DeviceId:               66fdd892bfcfacf75f88c8cc5dc93d3e7ba46888
  pre:
      CapsuleApplyMethod:nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader:False
  post:
      LastAttemptStatus:0x0
      LastAttemptVersion:0x0
[... 686 lines cut ...]
├── memtest86+ia32.efi
├── memtest86+x64.bin
├── memtest86+x64.efi
├── vmlinuz -> vmlinuz-6.11.0-21-generic
├── vmlinuz-6.11.0-19-generic
├── vmlinuz-6.11.0-21-generic
└── vmlinuz.old -> vmlinuz-6.11.0-19-generic

16 directories, 323 files
```

**Additional questions**

- Operating system and version:
  Ubuntu 24.04.2 LTS
- Have you tried rebooting?
  Only about a dozen times.
- Is this a regression?
  No idea.
- Are you using an NVMe disk?
  I don't know what that is, but `lsblk` says my disk is named "nvme0n1", so probably.
- Is secure boot enabled?
  Yes.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
  No. (It is Lenovo, but I made sure Boot Order Lock was turned off. Doesn't really help much if there is no "Linux Firmware Updater" boot entry in the first place, though.)

ftmiranda, 2025-04-16:
Just a thought - reboot the computer, enter the BIOS and disable secure boot
Then try this process again

I' m having the same issue https://github.com/fwupd/fwupd/issues/8705
And Grok suggested disabling secure boot, which I did, but in my case it did not help at all

superm1, 2025-04-16:
As mentioned in #8702.

> What does sudo fwupdtool get-devices --plugins uefi_dbx -vv say for you please? I'm specifically looking for the Checksum line -- thanks. It's probably missing a line in https://github.com/fwupd/fwupd/blob/main/plugins/uefi-dbx/uefi-dbx.quirk#L4

ftmiranda, 2025-04-16:
> As mentioned in [#8702](https://github.com/fwupd/fwupd/issues/8702).

$ sudo fwupdtool get-devices --plugins uefi_dbx -vv
[sudo] password for fmiranda: 
19:48:09.009 FuDebug              verbose to debug (on console 1)
19:48:09.009 FuEngine             starting fwupd 1.9.28…
19:48:09.009 FuMain               locked /run/lock/fwupdtool
19:48:09.010 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
Loading…                 [ -                                     ]19:48:09.017 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
19:48:09.017 FuConfig             loading config /etc/fwupd/fwupd.conf
19:48:09.017 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
19:48:09.017 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
19:48:09.017 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
19:48:09.017 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
19:48:09.017 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
19:48:09.017 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
19:48:09.017 FuConfig             ::configuration loaded
19:48:09.017 FuRemoteList         loading /usr/share/fwupd/metainfo
19:48:09.017 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
19:48:09.017 XbSilo               file: 3209773d-ab01-ce56-7f8d-caffa9acaf10, current:3209773d-ab01-ce56-7f8d-caffa9acaf10, cached: (null)
19:48:09.017 XbSilo               loading silo with file contents
19:48:09.018 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
19:48:09.018 FuRemoteList         ::remote_list changed
19:48:09.018 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
19:48:09.018 FuRemoteList         ::remote_list changed
19:48:09.018 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
19:48:09.018 FuRemoteList         ::remote_list changed
19:48:09.018 FuEngine             priority local and vendor-directory is not download remote, so bumping
19:48:09.018 FuRemoteList         ordering lvfs-testing=lvfs+1
19:48:09.018 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]19:48:09.018 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
19:48:09.018 Jcat                 loaded 1 certificates
19:48:09.018 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
19:48:09.018 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
19:48:09.018 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
19:48:09.018 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
19:48:09.018 Jcat                 loaded 1 certificates
19:48:09.018 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
19:48:09.018 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
19:48:09.018 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
19:48:09.018 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]19:48:09.018 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
19:48:09.019 FuHistory            got schema version of 12
Loading…                 [*                                      ]19:48:09.021 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-1.9.28/libfu_plugin_modem_manager.so)
19:48:09.023 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
19:48:09.023 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
19:48:09.023 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
19:48:09.023 XbSilo               file: 2210ae90-8319-7895-fd7a-53a59b9df088, current:2210ae90-8319-7895-fd7a-53a59b9df088, cached: (null)
19:48:09.023 XbSilo               loading silo with file contents
Loading…                 [*                                      ]19:48:09.024 FuConfig             loading config /etc/fwupd/fwupd.conf
19:48:09.024 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
19:48:09.024 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
19:48:09.024 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
19:48:09.024 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
19:48:09.024 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
19:48:09.024 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
19:48:09.024 FuConfig             ::configuration loaded
19:48:09.024 FuStruct             FuStructSmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0x4f
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x4
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0x1bb3
  structure_table_addr: 0x4151a000
19:48:09.024 FuStruct             FuStructSmbiosStructure:
  type: 0x0
[... 812 lines cut ...]
  Flags:                ready

[... cut ...]
  Flags:                internal|updatable|supported|registered|needs-reboot|usable-during-update|only-version-upgrade|signed-payload
  Checksum:             SHA256(cdb7c90d3ab8833d5324f5d8516d41fa990b9ca721fe643fffaef9057d9f9e48)
  VendorId:             UEFI:Microsoft
  Version:              20241101
  VersionLowest:        20241101
  VersionFormat:        number
  Icon:                 computer
  InstallDuration:      1
  Created:              2025-04-16

19:48:10.669 FuPluginTest         destroy

superm1, 2025-04-16:
So it looks like for your case the existing checksum in the quirk file is listed.

You
```
 Guid[quirk]: fed1a551-5a99-5a8b-aa01-e4b63f6ae2d0 ← UEFI\CSUM_CDB7C90D3AB8833D5324F5D8516D41FA990B9CA721FE643FFFAEF9057D9F9E48
```

Quirk file
```
[UEFI\CSUM_CDB7C90D3AB8833D5324F5D8516D41FA990B9CA721FE643FFFAEF9057D9F9E48]
Version = 20241101
```

ftmiranda, 2025-04-16:
> So it looks like for your case the existing checksum in the quirk file is listed.
>
> You
>
> ```
>  Guid[quirk]: fed1a551-5a99-5a8b-aa01-e4b63f6ae2d0 ← [...]

Is this good? bad?  :-)

superm1, 2025-04-17:
> Is this good? bad? :-)

It means the root cause is not a missing line in that quirk file.

mauke, 2025-04-17:
> What does sudo fwupdtool get-devices --plugins uefi_dbx -vv say for you please? I'm specifically looking for the Checksum line -- thanks. It's probably missing a line in https://github.com/fwupd/fwupd/blob/main/plugins/uefi-dbx/uefi-dbx.quirk#L4

That's a lot of output. See below:

<details>
<summary>sudo fwupdtool get-devices --plugins uefi_dbx -vv</summary>

```
12:31:14.899 FuDebug              verbose to debug (on console 0)
12:31:14.899 FuEngine             starting fwupd 1.9.28...
12:31:14.900 FuMain               locked /run/lock/fwupdtool
12:31:14.900 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
12:31:14.902 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…: 0%
12:31:14.902 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for 'gio-vfs'
12:31:14.902 FuConfig             loading config /etc/fwupd/fwupd.conf
12:31:14.902 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
12:31:14.902 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
12:31:14.902 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
12:31:14.902 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
12:31:14.902 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
12:31:14.902 FuConfig             Failed to open file "/var/etc/fwupd/fwupd.conf": No such file or directory
12:31:14.903 FuConfig             ::configuration loaded
12:31:14.903 FuRemoteList         loading /usr/share/fwupd/metainfo
12:31:14.903 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
12:31:14.903 XbSilo               file: 2a1d7b33-9e7e-3686-2383-f5feb534542b, current:2a1d7b33-9e7e-3686-2383-f5feb534542b, cached: (null)
12:31:14.903 XbSilo               loading silo with file contents
12:31:14.903 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
12:31:14.903 FuRemoteList         ::remote_list changed
12:31:14.903 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
12:31:14.903 FuRemoteList         ::remote_list changed
12:31:14.903 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
12:31:14.903 FuRemoteList         ::remote_list changed
12:31:14.903 FuEngine             priority local and vendor-directory is not download remote, so bumping
12:31:14.903 FuRemoteList         ordering lvfs-testing=lvfs+1
12:31:14.903 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…: 1%
12:31:14.903 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
12:31:14.903 Jcat                 loaded 1 certificates
12:31:14.903 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:31:14.903 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
12:31:14.903 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
12:31:14.903 Jcat                 loaded 1 certificates
12:31:14.903 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
12:31:14.903 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
12:31:14.903 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
12:31:14.903 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
12:31:14.903 FuEngine             client certificate now exists: nothing to do
Loading…: 2%
12:31:14.903 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
12:31:14.904 FuHistory            got schema version of 12
Loading…: 3%
Loading…: 4%
12:31:14.905 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-1.9.28/libfu_plugin_modem_manager.so)
12:31:14.906 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
12:31:14.906 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
12:31:14.906 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
12:31:14.906 XbSilo               file: d650d8ab-e073-f2e8-201c-cff4613bd36c, current:d650d8ab-e073-f2e8-201c-cff4613bd36c, cached: (null)
12:31:14.906 XbSilo               loading silo with file contents
Loading…: 5%
12:31:14.907 FuConfig             loading config /etc/fwupd/fwupd.conf
12:31:14.907 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
12:31:14.907 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
12:31:14.907 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
12:31:14.907 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
12:31:14.907 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
12:31:14.907 FuConfig             Failed to open file "/var/etc/fwupd/fwupd.conf": No such file or directory
12:31:14.907 FuConfig             ::configuration loaded
12:31:14.907 FuStruct             FuStructSmbiosEp64:
  anchor_str: _SM3_
[... 972 lines cut ...]

Idle…: 0%
12:31:16.839 FuEngine             resetting update motd timeout
12:31:16.840 FuMain               FuUefiDbxDevice:
  DeviceId:             362301da643102b9f38477387e2193e57abaa590
  Name:                 UEFI dbx
[... cut ...]
  Flags:                internal|updatable|supported|registered|needs-reboot|usable-during-update|only-version-upgrade|signed-payload
  Checksum:             SHA256(13a1f37bedfb5417b6b737e2a3816c8fd587d74d836914b2b2edc9fd6ca30e58)
  VendorId:             UEFI:Microsoft
  Version:              20230501
  VersionLowest:        20230501
  VersionFormat:        number
  Icon:                 computer
  InstallDuration:      1
  Created:              2025-04-17

12:31:16.841 FuPluginTest         destroy
```

</summary>

Zhuwenxue2002, 2025-04-18:
I'm having the same issue, and while I won't have any other issues with it, it's also very annoying that fwupd keeps sending notifications (and can't block it) as Ubuntu's default software

hughsie, 2025-04-18:
What does this output please:

    wget https://fwupd.org/downloads/d661d4a0aaca09dfa9e56967ca2467b0575fc07cb704d182fa8c68225452957f-DBXUpdate-20241101-x64.cab
    sudo fwupdtool -vv --plugins uefi_dbx install d661d4a0aaca09dfa9e56967ca2467b0575fc07cb704d182fa8c68225452957f-DBXUpdate-20241101-x64.cab

It's probably failing to write the variable.

Zhuwenxue2002, 2025-04-18:
> What does this output please:
>
> ```
> wget https://fwupd.org/downloads/d661d4a0aaca09dfa9e56967ca2467b0575fc07cb704d182fa8c68225452957f-DBXUpdate-20241101-x64.cab
> sudo fwupdtool -vv --plugins uefi_dbx [...]

Although the final execution result was successful, failed to set release for component appeared many times in the program: release needs protocol org.uefi.dbx but device has org.uefi.dbx2,

mauke, 2025-04-22:
Again, a lot of output:

<details>
<summary><code>sudo fwupdtool -vv --plugins uefi_dbx install d661d4a0aaca09dfa9e56967ca2467b0575fc07cb704d182fa8c68225452957f-DBXUpdate-20241101-x64.cab</code></summary>

```
06:38:20.811 FuDebug              verbose to debug (on console 1)
06:38:20.811 FuEngine             starting fwupd 1.9.28…
Loading…                 [ -                                     ]Loading…                 [ -                                     ]06:38:20.812 FuMain               locked /run/lock/fwupdtool
06:38:20.812 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
06:38:20.814 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
06:38:20.815 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
06:38:20.815 FuConfig             loading config /etc/fwupd/fwupd.conf
06:38:20.815 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
06:38:20.815 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
06:38:20.815 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
06:38:20.815 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
06:38:20.816 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
06:38:20.816 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
06:38:20.816 FuConfig             ::configuration loaded
06:38:20.816 FuRemoteList         loading /usr/share/fwupd/metainfo
06:38:20.817 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
06:38:20.817 XbSilo               file: 85374c3c-23e9-ed71-db0b-cde19f9665a7, current:2a1d7b33-9e7e-3686-2383-f5feb534542b, cached: (null)
06:38:20.820 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
06:38:20.820 FuRemoteList         ::remote_list changed
06:38:20.820 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
06:38:20.821 FuRemoteList         ::remote_list changed
06:38:20.821 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
06:38:20.821 FuRemoteList         ::remote_list changed
06:38:20.821 FuEngine             priority local and vendor-directory is not download remote, so bumping
06:38:20.821 FuRemoteList         ordering lvfs-testing=lvfs+1
06:38:20.821 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
06:38:20.821 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
06:38:20.822 Jcat                 loaded 1 certificates
06:38:20.822 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
06:38:20.822 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
06:38:20.822 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
06:38:20.822 Jcat                 loaded 1 certificates
06:38:20.822 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
06:38:20.822 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
06:38:20.822 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
06:38:20.822 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
06:38:20.822 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]06:38:20.822 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
06:38:20.823 FuHistory            got schema version of 12
Loading…                 [                                       ]06:38:20.825 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-1.9.28/libfu_plugin_modem_manager.so)
06:38:20.829 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
06:38:20.830 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
06:38:20.830 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
06:38:20.830 XbSilo               file: d650d8ab-e073-f2e8-201c-cff4613bd36c, current:d650d8ab-e073-f2e8-201c-cff4613bd36c, cached: (null)
06:38:20.830 XbSilo               loading silo with file contents
06:38:20.831 FuConfig             loading config /etc/fwupd/fwupd.conf
06:38:20.831 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
06:38:20.831 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
06:38:20.831 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
06:38:20.831 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
06:38:20.831 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
06:38:20.831 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
06:38:20.831 FuConfig             ::configuration loaded
06:38:20.831 FuStruct             FuStructSmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0x8a
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x3
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0xa51
  structure_table_addr: 0xc7c4e000
[... 1420 lines cut ...]
  Modified:             2025-04-22
  UpdateState:          success
[... cut ...]
06:38:23.084 FuEngine             install phase now composite-cleanup
06:38:23.084 FuDeviceList         no replug or re-enumerate required
06:38:23.084 FuEngine             install phase now setup
06:38:23.084 FuEngine             resetting update motd timeout
06:38:23.086 FuIdle               uninhibiting: timeout,signals by update
06:38:23.086 FuIdle               now inhibited: none
06:38:23.086 FuEngine             clearing device-changed allowlist as transaction done

An update requires a reboot to complete. Restart now? [y|N]: 
```

</details>

hughsie, 2025-04-22:
So it looks like the dbx update deploys correctly -- but the UEFI firmware forgets that we've updated it (or resets it to the old version) on reboot. @mrhpearson does this ring any bells? Any future firmware updates that fix deploying the dbx on the ThinkPad L15 Gen 2a perhaps? Thanks.

mauke, 2025-04-22:
My main question is, shouldn't there be some sort of "Firmware Updater" entry listed in `efibootmgr`? (Because there isn't.)

superm1, 2025-04-22:
> My main question is, shouldn't there be some sort of "Firmware Updater" entry listed in `efibootmgr`? (Because there isn't.)

Not for dbx updates. Those are installed natively in OS.

mrhpearson, 2025-04-23:
Curious issue.
Created internal ticket LO-3708 and we'll see if we can reproduce and get input from the FW team as to what is happening

sinanawad, 2025-05-01:
Hi,
I'm having the same behavior on a new laptop I purchased - Chuwi minibook X N150 running kubuntu 25.04 - Currently installed  dbx is (20230501)

Output of `sudo fwupdtool get-devices --plugins uefi_dbx -vv` below

Hope this helps.
-----

05:09:46.020 FuDebug              verbose to debug (on console 0)
05:09:46.021 FuEngine             starting fwupd 2.0.7...
05:09:46.021 FuMain               locked /run/lock/fwupdtool
[GLoading…                 [ -                                     ][GLoading…                 [ -                                     ]05:09:46.034 FuConfig             loading config /etc/fwupd/fwupd.conf
05:09:46.034 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
05:09:46.034 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
05:09:46.034 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
05:09:46.034 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
05:09:46.034 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
05:09:46.034 FuConfig             Failed to open file "/var/etc/fwupd/fwupd.conf": No such file or directory
05:09:46.035 FuConfig             ::configuration loaded
05:09:46.035 FuRemoteList         loading /usr/share/fwupd/metainfo
05:09:46.035 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
05:09:46.035 XbSilo               file: 4f321eae-97fe-0190-20f0-91ac2bd5d05c, current:4f321eae-97fe-0190-20f0-91ac2bd5d05c
05:09:46.035 XbSilo               loading silo with existing file contents
05:09:46.035 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
05:09:46.036 FuRemoteList         ::remote_list changed
05:09:46.036 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
05:09:46.036 FuRemoteList         ::remote_list changed
05:09:46.036 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
05:09:46.036 FuRemoteList         ::remote_list changed
05:09:46.036 FuEngine             priority local and vendor-directory is not download remote, so bumping
05:09:46.036 FuRemoteList         ordering lvfs-testing=lvfs+1
05:09:46.036 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
[GLoading…                 [                                       ][GLoading…                 [                                       ]05:09:46.036 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
05:09:46.037 FuHistory            got schema version of 14
[GLoading…                 [*                                      ]05:09:46.042 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-2.0.7/libfu_plugin_modem_manager.so)
[GLoading…                 [*                                      ]05:09:46.044 FuQuirks             open database /var/cache/fwupd/quirks.db
05:09:46.044 FuQuirks             mtimes unchanged: quirks,pci.ids:1741486505,usb.ids:1736891486, doing nothing
05:09:46.044 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
05:09:46.045 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
05:09:46.045 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
05:09:46.045 XbSilo               file: ebb2755b-377a-3552-a34c-a3bc4bb74106, current:ebb2755b-377a-3552-a34c-a3bc4bb74106
05:09:46.045 XbSilo               loading silo with existing file contents
[GLoading…                 [*                                      ]05:09:46.047 FuConfig             loading config /etc/fwupd/fwupd.conf
05:09:46.047 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
05:09:46.047 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
05:09:46.047 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
05:09:46.047 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
05:09:46.047 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
05:09:46.047 FuConfig             Failed to open file "/var/etc/fwupd/fwupd.conf": No such file or directory
05:09:46.047 FuConfig             ::configuration loaded
05:09:46.047 FuStruct             FuStructSmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0x30
  entry_point_len: 0x18
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x6
  smbios_docrev: 0x0
  entry_point_rev: 0x1
  reserved0: 0x0
  structure_table_len: 0x895
  structure_table_addr: 0x73bd5000
05:09:46.048 FuStruct             FuStructSmbiosStructure:
  type: 0x0
  length: 0x1a
  handle: 0x0
05:09:46.048 FuStruct             FuStructSmbiosStructure:
  type: 0x1
  length: 0x1b
  handle: 0x1
05:09:46.048 FuStruct             FuStructSmbiosStructure:
[... 2464 lines cut ...]
  Name:                 uefi_dbx
  Flags:                ready

05:09:47.576 FuEngine             resetting update motd timeout
05:09:47.576 FuMain               FuUefiDbxDevice:
  DeviceId:             362301da643102b9f38477387e2193e57abaa590
  Name:                 UEFI dbx
  Guid:                 91aa5eb7-0f48-52ff-a68d-c01f25bc33a0 <- UEFI\CRT_9CD3A281B2EA0DE4E3D5FE17A9349C66A5256349FE36EF7A80A74051653443F0&ARCH_X64
  Guid:                 d07ff664-b0e1-5f4e-a723-d7fbcbfcb94f <- UEFI\CRT_3CD3F0309EDAE228767A976DD40D9F4AFFC4FBD5218F2E8CC3C9DD97E8AC6F9D&ARCH_X64
  Guid:                 f8ba2887-9411-5c36-9cee-88995bb39731 <- UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
[... cut ...]
  Checksum:             SHA256(13a1f37bedfb5417b6b737e2a3816c8fd587d74d836914b2b2edc9fd6ca30e58)
  VendorId:             UEFI:Microsoft
  Version:              20230501
  VersionLowest:        20230501
  VersionFormat:        number
  Icon:                 computer
  InstallDuration:      1
  Created:              2025-05-01

[G
05:09:47.576 FuProgress           raw timing data was { 0,001, 0,000, 0,001, 0,028, 0,000 } -- steps were set as [ 1 1 3 1 94 ] but should have been [ 5 0 3 92 0 ] at ../libfwupdplugin/fu-context.c:1101

hughsie, 2025-05-01:
@sinanawad can you provide us with the output of `sudo fwupdtool hwids` please. Then I'll add it to the blocked list.

sinanawad, 2025-05-01:
Sure, thanks for the quick response!
----

Computer Information
--------------------
BiosVendor: American Megatrends International, LLC.
BiosVersion: DNN20 V2.50
BiosMajorRelease: 2
BiosMinorRelease: 50
FirmwareMajorRelease: 00
FirmwareMinorRelease: 12
Manufacturer: CHUWI Innovation And Technology(ShenZhen)co.,Ltd
Family: Default string
ProductName: MiniBook X
ProductSku: Default string
EnclosureKind: a
BaseboardManufacturer: Default string
BaseboardProduct: Default string
Hardware IDs
------------
{374ea8d8-ef94-5f64-8314-bed536d6f1e2}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{04d24e12-8fc6-5eab-ae86-b3e9df1b5e54}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{595929ca-02f0-50d1-a28d-66550aeb0994}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{fe5d10c0-2367-58e0-98bd-671ff5faf232}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{cbd713fd-2b2d-5f71-85b9-747a9ba9a3ff}   <- Manufacturer + Family + ProductName + ProductSku
{aade8def-8d44-58b1-a8b9-d6fdfbb54d8a}   <- Manufacturer + Family + ProductName
{518bbcd2-106c-5a39-a81c-8a269298eb9b}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{1f173942-439d-5b32-a2e2-6222fb844484}   <- Manufacturer + ProductSku
{b3fc87b3-4b45-5f59-a39a-900e32c84d79}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{52e98239-4cfc-5a22-bf17-98539e439dd9}   <- Manufacturer + ProductName
{518bbcd2-106c-5a39-a81c-8a269298eb9b}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{1f173942-439d-5b32-a2e2-6222fb844484}   <- Manufacturer + Family
{567f4c96-c84e-5701-a3c5-533fe388716c}   <- Manufacturer + EnclosureKind
{d90a2fe5-715c-5290-aba9-e395bd261f89}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{368a4dd3-0d2c-519a-9535-cf6d004e9a22}   <- Manufacturer
Extra Hardware IDs
------------------
{0d8d4362-6275-52bf-ba1e-8005069377c4}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor
{21d102ac-0bc2-54a5-9c9c-cc747970333e}   <- Manufacturer + Family + ProductName + BiosVendor
{0166589d-f2d3-5111-825f-41357a6476a3}   <- Manufacturer + BiosVendor

Zhuwenxue2002, 2025-05-01:
> [@sinanawad](https://github.com/sinanawad) can you provide us with the output of `sudo fwupdtool hwids` please. Then I'll add it to the blocked list.您能否向我们提供 `sudo fwupdtool hwids` 的输出。然后我会将其添加到阻止列表中。

Computer Information
--------------------
BiosVendor: AMI
BiosVersion: F.19
BiosMajorRelease: 15
BiosMinorRelease: 19
FirmwareMajorRelease: 11
FirmwareMinorRelease: 1d
Manufacturer: HP
Family: 103C_5335KV HP OMEN
ProductName: OMEN Laptop 15-ek0xxx
ProductSku: 182P8PA#AB2
EnclosureKind: a
BaseboardManufacturer: HP
BaseboardProduct: 878C
Hardware IDs
------------
{cc65b9da-5aea-5be8-b32b-adf9ad512570}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{1a781c81-a6e2-5c70-bc99-95c23619e38f}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{f3685548-2716-5e7c-80b4-b2bfe677a512}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{bc29d072-06e1-57fa-ae46-4168df9a7de1}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{a926b04f-9a6f-5b8b-a94f-273556ec3d97}   <- Manufacturer + Family + ProductName + ProductSku
{b8116e6c-04ea-5cb1-9b52-da66ab5ba0f5}   <- Manufacturer + Family + ProductName
{7b75cd41-69c8-5d48-8471-52ed88b96574}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{7a9ccbe9-cdad-5f95-8f00-ff604676992d}   <- Manufacturer + ProductSku
{ee6ad401-7d31-546a-a573-0e980a2fde19}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{17667187-7d22-5543-8931-dea26ed4ea80}   <- Manufacturer + ProductName
{43c67e96-5d3f-5381-824e-d42f764328ae}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{ef3cb017-4e74-5f9c-a653-aa5a57839b09}   <- Manufacturer + Family
{f999e8e2-2bed-5ee7-85be-a69096b9544d}   <- Manufacturer + EnclosureKind
{4fc65dfe-4dd8-5d24-86f9-3d05e926f379}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{93f84748-c854-5d6b-b78a-13c2361e0758}   <- Manufacturer
Extra Hardware IDs
------------------
{34e7f8c2-0fc2-5bfb-885a-e22c736a0128}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor
{1e8e74b6-1a53-530b-b6ca-9bedb9abbafd}   <- Manufacturer + Family + ProductName + BiosVendor
{bf161834-5522-592a-ae09-48cb5d972497}   <- Manufacturer + BiosVendor

Excuse me, please add my device to the block list as well

hughsie, 2025-05-02:
@Zhuwenxue2002 done -- thanks.

harpogo, 2025-05-02:
Here exactly the same issue on a Thinkpad T460p running Kubuntu 24.04.2. What does adding to the blocklist mean? That the device will be excluded from this update or even from any future UEFI dbx-updates respectively these updates won't be offered any longer on these devices? Is this just a workaround for now? Wouldn't it be better to find a solution to get this update? 

Ok CVE-2024-7344 is seemingly a vulnerability which needs physical presence to be exploited locally - as far as I understood. But although it feels better when it could be fixed.

And: It seems Microsoft fixed this/deployed this Update with its January cumulative update. Does this mean, it's only fixed for Windows or shouldn't it be rather fixed for the whole secure boot functionality for all systems on a device? Could this be a reason for this 'bug' on dual boot systems, if this update was deployed through Microsoft update and (*)ubuntu may not be able to recognize that for some reason?

hughsie, 2025-05-04:
> Is this just a workaround for now? 

Yes.

> Wouldn't it be better to find a solution to get this update?

The only solution is a firmware update from the vendor. In the case of Lenovo you should file a bug with https://github.com/fwupd/firmware-lenovo and I can tag the right people. For other vendors not shipping firmware updates on the LVFS a workaround is the best we can do.

closed 2025-05-04 by hughsie

Zhuwenxue2002, 2025-05-05:
> [@Zhuwenxue2002](https://github.com/Zhuwenxue2002) done -- thanks. 

What should I do to apply the update? I'm still prompted that the project needs to be updated, as of now

[image]

hughsie, 2025-05-05:
@Zhuwenxue2002 what hardware is that? e.g. the output from "sudo fwupdtool hwids"

jprojivazquez, 2025-05-06:
HI! I'm having the same issue on my Huawei Matebook D15 2020 running Ubuntu 24.04.2 LTS. 

My hardware is a AMD R7 3700U, 8 gb ddr4 ram and 512 nvme drive.

Here's the output for "sudo fwupdtool hwids" and a screenshot of the "update":

Computer Information
--------------------
BiosVendor: HUAWEI
BiosVersion: 1.19
BiosMajorRelease: 1
BiosMinorRelease: 19
FirmwareMajorRelease: 01
FirmwareMinorRelease: 13
Manufacturer: HUAWEI
Family: MateBook D
ProductName: BOHK-WAX9X
ProductSku: C331
EnclosureKind: a
BaseboardManufacturer: HUAWEI
BaseboardProduct: BOHK-WAX9X-PCB
Hardware IDs
------------
{87acc954-9b7b-56f6-8485-fe3cd63719d9}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{763a51c7-4e60-5197-b13f-907c32bacb2d}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{066e57a9-f765-55bb-8c84-6fb55bee9848}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{345515cd-cd84-5590-b163-efc4d5e33507}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{b348da23-e0bc-5841-b59d-b97a5364c341}   <- Manufacturer + Family + ProductName + ProductSku
{44a78ffd-1007-5aa2-8b76-eb369d2ed7ae}   <- Manufacturer + Family + ProductName
{0db08c6a-5c36-51f4-b70f-7ca7fef8c975}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{496598a3-ce2c-5a54-a31b-4cbc2a4546ec}   <- Manufacturer + ProductSku
{f044caa2-5cc7-5934-918e-f667260d9752}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{da3c0716-2635-5c21-8c69-20cba6635a23}   <- Manufacturer + ProductName
{717c8756-3051-5278-979a-5685cd8e2b34}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{b9c06130-ca27-564c-a15d-abe34fb071ae}   <- Manufacturer + Family
{4b596832-4424-5130-97c5-2524773b6e97}   <- Manufacturer + EnclosureKind
{a98d46c9-ed25-54da-8510-ef9035cecdd1}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{0df4aa3f-2706-51d4-9296-80119e47f1e1}   <- Manufacturer
Extra Hardware IDs
------------------
{dc801927-e9ac-537a-be28-bff79b8ebb80}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor
{ddf7d296-0de7-5a6f-922a-d9bc7e670143}   <- Manufacturer + Family + ProductName + BiosVendor
{04798471-7c9b-5189-b99b-4d19e1a6fa89}   <- Manufacturer + BiosVendor

[image]

ftmiranda, 2025-05-06:
Same issue

Computer Information
--------------------
BiosVendor: LENOVO
BiosVersion: M4GKT25A
BiosMajorRelease: 1
BiosMinorRelease: 37
FirmwareMajorRelease: 01
FirmwareMinorRelease: 16
Manufacturer: LENOVO
Family: ThinkStation P360 Tiny
ProductName: 30FBS39B01
ProductSku: LENOVO_MT_30FB_BU_Think_FM_ThinkStation P360 Tiny
EnclosureKind: 23
BaseboardManufacturer: LENOVO
BaseboardProduct: 330E
Hardware IDs
------------
{0250582b-f857-55bb-b0d9-e5be68610f49}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{9a375b0c-1701-594d-bfa3-5ae58bc94334}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{ad527b69-1af3-5bb0-a100-03671ef159e2}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{b3097405-d7c7-5f54-ba3b-e33f6d479a00}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{36488102-7f7a-5f08-b3b0-a6f190f3d84c}   <- Manufacturer + Family + ProductName + ProductSku
{4504b509-4d4f-5221-8a87-e21262eee395}   <- Manufacturer + Family + ProductName
{75b4ed92-b339-5ed8-89fc-8310dc0d833c}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{c24763d8-8fdc-5c79-978b-2cec383ed9e8}   <- Manufacturer + ProductSku
{7cac3995-e3a7-5965-a488-f0558498fc73}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{4b4306a1-f8da-5055-9824-f17917d52e01}   <- Manufacturer + ProductName
{b3a93ac3-1722-5397-8a19-9e89d344b222}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{8bd04657-1fb8-598a-a2e2-903a0d6e08b6}   <- Manufacturer + Family
{78cfbbd7-c5d4-5d24-9ad2-df6114c7d37d}   <- Manufacturer + EnclosureKind
{55db0253-c51a-527d-a7ec-6a8fd1bd07eb}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{6de5d951-d755-576b-bd09-c5cf66b27234}   <- Manufacturer
Extra Hardware IDs
------------------
{c8b191f4-c457-5c18-b66f-77f11efe6a7b}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor
{7c495e92-1d95-5d11-9c74-70d3d0d3dcff}   <- Manufacturer + Family + ProductName + BiosVendor
{99431f53-09a1-5869-be79-65e2fa3f341d}   <- Manufacturer + BiosVendor

hughsie, 2025-05-06:
@mrhpearson the "ThinkStation P360 Tiny" is expected to accept a dbx update, correct?

mrhpearson, 2025-05-07:
> [@mrhpearson](https://github.com/mrhpearson) the "ThinkStation P360 Tiny" is expected to accept a dbx update, correct?

Yeah, should be. I'll check with the FW team in case there is something funky going on. Internal ticket LO-3725

@ftmiranda - might be worth raising a separate issue in https://github.com/fwupd/firmware-lenovo/issues . Tag me if you do.

mrhpearson, 2025-05-28:
For the original reporter - I heard back from the L14 G2 AMD FW team and they can't reproduce the problem. They asked if you can update to the latest BIOS (recently released) and try again please.

For Ideacentre - afraid that isn't in the Linux program so I have no strings I can pull on as it's not supported. You can try Lenovo support but they'll likely require you to update from Windows.

T460p - I don't have anything I can do there I'm afraid. Sorry

P360 Tiny - I was really hoping to get the solution from the L14 G2 team and go from there, but I can follow up with the P360 team separately.

EDU-ACACIO, 2025-06-08:
I'm having the same issue on my GalaxyBook2 Dualboot 
[image]

Any chance of a future update fix this?

Computer Information
--------------------
BiosVendor: American Megatrends International, LLC.
BiosVersion: P11RGK.050.250403.SX
BiosMajorRelease: 5
BiosMinorRelease: 25
FirmwareMajorRelease: ff
FirmwareMinorRelease: ff
Manufacturer: SAMSUNG ELECTRONICS CO., LTD.
Family: Galaxy Book2
ProductName: 550XED
ProductSku: SCAI-A5A5-A5A5-ADLP-PRGK
EnclosureKind: a
BaseboardManufacturer: SAMSUNG ELECTRONICS CO., LTD.
BaseboardProduct: NP550XED-KT3BR
Hardware IDs
------------
{80a1e0fd-9671-529e-ac32-e001f224e48f}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{07dce2a6-2327-5c78-a3b5-48d567c1f98c}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{71ab1bcd-b666-565f-8e8a-0082ce271f3f}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{909eaf8f-0532-587e-b7d2-101b4437f187}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{717e995f-7d12-564b-bf51-fca11ae288ce}   <- Manufacturer + Family + ProductName + ProductSku
{7ca19a92-3c4e-5eb0-948b-6bfc2b8491a9}   <- Manufacturer + Family + ProductName
{ad8b126d-85e9-5e14-8be4-32055c2101e8}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{82f4fdcf-ddb7-5842-a8da-ec8f2154e247}   <- Manufacturer + ProductSku
{12de2cb6-73a6-5e6b-ba31-ae14825bccb9}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{02eb7e92-b76a-5343-a341-9efc067647a8}   <- Manufacturer + ProductName
{8d9d0124-dd38-542a-b100-53cc2e9461fc}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{2f1af45d-3918-5580-a25f-adc0e774751f}   <- Manufacturer + Family
{b5ba5e91-fdc7-52c1-85b5-9a0da49c298f}   <- Manufacturer + EnclosureKind
{69327635-06bd-53eb-bccd-ed70f4c67bab}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{ca0158e4-038b-5a84-8037-1b3cbf948d8a}   <- Manufacturer
Extra Hardware IDs
------------------
{6d045d3a-a250-515a-b308-8fcf2668c6e5}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor
{dbc2ef75-1ce4-59a2-852b-d6f36b1cb101}   <- Manufacturer + Family + ProductName + BiosVendor
{b78f6180-e9e7-5276-a663-335ab8100f3b}   <- Manufacturer + BiosVendor

hughsie, 2025-06-09:
@EDU-ACACIO can you open a new issue please.

EDU-ACACIO, 2025-06-13:
Mine just updated to a new version 202550507 that came out, no issues apparently.

[image]

hughsie, 2025-06-13:
For machines that are failing, what's the output of `df -h | grep efivars` please.

Dan-Ptr, 2025-06-13:
@hughsie @EDU-ACACIO The most recent update, 202507, fixed it for me too. Thanks!

whtyger, 2025-06-15:
I observe the same problem on Dell Latitude 5591, Kubuntu 20.04 and fwupmgr version 1.7.9.
`df -h | grep efivars` reports nothing.
Should I post an output of `sudo fwupdtool hwids`?
Found a proposal somewhere to run `fwupdmgr update --force`, is it worth trying?

superm1, 2025-06-15:
> I observe the same problem on Dell Latitude 5591, Kubuntu 20.04 and fwupmgr version 1.7.9. BIOS version is the latest, 1.38.0. `df -h | grep efivars` reports nothing. Should I post an output of `sudo fwupdtool hwids`? Found a proposal somewhere to run `fwupdmgr update --force`, is it worth trying?

The version of fwupd you're using is EoL.  You should upgrade to a currently supported version to get help upstream with issues.
https://github.com/fwupd/fwupd/security/policy

whtyger, 2025-06-17:
> I installed all the latest Ubuntu updates and things seem to be well. UEFI dbx is at 20250507 now, even after a reboot.

You use Ubuntu 24.04, as I suppose. Previous Ubuntu versions, even with the latest updates, have got older fwupd and cannot upgrade it without recompiling and aren't proposed UEFI dbx version newer than 20241101.

harpogo, 2025-06-17:
> I installed all the latest Ubuntu updates and things seem to be well. UEFI dbx is at 20250507 now, even after a reboot.

Same here! I haven't recognized it until now after I read your post. Thanks for that. I didn't try a firmware-update again. It seems to be made automatically. 
I just did the usual distro-updates via apt. I'm not sure which update may have solved the issue. From the list of updates during the last weeks only the linux-firmware update shipped last week is what seem possibly related to this, but the changelog doesn't show any hint.

> You use Ubuntu 24.04, as I suppose.

Here it's on 24.04.

polo199, 2025-06-17:
Hi there,
I'm getting the same bug on a Lenovo e485 laptop running Kubuntu 24.04 with `Upgrade UEFI dbx from 20241101 to 20250507`. It does not upgrade.

PolyachenkoYA, 2025-07-01:
getting the same issue on Lenovo P16s, dual boot Ubuntu 22.04 + Win11

superm1, 2025-07-01:
> Getting the same issue on Lenovo P16s, dual boot Ubuntu 22.04 + Win11.
> Trying to upgrade to 20241101.
> 
> `$df -h | grep efivars`
> ```efivarfs        196K   80K  112K  42% /sys/firmware/efi/efivars```

22.04 has an EOL fwupd included. It's not supported upstream.

You can install the snap, flatpak or from source or complain to Canonical to update it.

ftmiranda, 2025-07-05:
> I have the same issue.
> 
> Why is this issue closed?

because no one will fix it

RamAvni, 2025-07-05:
@ftmiranda 
wdym?

hughsie, 2025-07-05:
> I have the same issue.

It's not the same issue. Your issue is nothing to do with the dbx. You need to open an issue in https://github.com/fwupd/firmware-lenovo/issues if you want Lenovo to help.

> because no one will fix it

That's not true, please don't resort to FUD.

ftmiranda, 2025-07-05:
> FUD

fair, I guess I was too hash on that comment :-)

ku4eto, 2025-08-06:
I am experiencing similar issue - if not hte same.

On a Lenovo Thinkpad T15, i keep getting one and same update for the UEFI dbx, but the version number seems messed up.

```
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  500
│       Minimum Version:  500
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            14503b3d-73ce-5d06-8137-77c68972a341 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649
│                         5971a208-da00-5fce-b5f5-1234342f9cf7 ← UEFI\CRT_A9087D1044AD18F7A94916D284CBC01827CF23CD8F60B79072C9CAA1FEF4D649&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│                         • Signed Payload
```

```
Devices that were not updated correctly:

 • UEFI dbx (500 → 20241101)

```

```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 500 to 20241101?                                       ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the list of forbidden signatures (the "dbx") to the latest      ║
║ release from Microsoft.                                                      ║
║                                                                              ║
║ An insecure version of Howyar's SysReturn software was added, due to a       ║
║ security vulnerability that allowed an attacker to bypass UEFI Secure Boot.  ║
║                                                                              ║
║ UEFI dbx and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

```

hughsie, 2025-08-06:
Ohh no, I'm mistaken; A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503 is 20230501 -- @ku4eto what fwupd version are you using?

ku4eto, 2025-08-06:
That would e 1.7.9.
I am aware its EOL, but i do not know if the issue is from the update package or the `fwupd` itself.

superm1, 2025-08-06:
You need newer fwupd. Upgrade please.

mrhpearson, 2025-08-06:
I'd missed my tag above - apologies.

As noted, for the Lenovo platforms - please create a ticket in https://github.com/fwupd/firmware-lenovo/issues for your platform with the details and feel free to tag me. I can review specifics with the FW team.