fwupd/fwupd #8699: UEFI db - Invalid

alb2talk, 2025-04-15:
**Describe the bug**
Just got update 2.0.8 from Tumbleweed distro, I see that a new option `UEFI db` has been added to me, but it is says invalid.

**fwupd version information**
```
$ fwupdmgr --version
compile   info.libusb                   1.0.28
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.3
compile   org.freedesktop.fwupd         2.0.8
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.14.1-1-default
runtime   org.freedesktop.fwupd         2.0.8
```

```shell
$ fwupdtool get-report-metadata
Loading…                 [*********************                  ]16:16:17.922 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5:1.0/host5/target5:0:0/5:0:0:0/block/sdb: failed to setup: failed to read version: command fail with status 2, senseKey 0x05, asc 0x24, ascq 0x00
Loading…                 [**********************                 ]16:16:17.933 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/host4/target4:0:0/4:0:0:0/block/sda: failed to setup: failed to read version: command fail with status 2, senseKey 0x05, asc 0x20, ascq 0x00
16:16:17.939 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/host4/target4:0:0/4:0:0:0/block/sda: failed to setup: no value for REV
16:16:18.011 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:17.0/ata4/host3/target3:0:0/3:0:0:0/block/sr0: failed to subclass open: failed to open /dev/sr0: No medium found
16:16:18.075 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:17.0/ata4/host3/target3:0:0/3:0:0:0/block/sr0: failed to subclass open: failed to open /dev/sr0: No medium found
Waiting…                 [*************************************  ]
BatteryLevel:           101
BatteryThreshold:       10
BootTime:               1744733018
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.fwupd): 2.0.8
CpuArchitecture:        x86_64
CpuModel:               Intel Core™ i7-10700 CPU @ 2.90GHz
DisplayState:           connected
DistroId:               opensuse-tumbleweed
DistroName:             openSUSE Tumbleweed
DistroPrettyName:       openSUSE Tumbleweed
DistroVersion:          20250414
FwupdSupported:         True
HostBaseboardManufacturer: Dell Inc.
HostBaseboardProduct:   32W55
HostBiosMajorRelease:   01
HostBiosMinorRelease:   1d
HostBiosVendor:         Dell Inc.
HostBiosVersion:        1.29.0
HostEnclosureKind:      3
HostFamily:             OptiPlex
HostFirmwareMajorRelease: ff
HostFirmwareMinorRelease: ff
HostProduct:            OptiPlex 5080
HostSku:                9A6
HostVendor:             Dell Inc.
KernelCmdline:          selinux=1 systemd.machine_id=36c21dda967847a69b1f9ddc4badaba2 intel_iommu=on enforcing=1 mmio_stale_data=full,nosmt
KernelName:             Linux
KernelVersion:          6.14.1-1-default
LidState:               unknown
PlatformArchitecture:   x86_64
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.8
RuntimeVersion(org.freedesktop.fwupd-efi): 1.7
RuntimeVersion(org.kernel): 6.14.1-1-default
DeviceId:               3956d699b87ea6a7250cf559effb015f36f6307d
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
[... 360 lines cut ...]
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Can tag for emulation
```
```

</details>

**Additional questions**

I use systemd-boot for my Tumbleweed.

```
Operating System: openSUSE Tumbleweed 20250414
KDE Plasma Version: 6.3.4
KDE Frameworks Version: 6.12.0
Qt Version: 6.9.0
Kernel Version: 6.14.1-1-default (64-bit)
Graphics Platform: Wayland
Processors: 8 × Intel® Core™ i7-10700 CPU @ 2.90GHz
Memory: 30.9 GiB of RAM
Graphics Processor: Intel® UHD Graphics 630
Manufacturer: Dell Inc.
Product Name: OptiPlex 5080
```

closed 2025-04-18 by alb2talk

de-soot, 2025-04-21:
how did you fix it?

alb2talk, 2025-04-21:
I haven't fixed yet so far, as far as i understand i may need to configure some settings on the very bottom of this page.
https://en.opensuse.org/openSUSE:UEFI

This is my current status
[image]
First I have to activate expert key management, then i have to continue with any of those settings, or exactly "db" from what I read, it's not that easy, it can be done, but for me it would be best to do it on a new installation.

In short, I've given up for now, I have my system secured hdd/swap with passkey. I'm afraid it will break some existing configuration and then it will get out of control.