fwupd/fwupd #8701: Migration to 2.0.7 - get-devices - DELL WD19TB and WD22TB shows up with fwupdtool but not with fwupdmgr

bbo-igel, 2025-04-16:
**Describe the question**
Another issue that came up with the switch from 1.9.26 to 2.0.7 in IGEL OS.

The DELL WD19TB and WD22TB are not shown as devices with fwupdmgr get-devices, but fwupdtool get-devices does show them.

<details>
  <summary>fwupdmgr get-devices -v</summary>

```shell
(fwupdmgr:12512): GVFS-RemoteVolumeMonitor-DEBUG: 11:21:07.904: Error: The connection is closed

(fwupdmgr:12512): GLib-GIO-DEBUG: 11:21:07.906: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: unwatch_fast: "/system/proxy/" (active: 1, establishing: 0)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: unwatch_fast: "/system/proxy/http/" (active: 1, establishing: 0)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.906: unwatch_fast: "/system/proxy/https/" (active: 1, establishing: 0)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.907: unwatch_fast: "/system/proxy/ftp/" (active: 1, establishing: 0)
(fwupdmgr:12512): dconf-DEBUG: 11:21:07.907: unwatch_fast: "/system/proxy/socks/" (active: 1, establishing: 0)
(fwupdmgr:12512): GLib-GIO-DEBUG: 11:21:07.930: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:12512): Fwupd-DEBUG: 11:21:07.934: Emitting ::status-changed() [idle]
FuMain-INFO: 11:21:07.942: FwupdDevice:
  DeviceId:             f3a85073aaf5c28b929c7568e2fc1b724677cadb
  Name:                 System Firmware
  Guid:                 b68b18a3-adb2-44c8-9c22-b686cceb9c08
  Summary:              UEFI System Resource Table device (updated via NVRAM)
  Plugin:               uefi_capsule
  Protocol:             org.uefi.capsule
  Flags:                internal|updatable|require-ac|supported|needs-reboot|usable-during-update
  RequestFlags:         allow-generic-message
  Vendor:               Dell
  VendorId:             DMI:Dell Inc.
  Version:              1.28.0
  VersionLowest:        1.28.0
  VersionFormat:        dell-bios
  VersionRaw:           0x00011c00
  VersionLowestRaw:     0x00011c00
  Icon:                 computer
  Created:              2025-04-16
  UpdateState:          success

FuMain-INFO: 11:21:07.942: FwupdDevice:
  DeviceId:             362301da643102b9f38477387e2193e57abaa590
  CompositeId:          f3a85073aaf5c28b929c7568e2fc1b724677cadb
  Name:                 UEFI dbx
  Guid:                 4a6cd2cb-8741-5257-9d1f-89a275dacca7 ← UEFI\CRT_E28D59CA489BD2AD580F2EA5D62D6A29BB9C02AE5A818434A37DA7FC11DFF9E9&ARCH_X64
  Guid:                 f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
  Summary:              UEFI revocation database
  Plugin:               uefi_dbx
  Protocol:             org.uefi.dbx2
  Flags:                internal|updatable|supported|needs-reboot|can-verify|can-verify-image|usable-during-update|only-version-upgrade|signed-payload|can-emulation-tag
  Checksum:             SHA256(13a1f37bedfb5417b6b737e2a3816c8fd587d74d836914b2b2edc9fd6ca30e58)
  VendorId:             UEFI:Microsoft
  Version:              20230501
  VersionLowest:        20230501
  VersionFormat:        number
  Icon:                 computer
  InstallDuration:      1
  Created:              2025-04-16
``` 
</details>

[... 441 lines cut ...]
PlatformArchitecture:   x86_64
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat): 0.2.0
RuntimeVersion(com.hughsie.libxmlb): 0.3.21
RuntimeVersion(org.freedesktop.fwupd): 2.0.7
RuntimeVersion(org.freedesktop.fwupd-efi): 1.7
RuntimeVersion(org.kernel): 6.12.20
DeviceId:               f3a85073aaf5c28b929c7568e2fc1b724677cadb
  pre:
      CapsuleApplyMethod: nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader: False
  post:
      LastAttemptStatus: 0x0
      LastAttemptVersion: 0x0
DeviceId:               362301da643102b9f38477387e2193e57abaa590
  pre:
      EfivarsNvramUsed: 70274
      LinuxLockdown:    integrity
      EfivarsNvramUsed: 70274
      SecureBoot:       Disabled
      UEFIUXCapsule:    Enabled

```

bbo-igel, 2025-04-16:
Hi @hughsie  @superm1  Any suggestion where to look or what might go wrong here?

superm1, 2025-04-16:
I would start with systemd lockdown stuff like that other bug. Narrow it down.

superm1, 2025-04-16:
Actually I see this;

> Apr 16 11:09:06 ITCA4BB6D09CC1B fwupd[11920]: 09:09:06.695 FuEngine             failed to setup backend udev: failed to set up netlink: bind to udev socket failed: Address already in use

I'm pretty sure that happens from an old dependency being installed. I remember reports of it on new fwupd with old Ubuntu too. I don't remember the specific dependency though.

@hughsie do you?

superm1, 2025-04-16:
What version of libudev?  See https://github.com/fwupd/fwupd/issues/8083 for context on why I'm asking.

bbo-igel, 2025-04-16:
> What version of libudev? See [#8083](https://github.com/fwupd/fwupd/issues/8083) for context on why I'm asking.

libudev-so.1.7.7 from libudev1 254.22-1~bpo12+1 (debian backport)

superm1, 2025-04-16:
Yeah, I _meant_ to ask for both packages versions.  Can you try a newer libusb?

superm1, 2025-04-16:
I _think_ you'll want libusb 1.0.27 or later.  Maybe we should be guarding against this @hughsie?

hughsie, 2025-04-16:
> I'm pretty sure that happens from an old dependency being installed.

Delete the venv and re-run ./contrib/setup normally fixes this.

closed 2025-04-16 by superm1

bbo-igel, 2025-04-17:
FYI - I did a test branch where I updated to libusb 1.0.27 (which required a GLIBC update)
The issue is still there, I now focusing on our OS to find out what goes wrong.

superm1, 2025-04-17:
> FYI - I did a test branch where I updated to libusb 1.0.27 (which required a GLIBC update)
> The issue is still there, I now focusing on our OS to find out what goes wrong.

Did you rebuild fwupd against the newer version or just add updated library to link with?