fwupd/fwupd #8702: Unable to update dbx

RobusTetus, 2025-04-16:
**Describe the bug**
On a fresh install of Fedora 42, trying to update UEFI dbx, fwupd does not report anything amiss when using `update`, either in terminal or GNOME software and looks like there is no update available.

Upon closer inspection (me knowing that dbx is not in fact up to date, since I reset Secure Boot in bios and ran `mokutil --reset` before when I was trying to debug #8695) and installing .cab file directly after downloaded from lvfs and installed with `sudo fwupdmgr install <path-to-dbx-update.cab>` output is just this

```
Device UEFI dbx [362301da643102b9f38477387e2193e57abaa590] has no firmware version
```

When running `sudo fwupdmgr get-details` on the .cab following is the output.

```
LENOVO 82TK
│
└─UEFI dbx:
  │   Identifikátor zařízení: 362301da643102b9f38477387e2193e57abaa590
  │   Souhrn:             UEFI revocation database 10d45fcb
  │   Výrobce:            UEFI:Microsoft
  │   Délka instalace:    1 sekunda
  │   Chyba při aktualizaci: Device UEFI dbx [362301da643102b9f38477387e2193e57abaa590] has no firmware version
  │   GUID:               f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
  │   Příznaky zařízení:  • Vestavěné zařízení
  │                       • Podporováno na vzdáleném serveru
  │                       • Po instalaci vyžaduje restart
  │                       • Je k dispozici kryptografické ověření otisku dat
  │                       • Zařízení je možné používat i v průběhu aktualizace
  │                       • Možné aktualizovat
  │                       • Zařízení umožňuje pouze aktualizace na novější verze
  │                       • Podepsaný obsah
  │                       • Může označit pro emulaci
  │ 
  └─Aktualizace nastavení pro Secure Boot dbx:
        Nová verze:       20241101
        Souhrn:           UEFI Secure Boot Forbidden Signature Database
        Varianta:         x64
        Licence:          Uzavřená
        Velikost:         15,1 kB
        Naléhavost:       Vysoká
        Výrobce:          Linux Foundation
        Příznaky vydání:  • Důvěryhodný obsah
                          • Důvěryhodná metadata
        Popis:            
        This updates the list of forbidden signatures (the "dbx") to the latest release from Microsoft.
        
        An insecure version of Howyar's SysReturn software was added, due to a security vulnerability that allowed an attacker to bypass UEFI Secure Boot.
        Problémy:         CVE-2024-7344
                          529659
        Kontrolní součet: d661d4a0aaca09dfa9e56967ca2467b0575fc07cb704d182fa8c68225452957f

```

**Steps to Reproduce**
Not sure how it happened, but I suspect I might have borked something when running `mokutil --reset`. It never happened before, even when reseting sb keys in bios, the dbx would just roll back to version 77 or so.

**Expected behavior**
When installing the file directly with `fwupd` and `--force` flag should just ignore firmware version and install it anyway, thus restoring dbx into firmware version that is known by lvfs.

**fwupd version information**
Please provide the version of the daemon and client.

**Also observed same behaviour on fwupd 2.0.7.**

```shell
rob@fedora:~$ sudo fwupdtool --version
Načítání…                [************************************** ]
VAROVÁNÍ: Nezjištěn (či nenastaven) UEFI ESP oddíl
Podrobnosti viz https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found.
compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.8
compile   com.hughsie.libxmlb           0.3.22
[... 844 lines cut ...]
│                         • Po instalaci vyžaduje restart
│                         • Zařízení je možné používat i v průběhu aktualizace
│     Požadavky ze zařízení: • Zpráva
│   
└─USB4 Retimer:
      Identifikátor zařízení: 23f75b8fe7e0d7d43582bccecaf3c9a0472e72b5
      Souhrn:             A physical layer protocol-aware, software-transparent extension device that forms two separate electrical link segments
      Stávající verze:    10.00
      Výrobce:            THUNDERBOLT:0x8087
      GUID identifikátory: e6f2e4ac-f76a-5fc4-b36a-6f9cb548689e ← THUNDERBOLT\VEN_8087&DEV_15EE
                          cd520904-a41c-502f-9233-2746479e7c42 ← TBT-808715ee-retimer1-0:3.1
      Příznaky zařízení:  • Vestavěné zařízení
                          • Možné aktualizovat
                          • Zařízení podporuje rozfázované aktualizace
                          • Nepodepsaný obsah
                          • Může označit pro emulaci
```

</details>

**Additional questions**

- Operating system and version: Fedora Silverblue 42.20250416.0 (freshly installed)
- Have you tried rebooting? - Countless times
- Is this a regression? - nope, since version 2.0.7 worked before, now even that version does not work

hughsie, 2025-04-16:
What does `sudo fwupdtool get-devices --plugins uefi_dbx -vv` say for you please? I'm specifically looking for the `Checksum` line -- thanks. It's probably missing a line in https://github.com/fwupd/fwupd/blob/main/plugins/uefi-dbx/uefi-dbx.quirk#L4

RobusTetus, 2025-04-16:
> What does `sudo fwupdtool get-devices --plugins uefi_dbx -vv` say for you please? I'm specifically looking for the `Checksum` line -- thanks. It's probably missing a line in https://github.com/fwupd/fwupd/blob/main/plugins/uefi-dbx/uefi-dbx.quirk#L4

Here it is.

```
FuUefiDbxDevice:
  DeviceId:             362301da643102b9f38477387e2193e57abaa590
  Name:                 UEFI dbx
  Guid:                 f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
  Summary:              UEFI revocation database 10d45fcb
  Plugin:               uefi_dbx
  Protocol:             org.uefi.dbx2
  Flags:                internal|updatable|supported|needs-reboot|can-verify|can-verify-image|usable-during-update|only-version-upgrade|signed-payload|can-emulation-tag
  Checksum:             SHA256(10d45fcba396aef3153ee8f6ecae58afe8476a280a2026fc71f6217dcf49ba2f)
  VendorId:             UEFI:Microsoft
  VersionFormat:        number
  Icon:                 application-certificate
  InstallDuration:      1
  Created:              2025-04-16
  InstanceId[qu]:       254c69ad-6a50-577c-ac2e-8e7fb9026c35 ← UEFI\GUID_8be4df61-93ca-11d2-aa0d-00e098032b8c&NAME_KEK
  InstanceId[qu]:       c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
  InstanceId[vi,qu]:    f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
  InstanceId[qu]:       cfc4f9b5-023c-58bd-9db1-72f1d8b38c23 ← UEFI\CSUM_10D45FCBA396AEF3153EE8F6ECAE58AFE8476A280A2026FC71F6217DCF49BA2F
  PhysicalId:           dbx
  BackendId:            8be4df61-93ca-11d2-aa0d-00e098032b8c-KEK
  AcquiesceDelay:       50
  FirmwareGType:        FuEfiSignatureList
  PossiblePlugin:       uefi_dbx
  PrivateFlags:         md-only-checksum,md-set-version,host-firmware-child,is-open,update-pending,registered
  Guid:                 8be4df61-93ca-11d2-aa0d-00e098032b8c
  Name:                 KEK

```

closed 2025-05-03 by hughsie