fwupd/fwupd #8705: Lenovo P360 Tiny Workstation Ubuntu 24.04 can't update drivers/firmware correctly

ftmiranda, 2025-04-16:
**Describe the bug**
Using a Lenovo P360 Tiny workstation and Ubuntu 24.04 LTS
When I run $ sudo fwupdmgr update command it shows updates and it does show the process of updating, but after a recommended reboot, no updates were applied.

**Steps to Reproduce**
Use the Ubuntu 24.04 LTS command $ sudo fwupdmgr update to update the outdated components:
$ sudo fwupdmgr update
Devices with no available firmware updates:
 • SSD 990 EVO Plus 2TB
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Embedded Controller from 0.1.22 to 0.1.24?                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Fix some EC Setup issues.                                                  ║
║                                                                              ║
║ 30FBS39B01 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Intel Management Engine from 10.0.15.1662 to 10.1.30.2330?           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Enhancement to address security vulnerability.                             ║
║                                                                              ║
║ 30FBS39B01 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
Updating Intel Management Engine…                                ]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.37 to 0.1.50?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Fix some BIOS Setup issues.                                                ║
║                                                                              ║
║ 30FBS39B01 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
Updating System Firmware…[ -                                     ]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
Do not turn off your computer or remove the AC adapter while the update is in progress.
Devices with the latest available firmware version:
 • UEFI dbx

Then reboot

**Expected behavior**
After the reboot all the updated devices are running the updated version, but instead it remains on the old versions

**fwupd version information**
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.28
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
[... 272 lines cut ...]
│     GUID:               0c208ff1-6a42-45e1-887a-7694815a9981
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            LENOVO
      GUID:               3110ca52-44a7-5bdd-ba77-8a8de8eb6a2b ← UEFI\CRT_64ADF813F63132B2F8E8499FBD00D90BCF762DC3

</details>

**Additional questions**

- Operating system and version:
$ uname -a
Linux ThinkStation-P360 6.11.0-21-generic #21~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Feb 24 16:52:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 24.04 LTS
- Have you tried rebooting?
yes
- Is this a regression?
not sure

ftmiranda, 2025-04-17:
It looks like the issue is between the fwupdmgr shipped with Ubuntu 24.04.2 LTS
I have done several troubleshooting steps.. and I noticed that that issue lies with the inability of the fwupdmgr to create an boot entry with the proper configuration: 
$ fwupdmgr update --verbose
(pkttyagent:4768): GLib-GIO-DEBUG: 07:36:14.538: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:4766): GLib-GIO-DEBUG: 07:36:14.547: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:4766): dconf-DEBUG: 07:36:14.547: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:4766): dconf-DEBUG: 07:36:14.547: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:4766): dconf-DEBUG: 07:36:14.547: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:4766): dconf-DEBUG: 07:36:14.547: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:4766): dconf-DEBUG: 07:36:14.547: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:4766): GLib-GIO-DEBUG: 07:36:14.547: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:4766): GLib-GIO-DEBUG: 07:36:14.548: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:4766): GLib-GIO-DEBUG: 07:36:14.548: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:4766): dconf-DEBUG: 07:36:14.550: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:4766): dconf-DEBUG: 07:36:14.550: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:4766): dconf-DEBUG: 07:36:14.550: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:4766): dconf-DEBUG: 07:36:14.550: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:4766): dconf-DEBUG: 07:36:14.550: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:4766): Fwupd-DEBUG: 07:36:14.551: Emitting ::status-changed() [idle]
Devices with no available firmware updates: 
 • SSD 990 EVO Plus 2TB
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Embedded Controller from 0.1.22 to 0.1.24?                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Fix some EC Setup issues.                                                  ║
║                                                                              ║
║ 30FBS39B01 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Fwupd-INFO: 07:36:18.018: downloading https://fwupd.org/downloads/a844840f4cea4ac4e237e3a84c2fa6c98aa7c03f20e64d4f02da0706d1997055-ThinkCentre-Tiny8-ECFirmware-M4GCT18A.cab
(fwupdmgr:4766): Fwupd-DEBUG: 07:36:18.018: Emitting ::status-changed() [downloading]
Downloading…             [         |                             ]Fwupd-INFO: 07:36:18.377: download progress: 100%
Downloading…             [***************************************]Fwupd-INFO: 07:36:18.571: download progress: 2%
Downloading…             [                                       ]Fwupd-INFO: 07:36:18.571: download progress: 5%
Downloading…             [*                                      ]Fwupd-INFO: 07:36:18.574: download progress: 8%
Downloading…             [***                                    ]Fwupd-INFO: 07:36:18.574: download progress: 9%
Downloading…             [***                                    ]Fwupd-INFO: 07:36:18.575: download progress: 12%
Downloading…             [****                                   ]Fwupd-INFO: 07:36:18.576: download progress: 15%
Downloading…             [*****                                  ]Fwupd-INFO: 07:36:18.578: download progress: 17%
Downloading…             [******                                 ]Fwupd-INFO: 07:36:18.578: download progress: 20%
Downloading…             [*******                                ]Fwupd-INFO: 07:36:18.578: download progress: 21%
Downloading…             [********                               ]Fwupd-INFO: 07:36:18.578: download progress: 26%
Downloading…             [**********                             ]Fwupd-INFO: 07:36:18.578: download progress: 27%
Downloading…             [**********                             ]Fwupd-INFO: 07:36:18.579: download progress: 29%
Fwupd-INFO: 07:36:18.579: download progress: 30%
Downloading…             [***********                            ]Fwupd-INFO: 07:36:18.579: download progress: 33%
Downloading…             [************                           ]Fwupd-INFO: 07:36:18.579: download progress: 34%
Downloading…             [*************                          ]Fwupd-INFO: 07:36:18.584: download progress: 37%
Downloading…             [**************                         ]Fwupd-INFO: 07:36:18.585: download progress: 41%
Downloading…             [***************                        ]Fwupd-INFO: 07:36:18.585: download progress: 44%
[... cut ...]
/boot/grub/x86_64-efi
/boot/grub/x86_64-efi/grub.efi
/boot/grub/x86_64-efi/core.efi
find: ‘/boot/lost+found’: Permission denied
/boot/efi
/boot/efi/EFI/ubuntu/grubx64.efi
/boot/efi/EFI/ubuntu/shimx64.efi
/boot/efi/EFI/ubuntu/mmx64.efi
/boot/efi/EFI/BOOT/fbx64.efi
/boot/efi/EFI/BOOT/mmx64.efi
/boot/memtest86+x64.efi
/boot/memtest86+ia32.efi

That means there is something fundamentally wrong between the Lenovo P360 Tiny Workstation, which is a fairly new device, still under warranty, using a 12 Gen Intel CPU.

ftmiranda, 2025-04-17:
I also tried to download manually the cab files and try to update.... no success...

$ sudo fwupdmgr install ThinkCentre-Tiny8-ECFirmware-M4GCT18A.cab
[sudo] password for fmiranda: 
Waiting…                 [***************************************]
Successfully installed firmware
An update requires a reboot to complete. Restart now? [y|N]: n
fmiranda@ThinkStation-P360:~/Downloads$ sudo fwupdmgr install ThinkCentre-M90q-5-ME-Firmware2330.cab -vv
(pkttyagent:5471): GLib-GIO-DEBUG: 07:47:44.110: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:5469): GLib-GIO-DEBUG: 07:47:44.118: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:5469): dconf-DEBUG: 07:47:44.118: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:5469): dconf-DEBUG: 07:47:44.119: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:5469): dconf-DEBUG: 07:47:44.119: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:5469): dconf-DEBUG: 07:47:44.119: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:5469): dconf-DEBUG: 07:47:44.119: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:5469): GLib-GIO-DEBUG: 07:47:44.119: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:5469): dconf-DEBUG: 07:47:44.121: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:5469): dconf-DEBUG: 07:47:44.121: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:5469): dconf-DEBUG: 07:47:44.121: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:5469): dconf-DEBUG: 07:47:44.121: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:5469): dconf-DEBUG: 07:47:44.121: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:5469): GLib-GIO-DEBUG: 07:47:44.122: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.202: Emitting ::status-changed() [idle]
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.219: Emitting ::status-changed() [decompressing]
Decompressing…           [    \                                  ](fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.363: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [    \                                  ](fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.381: Emitting ::status-changed() [device-busy]
Waiting…                 [     \                                 ](fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.381: Emitting ::device-changed(add7bba879f963696bdc16794507abcf728f9d2a)
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.381: Emitting ::device-changed(add7bba879f963696bdc16794507abcf728f9d2a)
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.385: Emitting ::device-changed(add7bba879f963696bdc16794507abcf728f9d2a)
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.385: Emitting ::device-changed(add7bba879f963696bdc16794507abcf728f9d2a)
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.388: Emitting ::status-changed() [device-write]
Writing…                 [                                       ](fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.388: Emitting ::status-changed() [device-restart]
Restarting device…       [                                       ](fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.388: Emitting ::status-changed() [device-write]
Writing…                 [                                       ](fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.388: Emitting ::status-changed() [scheduling]
Scheduling…              [                                       ](fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.388: Emitting ::status-changed() [decompressing]
Decompressing…           [                                       ](fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.420: Emitting ::status-changed() [device-restart]
Restarting device…       [************************************** ](fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.420: Emitting ::status-changed() [device-busy]
Waiting…                 [************************************** ](fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.427: Emitting ::device-changed(add7bba879f963696bdc16794507abcf728f9d2a)
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.431: Emitting ::device-changed(add7bba879f963696bdc16794507abcf728f9d2a)
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.431: Emitting ::device-changed(add7bba879f963696bdc16794507abcf728f9d2a)
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.431: Emitting ::device-changed(add7bba879f963696bdc16794507abcf728f9d2a)
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.432: Emitting ::device-changed(add7bba879f963696bdc16794507abcf728f9d2a)
Waiting…                 [***************************************](fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.432: Emitting ::device-changed(add7bba879f963696bdc16794507abcf728f9d2a)
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.436: Emitting ::device-changed(add7bba879f963696bdc16794507abcf728f9d2a)
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.436: Emitting ::status-changed() [unknown]
(fwupdmgr:5469): Fwupd-DEBUG: 07:47:44.436: Emitting ::changed()

Successfully installed firmware
An update requires a reboot to complete. Restart now? [y|N]: n
fmiranda@ThinkStation-P360:~/Downloads$ ls *.cab -l
-rwxrwxr-x 1 fmiranda fmiranda  9587448 Apr 14 22:00 ThinkCentre-M90q-5-ME-Firmware2330.cab
-rwxrwxr-x 1 fmiranda fmiranda   295910 Mar 11 12:25 ThinkCentre-Tiny8-ECFirmware-M4GCT18A.cab
[... cut ...]
Boot000C* UEFI: PXE IPv6 Realtek USB Ethernet Controller	PciRoot(0x0)/Pci(0x14,0x0)/USB(16,0)/MAC(6c1ff71be551,0)/IPv6([::]:<->[::]:,0,0)0000424f
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 14 / 03 05 06 00 10 00 / 03 0b 25 00 6c 1f f7 1b e5 51 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0d 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 00 00 42 4f

hughsie, 2025-04-18:
> CapsuleApplyMethod:cod

According to the system firmware, the firmware wants us to use Capsule-on-Disk rather than using fwupd-efi. 

You might want to try `DisableCapsuleUpdateOnDisk=true` -- e.g. see `man fwupd.conf` -- if that works I can ping the ThinkStation team and ask if they want to change the BIOS setting or add the quirk via a DMI match.

ftmiranda, 2025-04-18:
> According to the system firmware, the firmware wants us to use Capsule-on-Disk rather than using fwupd-efi.
>
> You might want to try `DisableCapsuleUpdateOnDisk=true` -- e.g. see `man fwupd.conf` -- [...]

I added that parameter inside the fwupd.conf, rebooted then tried the process to update again, it completes but after another reboot we are back to square 1. No updates were actually applied.

ftmiranda, 2025-05-01:
After some recent updates on Ubuntu 24.04.2, new kernel, now,  6.11.0-24-generic, new version of fwupd
when I check the status of this service I see this:
$ systemctl status fwupd 
● fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: active (running) since Thu 2025-05-01 07:39:12 CDT; 6min ago
       Docs: https://fwupd.org/
   Main PID: 3242 (fwupd)
      Tasks: 6 (limit: 76437)
     Memory: 25.3M (peak: 26.8M)
        CPU: 500ms
     CGroup: /system.slice/fwupd.service
             └─3242 /usr/libexec/fwupd/fwupd

May 01 07:39:10 ThinkStation-P360 systemd[1]: Starting fwupd.service - Firmware update daemon...
May 01 07:39:12 ThinkStation-P360 fwupd[3242]: 12:39:12.558 FuEngine             failed to update history database: Error opening file>
May 01 07:39:12 ThinkStation-P360 fwupd[3242]: 12:39:12.572 FuEngine             failed to update history database: Error opening file>
May 01 07:39:12 ThinkStation-P360 fwupd[3242]: 12:39:12.586 FuEngine             failed to update history database: Error opening file>
May 01 07:39:12 ThinkStation-P360 fwupd[3242]: 12:39:12.599 FuMain               Daemon ready for requests (locale en_US.UTF-8)
May 01 07:39:12 ThinkStation-P360 systemd[1]: Started fwupd.service - Firmware update daemon.

I also tried the with or without DisableCapsuleUpdateOnDisk=true - nothing

I guess this is  broken either on Ubuntu or Lenovo for this PC model

ElCoyote27, 2025-05-03:
Similar issue on Lenovo P350 ThinkStation Tiny.

ElCoyote27, 2025-05-03:
# fwupdmgr update
Devices with no available firmware updates: 
 • SSD 970 EVO 1TB
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.64 to 1.65?                                   ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Enhancement to address security vulnerability.                             ║
║ • Fix boot order exclusion under Windows Advanced Startup.                   ║
║ • Fix Lenovo Cloud Boot failure issue.                                       ║
║ • Enhance the compatibility of NVMe SSD.                                     ║
║ • Support Class WMI Lenovo systemDeploymentBootMode.                         ║
║                                                                              ║
║ 30EF0009GE must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
Devices with the latest available firmware version:
 • UEFI dbx
An update requires a reboot to complete. Restart now? [y|N]: 

Happy to test out some stuff (as long as it does not brick my machine permanently)

ftmiranda, 2025-05-03:
@hughsie and @ElCoyote27 
Same here, let me know any tests or information from my system you might need to help debug this issue.
Thank you

ftmiranda, 2025-05-09:
@iranmarcius 
 I created an issue on lenovo's github too https://github.com/fwupd/firmware-lenovo/issues/510#issuecomment-2859954168
but no resolution yet...

iranmarcius, 2025-05-09:
For what I saw at Lenovo site, looks like there's no bios update for my machine. The link I posted before pointed to a recommendation to update the bios... :/

Dead end.

vitaly-castLabs, 2025-07-30:
Same issue with Lenovo ThinkCentre M70q-4 and Ubuntu 24.04.2 - got stuck at 0.1.38 months ago, no new updates would install:
```
fwupdmgr get-updates
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • MZVL21T0HDLU-00BL7
LENOVO 12E3CTO1WW
│
└─System Firmware:
  │   Device ID:          8a3bd64788bb0b0e8e4daa89392d48201ee98625
  │   Summary:            UEFI System Resource Table device (Updated via capsule-on-disk)
  │   Current version:    0.1.38
  │   Minimum Version:    0.1.38
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Failed
  │   Update Error:       failed to update to 0: error-invalid-format
  │   GUID:               d64ea3d9-4e0e-4cb5-80d2-8ee22bc34c80
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │   Device Requests:    • Message
  │ 
  ├─ThinkCentre M70q-4, ThinkCentre M750q System Update:
  │     New version:      0.1.42
  │     Remote ID:        lvfs
  │     Release ID:       116467
  │     Summary:          System Firmware for ThinkCentre M70q-4, ThinkCentre M750q
  │     License:          Proprietary
  │     Size:             16.9 MB
  │     Created:          2025-06-19
  │     Urgency:          Low
  │     Tested by Lenovo:
  │       Tested:         2025-06-20
  │       Distribution:   ubuntu 24.04
  │       Old version:    0.1.41
  │       Version[fwupd]: 1.9.16
  │     Vendor:           Lenovo
  │     Release Flags:    • Trusted metadata
  │                       • Is upgrade
  │                       • Tested by trusted vendor
  │     Description:      
  │     This stable release fixes the following issues:
  │     
  │     • Update DPTF Pop-up Information.
  │     • Update UEFI Diagnostic Driver to 04.39.000.
  │     • Enhancement to address security vulnerability.
  │     • Add microsoft_uefi_ca_2023 and microsoft_option_rom_uefi_ca_2023.
  │     • Fix a BIOS Update issue.
  │     Issues:           CVE-2024-38796
  │                       CVE-2025-22830
  │                       CVE-2025-22834
  │                       CVE-2024-6119
  │                       CVE-2024-5535
  │                       CVE-2023-6237
  │                       CVE-2025-20054
  │                       CVE-2025-20623
  │                       CVE-2024-45332
  │                       CVE-2024-43420
  │                       CVE-2025-20109
  │                       CVE-2025-20044
  │                       CVE-2024-9143
  │                       CVE-2023-52353
[... 63 lines cut ...]
        New version:      0.1.39
        Remote ID:        lvfs
        Release ID:       100661
        Summary:          System Firmware for ThinkCentre M70q-4, ThinkCentre M750q
        License:          Proprietary
        Size:             16.9 MB
        Created:          2024-10-17
        Urgency:          Low
        Tested by Lenovo:
          Tested:         2024-10-29
          Distribution:   rhel 9.4
          Old version:    0.1.38
          Version[fwupd]: 1.9.13
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
                          • Is upgrade
                          • Tested by trusted vendor
        Description:      
        This stable release fixes the following issues:
        
        • Update CPU microcode 0x12B for Intel 13th and 14th generation CPU instability issue.
        • (Fix) Fix some issues.
        Issue:            LEN-167071
        Checksum:         43e9b6fa4bfc1890871e977ae2a99bc80942e3ae4966a23038b409195d91f6e2
```