fwupd/fwupd #8707: Crashes showing up in Ubuntu error tracker

superm1, 2025-04-17:
In errors.ubuntu.com (similar to ABRT) there is a set of segfault errors showing up on 2.0.7 with the following signature.  

```
#0  g_str_hash (v=0x0) at ../../../glib/ghash.c:2474
        p = 0x0
        h = 5381
#1  0x000076922e65eee1 in g_hash_table_lookup_node (hash_table=0x5dbe8e8386e0, key=0x0, hash_return=<synthetic pointer>) at ../../../glib/ghash.c:421
        node_hash = <optimized out>
        hash_value = <optimized out>
        have_tombstone = 0
        step = 0
        node_index = <optimized out>
        first_tombstone = 0
        node_index = <optimized out>
        node_hash = <optimized out>
        hash_value = <optimized out>
        first_tombstone = <optimized out>
        have_tombstone = <optimized out>
        step = <optimized out>
        _g_boolean_var_14 = <optimized out>
        node_key = <optimized out>
#2  g_hash_table_insert_internal (hash_table=0x5dbe8e8386e0, key=0x0, value=0x0, keep_new_key=0) at ../../../glib/ghash.c:1575
        key_hash = <optimized out>
        node_index = <optimized out>
        key_hash = <optimized out>
        node_index = <optimized out>
        __func__ = <optimized out>
        _g_boolean_var_36 = <optimized out>
#3  g_hash_table_insert (hash_table=0x5dbe8e8386e0, key=0x0, value=0x0) at ../../../glib/ghash.c:1606
No locals.
#4  0x000076922eb961ac in fu_udev_device_read_property (self=self@entry=0x5dbe8e81b910, key=key@entry=0x76922ebb58c4 "DEVTYPE", error=error@entry=0x0) at /usr/include/glib-2.0/glib/gmem.h:245
        i = 2
        kvs = <optimized out>
        str = 0x5dbe8e5a2bc0 "DRIVER=snd_hda_codec_realtek\nMODALIAS=hdaudio:v10EC0298r00100103a01\n"
        uevent_lines = 0x5dbe8e7cea80
        priv = <optimized out>
        event = 0x0
        event_id = <optimized out>
        value = 0x0
        __func__ = "fu_udev_device_read_property"
#5  0x000076922eb96bc1 in fu_udev_device_probe (device=0x5dbe8e81b910, error=0x7ffeca844178) at ../libfwupdplugin/fu-udev-device.c:360
        self = <optimized out>
        priv = 0x5dbe8e81b620
        subsystem = 0x0
        attr_device = 0x0
        attr_vendor = 0x0
#6  0x000076922eb441be in fu_device_probe (self=self@entry=0x5dbe8e81b910, error=error@entry=0x7ffeca844178) at ../libfwupdplugin/fu-device.c:5735
        priv = <optimized out>
        device_class = 0x5dbe8e8272b0
        __func__ = "fu_device_probe"
```

superm1, 2025-04-17:
I'm wondering if this is actually a race condition with the kernel of some sort because fwupd is picking up the event very quickly.  I looked at the logs from a few of the reports.

```
(udev-worker)[10462]: controlC0: Process '/usr/sbin/alsactl -E HOME=/run/alsa -E XDG_RUNTIME_DIR=/run/alsa/runtime restore ' failed with exit code 99.
kernel: snd_soc_avs 0000:00:1f.3: Direct firmware load for intel/avs/hda-10ec0298-tplg.bin failed with error -2
kernel: snd_soc_avs 0000:00:1f.3: request topology "intel/avs/hda-10ec0298-tplg.bin" failed: -2
kernel: avs_hdaudio avs_hdaudio.0: ASoC: Parent card not yet available, widget card binding deferred
kernel: snd_soc_avs 0000:00:1f.3: Direct firmware load for intel/avs/hda-8086280b-tplg.bin failed with error -2
kernel: snd_soc_avs 0000:00:1f.3: request topology "intel/avs/hda-8086280b-tplg.bin" failed: -2
kernel: avs_hdaudio avs_hdaudio.2: ASoC: Parent card not yet available, widget card binding deferred
(udev-worker)[10464]: controlC1: Process '/usr/sbin/alsactl -E HOME=/run/alsa -E XDG_RUNTIME_DIR=/run/alsa/runtime restore ' failed with exit code 99.
(udev-worker)[10541]: controlC2: Process '/usr/sbin/alsactl -E HOME=/run/alsa -E XDG_RUNTIME_DIR=/run/alsa/runtime restore ' failed with exit code 99.
gsd-media-keys[2931]: gvc_mixer_stream_get_port: assertion 'stream->priv->ports != NULL' failed
systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
systemd[1]: fwupd.service: Failed with result 'core-dump'.
```

superm1, 2025-04-17:
Another one looks similar with a different type of audio.

```
kernel: cs42l43 sdw:0:0:01fa:4243:01: Failed to request IRQ 193 for cs42l43: -38
kernel: cs42l43 sdw:0:0:01fa:4243:01: Failed to add IRQ chip: -38
kernel: cs35l56 sdw:0:3:01fa:3556:01:1: Slave 1 state check1: UNATTACHED, status was 1
kernel: cs35l56 sdw:0:3:01fa:3556:01:0: Slave 2 state check1: UNATTACHED, status was 1
systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
systemd[1]: fwupd.service: Failed with result 'core-dump'.
```

superm1, 2025-04-17:
Another:

```
kernel: cs42l43 sdw:0:0:01fa:4243:01: Failed to read otp rev: -110
(udev-worker)[11042]: controlC1: Process '/usr/sbin/alsactl -E HOME=/run/alsa -E XDG_RUNTIME_DIR=/run/alsa/runtime restore ' failed with exit code 99.
kernel: soundwire_intel soundwire_intel.link.2: IO transfer timed out, cmd 3 device 2 addr c1 len 1
kernel: soundwire sdw-master-0-2: trf on Slave 2 failed:-110 write addr c1 count 0
systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
systemd[1]: fwupd.service: Failed with result 'core-dump'.
kernel: soundwire_intel soundwire_intel.link.2: IO transfer timed out, cmd 2 device 2 addr c0 len 1
kernel: soundwire sdw-master-0-2: trf on Slave 2 failed:-110 read addr c0 count 0
gsd-media-keys[3217]: Unable to get default sink
kernel: soundwire_intel soundwire_intel.link.2: IO transfer timed out, cmd 3 device 2 addr c0 len 1
kernel: soundwire sdw-master-0-2: trf on Slave 2 failed:-110 write addr c0 count 0
```

hughsie, 2025-04-17:
I think the `g_str_hash (v=0x0)` is the clue; somehow there must be a uevent with no key (or value). I don't see much wrong with `DRIVER=snd_hda_codec_realtek\nMODALIAS=hdaudio:v10EC0298r00100103a01\n` -- other than maybe we ought to ignore the empty last line -- but that wouldn't explain this bug at all.

superm1, 2025-04-17:
I think we either need a check here after the split on the number of items or a line before that which makes sure there really is a `=` in the line.

```c
			g_autofree gchar **kvs = g_strsplit(uevent_lines[i], "=", 2);
			g_hash_table_insert(priv->properties,
					    g_steal_pointer(&kvs[0]),
					    g_steal_pointer(&kvs[1]));
```

hughsie, 2025-04-17:
Yup, agreed. I wrote some test code and managed to get:

	# FuUdevDevice-DEBUG: key=DRIVER, value=snd_hda_codec_realtek
	# FuUdevDevice-DEBUG: key=MODALIAS, value=hdaudio:v10EC0298r00100103a01
	# FuUdevDevice-DEBUG: key=(null), value=(null)
	Segmentation fault (core dumped)

superm1, 2025-04-17:
> Yup, agreed. I wrote some test code and managed to get:
> 
> ```
> # FuUdevDevice-DEBUG: key=DRIVER, value=snd_hda_codec_realtek
> # FuUdevDevice-DEBUG: key=MODALIAS, value=hdaudio:v10EC0298r00100103a01
> # FuUdevDevice-DEBUG: key=(null), value=(null)
> Segmentation fault (core dumped)
> ```

Ah, great!

closed 2025-04-17 by superm1