fwupd/fwupd #8713: Aorus B650E PCR0 reconstruction went from Valid to Invalid

edwintorok, 2025-04-19:
**Describe the bug**
`sudo fwupdmgr security` shows
```
2025-04-15 18:11:11: TPM PCR0 reconstruction changed: Valid → Invalid
```

The date and time seems to coincide with the Fedora 41 -> 42 upgrade (although it is off by 1h, perhaps the timezone is UTC instead of local?)

```
$ sudo journalctl -b -15 |head -n1
Apr 15 19:10:28 fedora kernel: Linux version 6.14.2-300.fc42.x86_64 (mockbuild@1e0fabd13e6844ef858c949ecea55e64) (gcc (GCC) 15.0.1 20250329 (Red Hat 15.0.1-0), GNU ld version 2.44-3.fc42) #1 SMP PREEMPT_DYNAMIC Thu Apr 10 21:50:55 UTC 2025
$ sudo journalctl -b -16 |head -n1
Apr 15 18:54:48 fedora kernel: Linux version 6.13.10-200.fc41.x86_64 (mockbuild@317985c0afde4e3ab48d4cc864aa7430) (gcc (GCC) 14.2.1 20250110 (Red Hat 14.2.1-7), GNU ld version 2.43.1-5.fc41) #1 SMP PREEMPT_DYNAMIC Mon Apr  7 19:01:38 UTC 2025
```

According to https://packages.fedoraproject.org/pkgs/fwupd/fwupd/ this means that fwupd went from 1.9.29-1 to 2.0.8-2 (and of course grub/kernel changed too).

I tried rebuilding the 1.9.29-1 in a fedora41 distrobox, but that one calculates the same PCR0 as the latest (and doesn't match tpm2_pcrread), so perhaps the bug is in some other component that changed between Fedora41 and Fedora42.

**Steps to Reproduce**
```
sudo fwupdmgr security
```

<details>

```
Host Security ID: HSI:1 (v2.0.8)

HSI-1
✔ SMM locked down:               Locked
✔ BIOS firmware updates:         Enabled
✔ Fused platform:                Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI secure boot:              Enabled

HSI-2
✔ SPI write protection:          Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✘ TPM PCR0 reconstruction:       Invalid

HSI-3
✔ CET Platform:                  Supported
✘ SPI replay protection:         Not supported
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✔ SMAP:                          Enabled
✘ Processor rollback protection: Disabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ CET OS Support:                Supported
✔ fwupd plug-ins:                Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted

The TPM PCR0 differs from reconstruction.
 » https://fwupd.github.io/hsi.html#pcr0-tpm-event-log-reconstruction

Host Security Events
  2025-04-19 16:07:19:  ✔ CET OS Support changed: Not supported → Supported
  2025-04-19 15:56:59:  ✘ CET OS Support changed: Supported → Not supported
[... 539 lines cut ...]
VERSION_CODENAME=""
PLATFORM_ID="platform:f42"
PRETTY_NAME="Fedora Linux 42.20250417.0 (Kinoite)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:42"
DEFAULT_HOSTNAME="fedora"
HOME_URL="https://kinoite.fedoraproject.org"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora-kinoite/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://pagure.io/fedora-kde/SIG/issues"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=42
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=42
SUPPORT_END=2026-05-13
VARIANT="Kinoite"
VARIANT_ID=kinoite
OSTREE_VERSION='42.20250417.0'
```
<details>

- Have you tried rebooting? Yes
 
- Is this a regression? Yes, appears to coincide with Fedora 41 to Fedora 42 upgrade

hughsie, 2025-05-03:
If you build fwupd, what does `sudo ./plugins/tpm/fwupdtpmevlog` say please?

edwintorok, 2025-05-03:
> If you build fwupd, what does `sudo ./plugins/tpm/fwupdtpmevlog` say please?

I've included that in the 'details' tag above, see "I've built fwupdtpmevlog from latest git and here is the output:", click on it to expand the full output.

edwintorok, 2025-05-05:
Tried changing a BIOS setting, and setting it to 'ASP fTPM level 2', instead of `Pluton fTPM level 3' made 'TPM PCR0 reconstruction is valid` message appear (well either that, or the fact that changing these settings forces a TPM reset).

There are far fewer events in the tpm2 log now for PCR0, so perhaps something is buggy either in the firmware or in the Linux kernel / tpm2 tools when interpreting the other event log:
```
sudo venv/build/plugins/tpm/fwupdtpmevlog -p 0
PCR:                    BIOS (0)
Type:                   0x8
Description:            ev-s-crtm-version
ChecksumSha256:         d681837b3b8f5fd685bfad042ca78e6884e10e1c8f272dbccd2fd10be738a7fa
BlobStr:                BSsQp8fZZUGBQCrd6Ur2PA==

PCR:                    BIOS (0)
Type:                   0x8000000a
ChecksumSha256:         70d0f9c08da67b218604edf2e7a1b3159aac7dee75357b4d40f92e5c2f33e1f2
BlobStr:                D0ZWX01BSU5fV1JBUFBFUgAAAAEAAAAAAPC6AAAAAAA=

PCR:                    BIOS (0)
Type:                   0x4
Description:            ev-separator
ChecksumSha256:         df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119
BlobStr:                AAAAAA==

Reconstructed PCRs:
  PCR 0:                SHA256(1f0aeb882c2ccf1469c15020422795b59b3b6ed28e773484f86973a65a597638)
```

Anyway it works now, just leaving this comment here, so that others can find it via Google, in my case turning off Pluton avoids whatever the bug is.