fwupd/fwupd #8714: ThinkPad T14s Gen 3 AMD refuses to update

Jackaed, 2025-04-19:
**Describe the bug**
When updating system firmware on a thinkpad t14s gen 3 amd from 0.1.44 to 0.1.45, the update fails to apply.

**Steps to Reproduce**
Run `fwupdmgr upgrade` on a ThinkPad T14s Gen 3 AMD with secure boot disabled, reboot, system briefly flashes into fwupd-efi before immediately exiting and booting into system.

**Expected behavior**
Update is correctly applied

**fwupd version information**
```
compile   info.libusb                   1.0.28
compile   com.hughsie.libxmlb           0.3.22
compile   com.hughsie.libjcat           0.2.3
compile   org.freedesktop.fwupd         2.0.8
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.12.23
runtime   org.freedesktop.fwupd         2.0.8
```

```shell
fwupdtool get-report-metadata
fwupdtool hwids
```

```
BatteryLevel:           101
BatteryThreshold:       25
BootTime:               1744658117
CompileVersion(com.hughsie.libjcat): 0.2.3
CompileVersion(com.hughsie.libxmlb): 0.3.22
CompileVersion(info.libusb): 1.0.28
CompileVersion(org.freedesktop.fwupd): 2.0.8
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 7 PRO 6850U with Radeon Graphics
DisplayState:           connected
DistroId:               nixos
DistroName:             NixOS
DistroPrettyName:       NixOS 25.05 (Warbler)
DistroVariant:
DistroVersion:          25.05
FwupdSupported:         True
HostBaseboardManufacturer: LENOVO
HostBaseboardProduct:   21CQCTO1WW
HostBiosMajorRelease:   01
HostBiosMinorRelease:   2c
HostBiosVendor:         LENOVO
HostBiosVersion:        R22ET74W (1.44 )
HostEnclosureKind:      a
HostFamily:             ThinkPad T14s Gen 3
HostFirmwareMajorRelease: 01
HostFirmwareMinorRelease: 20
HostProduct:            21CQCTO1WW
HostSku:                LENOVO_MT_21CQ_BU_Think_FM_ThinkPad T14s Gen 3
HostVendor:             LENOVO
KernelCmdline:          workqueue.power_efficient=true vt.default_grn=0x20,0x49,0xbb,0xbd,0xa5,0x86,0xc0,0xc4,0x5c,0x49,0xbb,0xbd,0xa5,0x86,0xc0,0xdb init=/nix/store/3phjs5gr8l3nbwq9svhhga641l9ip53d-nixos-system-nixos-25.05.20250412.2631b0b/init vt.default_red=0x1d,0xfb,0xb8,0xfa,0x83,0xd3,0x8e,0xd5,0x66,0xfb,0xb8,0xfa,0x83,0xd3,0x8e,0xeb vt.default_blu=0x21,0x34,0x26,0x2f,0x98,0x9b,0x7c,0xa1,0x54,0x34,0x26,0x2f,0x98,0x9b,0x7c,0xb2
KernelName:             Linux
KernelVersion:          6.12.23
LidState:               unknown
PlatformArchitecture:   x86_64
PowerState:             unknown
RuntimeVersion(com.hughsie.libjcat): 0.2.3
RuntimeVersion(com.hughsie.libxmlb): 0.3.22
RuntimeVersion(org.freedesktop.fwupd): 2.0.8
RuntimeVersion(org.kernel): 6.12.23
DeviceId:               87792b07537e5af953ad48ab06eaa0ddced64f2d
  pre:
      CapsuleApplyMethod: nvram
[... 483 lines cut ...]
└── loader
    ├── entries
    │   ├── nixos-generation-26.conf
    │   └── nixos-generation-27.conf
    ├── entries.srel
    ├── loader.conf
    └── random-seed
```

**Additional questions**

- Operating system and version:
NixOS Unstable latest
- Have you tried rebooting?
Yes
- Is this a regression?
Yes
- Are you using an NVMe disk?
Yes
- Is secure boot enabled?
No (yes at the time of writing the issue (i.e. logs may show secure boot) but no when I ran the update)
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
No

Prior Issue created for lenovo-specific repo (I figured it might be more relevant here) https://github.com/fwupd/firmware-lenovo/issues/505

Jackaed, 2025-04-20:
And now on `fwupd-git` neither the update nor the device appear at all. Any ideas @superm1 ? I'm very confused. Seems like Lenovo have been doing some weird shit, looking at the recent issues. Is this a problem on their end or on the fwupd end? Let me know what logs you need.

Jackaed, 2025-04-20:
On UEFI-git I get "failed to add device /sys/devices/pci0000:00/0000:00:08.1/0000:33:00.0: failed to setup: cannot parse datastream" if that helps.

hughsie, 2025-04-22:
> flashes into fwupd-efi before immediately exiting

If you record with your phone camera, do you see any messages even for a split second? My gut instinct would be something like the gnu-efi version or something like that. Would it be possible to test with a Fedora install and rule out software vs hardware issue?

hughsie, 2025-06-06:
@Jackaed you need to either downgrade gnu-efi to 3.0.x or upgrade fwupd-efi to 1.7 (which sets GNU_EFI_3_0_COMPAT) -- this is not a fwupd bug, but is a result of the ABI behaviour changing in https://github.com/ncroxon/gnu-efi/blob/master/docs/README.ABI.md

hughsie, 2025-06-07:
Ohh the gnu-efi ABI change is responsible for the failure to update, the weird ESRT entry is something else entirely.