fwupd/fwupd #8740: Dell XPS 15 9530 BIOS updates not detected by fwupd

philipmagnusson, 2025-04-29:
## Describe the bug
I have a Dell XPS 15 9530 laptop, and fwupd is not detecting available BIOS updates that are shown on Dell's support website. The Dell support site shows a critical BIOS update (version 1.21.0) released on April 11, 2025, but fwupd reports "No supported devices" when running `fwupdmgr get-releases`.

## Hardware Information
- Device: Dell XPS 15 9530
- Current BIOS version (according to BIOS itself): 1.20.0
- Current BIOS version (according to fwupd): 70656
- Latest BIOS version on Dell support site: 1.21.0 (released April 2025, marked as critical)

## Steps to Reproduce
1. Run `sudo fwupdmgr refresh --force`
2. Run `sudo fwupdmgr get-releases`
   - Result: "No supported devices"
3. Run `sudo fwupdmgr get-updates`
   - Result: "No updatable devices"

## Expected behavior
fwupd should detect and offer the 1.21.0 BIOS update that's available on Dell's support site, especially since it's marked as critical.

## fwupd version information
```
sudo fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.8.0-58-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

## Important Version Compatibility Issue Found
When running `fwupdtool get-updates`, I discovered the following error:
```
failed to get releases for UEFI dbx: No releases found: Not compatible with org.freedesktop.fwupd version 1.7.9, requires >= 1.9.1, Not compatible with org.freedesktop.fwupd version 1.7.9, requires >= 1.9.28
```
This indicates that my fwupd version (1.7.9) is too old to support the firmware updates for my device, which require fwupd 1.9.28 or newer.

Please note how you installed it: `apt` (standard Linux Mint repositories)

<details>
<summary>fwupd device information</summary>

Output of the Dell XPS 15 9530 System Firmware:
```
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    70656
│ │   Minimum Version:    70656
│ │   Vendor:             Dell Inc. (DMI:Dell Inc.)
│ │   Update State:       Success
│ │   GUIDs:              b902afc2-93ea-46b3-bb43-d0e9494bf1c4
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Full disk encryption secrets may be invalidated when updating
```
</details>

## Additional information about my system
```
$ inxi -Fxz
System:
  Kernel: 6.8.0-58-generic arch: x86_64 bits: 64 compiler: N/A
    Desktop: Cinnamon v: 6.0.4 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
Machine:
  Type: Laptop System: Dell product: XPS 15 9530 v: N/A
    serial: <superuser required>
  Mobo: Dell model: 01WV13 v: A00 serial: <superuser required> UEFI: Dell
    v: 1.20.0 date: 02/11/2025
```

## Additional questions
- Operating system and version: Linux Mint 21.3 Victoria (based on Ubuntu 22.04)
- Have you tried rebooting? Yes
- Is this a regression? No, first time trying to use fwupd for firmware updates
- Are you using an NVMe disk? Yes, PC801 NVMe SK hynix 1TB
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No, this is a Dell system

hughsie, 2025-04-29:
1.7.x is EOL, can you try with something newer like1.9.x or 2.0.x please:  https://github.com/fwupd/fwupd/blob/main/SECURITY.md

philipmagnusson, 2025-04-29:
I understand that 1.7.x is EOL from the fwupd project's perspective, but this creates a challenging situation for users of Linux Mint 21.3 (released in January 2024 and supported until 2027) which ships with fwupd 1.7.9 in its official repositories.
This seems to create a gap where:

Users on supported LTS distributions can't get firmware updates
The only options are to either:

Use unofficial PPAs (potentially compromising system stability)
Boot to Windows for firmware updates (defeating the purpose of fwupd)
Upgrade to a newer distribution (not always feasible for many users)

Is there any consideration for backporting critical firmware update capability to the 1.7.x branch for users on current LTS distributions? Or perhaps creating a standalone update tool that doesn't require the newest fwupd versions?
I tried looking for a PPA to upgrade fwupd (the standard ppa:fwupd-team/fwupd-stable doesn't seem to exist anymore), but I'm not sure which PPA would be recommended for Ubuntu 22.04/Linux Mint 21.3 users. Could you please advise on the best way to get a newer fwupd version (1.9.x or 2.0.x) on my current distribution?
I wanted to highlight this issue as it affects many users on popular LTS distributions that are still well within their support window. While I understand the security and feature benefits of newer versions, there's currently a gap for users on supported LTS distributions who want to keep their firmware updated

superm1, 2025-04-29:
If you're going to track Ubuntu as a base you need to convince them to fix bugs or carry fixes yourself.

Here is the bug tracking the sru.

https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/2056548

I did the packaging work already and it was totally ignored.

closed 2025-04-29 by superm1

philipmagnusson, 2025-04-29:
Thanks for the link to the bug. Pretty frustrating that your SRU work was ignored. I'll keep an eye on that bug and maybe add my case there too.
For now I'll just use Windows for firmware updates. Thanks for your help and for the work on fwupd.

philipmagnusson, 2025-04-29:
@superm1 
Thanks for pointing me to that bug report! I found the PPA mentioned there, added it and updated to fwupd 1.9.16, and now I can see updates.
It found a UEFI dbx update (from 270 to 371) but no BIOS update yet. I'm guessing the 1.21.0 BIOS from Dell's site just isn't in LVFS yet.
At least now I'll be able to get firmware updates directly in Linux going forward. Thanks again for your help.