fwupd/fwupd #8752: Quectel EM061KGL: "failed to detach: failed to detach using modem_manager: suitable port not found"

suiaji, 2025-05-02:
**Describe the question**
Trying to upgrade the FW of a Quecktel EM061K-GL modem in a lenovo T14 laptop.

```
06:01:07.733 FuCabinet            processing file: firmware.metainfo.xml
06:01:07.734 FuCabinet            processing release: EM061KGLAAR01A02M2G_01.013.01.013
06:01:08.443 FuCabinet            verified indirect payload firmware.zip: 3
06:01:08.453 FuEngine             composite update 1: 02861494e3c701c896953cd442bd2f1ad867e1eb EM061KGLAAR01A02M2G_01.008.01.008->EM061KGLAAR01A02M2G_01.013.01.013 (n/a, order:0: priority:0)
06:01:08.453 FuEngine             install phase now composite-prepare
06:01:08.454 FuDeviceList         no replug or re-enumerate required
Waiting…                 [*******************                    ]06:01:08.471 FuEngine             install phase now prepare
06:01:08.471 FuDeviceList         no replug or re-enumerate required
06:01:08.475 FuEngine             prepare -> FuMmDevice:
  DeviceId:             02861494e3c701c896953cd442bd2f1ad867e1eb
  Name:                 EM061K-GL
  Status:               device-busy
  Guid:                 1d91bd8e-206e-5cad-aaeb-5b90df522f5f ← USB\VID_2C7C&PID_6008&REV_0504&CARRIER_GERMANY-VOLTE-VODAFONE&NAME_EM061K-GL
  Guid:                 6f3c226b-f146-50d9-addd-09699af95b7d ← USB\VID_2C7C&PID_6008&REV_0504&CARRIER_GERMANY-VOLTE-VODAFONE
  Guid:                 e307a004-f1d0-5e6b-9176-b38687fad115 ← USB\VID_2C7C&PID_6008&REV_0504
  Guid:                 2bdac371-5939-5239-9dfa-eb4d4ff12bc8 ← USB\VID_2C7C&PID_6008
  Summary:              Mobile broadband device
  Plugin:               modem_manager
  Protocol:             com.qualcomm.firehose
  Flags:                require-ac|supported|updatable-hidden|unsigned-payload
  Problems:             update-in-progress
  Vendor:               Quectel
  VendorId:             USB:0x2C7C
  Version:              EM061KGLAAR01A02M2G_01.008.01.008
  VersionFormat:        plain
  Icon:                 modem
  Created:              2025-05-02
  Modified:             2025-05-02
  UpdateError:          An update is in progress
  InstanceId[vi,qu]:    1d91bd8e-206e-5cad-aaeb-5b90df522f5f ← USB\VID_2C7C&PID_6008&REV_0504&CARRIER_GERMANY-VOLTE-VODAFONE&NAME_EM061K-GL
  InstanceId[vi,qu]:    6f3c226b-f146-50d9-addd-09699af95b7d ← USB\VID_2C7C&PID_6008&REV_0504&CARRIER_GERMANY-VOLTE-VODAFONE
  InstanceId[vi,qu]:    e307a004-f1d0-5e6b-9176-b38687fad115 ← USB\VID_2C7C&PID_6008&REV_0504
  InstanceId[vi,qu]:    2bdac371-5939-5239-9dfa-eb4d4ff12bc8 ← USB\VID_2C7C&PID_6008
  InstanceId[qu]:       22ae45db-f68e-5c55-9c02-4557dca238ec ← USB\VID_2C7C
  PhysicalId:           /sys/devices/pci0000:00/0000:00:14.0/usb3/3-8
  AcquiesceDelay:       50
  Order:                0
  PrivateFlags:         use-runtime-version,replug-match-guid,add-instance-id-rev,update-pending,registered
  Inhibit:              [update-in-progress] An update is in progress
  MbimPort:             /dev/cdc-wdm0

06:01:08.475 FuDeviceList         no replug or re-enumerate required
Writing…                 [*******************                    ]06:01:08.475 FuDeviceList         no replug or re-enumerate required
Decompressing…           [*******************                    ]06:01:08.475 FuDeviceList         no replug or re-enumerate required
06:01:08.475 FuDeviceList         no replug or re-enumerate required
Restarting device…       [*******************                    ]06:01:08.475 FuEngine             install phase now detach
06:01:08.475 FuDeviceList         no replug or re-enumerate required
06:01:08.475 FuEngine             detach -> FuMmDevice:
  DeviceId:             02861494e3c701c896953cd442bd2f1ad867e1eb
  Name:                 EM061K-GL
  Status:               device-restart
  Percentage:           1
  Guid:                 1d91bd8e-206e-5cad-aaeb-5b90df522f5f ← USB\VID_2C7C&PID_6008&REV_0504&CARRIER_GERMANY-VOLTE-VODAFONE&NAME_EM061K-GL
  Guid:                 6f3c226b-f146-50d9-addd-09699af95b7d ← USB\VID_2C7C&PID_6008&REV_0504&CARRIER_GERMANY-VOLTE-VODAFONE
  Guid:                 e307a004-f1d0-5e6b-9176-b38687fad115 ← USB\VID_2C7C&PID_6008&REV_0504
  Guid:                 2bdac371-5939-5239-9dfa-eb4d4ff12bc8 ← USB\VID_2C7C&PID_6008
  Summary:              Mobile broadband device
  Plugin:               modem_manager
  Protocol:             com.qualcomm.firehose
  Flags:                require-ac|supported|updatable-hidden|unsigned-payload
  Problems:             update-in-progress
  Vendor:               Quectel
  VendorId:             USB:0x2C7C
  Version:              EM061KGLAAR01A02M2G_01.008.01.008
  VersionFormat:        plain
  Icon:                 modem
[... 402 lines cut ...]
      0x00000001 82012974564c5f378239e766825b4895343f6c464d78d15333ff4ca9c9fb459e [QgBvAG8AdAAgAEcAdQBhAHIAZAAgAE0AZQBhAHMAdQByAGUAZAAgAEkAQgBCAAAAAAA=]
      0x00000001 6723277c305f4f6506ac13e9ff819601193519f2305f8ce797c26f27c3ed348f [QgBJAE8AUwAgAE0AZQBhAHMAdQByAGUAZAAgAEIAbwBvAHQAIABHAHUAYQByAGQAIABQAG8AbABpAGMAeQAAAAAA]
      0x80000008 17e59e6af8b54f4329ff1443df3d67c9ef93f1c298da9f01cf3903afacc6e199 [AADA/wAAAAAA6AcAAAAAAA==]
      0x80000008 3424a975e2f64b0241f4800aead54fb73b262c40671772d533f8a86cb3f1ade8 [AAC6/wAAAABoEgEAAAAAAA==]
      0x80000008 1c9d38e201023b951eed880dfa7d9c581594ec4d14eac8aa3ea2b0e6e0d6f944 [AACV/wAAAAAAvwkAAAAAAA==]
      0x80000008 69b4ffdaf0e1681b5ce4f4793a6f4ca0dfe410157261e7e1acf79f828bc3e8a0 [AACJ/wAAAABgOAMAAAAAAA==]
      0x80000008 8a7825f01ab3b7f133cb92789738ea3fc5911059b0bbcfa7c287374f3779c41e [AAAb/wAAAAAIq1sAAAAAAA==]
      0x80000008 7ed220ec98c6329919bdb91fad9fec07540bd578d28b0dbaa86c832844ce4375 [AAB9/wAAAAB4wQgAAAAAAA==]
      0x80000008 0141a0cea13ff84739dd3b0f617894a2157c9ea24527d6ad67eb7fc895156688 [AADI/wAAAAAA4AoAAAAAAA==]
      0x00000001 d07b290b36b467f2ad300b907cf91f6a7ce2dcd94c3898ca90ee7446b9c759ee [QUNQSSBEQVRB]
      0x00000001 608072d8953921f15718897cbb0a47623e0c29fa7286d20beb0d733756acb643 [QUNQSSBEQVRB]
      0x00000011 260eec37b117e76a7f39b33f55ea26775d62684e450efe22c52e6e1cf5ce1af2 [SW50ZWxDU3hFRXZlbnQwMQAAAAAEAAAAAAMAADAAAAAPosJsSvJlNrSoViZ0PXBki/pHATiPNEfoKVI615jTyFlpNe1A8DnZujKpFu9InkICAwAAFgAAABIAAAAPANMJAQAAAAIAAAACAAAAAQACLQAAFgAAAAEAAAAAAAEAAQAAAAEAAAAAAAAAAQABAgAAMAAAAEOESoEm+jSSP6TW7ai+/rUIMtXjXZnCJuzGlrnp5gy3btsjy5t6RaomA78r5JNGQQICAAAWAAAACAcPAAAAJgQAAAAAAAAAAAAAAAABAAEOAAAwAAAAHi4oEPIxqbclIoOaRdQ7S2svPKzkSqgr/hyKcha2aiqUUcx3v1lyQo7F7CGIWAJeAg4AABYAAAAIBw8AAAANBAAAAAAAAAAAAAAAAAEAAwQAAAEAAAABAwAAAAQAAAA3AAAAAQYAADAAAABRVOaQhG60Uq+bcWcrCXuNehSux3+PfUe+72jyTkERvrWKqBRn0MdILKSflKFOtzkCBgAAFgAAADAAHAAAAAAAAAAAAAAAAAAAAAAAAQABBwAAMAAAAOVnCCxoAllpJzyux4NIqNJFJqRL0jggND5OUu1p/r3U/nBejGFP3zsC0xuB6nLbuQIHAAAWAAAAEgABAAAAZxsAAAAAAAAAAAAAAAABAAENAAAwAAAACa4eFWv/FdLTF+biHg8wTwQ51TuM2iA4vq0Q9q4LXvbVPd3C3EJa/8uv8611Lw4nAg0AABYAAAASAAEAAQCNGwAAAAAAAAAAAAAAAAEAAQgAADAAAAAQtxthWqxaepo30Rs8MKGMdJ4OejwzJ1uykjvtD/SC78tPNwba8CE9+fpy/Jv+ybsCCAAAFgAAABIAAQAPAHkFAAAAAAAAAAAAAAAAAQABJAAAMAAAAKQRDk55c4izWSj6DnNg6Q6oY4ibGl96MbK71BGz92O+SiRwbeEFfakvIYS/HneMzAIkAAAWAAAAFAAoAMsFAAAAAAAAAAAAAAAAAAABAAHAAAAwAAAACAQulsaElrmt9yYQyD9R3mlBrD+w/4lsHO8tdAY+w3aDSnLz9l5+zLA6KAHF7WJSAsAAABYAAADoAwEACgBeCAEAAAABAAAAAAAAAAEAAcEAADAAAAAU+IHZ/rpJx2ewb31AKm7RCkcCCjLr8H1MAuCaxeE19nedhPkGpmIXjLWVsrAUE74CwQAAFgAAAAKBfoAAAFBQAQAAAAEAAAAAAAAAAQABwgAAMAAAAFado9DsRgNDFFBhleobc0Xco6cfpP+qWG3M1SO9F6mov8EW9pVkNeS3XfyA3r6iQgLCAAAWAAAAAgF+gAAAUFABAAAAAQAAAAAAAAABAANkAAAEAAAAARAAAAMDAAABAAAAAA==]
      0x00000011 2046f8405292c0a34c3e2ca758b338643a97a51a1713991e6acbad36ef7266c1 [SW50ZWxDU3hFSW5mb0V2ZW50AAABAAAAhoBwfgEAAAA=]
      0x00000004 df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119 [AAAAAA==]
      PCR0: 3347bcfc1b79a6bc9691652b4b223da1dd98e32b19c9baedce24473db5ec7a4e
      TpmFamily:        2.0
      BootMgrDesc:      legacy
      EfivarsNvramUsed: 80261
      SecureBoot:       Disabled
      UEFIUXCapsule:    Disabled
```

Also tried non hardened kernel, no difference.

any ideas on how to proceed here? thanks a lot!

Queztion, 2025-05-16:
Hello @suiaji , I am curious as I encountered similar "suitable port not found" issue in Quectel RM520N-GL.

Did you try with fwupdmgr 1.9.30 before going for fwupd  2.0.8?

I've noticed this after downgrading from 2.0.9 to 1.9.30.

In 1.9.30, **fwupd** and **fwupd-efi** both displays 1.9.30 as its version. And that is when I first encountered the suitable port not found issue.

Note, there is an ongoing issue with RM520N-GL and EM061K and EM160R modems for fwupd 2.0.+
https://github.com/fwupd/fwupd/issues/8382

suiaji, 2025-05-16:
Thanks for you reply @Queztion .

> Did you try with fwupdmgr 1.9.30 before going for fwupd 2.0.8?

nope, just using whatever is current in the Arch Linux repos.

meanwhile Arch updated to `compile   org.freedesktop.fwupd         2.0.9`
and the card is not even showing up anymore (`sudo fwupdmgr get-devices` -> no quectel to be seen; modemmanager still "1.24.0" ) while

```
$ sudo lsusb | grep Quectel
Bus 003 Device 003: ID 2c7c:6008 Quectel Wireless Solutions Co., Ltd. EM061K-GL
```

the device is still there  and perfectly working. 🤷

hughsie, 2025-05-16:
> and the card is not even showing up anymore (sudo fwupdmgr get-devices -> no quectel to be seen; modemmanager still "1.24.0" ) while

On 2.0.9 what does `sudo fwupdtool get-devices --plugins modem_manager -vv` say? I wonder if you've also got a very new ModemManager that needs https://github.com/fwupd/fwupd/commit/56b643cb15da54b656a1013d7824b9895aa7e42f too.

suiaji, 2025-05-16:
modemmanager 1.24.0-1 (1.24 was released in april)   https://archlinux.org/packages/extra/x86_64/modemmanager/

> On 2.0.9 what does sudo fwupdtool get-devices --plugins modem_manager -vv say?

[log.txt](https://github.com/user-attachments/files/20239779/log.txt)

hughsie, 2025-05-16:
So really, this:

    07:16:02.083 FuPluginMm           ignoring, and trying legacy fallback: no explicit GType for USB\VID_2C7C&PID_6008&REV_0504&CARRIER_GERMANY-VOLTE-VODAFONE&NAME_EM061K-GL
, USB\VID_2C7C&PID_6008&REV_0504&CARRIER_GERMANY-VOLTE-VODAFONE, USB\VID_2C7C&PID_6008&REV_0504, USB\VID_2C7C&PID_6008, USB\VID_2C7C
    07:16:02.083 FuPluginMm           found port cdc-wdm0: mbim
    07:16:02.083 FuPluginMm           found port wwp0s20f0u8: net
    07:16:02.083 FuPluginMm           ignoring: update method firehose, sahara not supported

@suiaji if you add something like this to the modem-manager.quirk file (you can also just create a /var/lib/fwupd/quirks.d/test.quirk file with the contents for text) -- does the device get detected correctly?

    # Quecktel EM061K-GL
    [USB\VID_2C7C&PID_6008]
    GType = FuMmFirehoseDevice

I'm a bit confused because we do map `MM_MODEM_FIRMWARE_UPDATE_METHOD_FIREHOSE | MM_MODEM_FIRMWARE_UPDATE_METHOD_SAHARA` to `FuMmFirehoseDevice` -- but we expect an `MM_MODEM_PORT_TYPE_AT` and it looks like we only found MBIM. i.e. ModemManager is telling us it's simultaneously a firehose+sahara and MBIM. In which case, the above won't work, and we need to do something like:

	[USB\VID_2C7C&PID_6008]
	GType = FuMmMhiQcdmDevice
	ModemManagerFirehoseProgFile = prog_firehose_sdx24.mbn

...assuming the ModemManagerFirehoseProgFile is the same name as the others -- you can check the firmware archive and look for the `prog_firehose_*` filename. If that works then we should map `MM_MODEM_FIRMWARE_UPDATE_METHOD_FIREHOSE | MM_MODEM_FIRMWARE_UPDATE_METHOD_SAHARA` (with a MM_MODEM_PORT_TYPE_MBIM) to `FuMmFirehoseDevice` to avoid needing the quirk -- although the quirk is probably a good idea to avoid using the heuristic anyway.

suiaji, 2025-05-16:
thanks a lot for your suggestions @hughsie !

the quirks file makes the modem even disappear from `fwupdtool get-devices --plugins modem_manager -vv`:

without quirks:

```
$ sudo fwupdtool get-devices --plugins modem_manager -vv 2>&1 | grep found
Loading…                 [ -                                     ]11:41:13.351 FuEngine             ignoring: no devices were found for emulation tag
11:41:13.368 FuPluginMm           found port cdc-wdm0: mbim
11:41:13.368 FuPluginMm           found port wwp0s20f0u8: net
```
with 
```
# Quecktel EM061K-GL
[USB\VID_2C7C&PID_6008]
GType = FuMmFirehoseDevice
```

```
$ sudo fwupdtool get-devices --plugins modem_manager -vv 2>&1 | grep found
Loading…                 [ -                                     ]11:41:26.909 FuEngine             ignoring: no devices were found for emulation tag
```

with
```
[USB\VID_2C7C&PID_6008]
GType = FuMmMhiQcdmDevice
ModemManagerFirehoseProgFile = prog_firehose_sdx24.mbn
```

```
$ sudo fwupdtool get-devices --plugins modem_manager -vv 2>&1 | grep found
Loading…                 [ -                                     ]11:41:32.576 FuEngine             ignoring: no devices were found for emulation tag
```

> you can check the firmware archive and look for the prog_firehose_* filename.
sorry but no idea where to look!

`$ sudo find / -iname '*.mbn'` -> empty

hughsie, 2025-05-16:
Gah, I misread. We want `FuMmMbimDevice` for `Port[mbim]:           /dev/cdc-wdm0` -- apologies. Can you try with:

    # Quecktel EM061K-GL
    [USB\VID_2C7C&PID_6008]
    GType = FuMmMbimDevice

suiaji, 2025-05-19:
Hey @hughsie,

it works:

```
$ sudo fwupdtool get-devices --plugins modem_manager
Loading…                 [************************************** ]
LENOVO 21ML0098GE
│
└─EM061K-GL:
      Device ID:          02861494e3c701c896953cd442bd2f1ad867e1eb
      Summary:            Mobile broadband device
      Current version:    EM061KGLAAR01A02M2G_01.008.01.008
      Vendor:             Quectel (USB:0x2C7C)
      GUIDs:              1d91bd8e-206e-5cad-aaeb-5b90df522f5f ← USB\VID_2C7C&PID_6008&REV_0504&CARRIER_GERMANY-VOLTE-VODAFONE&NAME_EM061K-GL
                          6f3c226b-f146-50d9-addd-09699af95b7d ← USB\VID_2C7C&PID_6008&REV_0504&CARRIER_GERMANY-VOLTE-VODAFONE
                          e307a004-f1d0-5e6b-9176-b38687fad115 ← USB\VID_2C7C&PID_6008&REV_0504
                          2bdac371-5939-5239-9dfa-eb4d4ff12bc8 ← USB\VID_2C7C&PID_6008
      Device Flags:       • Updatable
                          • System requires external power source
                          • Unsigned Payload
                          • Can tag for emulation
```

although it resides in `Loading…                 [***********************************    ]` state for ~ 10s.

same delay with `sudo fwupdtool update` but the modem appears again:
```
Devices with no available firmware updates:
 • EM061K-GL
```

The device currently runs `Current version:    EM061KGLAAR01A02M2G_01.008.01.008`, 
initially (see 1st post) I wanted to update to `processing release: EM061KGLAAR01A02M2G_01.013.01.013`  but updating is not shown as an option anymore. Any ideas? Thanks!

Queztion, 2025-05-28:
Hello @hughsie , @suiaji ,

This is just to give information on my recent tests as it might help with this issue:

**EM061K** 
**fwupdmgr 2.0.10 and 2.0.9**
Modified quirk file from project folder and compiled:
fwupd/plugins/modem-manager/modem-manager.quirk

**Result:** 
fwupdmgr get-devices detects the modem
fwupdmgr get-releases does not detect the modem
fwupdmgr update has no update for modem

**fwupdmgr 2.0.8**
Even if quirk file is not modified, modem is detected.
But fails to update.

When we added var/lib/fwupd/quirks.d/test.quirk
```
# Quecktel EM061K-GL
[USB\VID_2C7C&PID_6008]
GType = FuMmMbimDevice
```

Suitable port not found issue is reproduced

Queztion, 2025-07-29:
Hello @hughsie ,

Please refer below for latest test on Fedora 42 with EM061K, this is using latest fwupdmgr 2.0.14 from
 https://copr.fedorainfracloud.org/coprs/rhughes/fwupd

**TL;DR:**  There was no need to add a quirk file, modem is detected from get-releases.
Updates are detected as well.

However, firmware update fails and and modemmanager crashes after attempt to update:

```
$ sudo fwupdmgr downgrade
[sudo] password for eiger1-intel-pe-em061k: 
0.	Cancel
1.	02861494e3c701c896953cd442bd2f1ad867e1eb (EM061K-GL)
2.	ee3ccc9fd1f3a2ea6ef94e16ab61dafa1cf782e6 (Embedded Controller)
3.	371581745d6f93adaeae648c7fdf18e7b3c5ccaf (Prometheus)
4.	65841ecb4a989a55f43a02c5303bd22b9d1f0da6 (System Firmware)
5.	5bc922b7bd1adb5b6f99592611404036bd9f42d0 (UEFI CA)
6.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
Choose device [0-6]: 1
0.	Cancel
1.	EM061KGLAAR01A02M2G_01.013.01.013
2.	EM061KGLAAR01A02M2G_01.011.01.011
Choose release [0-2]: 1
╔══════════════════════════════════════════════════════════════════════════════╗
║ Downgrade EM061K-GL from EM061KGLAAR01A02M2G_01.015 to                       ║
║ EM061KGLAAR01A02M2G_01.013.01.013?                                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Firmware release EM061KGLAAR01A02M2G_01.013.01.013 packaged for LVFS.        ║
║                                                                              ║
║ • Performance improvements                                                   ║
║ • Stability improvements                                                     ║
║                                                                              ║
║ EM061K-GL and all connected devices may not be usable while updating.        ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Downgrading Gobi Wireless Modem (QDL mode)…                      ] 3 minutes remaining…
Waiting…                 [*                                      ]
failed to write-firmware: failed to get device response: failed to do bulk transfer (read): USB error: Operation timed out [-7]

```

Logs are added as well.

[fwupdmgr4_em061.zip](https://github.com/user-attachments/files/21481820/fwupdmgr4_em061.zip)

hughsie, 2025-07-29:
@Queztion can you link to the payload you're trying to deploy please? I see in your logs:

    Jul 29 14:17:45 fedora fwupd[3417]: 05:17:45.221 FuEngine             failed to attach device after failed update: failed to attach using qc_firehose: failed to compile a42c6cf1de3abfdea9b95f34687cbbe92b9a7383: Error on line 1 char 1: Document must begin with an element (e.g. <book>)

i.e. is `firehose-rawprogram.xml|rawprogram_*.xml` in the payload archive a valid XML document?

Mank-Wang, 2025-08-14:
@hughsie 
The *.xml file in the payload must be a valid XML document.
These firmware packages have been upgraded many times by the fwupd 1.9.xx .

Queztion, 2025-11-11:
Hello @hughsie ,

I've tested latest fwupdmgr 2.0.17 in fedora, but the same issue and result as above 2 comments for 2.0.16 are found.

Logs are below:

[2.0.17_fwupd.zip](https://github.com/user-attachments/files/23475805/2.0.17_fwupd.zip)

Also, as a sidenote,
there was an issue with ModemManager 1.22.0 where the FW version, displayed as 
**EM061KGLAAR01A02M2G_01.015**  instead of 
**EM061KGLAAR01A02M2G_01.015.01.015**

The issue appears intermittently, and is already fixed in ModemManager 1.23.11 and onwards.
I don't know if that is related or if that affects this issue.
But is ModemManager update in Fedora coming soon? 
I'll create another ticket for this one and will provide the patches if needed.

Queztion, 2025-11-19:
Hello @hughsie 

We have performed the following tests:

> reboot laptop from Ubuntu 25.10 to Fedora 43
> tried to perform fwupdate before 1 minute
> start firmware update
> *success*
> start 2nd firmware update
> fw update fails due to failed to attach replug:

modem is still missing after we tried running the following:

sudo fwupdtool --plugins qc-firehose attach

```

Choose release [0-2]: 1
╔══════════════════════════════════════════════════════════════════════════════╗
║ Downgrade EM061K-GL from EM061KGLAAR01A02M2G_01.015.01.015 to                ║
║ EM061KGLAAR01A02M2G_01.013.01.013?                                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Firmware release EM061KGLAAR01A02M2G_01.013.01.013 packaged for LVFS.        ║
║                                                                              ║
║ • Performance improvements                                                   ║
║ • Stability improvements                                                     ║
║                                                                              ║
║ EM061K-GL and all connected devices may not be usable while updating.        ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Restarting device…       [***********************                ] Less than one minute remaining…
failed to attach: failed to wait for attach replug: device 411c4b86d5f7216b21b278c743d25e0e6a90bf34 did not come back

```

Logs are attached for this current test:
failed_attached_replug3

[failed_attached_replug3.zip](https://github.com/user-attachments/files/23626357/failed_attached_replug3.zip)

Previous logs also attached:

[failed_attached_replug.zip](https://github.com/user-attachments/files/23626350/failed_attached_replug.zip)