fwupd/fwupd #8779: Dell OptiPlex SFF Plus 7020 - System Update not offered

realsobek, 2025-05-09:
**Describe the bug**
PC hardware (Dell, OptiPlex Small Form Factor (SFF) Plus 7020) supported: https://fwupd.org/lvfs/search?value=7020
but firmware updates not offered for installation. Downloading CAB file and installing it works without problem.

**Steps to Reproduce**
Update not offered:
```
# fwupdmgr get-remotes | grep -B 3 Enabled
│
├─  Remote ID:          vendor-directory
│     Type:               directory
│     Enabled:            true
--
│   
├─  Remote ID:          lvfs-testing
│     Type:               download
│     Enabled:            false
--
│   
└─  Remote ID:          lvfs
      Type:               download
      Enabled:            true

# fwupdmgr get-updates
Devices with no available firmware updates: 
 • Bios DB Key
 • Bios FW Aux Authority
 • KEK CA
 • Key Exchange Key
 • PC SN810 NVMe WDC 512GB
 • SBAT
 • System Firmware
 • TPM
 • UEFI CA
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • Windows Production PCA
No updatable devices

# fwupdmgr refresh --force
Updating lvfs
Downloading…             [  -                                    ]
Successfully downloaded new metadata: Updates have been published for 0 of 13 local devices
```

CAB file download and installation:
```
# dmidecode -s bios-version
1.12.0
# wget https://fwupd.org/downloads/162956764e85d66eb6dccc47cfd8a53ce92fdaf9fe8b94718164f81a30ba5e28-OptiHighMlkRpl_16.1.35.2557_DVT_firmware.cab
# fwupdmgr local-install 162956764e85d66eb6dccc47cfd8a53ce92fdaf9fe8b94718164f81a30ba5e28-OptiHighMlkRpl_16.1.35.2557_DVT_firmware.cab 
Waiting…                 [***************************************]
Successfully installed firmware
An update requires a reboot to complete. Restart now? [y|N]: y
# dmidecode -s bios-version
1.13.2
```

**Expected behavior**
`fwupdmgr get-updates` to offer update and `fwupdmgr update` to install it.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
# fwupdtool get-report-metadata
Waiting…                 [*************************************  ]
BatteryLevel:           101
BatteryThreshold:       10
[... 434 lines cut ...]
│   │   └── unicode.pf2
│   ├── grub.cfg
│   └── grubenv
├── initramfs-0-rescue-f681192259b946f0aa04591dde44989c.img
├── initramfs-6.14.3-300.fc42.x86_64.img
├── loader
│   └── entries
│       ├── f681192259b946f0aa04591dde44989c-0-rescue.conf
│       └── f681192259b946f0aa04591dde44989c-6.14.3-300.fc42.x86_64.conf
├── symvers-6.14.3-300.fc42.x86_64.xz -> /lib/modules/6.14.3-300.fc42.x86_64/symvers.xz
├── System.map-6.14.3-300.fc42.x86_64
├── vmlinuz-0-rescue-f681192259b946f0aa04591dde44989c
└── vmlinuz-6.14.3-300.fc42.x86_64

13 directories, 23 files
```

**Additional questions**

- Operating system and version: GNU/Linux, Fedora Workstation 42
- Have you tried rebooting? yes, to no avail
- Is this a regression? I guess not, because another computer with Fedora 40 and BIOS 1.4.1 has the same problem
- Are you using an NVMe disk? yes
- Is secure boot enabled? yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? no

realsobek, 2025-05-09:
It is a regression. According to documentation of that other computer: on 2024-06-03 with Fedora 40 I used `fwupdmgr` (version unknown) to update system firmware 1.2.1 -> 1.4.1 and UEFI dbx 220 -> 371. But today on 2025-05-09 that same computer with Fedora 40 (with latest updates) does not offer firmware update.

realsobek, 2025-05-09:
I reinstalled the computer with vanilla Fedora Workstation 40 without updates. firmware updates are offered. Same with latest Fedora 40 updates and after upgrading to Fedora 42.

I suspect a local, custom installation to be the culprit. Other LVFS supported hardware that is not using the latest firmware does not get it offered by `fwupdmgr`. The differences in above commands outputs do not cause the problem. Hence I must bisect the local, custom installation script.

hughsie, 2025-05-09:
Can you attach the output of `sudo fwupdtool get-devices --plugins uefi_capsule -vv` please? That might give us some clues.

realsobek, 2025-05-09:
culprit is a file from snapgene-8.0-3.x86_64.rpm from https://www.snapgene.com; minimal reproducible (I included only the problematic part of the 15 kB XML file):
```
# cat > /usr/share/mime/packages/application-x-snapgene.xml << 'EOF'
<?xml version='1.0' encoding='UTF-8'?>
<mime-info xmlns='http://www.freedesktop.org/standards/shared-mime-info'>
    <mime-type type='application/vnd.jellyfish.xml'>
        <comment>Jellyfish File</comment>
        <weight>40</weight>
        <icon name='snapgene-document_generic'/>
        <glob pattern='*.xml'/>
    </mime-type>
</mime-info>
EOF
# update-mime-database /usr/share/mime # executed in `rpm -q --scripts snapgene`
```

Changing the XML glob pattern (f.i. to xml2) makes `fwupdmgr` work again as expected; quick testing with `vim /usr/share/mime/packages/application-x-snapgene.xml ; update-mime-database /usr/share/mime ; rm -rf /var/lib/fwupd/* /var/cache/fwupd/* ; fwupdmgr get-updates -y`

output of `sudo fwupdtool get-devices --plugins uefi_capsule -vv`:
<details>

<summary>**with the culprit XML file**</summary>

```shell
22:54:34.885 FuDebug              verbose to debug (on console 1)
22:54:34.885 FuMain               locked /run/lock/fwupdtool
Loading…                 [ -                                     ]22:54:34.894 FuCommon             creating path /var/lib/fwupd/quirks.d
22:54:34.894 FuCommon             creating path /var/lib/fwupd/remotes.d
22:54:34.894 FuConfig             loading config /etc/fwupd/fwupd.conf
22:54:34.894 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
22:54:34.894 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
22:54:34.894 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
22:54:34.894 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
22:54:34.894 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
22:54:34.894 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
22:54:34.894 FuConfig             ::configuration loaded
22:54:34.895 FuRemoteList         loading /usr/share/fwupd/metainfo
22:54:34.895 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
22:54:34.895 XbSilo               failed to load silo: Failed to open file “/var/cache/fwupd/metainfo.xmlb”: open() failed: No such file or directory
22:54:34.895 XbSilo               detected content type of org.freedesktop.fwupd.remotes.lvfs-testing.metainfo.xml to be application/xml
22:54:34.896 XbSilo               detected content type of org.freedesktop.fwupd.remotes.lvfs.metainfo.xml to be application/xml
22:54:34.897 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
22:54:34.897 FuRemoteList         ::remote_list changed
22:54:34.897 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
22:54:34.897 FuRemoteList         ::remote_list changed
22:54:34.897 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
22:54:34.897 FuRemoteList         ::remote_list changed
22:54:34.897 FuEngine             priority local and vendor-directory is not download remote, so bumping
22:54:34.897 FuRemoteList         ordering lvfs-testing=lvfs+1
22:54:34.897 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]22:54:34.897 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
22:54:34.898 FuHistory            no schema version: no such table: schema
22:54:34.898 FuHistory            got schema version of 0
22:54:34.898 FuHistory            building initial database
Loading…                 [*                                      ]22:54:34.934 FuPlugin             load(/usr/lib64/fwupd-2.0.8/libfu_plugin_modem_manager.so)
Loading…                 [*                                      ]22:54:34.936 FuQuirks             open database /var/cache/fwupd/quirks.db
22:54:34.938 FuQuirks             indexing vendor IDs from /usr/share/hwdata/pci.ids
22:54:34.995 FuQuirks             indexing vendor IDs from /usr/share/hwdata/usb.ids
22:54:35.073 FuQuirks             indexing vendor IDs from /usr/share/hwdata/pnp.ids
22:54:35.083 FuQuirks             indexing vendor IDs from /usr/share/hwdata/oui.txt
22:54:35.267 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
22:54:35.267 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
22:54:35.267 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
22:54:35.268 XbSilo               failed to load silo: Failed to open file “/var/cache/fwupd/quirks.xmlb”: open() failed: No such file or directory
22:54:35.268 XbSilo               detected content type of builtin.quirk.gz to be application/gzip
22:54:35.268 XbSilo               detected content type of builtin.quirk to be text/plain
Loading…                 [*                                      ]22:54:35.281 FuConfig             loading config /etc/fwupd/fwupd.conf
22:54:35.281 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
22:54:35.281 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
22:54:35.281 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
[... 2489 lines cut ...]
  Created:              2025-05-09
  UpdateState:          success

22:54:35.971 FuMain               FuUefiNvramDevice:
[... cut ...]
  VendorId:             DMI:Dell Inc.
  Version:              1636902180
  VersionLowest:        1636902180
  VersionFormat:        number
  VersionRaw:           0x61912524
  VersionLowestRaw:     0x61912524
  Created:              2025-05-09
  UpdateState:          success

22:54:35.971 FuProgress           raw timing data was { 0.000, 0.000, 0.000, 0.135, 0.002 } -- steps were set as [ 1 8 64 26 1 ] but should have been [ 0 0 0 98 2 ] at ../plugins/uefi-capsule/fu-uefi-capsule-plugin.c:1029
```

</details>

realsobek, 2025-05-09:
(putting all in one message resulted in "problem saving your message" message)

output of `sudo fwupdtool get-devices --plugins uefi_capsule -vv`:
<details>

<summary>**without the culprit XML file**</summary>

```shell
22:56:52.740 FuDebug              verbose to debug (on console 1)
22:56:52.740 FuMain               locked /run/lock/fwupdtool
Loading…                 [ -                                     ]22:56:52.748 FuConfig             loading config /etc/fwupd/fwupd.conf
22:56:52.748 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
22:56:52.748 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
22:56:52.748 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
22:56:52.748 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
22:56:52.748 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
22:56:52.748 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
22:56:52.748 FuConfig             ::configuration loaded
22:56:52.748 FuRemoteList         loading /usr/share/fwupd/metainfo
22:56:52.748 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
22:56:52.748 XbSilo               file: 1ab6452c-34ea-b67c-ea4b-f43bfd0e9981, current:1ab6452c-34ea-b67c-ea4b-f43bfd0e9981
22:56:52.748 XbSilo               loading silo with existing file contents
22:56:52.748 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
22:56:52.748 FuRemoteList         ::remote_list changed
22:56:52.748 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
22:56:52.748 FuRemoteList         ::remote_list changed
22:56:52.748 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
22:56:52.749 FuRemoteList         ::remote_list changed
22:56:52.749 FuEngine             priority local and vendor-directory is not download remote, so bumping
22:56:52.749 FuRemoteList         ordering lvfs-testing=lvfs+1
22:56:52.749 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…                 [                                       ]22:56:52.749 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
22:56:52.749 FuHistory            got schema version of 14
Loading…                 [*                                      ]22:56:52.754 FuPlugin             load(/usr/lib64/fwupd-2.0.8/libfu_plugin_modem_manager.so)
Loading…                 [*                                      ]22:56:52.755 FuQuirks             open database /var/cache/fwupd/quirks.db
22:56:52.755 FuQuirks             mtimes unchanged: quirks,pci.ids:1743465600,usb.ids:1743465600,pnp.ids:1743465600,oui.txt:1743465600, doing nothing
22:56:52.755 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
22:56:52.755 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
22:56:52.755 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
22:56:52.755 XbSilo               file: 314b5563-ee56-c1ca-aa91-ef1d93f465f6, current:314b5563-ee56-c1ca-aa91-ef1d93f465f6
22:56:52.755 XbSilo               loading silo with existing file contents
Loading…                 [*                                      ]22:56:52.756 FuConfig             loading config /etc/fwupd/fwupd.conf
22:56:52.756 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
22:56:52.756 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
22:56:52.756 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
22:56:52.756 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
22:56:52.756 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
22:56:52.756 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
22:56:52.756 FuConfig             ::configuration loaded
22:56:52.756 FuStruct             FuStructSmbiosEp32:
  anchor_str: _SM_
  entry_point_csum: 0xed
  entry_point_len: 0x1f
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x6
  max_structure_sz: 0x18c
  entry_point_rev: 0x0
  intermediate_anchor_str: _DMI_
  intermediate_csum: 0x6a
  structure_table_len: 0x1903
  structure_table_addr: 0x5d44a000
  number_smbios_structs: 0x6b
  smbios_bcd_rev: 0x36
22:56:52.756 FuStruct             FuStructSmbiosStructure:
  type: 0xda
  length: 0xfb
  handle: 0xda00
22:56:52.756 FuStruct             FuStructSmbiosStructure:
  type: 0xda
  length: 0xfb
[... 2475 lines cut ...]
  UpdateState:          success

22:56:53.474 FuMain               FuUefiNvramDevice:
  DeviceId:             c2df1116b6d29b6fad71b4444c4c2f216fcbb6d2
  Name:                 UEFI Device Firmware
  Guid:                 a7486a46-2c93-48cb-a1fd-5b89a1de9532
  Summary:              UEFI System Resource Table device (updated via NVRAM)
  Plugin:               uefi_capsule
  Protocol:             org.uefi.capsule
  Flags:                internal|updatable|require-ac|needs-reboot|usable-during-update
  RequestFlags:         allow-generic-message
  VendorId:             DMI:Dell Inc.
  Version:              1636902180
  VersionLowest:        1636902180
  VersionFormat:        number
  VersionRaw:           0x61912524
  VersionLowestRaw:     0x61912524
  Created:              2025-05-09
  UpdateState:          success

22:56:53.474 FuProgress           raw timing data was { 0.001, 0.000, 0.000, 0.003, 0.534 } -- steps were set as [ 1 1 3 1 94 ] but should have been [ 0 0 0 1 99 ] at ../libfwupdplugin/fu-context.c:1101
22:56:53.474 FuProgress           raw timing data was { 0.000, 0.000, 0.000, 0.140, 0.002 } -- steps were set as [ 1 8 64 26 1 ] but should have been [ 0 0 0 98 1 ] at ../plugins/uefi-capsule/fu-uefi-capsule-plugin.c:1029
```

</details>

superm1, 2025-05-10:
I don't believe there is anything we can do about this. It should be reported to the owner of that package.

closed 2025-05-10 by superm1