fwupd/fwupd #8780: After latest update dbx still cannot be updated

RobusTetus, 2025-05-10:
I am opening a new issue since I do not have permission to reopen #8702 

I have tried the newest update 2.0.9 which should contain added quirk, but it would seem that it is still happening. When trying to manually install latest [cab archive](https://fwupd.org/lvfs/devices/com.microsoft.dbx.x64.firmware) I get `Device UEFI dbx [362301da643102b9f38477387e2193e57abaa590] has no firmware version` 

Running `get-details` on the archive shows the same error during upgrade.

```
LENOVO 82TK
│
└─UEFI dbx:
  │   Identifikátor zařízení: 362301da643102b9f38477387e2193e57abaa590
  │   Souhrn:             UEFI revocation database 10d45fcb
  │   Výrobce:            UEFI:Microsoft
  │   Délka instalace:    1 sekunda
  │   Chyba při aktualizaci: Device UEFI dbx [362301da643102b9f38477387e2193e57abaa590] has no firmware version
  │   GUID:               f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
  │   Příznaky zařízení:  • Vestavěné zařízení
  │                       • Podporováno na vzdáleném serveru
  │                       • Po instalaci vyžaduje restart
  │                       • Je k dispozici kryptografické ověření otisku dat
  │                       • Zařízení je možné používat i v průběhu aktualizace
  │                       • Možné aktualizovat
  │                       • Zařízení umožňuje pouze aktualizace na novější verze
  │                       • Podepsaný obsah
  │                       • Může označit pro emulaci
  │ 
  └─Aktualizace nastavení pro Secure Boot dbx:
        Nová verze:       20241101
        Souhrn:           UEFI Secure Boot Forbidden Signature Database
        Varianta:         x64
        Licence:          Uzavřená
        Velikost:         15,1 kB
        Naléhavost:       Vysoká
        Výrobce:          Linux Foundation
        Příznaky vydání:  • Důvěryhodný obsah
                          • Důvěryhodná metadata
        Popis:            
        This updates the list of forbidden signatures (the "dbx") to the latest release from Microsoft.
        
        An insecure version of Howyar's SysReturn software was added, due to a security vulnerability that allowed an attacker to bypass UEFI Secure Boot.
        Problémy:         CVE-2024-7344
                          529659
        Kontrolní součet: 70ce9f7793c7b8c5f1f2f296d6e3fae3491e71d5fd2bc6faebd186c96f226b1a

```

Only thing that is indeed fixed on the 2.0.9 release for me is #8695

RobusTetus, 2025-05-10:
I have also manually verified that the quirk is indeed present in `/var/lib/fwupd/quirks.d/builtin.quirk.gz`

hughsie, 2025-05-10:
I'm a it confused why a45fc03ea72401fd3517e01ac80f85044c28a299 didn't work. Do you see anything weird from `sudo fwupdtool get-devices --plugins uefi_dx` -- maybe I typod the checksum?

RobusTetus, 2025-05-10:
> I'm a it confused why [a45fc03](https://github.com/fwupd/fwupd/commit/a45fc03ea72401fd3517e01ac80f85044c28a299) didn't work. Do you see anything weird from `sudo fwupdtool get-devices --plugins uefi_dx -vv` -- maybe I typod the checksum?

There is no typo, I already checked, the checksum added is correct. In case I missed something I am adding full output of `sudo fwupdtool get-devices --plugins uefi_dbx -vv` as a file. [fwupd.log](https://github.com/user-attachments/files/20140946/fwupd.log)

hughsie, 2025-05-11:
> mtimes unchanged: quirks,pci.ids:0,usb.ids:0,pnp.ids:0,oui.txt:0, doing nothing

This looks odd -- why would the mtimes all be zero? If you do `sudo rm /var/cache/fwupd/quirks.*` and force a silo refresh does it start working?

RobusTetus, 2025-05-11:
> > mtimes unchanged: quirks,pci.ids:0,usb.ids:0,pnp.ids:0,oui.txt:0, doing nothing
> 
> This looks odd -- why would the mtimes all be zero? If you do `sudo rm /var/cache/fwupd/quirks.*` and force a silo refresh does it start working?

That fixed it. Deleting the cache and then running `sudo fwupdtool update` instantly recognized the update and I successfully updated the dbx. Thank you for your help.

closed 2025-05-11 by RobusTetus