fwupd/fwupd #8793: Unable to update UEFI dbx on HP Elitebook 845 Gen 10

adhik, 2025-05-14:
**Describe the bug**
I faced a very similar issue like this one https://github.com/fwupd/fwupd/issues/8710, on an HP Elitebook 845 gen 10 laptop with Ubuntu 24.04.02 LTS. Does the `BaseboardProduct=8B6E` need to be added as well? 

**Steps to Reproduce**
1. Tried using the firmware updater (the Gnome app), it said it just need to reboot but then the updates didn't get installed correctly.
2. Also tried to update by using the terminal command `sudo fwupdmgr update --force -y`, which was also said to reboot, but then the UEFI dbx were not updated as well, and the gnome notification (for updates) appeared again. 

**Expected behavior**
UEFI dbx to be updated / installed correctly.

**fwupd version information**
```
compile   org.freedesktop.fwupd         1.9.29
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.29
runtime   org.kernel                    6.11.0-25-generic
```

```shell
fwupdtool get-report-metadata

BatteryLevel:           98
BatteryThreshold:       50
BootTime:               1747221417
CompileVersion(com.hughsie.libjcat):0.2.0
CompileVersion(com.hughsie.libxmlb):0.3.18
CompileVersion(org.freedesktop.fwupd):1.9.29
CompileVersion(org.freedesktop.gusb):0.4.8
CpuArchitecture:        x86_64
CpuModel:               Advanced Micro Devices, Inc. AMD Ryzen 7 PRO 7840HS w/ Radeon 780M Graphics
DisplayState:           connected
DistroId:               ubuntu
DistroName:             Ubuntu
DistroPrettyName:       Ubuntu 24.04.2 LTS
DistroVersion:          24.04
FwupdSupported:         True
HostBaseboardManufacturer:HP
HostBaseboardProduct:   8B6E
HostBiosMajorRelease:   08
HostBiosMinorRelease:   00
HostBiosVendor:         HP
HostBiosVersion:        V82 Ver. 01.08.00
HostEnclosureKind:      a
HostFamily:             103C_5336AN HP EliteBook
HostFirmwareMajorRelease:60
HostFirmwareMinorRelease:35
HostProduct:            HP EliteBook 845 14 inch G10 Notebook PC
HostSku:                84G22PC#AB2
HostVendor:             HP
KernelName:             Linux
KernelVersion:          6.11.0-25-generic
LidState:               open
PowerState:             ac-charging
RuntimeVersion(com.hughsie.libjcat):0.2.0
RuntimeVersion(com.hughsie.libxmlb):0.3.x
RuntimeVersion(org.freedesktop.fwupd):1.9.29
RuntimeVersion(org.freedesktop.fwupd-efi):1.4
RuntimeVersion(org.freedesktop.gusb):0.4.8
RuntimeVersion(org.kernel):6.11.0-25-generic
DeviceId:               307a00f856126b8dca325acd8b542cd28c9002c3
  pre:
      CapsuleApplyMethod:nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader:False
  post:
      LastAttemptStatus:0x0
      LastAttemptVersion:0x1080000
DeviceId:               b8e2f4eb5953baa0de8e6075d2ecefb92cf23616
  pre:
      CapsuleApplyMethod:nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader:False
  post:
      LastAttemptStatus:0x0
      LastAttemptVersion:0x18160000
DeviceId:               b970edafffcf075cf953e5efebbf2595bf98f1db
  pre:
      CapsuleApplyMethod:nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader:False
  post:
      LastAttemptStatus:0x0
      LastAttemptVersion:0x17030000
DeviceId:               e97608be1da5b9d742670ca8c3c4f515bc10646a
  pre:
      CapsuleApplyMethod:nvram
      EspKind:          c12a7328-f81f-11d2-ba4b-00a0c93ec93b
      EspPath:          /boot/efi
      MissingCapsuleHeader:False
  post:
      LastAttemptStatus:0x0
      LastAttemptVersion:0x37c42f06
DeviceId:               362301da643102b9f38477387e2193e57abaa590
  pre:
      EfivarNvramUsed:  87244
      LinuxLockdown:    none
      Pcr0_SHA256:      053c46e07e25caf9e6f264143b501c819060fd68fabf163800cbaec4cd7b0e64
      TpmEventLog:      0x00000008 5c0fbf4e5ff5fdc9b5f00a0af777df7ececedc01a17d859a95e9c39ec930b96a [VjgyAAAAAAAAAAAAAAAAAAgAAAAAAAAABAAAAAEAAAAyZDc2M2EwNzAyZDZjODg1ZTU3Y2U2MjA5MzMxMjM4NTZkMjRhZjAxg7KXfvezcCKy0TKUB2gRMxBk1WI4lQYYIn8/V4zbKWdXvKiXnRpl5oJgpLNOMLEeSFgGE2KRX6vVZQ5XXGZoijegI8lSbOwplf99vioXatXs7rFO7vXxmvFo471o4cdwNV0DM0pQCa03olKg5kZyu0LnDcRuaD84h4sZegJIXrnFCk8XgZ5Sr86b5FGQaj0HEXqFksi4S17S5p/MIIdm/FSJfLr0vTNrikzachSOAJr0aWwGe+l98MHr06F01c9chOOC59PQCcbpm2I0GXpa3SUdhBtWPHKDPNb1RYlaj/7YfLOthC4TqwvVBrD6cpHhZzltTKuOMrCI09/g2M/XgKp/G+UhCvWvMXmP/udU9qpYcRWSPpJXpEp8pMMawLEtr+uGL/mkcmHfVxdCfWmIvHvvduKhZ4vkzQqtzhofntmapEqhmyISKct3eFNKEGUk+Md7CdkkXWYQcCPmQprQi0MYDO6NKCXNsc6h3xZgn+Mo2g8/lH/n3c7c/hh3PrKf]
      0x80000008 ae49fe84dd4406af74888bc60cbcad09f45c1b6cd1275972ae597927c58b9337 [ACAG/wAAAAAAoLkAAAAAAA==]
      0x00000001 b4513952cf0a47256fbb0e044995e8c07c58772d6333ec4c1030295a0b30b5cf [QUNQSSBEQVRB]
      0x00000001 608072d8953921f15718897cbb0a47623e0c29fa7286d20beb0d733756acb643 [QUNQSSBEQVRB]
      0x00000004 df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119 [AAAAAA==]
      PCR0: 053c46e07e25caf9e6f264143b501c819060fd68fabf163800cbaec4cd7b0e64
      TpmFamily:        2.0
      EfivarNvramUsed:  87244
      SecureBoot:       Disabled
      UEFIUXCapsule:    Enabled
```

```shell
fwupdtool hwids

Computer Information
--------------------
BiosVendor: HP
BiosVersion: V82 Ver. 01.08.00
Manufacturer: HP
Family: 103C_5336AN HP EliteBook
ProductName: HP EliteBook 845 14 inch G10 Notebook PC
ProductSku: 84G22PC#AB2
EnclosureKind: 10
BaseboardManufacturer: HP
BaseboardProduct: 8B6E
Hardware IDs
------------
not available as 'BiosMajorRelease' unknown
not available as 'BiosMajorRelease' unknown
not available as 'BiosMajorRelease' unknown
{84e38085-d9f2-548f-a69f-7d233a4236c1}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{28a4c321-dfcf-5fc1-a8a3-e373398a7679}   <- Manufacturer + Family + ProductName + ProductSku
{3a985c0b-ad82-5ffc-a078-508dedf9104d}   <- Manufacturer + Family + ProductName
{c8d1f3c4-60ff-5299-ad8e-9ffc784a6c56}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{74d1b0c3-ea16-5cc6-9ee5-e35d6f69e4f1}   <- Manufacturer + ProductSku
{6f99cca2-d5d9-5167-b07e-52e8fefbf185}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{4a542577-5394-5136-ad8a-1ed8bbbf6e6b}   <- Manufacturer + ProductName
{843d0f62-ee61-5c11-95a0-483c94376cc1}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{19fe40b5-1900-5d4a-a48f-b9f9938f3e69}   <- Manufacturer + Family
{4c84c882-b54c-58d5-b412-26bcdf254c3e}   <- Manufacturer + EnclosureKind
{7121aa03-6ee3-5987-b04a-84aa6866a580}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{93f84748-c854-5d6b-b78a-13c2361e0758}   <- Manufacturer
Extra Hardware IDs
------------------
{fc5c43cd-5d79-51fe-9485-14ca426e2f86}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor
{12413025-64fe-5055-8d1f-513ba5017bb1}   <- Manufacturer + Family + ProductName + BiosVendor
{90eeff90-019d-5e49-ae7b-101135ba91d4}   <- Manufacturer + BiosVendor

```

**Additional questions**

- Operating system and version: Ubuntu 24.04.2 LTS
- Have you tried rebooting? yes
- Is this a regression? Not sure, got this machine recently
- Are you using an NVMe disk? yes
- Is secure boot enabled? no
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? no

hughsie, 2025-05-14:
> Does the BaseboardProduct=8B6E need to be added as well

Highly likely. Would you be able to do a pull request to do that please?

adhik, 2025-05-14:
> > Does the BaseboardProduct=8B6E need to be added as well
> 
> Highly likely. Would you be able to do a pull request to do that please?

Here: https://github.com/fwupd/fwupd/pull/8794 ? 
note: it was my first time contributing to an open source project haha

hughsie, 2025-05-14:
> it was my first time contributing to an open source project

Well, you did a really good job. Welcome to the club!

PR #8794 (2025-05-14): uefi-dbx: Do not allow updates on the HP 84G22PC (Elitebook 845 Gen10)
fixes: #8793

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [x] Feature
- [ ] Documentation
commit 59c6ddda (2025-05-14): uefi-dbx: Do not allow updates on the HP 84G22PC (Elitebook 845 Gen10)
fixes: #8793
closed 2025-05-14 by hughsie