fwupd/fwupd #8808: ThinkPad T14 G2 AMD - KEK update fails

verpejas, 2025-05-19:
**Describe the bug**
fwupdmgr tries to update the KEK in DBX but fails to do so with this error message: 
`failed to write-firmware: failed to write dec64d7746d983db3774829a00bf829d9f19e9cf: failed to write data to efivarsfs: Error writing to file descriptor: Invalid argument
`

```
verpejas@cezanne:~$ sudo fwupdmgr update
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade KEK CA from 2011 to 2023?                                            ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the UEFI Signature Database (the "KEK") to the latest release   ║
║ from Microsoft, signed by Lenovo Ltd. PK CA 2012.                            ║
║                                                                              ║
║ KEK CA and all connected devices may not be usable while updating.           ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Writing…                 [******************************         ]
Successfully uploaded report
failed to write-firmware: failed to write dec64d7746d983db3774829a00bf829d9f19e9cf: failed to write data to efivarsfs: Error writing to file descriptor: Invalid argument

verpejas@cezanne:~$ sudo fwupdmgr get-history
LENOVO 20XK0016US
│
└─KEK CA:
  │   Device ID:          b7a1d3d90faa1f6275d9a98da4fb3be7118e61c7
  │   Previous version:   2011
  │   Update State:       Failed
  │   Update Error:       failed to write-firmware: failed to write dec64d7746d983db3774829a00bf829d9f19e9cf: failed to write data to efivarsfs: Error writing to file descriptor: Invalid argument
  │   Last modified:      2025-05-19
  │   GUID:               814e950f-1449-566a-a190-42c9d3a3a2df
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Signed Payload
  │                       • Can tag for emulation
  │ 
  └─Secure Boot KEK Configuration Update:
        New version:      2023
        Remote ID:        lvfs
        Release ID:       113555
        Summary:          UEFI Secure Boot Key Exchange Key
        Variant:          Lenovo
        License:          Proprietary
        Size:             2.7 kB
        Created:          2025-04-29
        Urgency:          High
        Vendor:           Linux Foundation
        Release Flags:    • Trusted payload
                          • Trusted metadata
        Description:      
        This updates the UEFI Signature Database (the "KEK") to the latest release from Microsoft, signed by LENOVO.
        Checksum:         8bd7129b2fbadd03a75fa9f81bdd9f7094e9392600791c61c578bd5102822725
      

```

**Steps to Reproduce**
Update the KEK using fwupdmgr on a ThinkPad T14 G2 AMD 20XK

**Expected behavior**
The update finished succesfully

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdtool get-report-metadata

[... 544 lines cut ...]
│   └── entries  [error opening dir]
├── lost+found  [error opening dir]
├── symvers-6.14.4-300.fc42.x86_64.xz -> /lib/modules/6.14.4-300.fc42.x86_64/symvers.xz
├── symvers-6.14.5-300.fc42.x86_64.xz -> /lib/modules/6.14.5-300.fc42.x86_64/symvers.xz
├── symvers-6.14.6-300.fc42.x86_64.xz -> /lib/modules/6.14.6-300.fc42.x86_64/symvers.xz
├── System.map-6.14.4-300.fc42.x86_64
├── System.map-6.14.5-300.fc42.x86_64
├── System.map-6.14.6-300.fc42.x86_64
├── vmlinuz-0-rescue-4c52b1d7cbd74003a0f7c70b66e3fc54
├── vmlinuz-6.14.4-300.fc42.x86_64
├── vmlinuz-6.14.5-300.fc42.x86_64
└── vmlinuz-6.14.6-300.fc42.x86_64

6 directories, 17 files

```

**Additional questions**

- Operating system and version: Fedora 42 KDE
- Have you tried rebooting? Yes
- Is this a regression? -not applicable
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? no boot order lock.

hughsie, 2025-05-19:
This is https://github.com/fwupd/fwupd/pull/8807 I'm afraid. If you build fwupd with that patch it should work! I've pulled the update from the LVFS and will re-upload after the 2.0.10 release.

skymoo, 2025-05-28:
I was running into this on on Dell Precision 7560, my OS updated to 2.0.10 a couple of days ago and today it looks like it successfully installed this update.

closed 2025-05-28 by superm1