fwupd/fwupd #8821: Metadata signature check fails (likely) due to wrong /dev/mem mapping

marmarek, 2025-05-21:
**Describe the bug**
The user visible message is:
```
root@dom0:~# fwupdmgr refresh /var/cache/fwupd/qubes/metadata/firmware.xml.zst{,.jcat} lvfs
Failed to update metadata for lvfs: signatures were required, but none supplied
```
But this was very misleading. 

First of all, the signature is good:
```
root@dom0:~# jcat-tool --prefix=/var/cache/fwupd/qubes/metadata/  --public-keys=/etc/pki/fwupd  verify /var/cache/fwupd/qubes/metadata/firmware.xml.zst.jcat 
firmware.xml.zst:
    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
    PASSED sha256: OK
```

The actual problem is failure to initialize gpg, can be seen by using GPGME_DEBUG:
```
2025-05-21 13:37:29 gpgme[7380.1cd4] gpgme_debug: level=9
2025-05-21 13:37:29 gpgme[7380.1cd4] gpgme_check_version: call: req_version=(null), VERSION=1.23.2
2025-05-21 13:37:29 gpgme[7380.1cd4] gpgme_check_version_internal: call: req_version=(null), offset_sig_validity=60
2025-05-21 13:37:29 gpgme[7380.1cd4] gpgme_new: enter: r_ctx=0x00005821685174c8 
2025-05-21 13:37:29 gpgme[7380.1cd4] gpgme-dinfo: gpgconf='/usr/bin/gpgconf'
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_pipe: enter: inherit_idx=1 (GPGME uses it for reading)
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_pipe: leave: read fd=22 write fd=23
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn: enter: path=/usr/bin/gpgconf
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn: check: argv[ 0] = /usr/bin/gpgconf
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn: check: argv[ 1] = --list-dirs
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn: check: fd[0] = 0x17 -> 0x1
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn:550: error: Cannot allocate memory (12)\n
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_close: enter: fd=22
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_close: leave: result=0
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_close: enter: fd=23
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_close: leave: result=0
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_pipe: enter: inherit_idx=1 (GPGME uses it for reading)
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_pipe: leave: read fd=22 write fd=23
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn: enter: path=/usr/bin/gpgconf
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn: check: argv[ 0] = /usr/bin/gpgconf
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn: check: argv[ 1] = --list-components
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn: check: fd[0] = 0x17 -> 0x1
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn:550: error: Cannot allocate memory (12)\n
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_close: enter: fd=22
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_close: leave: result=0
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_close: enter: fd=23
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_close: leave: result=0
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_pipe: enter: inherit_idx=1 (GPGME uses it for reading)
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_pipe: leave: read fd=22 write fd=23
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn: enter: path=/usr/bin/gpgconf
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn: check: argv[ 0] = /usr/bin/gpgconf
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn: check: argv[ 1] = --version
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn: check: fd[0] = 0x17 -> 0x1
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_spawn:550: error: Cannot allocate memory (12)\n
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_close: enter: fd=22
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_close: leave: result=0
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_close: enter: fd=23
2025-05-21 13:37:29 gpgme[7380.1cd4]   _gpgme_io_close: leave: result=0
2025-05-21 13:37:29 gpgme[7380.1cd4] gpgme_new: leave: ctx=0x0000582168871af0
2025-05-21 13:37:29 gpgme[7380.1cd4] gpgme_set_protocol: enter: ctx=0x0000582168871af0 protocol=0 (OpenPGP)
2025-05-21 13:37:29 gpgme[7380.1cd4] gpgme_set_protocol: leave: 
2025-05-21 13:37:29 gpgme[7380.1cd4] gpgme_ctx_set_engine_info: enter: ctx=0x0000582168871af0 protocol=0 (OpenPGP), file_name=(default), home_dir=/var/lib/fwupd/gnupg
2025-05-21 13:37:29 gpgme[7380.1cd4] engine.c:413: returning error: Invalid crypto engine\n
2025-05-21 13:37:29 gpgme[7380.1cd4] gpgme_ctx_set_engine_info:1166: error: Invalid crypto engine <GPGME>\n
2025-05-21 13:39:18 gpgme[7380.1cd4] gpgme_data_new_from_file: enter: r_dh=0x00007fffd5bb0dd0 file_name=/etc/pki/fwupd/GPG-KEY-Linux-Foundation-Firmware, copy=1 (yes)
2025-05-21 13:39:18 gpgme[7380.1cd4]   gpgme_data_new_from_filepart: enter: r_dh=0x00007fffd5bb0dd0 file_name=/etc/pki/fwupd/GPG-KEY-Linux-Foundation-Firmware, stream=0x0000000000000000, offset=0, length=2169
```

`Cannot allocate memory`? Weird. But at the very same time, kernel prints:
```
[  316.814111] x86/PAT: fwupd:7476 map pfn expected mapping type uncached for [mem 0x76896000-0x76897fff], got write-back
[  316.814912] x86/PAT: fwupd:7476 map pfn expected mapping type uncached for [mem 0x76896000-0x76897fff], got write-back
[... 147 lines cut ...]
│       GUIDs:            f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│                         d07ff664-b0e1-5f4e-a723-d7fbcbfcb94f ← UEFI\CRT_3CD3F0309EDAE228767A976DD40D9F4AFFC4FBD5218F2E8CC3C9DD97E8AC6F9D&ARCH_X64
│                         2c456282-9d22-5324-aad3-2de617687188 ← UEFI\CRT_E5BE3E64C6E66A281457ECDECE0D6D0787577AAD2A3A0144262C10C14BA8D8F1&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Microsoft RSA Third Party PCA 2023
      Vendor:             Microsoft Corporation
      GUID:               fcde00d4-4821-54ce-88d9-be8c13c8415f ← UEFI\CRT_19936DBF64894773BD31EB76D9EF0417FEDDD8EA
```

</details>

**Additional questions**

- Operating system and version: Qubes OS 4.3
- Have you tried rebooting? Yes
- Is this a regression? Unsure, possibly no. But it does work in Starlabs StarBook MkVI which also uses flashrom plugin.

hughsie, 2025-05-21:
I have no idea I'm afraid. If you simplify things a bit (i.e. a Fedora install with out xen or Qubes in the mix) do you see the same thing? Long term I think we have to deprecate flashrom direct hardware access and get vendors to support and use the mtd interface for bios updates.

marmarek, 2025-05-21:
I have an idea what is going on here. I guess `fork()` is not happy about copying those `/dev/mem` mappings to the new process. If they are indeed needed to remain mapped all the time in the main `fwupd` process, I think they should be marked with `MADV_DONTFORK`. `gpg` (or any other child process) shouldn't really have them anyway.

hughsie, 2025-05-22:
That all sounds plausible; but we don't manually fork -- we rely on systemd to manage the service startup. But back a bit; if GPG doesn't work we're supposed to use PKCS#7 signatures too -- did you compile jcat without gnutls support perhaps?

hughsie, 2025-05-22:
Okay, I guess that's fine -- but I think you're going to have to submit the fix I'm afraid.

marmarek, 2026-03-02:
> I think I've updated all the qubes stuff to use JCat instead.

While dropping GPG support is totally understandable, I'm not very happy about JCat either... The qubes wrapper verifies its selected parts separately for a reason (for example to avoid issues like https://github.com/advisories/GHSA-phhj-vpf5-5666). Plus PKCS#7 also isn't exactly easy to get right. How test coverage of jcat-tool looks like?