fwupd/fwupd #8831: Unable to update UEFI dbx on Framework Laptop 13th Gen

dLopreiato, 2025-05-23:
**Describe the bug**
No matter how many times I attempt to update the UEFI dbx on my machine, after reboot, the version is still 20230501, instead of the newest version.

**Steps to Reproduce**
1. `sudo fwupdmgr update`
2. "Upgrade UEFI dbx from 20230501 to 20241101?" and "Perform operation?" `Y`
3. "An update requires a reboot to complete. Restart now?" `y`
4. Log back in
5. `sudo fwupdmgr update`
6. The UEFI dbx update from 20230501 to 20241101 is offered again.

**Expected behavior**
The UEFI dbx to be on version 20241101 after updating and rebooting.

**fwupd version information**
```
$ sudo fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.29
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.29
runtime   org.kernel                    6.11.0-26-generic
```

```
$ sudo fwupdtool get-report-metadata
BatteryLevel:           63
BatteryThreshold:       10
BootTime:               1748040165
CompileVersion(com.hughsie.libjcat):0.2.0
CompileVersion(com.hughsie.libxmlb):0.3.18
CompileVersion(org.freedesktop.fwupd):1.9.29
CompileVersion(org.freedesktop.gusb):0.4.8
CpuArchitecture:        x86_64
CpuModel:               Intel 13th Gen Intel Core™ i7-1360P
DisplayState:           connected
DistroId:               ubuntu
DistroName:             Ubuntu
DistroPrettyName:       Ubuntu 24.04.2 LTS
DistroVersion:          24.04
FwupdSupported:         True
HostBaseboardManufacturer:Framework
HostBaseboardProduct:   FRANMCCP06
HostBiosMajorRelease:   03
HostBiosMinorRelease:   07
HostBiosVendor:         INSYDE Corp.
HostBiosVersion:        3.07
HostEnclosureKind:      a
HostFamily:             13in Laptop
HostFirmwareMajorRelease:ff
HostFirmwareMinorRelease:ff
HostProduct:            Laptop (13th Gen Intel Core)
HostSku:                FRANDECP06
HostVendor:             Framework
KernelName:             Linux
KernelVersion:          6.11.0-26-generic
LidState:               open
PowerState:             battery-discharging
RuntimeVersion(com.hughsie.libjcat):0.2.0
RuntimeVersion(com.hughsie.libxmlb):0.3.x
RuntimeVersion(org.freedesktop.fwupd):1.9.29
RuntimeVersion(org.freedesktop.fwupd-efi):1.4
RuntimeVersion(org.freedesktop.gusb):0.4.8
RuntimeVersion(org.kernel):6.11.0-26-generic
DeviceId:               d8f7c2e5ca0f897ff6d20b6e5a9945ca7c320ea9
[... 568 lines cut ...]
      dp: 7f ff 04 00
    data: 52 43
```

```shell
$ efivar -l | grep fw
```
There was nothing output from the above command.

```shell
$ tree /boot
locales-launch: Data of en_US locale not found, generating, please wait...
/boot  [error opening dir]

0 directories, 1 file
```

**Additional questions**

- Operating system and version: Ubuntu 24.04.2 LTS
- Have you tried rebooting? Yes
- Is this a regression? (I don't know, this is my first time back in the Linux world for a decade.)
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

JohnAZoidberg, 2025-06-02:
I just tried with 1.9.29 on Framework 13 Intel Core Ultra and the update succeeded.
I tried once with SecureBoot disabled and once with enabled, both ok.
@quinchou77 could you please try on 13th Gen?

@dLopreiato I assume you have not enrolled your own keys?
Could you please try the following:

1. Go into the BIOS settings `sudo systemctl reboot --firmware-setup`
2. Enter "Administer Secure Boot"
3. Go down to "DB Options" and select with Enter
4. This should show 4 keys from Microsoft and one from Framework
5. ESC back to Secureboot menu
6. Select "Restore Secure Boot to Factory Settings" and select "Enabled" (This is a one-time flag)
7. Reboot back into Ubuntu and try the update again

This is how I tested the update multiple times.

On windows you could also try https://github.com/cjee21/Check-UEFISecureBootVariables to double-check whether the update has been applied or not.

kaicataldo, 2025-06-12:
Got it, thanks for the quick response! Is GRUB + shim the recommended/blessed path?

hughsie, 2025-06-12:
> Is GRUB + shim the recommended/blessed path

It's what we test, yes. Agreed with @superm1 -- dbx updates should be supported either way.